Freetz-NG/freetz-ng #304: Can't build image with iptables/redirect, ipt_REDIRECT.ko missing

dekiesel, 2021-05-19:
Hi,

I am using ` freetz-ng 18317M-b20502a8c master 2021-05-17` to build an image for my 7390 that needs iptables with redirect (the reason being that I need to redirect rogue DNS requests to my DNS server. 

Here is my [config](https://pastebin.com/jtfKRWju).

Which gives me this during `make`

```
WARNING: File of kernel module not found: ipt_REDIRECT.ko
WARNING: File of kernel module not found: iptable_nat.ko
WARNING: File of kernel module not found: nf_conntrack.ko
WARNING: File of kernel module not found: nf_conntrack_ftp.ko
WARNING: File of kernel module not found: nf_conntrack_ipv4.ko
WARNING: File of kernel module not found: nf_defrag_ipv4.ko
WARNING: File of kernel module not found: nf_nat.ko
WARNING: File of kernel module not found: nf_nat_ftp.ko
WARNING: File of kernel module not found: xt_physdev.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
```
Searching for "REDIRECT" in `make kernel-menuconfig` gives me:

```
  │ Symbol: IP_NF_TARGET_REDIRECT [=n]                                                                                                                                                                      │
  │ Prompt: REDIRECT target support                                                                                                                                                                         │
  │   Defined at net/ipv4/netfilter/Kconfig:202                                                                                                                                                             │
  │   Depends on: NET && INET && NETFILTER && IP_NF_IPTABLES && NF_NAT && NETFILTER_ADVANCED                                                                                                                │
  │   Location:                                                                                                                                                                                             │
  │     -> Networking support (NET [=y])                                                                                                                                                                    │
  │       -> Networking options                                                                                                                                                                             │
  │         -> Network packet filtering framework (Netfilter) (NETFILTER [=y])                                                                                                                              │
  │           -> IP: Netfilter Configuration                                                                                                                                                                │
  │             -> IP tables support (required for filtering/masq/NAT) (IP_NF_IPTABLES [=m])                                                                                                                │
  │               -> Full NAT (NF_NAT [=n])  
```

but there is not Full NAT in the menuconfig:

[image]

Can anybody tell me what I am doing wrong?

Thanks!

fda77, 2021-05-19:
The source is the kernel-menuconfig. Select there as M what you need. Then select this in freetz menuconfig, if there are some missing they have to be added to the menuconfig, or for a quick test add the missing to the "own modules"

dekiesel, 2021-05-20:
I selected everything (M) in `Netfilter Configuration` but `make` still exists with 

```WARNING: File of kernel module not found: cmac.ko
WARNING: File of kernel module not found: iptable_nat.ko
WARNING: File of kernel module not found: md4.ko
WARNING: File of kernel module not found: nf_conntrack.ko
WARNING: File of kernel module not found: nf_conntrack_ftp.ko
WARNING: File of kernel module not found: nf_conntrack_ipv4.ko
WARNING: File of kernel module not found: nf_defrag_ipv4.ko
WARNING: File of kernel module not found: nf_nat.ko
WARNING: File of kernel module not found: nf_nat_ftp.ko
WARNING: File of kernel module not found: xt_physdev.ko
```
Searching for `nat` (as an example) gives me this 

```

  │ Symbol: NF_NAT [=n]                                                                            │
  │ Prompt: Full NAT                                                                               │
  │   Defined at net/ipv4/netfilter/Kconfig:162                                                    │
  │   Depends on: NET && INET && NETFILTER && IP_NF_IPTABLES && NF_CONNTRACK_IPV4                  │
  │   Location:                                                                                    │
  │     -> Networking support (NET [=y])                                                           │
  │       -> Networking options                                                                    │
  │         -> Network packet filtering framework (Netfilter) (NETFILTER [=y])                     │
  │           -> IP: Netfilter Configuration                                                       │
  │             -> IP tables support (required for filtering/masq/NAT) (IP_NF_IPTABLES [=m]) 
````
But as the picture shows there is no `nat` or `nf_nat` in `Netfilter Configuration`.

Apologies if this is something obvious and if you have already answered this!

fda77, 2021-05-20:
If a module does not exist in the kernel, dont select it in freetz.
And if a moduel in the kernel should be included in the image which dies not exist in freetz, add it (first) to "own modules"
The freetz menuconfig maybe not correct for all the kernel versions and may need be fixed

dekiesel, 2021-05-22:
Thank you for your continued support! What I understood is I should in `make kernel-menuconfig` create the modules I need and then add them to `own modules`.

So as a test I marked everything with a `M` and called `make kernel-precompiled`.- I then added one of the modules I am missing (`ipt_REDIRECT.ko`) to `own modules.`, but that just lead to the same row appearing twice when calling `make`:

```
WARNING: File of kernel module not found: ipt_REDIRECT.ko       <---------
WARNING: File of kernel module not found: iptable_nat.ko
WARNING: File of kernel module not found: nf_conntrack.ko
WARNING: File of kernel module not found: nf_conntrack_ftp.ko
WARNING: File of kernel module not found: nf_conntrack_ipv4.ko
WARNING: File of kernel module not found: nf_defrag_ipv4.ko
WARNING: File of kernel module not found: nf_nat.ko
WARNING: File of kernel module not found: nf_nat_ftp.ko
WARNING: File of kernel module not found: xt_physdev.ko
WARNING: File of kernel module not found: ipt_REDIRECT.ko         <---------
```

I then checked to see whether the module has been build somewhere:

```
d@serv:~/freetz/freetz-ng$ find . -iname "*redirect*ko"
d@serv:~/freetz/freetz-ng$ 
```

And I take this to mean that the module hasn't been build so I need to find out why, correct?

[Here](https://pastebin.com/4MJuaSb6) is my kernel config (I think, sorry, really not too firm with C).

dekiesel, 2021-05-22:
I got it to built :)

What I didn't realize was that sometimes the indicator of whether or not something is visible (`=y/n`) is missing like with physev here:

```
 Symbol: NETFILTER_XT_MATCH_PHYSDEV [=n]                                                                                                                         │  
  │ Prompt: "physdev" match support                                                                                                                                 │  
  │   Defined at net/netfilter/Kconfig:734                                                                                                                          │  
  │   Depends on: NET && INET && NETFILTER && NETFILTER_XTABLES && BRIDGE && BRIDGE_NETFILTER && NETFILTER_ADVANCED                                                 │  
  │   Location:                                                                                                                                                     │  
  │     -> Networking support (NET [=y])                                                                                                                            │  
  │       -> Networking options                                                                                                                                     │  
  │         -> Network packet filtering framework (Netfilter) (NETFILTER [=y])                                                                                      │  
  │           -> Core Netfilter Configuration                                                                                                                       │  
  │             -> Netfilter Xtables support (required for ip_tables) (NETFILTER_XTABLES [=m])
```
so for me that meant if it wasn't there I needed to edit some other config files manually. Totally selfinflicted.

Now I "only" need to test whether the modules work. Thanks!

fda77, 2021-05-22:
In some cases are patches for the kernel are requires, as avm does not use vanilla kernel, but adds some stuff - which maybe breaks some other things which are not enabled by AVM. Additional freetz patches are in male/linux/patches/....

dekiesel, 2021-05-27:
Flashing the image to the box works, but the box doesn't work afterwards (info flashing red, have to use the recovery).

[menuconfig](https://pastebin.com/V4qxhAqk), [kernel-menuconfig](https://pastebin.com/LTPVHqYQ)

If you have any ideas what else I could try I'd be very happy. Otherwise I need to bite the bullet and buy another router :/

Anyway, thanks for your help!

fda77, 2021-05-27:
If the modules are not loaded it should not matter
Or have you replaced the kernel? Then output of serial console should show cause

dekiesel, 2021-05-28:
I don't have a soldering iron atm so I opted for building with the AVM kernel. The box boots, but I can't use MASQUERADE
```
root@fritz:/var/mod/root# iptables -t nat -A POSTROUTING -j MASQUERADE
modprobe: can't load module x_tables (kernel/net/netfilter/x_tables.ko): invalid module format
iptables v1.4.11.1: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@fritz:/var/mod/root#
```
dmesg says:
```
 4626.852000] x_tables: exports duplicate symbol xt_free_table_info (owned by kernel)
```

I tried an strace (which is unfortunately the end of my C/Kernel knowledge:

```
root@fritz:/var/mod/root# strace modprobe x_tables
execve("/sbin/modprobe", ["modprobe", "x_tables"], 0x7fee49d4 /* 280 vars */) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x2aaaf000
open("/usr/lib/freetz/libc.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/mod/lib/libc.so.0", O_RDONLY)    = -1 ENOENT (No such file or directory)
open("/mod/usr/lib/libc.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=512984, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x2aab0000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\343@\0\0\0004"..., 4096) = 4096
mmap(NULL, 602112, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x2aac0000
mmap(0x2aac0000, 506864, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x2aac0000
mmap(0x2ab4b000, 8044, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x7b000) = 0x2ab4b000
mmap(0x2ab4d000, 23260, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x2ab4d000
close(3)                                = 0
munmap(0x2aab0000, 4096)                = 0
stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0755, st_size=29828, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x2aab0000
set_thread_area(0x2aab7440)             = 0
mprotect(0x2ab4b000, 4096, PROT_READ)   = 0
mprotect(0x2aabe000, 4096, PROT_READ)   = 0
ioctl(0, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
getuid()                                = 0
brk(NULL)                               = 0x496000
brk(0x497000)                           = 0x497000
chdir("/lib/modules")                   = 0
uname({sysname="Linux", nodename="fritz.box", ...}) = 0
chdir("2.6.28.10")                      = 0
open("/proc/modules", O_RDONLY|O_LARGEFILE) = 3
ioctl(3, TCGETS, 0x7fc40210)            = -1 ENOTTY (Inappropriate ioctl for device)
brk(0x498000)                           = 0x498000
read(3, "ath_pktlog 17952 0 - Live 0x816f"..., 4096) = 2253
brk(0x499000)                           = 0x499000
read(3, "", 4096)                       = 0
close(3)                                = 0
open("modules.builtin", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
lstat64("/etc/modprobe.conf", {st_mode=S_IFREG|0644, st_size=330, ...}) = 0
open("/etc/modprobe.conf", O_RDONLY|O_LARGEFILE) = 3
ioctl(3, TCGETS, 0x7fc3ffa8)            = -1 ENOTTY (Inappropriate ioctl for device)
read(3, "# general bulk keys (disks, memo"..., 4096) = 330
read(3, "", 4096)                       = 0
close(3)                                = 0
lstat64("/etc/modprobe.d", {st_mode=S_IFDIR|0755, st_size=39, ...}) = 0
open("/etc/modprobe.d", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat64(3, {st_mode=S_IFDIR|0755, st_size=39, ...}) = 0
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
getdents64(3, /* 4 entries */, 1024)    = 112
brk(0x49a000)                           = 0x49a000
lstat64("/etc/modprobe.d/usb.conf", {st_mode=S_IFREG|0777, st_size=60, ...}) = 0
open("/etc/modprobe.d/usb.conf", O_RDONLY|O_LARGEFILE) = 4
ioctl(4, TCGETS, 0x7fc3feb8)            = -1 ENOTTY (Inappropriate ioctl for device)
read(4, "blacklist cdc_ncm\nblacklist cdc_"..., 4096) = 60
read(4, "", 4096)                       = 0
close(4)                                = 0
lstat64("/etc/modprobe.d/wlan.conf", {st_mode=S_IFREG|0777, st_size=207, ...}) = 0
open("/etc/modprobe.d/wlan.conf", O_RDONLY|O_LARGEFILE) = 4
ioctl(4, TCGETS, 0x7fc3feb8)            = -1 ENOTTY (Inappropriate ioctl for device)
read(4, "blacklist aae\nblacklist adf\nblac"..., 4096) = 207
read(4, "", 4096)                       = 0
close(4)                                = 0
getdents64(3, /* 0 entries */, 1024)    = 0
close(3)                                = 0
open("modules.dep", O_RDONLY|O_LARGEFILE) = 3
ioctl(3, TCGETS, 0x7fc401b0)            = -1 ENOTTY (Inappropriate ioctl for device)
read(3, "kernel/drivers/block/loop.ko:\nke"..., 4096) = 4096
close(3)                                = 0
open("/proc/cmdline", O_RDONLY|O_LARGEFILE) = 3
fstat64(3, {st_mode=S_IFREG|0444, st_size=0, ...}) = 0
read(3, "memsize=128 flash_start=0xBF0000"..., 1024) = 108
read(3, "", 916)                        = 0
close(3)                                = 0
open("kernel/net/netfilter/x_tables.ko", O_RDONLY|O_LARGEFILE) = 3
read(3, "\177E", 2)                     = 2
_llseek(3, -2, [0], SEEK_CUR)           = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=16844, ...}) = 0
brk(0x49d000)                           = 0x49d000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\1\0\10\0\0\0\1\0\0\0\0\0\0\0\0"..., 17408) = 16844
read(3, "", 564)                        = 0
close(3)                                = 0
init_module(0x498140, 16844, "")        = -1 ENOEXEC (Exec format error)
write(2, "modprobe: can't load module x_ta"..., 95modprobe: can't load module x_tables (kernel/net/netfilter/x_tables.ko): invalid module format
) = 95
exit_group(1)                           = ?
+++ exited with 1 +++
root@fritz:/var/mod/root# 
```

fda77, 2021-05-28:
`do you need to insmod` - have you??
Some few modules need a running kernel compiled with these modules enabled as module. I dont know only for sure this if for NFS things

dekiesel, 2021-05-28:
I have used insmod (I also used modprobe, as you can see), but the error remains the same.

dekiesel, 2021-05-28:
The issue is that I can't load all the modules:

```
iptables -t nat -A POSTROUTING -j MASQUERADE
modprobe: can't load module x_tables (kernel/net/netfilter/x_tables.ko): **invalid module format**
```

not, that I am not loading them.

I think that is because of this symbol (from dmesg)

```
x_tables: exports duplicate symbol xt_free_table_info (owned by kernel)

```

I also tried building xtables and iptables into the kernel (* instead of M), but that just lead to the box crashing when running

```
iptables -t nat -A POSTROUTING -j MASQUERADE
```

fda77, 2021-05-28:
Maybe because of avm's changes for PA? Have you disabled it? https://freetz-ng.github.io/freetz-ng/wiki/00_FAQ/FAQ.de.html#funktioniert-iptablesnatconntrack

dekiesel, 2021-05-31:
```
root@fritz:/var/mod/root# echo disable > /proc/net/avm_pa/control
root@fritz:/var/mod/root# iptables -t nat -A POSTROUTING -j MASQUERADE
iptables v1.4.11.1: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded
```
So the error disappears after disabling the PA, but the box also immeaditely reboots. Thank you for your help, I'll conclude for myself that this can't be done and will buy a router that can run iptables.

fda77, 2021-05-31:
Sry, wrong issue linked!

closed 2021-05-31 by fda77

fda77, 2021-05-31:
Yes, iptables "state" things dont correctly work with PA - or even without  nowas it seems
It's really bad that avm does not uses common firewall/routing tools, but closed source things. There are 3 new devices which use iptables, but i dont now yet what excatly: https://github.com/Freetz-NG/freetz-ng/commit/2063b5a9fd2dfdd851beee127495dad74fd40337
I would recommend you some device supported by OpenWRT. You have more functions and could save money and time

dekiesel, 2021-05-31:
Agreed, my next router is going to be one that can run openwrt, even though it's a bit overkill for me. I am eying a TP-Link Archer C7 atm. It's sad, I was always very happy with my fritzbox...

Again, thanks for the continued support, it's very much appreciated.

fda77, 2021-05-31:
Me too! But im also not sure which device