Freetz-NG/freetz-ng #310: Cannot compile gcc

jschwartzenberg, 2021-05-23:
I'm running into this atter a `make`:
```
PATH=/home/julius/checkouts/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/bin:/bin:/sbin:/usr/bin:/usr/sbin make -j2 MAKEINFO=true -C /home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build all-gcc
make[1]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/intl'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/intl'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/libiberty'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libiberty'
make[3]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libiberty/testsuite'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libiberty/testsuite'
make[3]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libiberty'
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/libiberty'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=-D_GNU_SOURCE -fno-stack-protector" "CXXFLAGS=-D_GNU_SOURCE -fno-stack-protector" "CFLAGS_FOR_BUILD=-D_GNU_SOURCE -fno-stack-protector" "CFLAGS_FOR_TARGET=-g -O2 -minterlink-mips16" "INSTALL=/bin/install -c" "INSTALL_DATA=/bin/install -c -m 644" "INSTALL_PROGRAM=/bin/install -c" "INSTALL_SCRIPT=/bin/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc " "LIBCFLAGS=-D_GNU_SOURCE -fno-stack-protector" "LIBCFLAGS_FOR_TARGET=-g -O2 -minterlink-mips16" "MAKE=make" "MAKEINFO=true --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/sh" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/julius/checkouts/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu" "infodir=/home/julius/checkouts/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/share/info" "libdir=/home/julius/checkouts/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/lib" "prefix=/home/julius/checkouts/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu" "tooldir=/home/julius/checkouts/freetz-ng/toolchain/build/mips_gcc-5.5.0/mips-unknown-linux-gnu/mips-unknown-linux-gnu" "AR=ar" "AS=as" "CC=gcc" "CXX=x86_64-pc-linux-gnu-g++" "LD=ld" "LIBCFLAGS=-D_GNU_SOURCE -fno-stack-protector" "NM=nm" "PICFLAG=" "RANLIB=ranlib" "DESTDIR=" DO=all multi-do # make
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libbacktrace'
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/zlib'
make  all-am
make[3]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libbacktrace'
true  DO=all multi-do # make
make[3]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libbacktrace'
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libbacktrace'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libdecnumber'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libdecnumber'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/lto-plugin'
make  all-am
make[3]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/lto-plugin'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/lto-plugin'
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/lto-plugin'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/fixincludes'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/fixincludes'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/build-x86_64-pc-linux-gnu/libcpp'
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/libcpp'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/fixincludes'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/fixincludes'
make[2]: Entering directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/gcc'
x86_64-pc-linux-gnu-g++ -c   -D_GNU_SOURCE -fno-stack-protector -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/. -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../include -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../libcpp/include -I/home/julius/checkouts/freetz-ng/tools/build/include -I/home/julius/checkouts/freetz-ng/tools/build/include -I/home/julius/checkouts/freetz-ng/tools/build/include  -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../libdecnumber -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../libbacktrace   -o reload1.o -MT reload1.o -MMD -MP -MF ./.deps/reload1.TPo /home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/reload1.c
x86_64-pc-linux-gnu-g++ -c   -D_GNU_SOURCE -fno-stack-protector -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/. -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../include -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../libcpp/include -I/home/julius/checkouts/freetz-ng/tools/build/include -I/home/julius/checkouts/freetz-ng/tools/build/include -I/home/julius/checkouts/freetz-ng/tools/build/include  -I/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0/gcc/../libdecnumber -I/home/julius/checkouts/fre [... cut ...]
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/gcc'
make[1]: *** [Makefile:4102: all-gcc] Error 2
make[1]: Leaving directory '/home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build'
make: *** [toolchain/make/kernel/gcc/gcc.mk:79: /home/julius/checkouts/freetz-ng/source/toolchain-mips_gcc-5.5.0/gcc-5.5.0-build/.compiled] Error 2
```

This is on Parabola. The gcc version used is: 11.1.

fda77, 2021-05-27:
You gcc11 on you system is to new to compile gcc5.
It need to be set a compatibility mode by a parameter somehere in the makefile of gcc5

jschwartzenberg, 2021-05-28:
Yeah that makes sense. Might it also be possible to upgrade gcc at some point or would that bring many extra complications?

fda77, 2021-05-28:
This gcc version is the one avm used. It could cause problems if you build the kernel with a new one but want to load closed-source modules for dect,wlan,dsl etc

closed 2021-06-05 by fda77

jschwartzenberg, 2021-06-27:
I can confirm this is all fine now with the precompiled version.