Freetz-NG/freetz-ng #316: umurmur compilation fails due to linker errors

hadis-git, 2021-05-31:
The package umurmur fails to compile for me on my native archlinux desktop and laptop and the freetz-ng linux vm from the README.

My only change to the .config after having freshly clone this repo is that I enabled the umurmur package with `make menuconfig`. I get one of two errors depending on which ssl backend I select for umurmur. 

OpenSSL: 
/home/hadis/Source/freetz-ng/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/lib/gcc/mips-linux-uclibc/8.3.0/../../../../mips-linux-uclibc/bin/ld: ssli_openssl.o: in function `SSLi_init':
ssli_openssl.c:(.text+0x234): undefined reference to `EC_KEY_new_by_curve_name'
/home/hadis/Source/freetz-ng/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/lib/gcc/mips-linux-uclibc/8.3.0/../../../../mips-linux-uclibc/bin/ld: ssli_openssl.c:(.text+0x254): undefined reference to `EC_KEY_free
collect2: error: ld returned 1 exit status`

Mbed TLS:
checking for mbedtls_ssl_init in -lmbedtls... no
configure: error: could not find libmbedtls

Here's the output of `find -name 'libssl*'`:
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/openssl-1.1.1j/libssl.map
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/openssl-1.1.1j/libssl.so.1.1
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/openssl-1.1.1j/libssl.a
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/openssl-1.1.1j/libssl.so
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/openssl-1.1.1j/libssl.pc
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/openssl-1.1.1j/util/libssl.num
./packages/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/root/usr/lib/freetz/libssl.so.1.1
./packages/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/root/usr/lib/freetz/libssl.so
./toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/lib/libssl.so.1.1
./toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/lib/libssl.a
./toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/lib/libssl.so
./toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/lib/pkgconfig/libssl.pc

and here's the output of `find -name 'mbedtls*'`
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mbedtls-2.25.0
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mbedtls-2.25.0/tests/scripts/mbedtls_test.py
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mbedtls-2.25.0/include/mbedtls
./source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mbedtls-2.25.0/doxygen/mbedtls.doxyfile
./make/libs/mbedtls
./make/libs/mbedtls/mbedtls.mk
./toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/include/mbedtls

hadis-git, 2021-05-31:
I should note that I can compile umurmur without problems for all of my host systems.

fda77, 2021-05-31:
`undefined reference to `EC_KEY_new_by_curve_name'`
It could help to enable Elyptic Curve in libs>crypto>openssl config.
If not the umurmur version in freetz is to old to use with you newer (?) openssl selected

hadis-git, 2021-05-31:
Of course, EC stands for Elyptic Curve.
I was able to build now with EC support in openssl, however only after switching from fakeroot to pseudo as I get the following error if I try to build with fakeroot selected:

fakeroot: preload library `libfakeroot.so' not found, aborting.
make: *** [Makefile:308: firmware-nocompile] Error 1

This also doesn't fix umurmur with mbedtls not working. As you can see in my first post mbedtls doesn't get compiled before umurmur does.

fda77, 2021-05-31:
Im not sure for which ssl lib the log in op was. Maybe you need "distclean" for the package, as a rebuild rule is missing?

hadis-git, 2021-05-31:
umurmur fails to build with mbedtls after:

git clone https://github.com/Freetz-NG/freetz-ng/
cd freetz-ng
make menuconfig
make

fda77, 2021-05-31:
You could try a newer version of umurmur, the one in freetz is 4 years old

hadis-git, 2021-05-31:
I take back that mbedtls doesn't compile, that was my fault.

Trying to compile with the most current umurmur however fails for the same reason:

`---> package/umurmur ... configuring ...`
[...]
`Automatically preparing build ... done`
[...]

checking for mbedtls/ssl.h... yes
checking for mbedtls/version.h... yes
checking for mbedtls_ssl_init in -lmbedtls... no
configure: error: could not find libmbedtls

`ERROR: Build failed.
make: *** [make/umurmur/umurmur.mk:28: source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/umurmur-0.2.17/.configured] Error 1`

hadis-git, 2021-05-31:
I figured it out. 
For umurmur+openssl I need to enable EC for openssl and for umurmur+openssl I need to enable blowfish cipher and rsa prime-number generator.

fda77, 2021-05-31:
Changed with https://github.com/Freetz-NG/freetz-ng/commit/ee9fc55b04c7d2c0e534b4be8cbf051f1779d52d
If you want a newer version: https://freetz-ng.github.io/freetz-ng/SUPPORT.html#package-bump

closed 2021-05-31 by fda77

hadis-git, 2021-05-31:
Didn't just fix it for mbedtls?

For OpenSSL we need "Support elliptic curve cryptography"

fda77, 2021-05-31:
https://github.com/Freetz-NG/freetz-ng/commit/0e2aa115ea750a711d6732b8ada4ac3c15f017ac