Freetz-NG/freetz-ng #325: 7412 bootloop with recent checkout

edeso, 2021-06-09:
maybe a duplicate of https://github.com/Freetz-NG/freetz-ng/issues/135 have no serial attached to confirm.

after hours of building and flashing i pinpointed https://github.com/Freetz-NG/freetz-ng/commit/a3fd9e5436595a11e073789c1542480a80f519e7 as the faulty commit.

first try to readd the exports initially gained profit although have no clue why it was removed or why it is needed here. here's the patch

```
diff --git a/make/mod/files/root/etc/init.d/modlibrc b/make/mod/files/root/etc/init.d/modlibrc
index 16a413ba3..d1687e3bf 100644
--- a/make/mod/files/root/etc/init.d/modlibrc
+++ b/make/mod/files/root/etc/init.d/modlibrc
@@ -18,6 +18,9 @@
 # PID_FILE              /var/run/${DAEMON_BIN}.pid             pid-file for DAEMON
 # DAEMON_CFGFILE        /tmp/flash/${DAEMON}/${DAEMON}.cfg     config file for the daemon (typically used by startdaemon)
 
+export PATH=/mod/sbin:/mod/bin:/mod/usr/sbin:/mod/usr/bin:/mod/etc/init.d:/sbin:/bin:/usr/sbin:/usr/bin
+export LD_LIBRARY_PATH=/mod/lib:/mod/usr/lib
+
 [ -r /var/env.daemon ] && . /var/env.daemon
 . /etc/init.d/loglibrc

fda77, 2021-06-09:
This patch really helps?
It was removed from make/mod/files/root/etc/init.d/modlibrc as make/mod/files/root/etc/init.d/rc.mod puts these 2 lines into /var/env.daemon which is read by make/mod/files/root/etc/init.d/modlibrc.

edeso, 2021-06-10:
sounds reasoning. maybe i confused something along the way. i will doublecheck and come back with my findings.

edeso, 2021-06-10:
ok, just tried a fresh clone. same findings. unpatched it bootloops. is it possible that `/etc/init.d/modlibrc` is somehow run before the `/var/env.daemon` is generated on a 6.86 based image?

changed the patch to only export if `/var/env.daemon` is unreadable or the source command fails like so
```
[ -r /var/env.daemon ] && . /var/env.daemon || {
export PATH=/mod/sbin:/mod/bin:/mod/usr/sbin:/mod/usr/bin:/mod/etc/init.d:/sbin:/bin:/usr/sbin:/usr/bin
export LD_LIBRARY_PATH=/mod/lib:/mod/usr/lib
}
```

fda77, 2021-06-18:
There are "wrapper" for some avm daemons: https://github.com/Freetz-NG/freetz-ng/tree/master/patches/scripts#wrapper
Scripts: https://github.com/Freetz-NG/freetz-ng/tree/master/make/mod/files/root/usr/bin/wrapper
Which run rc.-Scipts in https://github.com/Freetz-NG/freetz-ng/tree/master/make/mod/files/root/etc/init.d
In the Scrips modlibrc is used

The wrapper are started during system start by avm, before rc.mod, which is the last in order. (see serial console output)
So /var/env.* could be missing and creation should be moved from rc.mod to modlibrc itself - in case it works earlier during startup

fda77, 2021-06-18:
Please try if this works for you: d79192c91 & ab105cc4a

edeso, 2021-06-21:
works for me! thanks!

closed 2021-07-01 by fda77