Freetz-NG/freetz-ng #331: Compilation fails with openssl error

stefanpuch, 2021-06-18:
I did a fresh git clone yesterday. The build process fails at openssl for uClibc-1.0.37. It seems to me, that the lib files are not correctly copied from toolchain/build directory to packages directory.

```make[2]: Verzeichnis „/home/space/freetz/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/apps“ wird betreten
installing openssl
make[2]: Verzeichnis „/home/space/freetz/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/apps“ wird verlassen
installing libcrypto.a
installing libssl.a
installing libcrypto.so.1.0.0
installing libssl.so.1.0.0
make[2]: Verzeichnis „/home/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib“ wird betreten
make[3]: Verzeichnis „/home/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib“ wird betreten
make[3]: Verzeichnis „/home/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib“ wird verlassen
make[3]: Verzeichnis „/home/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib“ wird betreten
make[3]: Verzeichnis „/home/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib“ wird verlassen
make[2]: Verzeichnis „/home/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib“ wird verlassen
cp libcrypto.pc /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/pkgconfig
chmod 644 /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/pkgconfig/libcrypto.pc
cp libssl.pc /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/pkgconfig
chmod 644 /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/pkgconfig/libssl.pc
cp openssl.pc /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/pkgconfig
chmod 644 /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/pkgconfig/openssl.pc
make[1]: Verzeichnis „/home/space/freetz/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u“ wird verlassen
sed -i -r -e "s,^(prefix=)(['"'"'"]?)([^'"'"'"]*)(\2)$,\1\2/home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc\3\4,g"  \
	/home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/pkgconfig/{libcrypto,libssl,openssl}.pc
mkdir -p packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/root
if test -d make/openssl/files; then 	tools/tar-gnu -cf - -C make/openssl/files --exclude=.svn --exclude=.git --exclude=.gitignore --exclude=.build-prereq-checked --exclude=.unpacked --exclude=.configured --exclude=.compiled --exclude=.installed . | tools/tar-gnu -xf - -C packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u; fi
mkdir -p packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/root/usr/bin/; cp source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/apps/openssl packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/root/usr/bin/openssl; [ "y" == "y" ] && tools/patchelf --set-interpreter /usr/lib/freetz/ld-uClibc.so.1 packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/root/usr/bin/openssl || true; /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-strip --remove-section={.comment,.note,.pdr} packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/root/usr/bin/openssl;
chmod 755 /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/libssl.so.; mkdir -p packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/root/usr/lib/freetz/; rm -f ; cp -a  packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/root/usr/lib/freetz/; /home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-strip --remove-section={.comment,.note,.pdr} packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/root/usr/lib/freetz/libssl.so.;
chmod: Zugriff auf '/home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib/libssl.so.' nicht möglich: Datei oder Verzeichnis nicht gefunden
cp: Fehlender Zieldatei‐Operand hinter 'packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/root/usr/lib/freetz/'
„cp --help“ liefert weitere Informationen.
/home/stefan/space/freetz/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-strip: 'packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/root/usr/lib/freetz/libssl.so.': No such file
make: *** [make/openssl/openssl.mk:104: packages/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/root/usr/lib/freetz/libssl.so.] Fehler 1

user@host:~/space/freetz/freetz-ng/> find . -name libssl.so* -exec ls -SHl {} \;
-r-xr-xr-x 1 user user 381468 Jun 18 06:51 ./toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib/libssl.so.1.0.0
-r-xr-xr-x 1 user user 381468 Jun 18 06:51 ./toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/lib/libssl.so
-rwxr-xr-x 1 user user 381468 Jun 18 06:49 ./source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/libssl.so.1.0.0
-rwxr-xr-x 1 user user 381468 Jun 18 06:49 ./source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/openssl-1.0.2u/libssl.so
user@host:~/space/freetz/freetz-ng/> 

user@host:~/space/freetz/freetz-ng/> git describe --tags
ng21030-326-g84a329449
user@host
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/6674779/config.txt)
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/6674780/config.txt)
:~/space/freetz/freetz-ng/>

```

stefanpuch, 2021-06-18:
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/6674782/config.txt)

stefanpuch, 2021-06-18:
The reason seems to be within make/openssl/openssl.mk in line 28
```
$(PKG)_LIBNAMES_LONG := $($(PKG)_LIBNAMES_SHORT:%=%.so.$($(PKG)_LIB_VERSION)
```
The $($(PKG)_LIB_VERSION) does not add a valid value to the  LIBNAMES_SHORT variable

(PKG)_LIB_VERSION is defined in line 2 it says:
```
(PKG)_LIB_VERSION:=$(call qstrip,$(FREETZ_OPENSSL_SHLIB_VERSION))
```
To my understanding $(FREETZ_OPENSSL_SHLIB_VERSION) should be '1.0.0' because of 
```
        config FREETZ_OPENSSL_SHLIB_VERSION
                string
                default "0.9.8" if FREETZ_OPENSSL_VERSION_0
                default "1.0.0" if FREETZ_OPENSSL_VERSION_1
                default "1.1"   if FREETZ_OPENSSL_VERSION_2
                default "3"     if FREETZ_OPENSSL_VERSION_3
```

in make/openssl/Config.in.libs and .config has 'FREETZ_OPENSSL_VERSION_1=y'

As a proof of concept I set it hardcoded to 1.0.0 and the build process works:
```
user@host:~/space/freetz/freetz-ng/> git diff
diff --git a/make/openssl/openssl.mk b/make/openssl/openssl.mk
index e82eebd5f..af702bb73 100644
--- a/make/openssl/openssl.mk
+++ b/make/openssl/openssl.mk
@@ -1,5 +1,6 @@
 $(call PKG_INIT_BIN,$(if $(FREETZ_OPENSSL_VERSION_0),0.9.8zh,$(if $(FREETZ_OPENSSL_VERSION_1),1.0.2u,$(if $(FREETZ_OPENSSL_VERSION_3),3.0.0-alpha13,1.1.1k))))
 $(PKG)_LIB_VERSION:=$(call qstrip,$(FREETZ_OPENSSL_SHLIB_VERSION))
+$(PKG)_LIB_VERSION:=1.0.0
 $(PKG)_SOURCE:=$(pkg)-$($(PKG)_VERSION).tar.gz
 $(PKG)_SOURCE_SHA256_0.9:=f1d9f3ed1b85a82ecf80d0e2d389e1fda3fca9a4dba0bf07adbf231e1a5e2fd6
 $(PKG)_SOURCE_SHA256_1.0:=ecd0c6ffb493dd06707d38b14bb4d8c2288bb7033735606569d8f90f89669d16
user@host:~/space/freetz/freetz-ng/>
```

fda77, 2021-06-18:
Why the variable should be missing? Check with
`grep OPENSSL .config`
Btw, I've not build openssl 1.0 since a long time as it is not supported anymore since 2019

fda77, 2021-06-18:
Maybe you should select openssl in menuconfig???

stefanpuch, 2021-06-18:
My config file is attached above. 
```
user@host:~/space/freetz/freetz-ng/> grep OPENSSL .config
# FREETZ_PACKAGE_OPENSSL is not set
FREETZ_PACKAGE_OPENVPN_OPENSSL=y
FREETZ_OPENSSL_VERSION_1=y
# FREETZ_OPENSSL_VERSION_2 is not set
FREETZ_OPENSSL_VERSION_1_MIN=y
FREETZ_OPENSSL_VERSION_1_MAX=y
FREETZ_OPENSSL_VERSION_2_MAX=y
FREETZ_BUSYBOX_FEATURE_WGET_OPENSSL=y
FREETZ_BUSYBOX___V133_FEATURE_WGET_OPENSSL=y
FREETZ_AVM_HAS_OPENSSL=y
FREETZ_AVM_HAS_OPENSSL_VERSION_1=y
```
I did not select OpenSSL but it is obvious implicitly used from OpenVPN which I selected. Maybe I should start again with a clean .config file and test again? I used a clean git puss but my .config from end f last year

fda77, 2021-06-18:
Ah, you selected just openvpn "static" what does not select openssl. Your config:
```
# FREETZ_LIB_libcrypto is not set
# FREETZ_LIB_libssl is not set
```
... because openvpn is static
and so FREETZ_OPENSSL_SHLIB_VERSION is missing
Seems the "OpenSSL configuration" should be always visible, like "OpenSSL version"

fda77, 2021-06-18:
Update & run `make olddefconfig`

closed 2021-06-18 by fda77

stefanpuch, 2021-06-18:
Your fix works fine for me. I also ensured that FREETZ_OPENSSL_VERSION_2 is now used (with config from scratch) because openssl 1.0 is unsupported.
Thanks for the quick help!