Freetz-NG/freetz-ng #334: tools/push_firmware calls ifconfig without path

1ras, 2021-06-20:
* Warning: It seems your network is not able to reach
   192.168.178.1 directly. This command could help to fix:
tools/push_firmware: Zeile 565: ifconfig: Kommando nicht gefunden.
   $ sudo ifconfig enp0s25:0 192.168.178. up
   Proceed anyway? ([y]/n) n

ifconfig should be called as /sbin/ifconfig as a normal user might not have /sbin in the PATH (Debian Buster).

<pre>
--- a/tools/push_firmware
+++ b/tools/push_firmware
@@ -562,7 +562,7 @@
 		*)	echo " One(!) of these commands could help to fix:" ;;
 	esac
 	for x in $(ls /sys/class/net/ | grep -v "^lo$" ); do
-		echo -e  "   $ \033[4msudo ifconfig $x:0 ${ip%.*}.$(ifconfig $x | sed -rn 's/.*inet [0-9\.]*\.([0-9]*) .*/\1/p') up\033[0m"
+		echo -e  "   $ \033[4msudo ifconfig $x:0 ${ip%.*}.$(/sbin/ifconfig $x | sed -rn 's/.*inet [0-9\.]*\.([0-9]*) .*/\1/p') up\033[0m"
 	done
 	echo -n  "   Proceed anyway? ([y]/n) "
 	read -n 1 -s PROCEED_NETCFG
</pre>

fda77, 2021-07-01:
This is a bug of debian..
 - use a good linux distribution
 - set you path correctly, eg `PATH=/sbin:$PATH`
 - or read /usr/share/doc/util-linux/NEWS.Debian.gz `If you want to restore behaviour more similar to the previous one you can add 'ALWAYS_SET_PATH yes' in /etc/login.defs.`

Btw,
```
[freetz@fedora:~/]$ which ifconfig
/usr/sbin/ifconfig
```

closed 2021-07-01 by fda77

1ras, 2021-07-01:
1. This is not a bug in Debian, the traditional location of ifconfig is /sbin/ifconfig.
2. sbin is for system binaries and not for user binaries, users are not expected to have sbin in the path, see the Filesystem Hierarchy Standard: https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch03s16.html
3. This is not about /bin/su to set a path (ALWAYS_SET_PATH) as /bin/su is not involved in this bug.
4. Your which command returns /usr/sbin/ifconfig because of usr merge and not because that's the location of ifconfig: https://www.freedesktop.org/wiki/Software/systemd/TheCaseForTheUsrMerge/

fda77, 2021-07-01:
It is a bug of debian, as every other distrubution sets PATH properly
```
$ realpath /sbin/ifconfig
/usr/sbin/ifconfig
```
Would work atm, but unsure how long

But hardcoded pats will not work if you install it form source, eg "local" dirs

1ras, 2021-07-01:
Again, sbin is for system binaries and not for user binaries, users are not expected to have sbin in the path. If distribution XYZ is doing it differently then fine. The point is that you cannot rely on it.

And just for the records, the bug is surely not where someone follows a standard. The opposite is true.

If you look closer on the push_firmware script then you can see that it uses a mix of ip and ifconfig which makes anyway not much sense as both commands are for the same purpose. So a mid-term fix could be to eliminate ifconfig and rely completely on the ip command. Ip is considered as a user binary and does not have this problems.

1ras, 2021-07-01:
Removing ifconfig in the mid-term could also remove a dependency. On many systems net-tools are not installed anymore by default.

fda77, 2021-07-01:
If you check the history of the script you will notice that it is at least 14 years old. So ifconfigs are okay until sb updates it

1ras, 2021-07-01:
Yes it's clear that it has historical reason. Practically the same thing as why ifconfig is at /sbin/ifconfig and not considered as a user command.
Using "realpath /sbin/ifconfig" should do it as well. When I opened the bug, I was just wondering if it makes sense to fix the ifconfig error or if it wouldn't be better to switch to ip as the script already uses ip.

1ras, 2021-07-01:
Oh no, I think "realpath /sbin/ifconfig" wont work with local installations, if that's your requirement.

fda77, 2021-07-01:
The problem source is that debian changed the default behaviour and now the script not working anymore. So i think the installed debian should set to behave like all other. Same with openwrt where fedora34 broke the "which" command - i changed my system to behave like other

1ras, 2021-07-01:
No, Debian has not changed the default behavior, Debian is still using it. Others have softened the rules by putting sbin in the path for users or by installing ifconfig into bin instead of sbin, not Debian.
The default behavior is that ifconfig is a system command intended for the root user and located at /sbin/ifconfig.
The default is that a user is not expected to have sbin in the path as he is not expected to use system commands.

fda77, 2021-07-01:
Sure debian broke it, as i wrote above
Read line ~~15 of https://wiki.debian.org/NewInBuster

fda77, 2021-07-01:
Btw, the debian NEWS entry i quoted in https://github.com/Freetz-NG/freetz-ng/issues/334#issuecomment-872338053 `ALWAYS_SET_PATH yes` is wrong and yet fixed by debian. Correct entry  `ALWAYS_SET_PATH=yes`

1ras, 2021-07-01:
No, this is **not** related to a change in Debian Buster. The change regarding the ALWAYS_SET_PATH option as described at https://wiki.debian.org/NewInBuster or at /usr/share/doc/util-linux/NEWS.Debian.gz is in a completely different area of the system, changes a completely different behavior and is **not at all** related to this bug report.

If ALWAYS_SET_PATH is set to yes, then the /bin/su commands initializes the PATH, even if you don't request a login shell. This is a documented behavior of the su command: https://man7.org/linux/man-pages/man1/su.1.html

push_firmware is not calling su at all, it's completely unrelated!

1ras, 2021-07-01:
This is what it does:

```
andi$ echo $PATH
/home/andi/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
andi$ su
Passwort:
root# echo $PATH
/home/andi/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
root# exit
exit
andi$ su -
Passwort:
root# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
root# echo ALWAYS_SET_PATH yes >> /etc/login.defs
root# exit
exit
andi$ su
Passwort:
root# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
# exit
exit
andi$ echo $PATH
/home/andi/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
```

It is not related to the PATH of the user account. It is a question which PATH the users gets **after** using the su command. Does he keep his old user PATH environment or does he get the PATH initialized with the environment of the new user?

If you request a login shell (su -) you get the PATH initialized, if not then not. if you want to always get the PATH initialized independently how you call su, then you need the ALWAYS_SET_PATH option.

In no case does it affect the path of the user account. My user path has not changed, compare the last line with the first one.

All this nice functionality is dropped by a distribution that sets the same path for everyone.

Hope that makes more sense now.

fda77, 2021-07-01:
I dont use Debian anymore as it has to much such on-default things. If "ALWAYS_SET_PATH" is not the correct option tel lthe correct one or just set the PATH in profile properly

1ras, 2021-07-01:
My PATH is set properly as users are not expected to have sbin on the path.
The correct behavior of a user script which wants to use a system command is to call it with the full path.

The ip command is not a system command and can replace ifconfig. However I had now a closer look on how this fancy logic works that suggest the IP address for the interface and that logic is faulty. It just uses ifconfig to get the last octet of the current IP address of the interface and uses this for the 192.168.178.x address. Guess what happens if the last octet of the current IP address is 1?

Yes, it wants me to setup my interface with "sudo ifconfig enp0s25:0 192.168.178.1 up". That's just great because then my interface has the identical IP address as the Fritzbox bootloader...

That whole fancy logic is for nothing ;-)

I can send you a patch that replaces ifconfig with ip, but I am no longer convinced that this logic makes any sense.

fda77, 2021-07-02:
There can nothing be done against stupidity of users...