Freetz-NG/freetz-ng #335: replace_kernel not booting on 6840

squelcher, 2021-06-22:
While the original kernel is booting fine with a modified image, a replace_kernel cannot be loaded:

`Eva_AVM >go

........................................................[] set PHYs in Reset 0x1

Lantiq xDSL CPE VR9
phym = 08000000, mem = 08000000, max_pfn = 00008000
Reserving memory for CP1 @0xa8000000, size 0x00000000
plat_device_tree_setup: AVM hardware subrevision 5
plat_device_tree_setup: using Fallback device-tree of AVM hardware subrevision 2
DT: d0 0d fe ed 00 00 10 a5

CAUSE    = 0x00000010  Address Alignment Load
STATUS   = 0x1100FF03      EPC      = 0x80854974
BADVADDR = 0x808411A6      ERROREPC = 0x00000000

$ 0(zr):0x00000000  $ 8(t0):0x0000000D  $16(s0):0x808411A6  $24(t8):0x00000000
$ 1(at):0x00000000  $ 9(t1):0x00000003  $17(s1):0x80590000  $25(t9):0x0000FCD8
$ 2(v0):0x808A27AC  $10(t2):0x00000001  $18(s2):0x80000FFF  $26(k0):0x00000000
$ 3(v1):0xD00D0000  $11(t3):0x00000000  $19(s3):0x80707EE0  $27(k1):0x00000000
$ 4(a0):0x808411A6  $12(t4):0x00000000  $20(s4):0x810C25B4  $28(gp):0x80704000
$ 5(a1):0x8065B423  $13(t5):0x00000002  $21(s5):0x810C2524  $29(sp):0x80707E30
$ 6(a2):0x808A27AC  $14(t6):0x00000000  $22(s6):0x810C26BC  $30(s8):0x00000000
$ 7(a3):0x0000000A  $15(t7):0x0000000F  $23(s7):0x0000000D  $31(ra):0x808528A8`

It seems there is something wrong with the memory addressing.

As said, the original kernel and also an OpenWRT kernel is loading fine.

fda77, 2021-07-01:
Did this ever worked - with freetz, with older FOS version?
6840 uses 7490 sources, you could download both and `diff` the .config of the kernel

squelcher, 2021-07-14:
I don't think that ever worked, I tried that a few years ago and gave up on that.

fda77, 2021-07-15:
In this case you should first try with avms config fom its sources package (if available, or if not write an email and beg for it)
And to a diff of 7490/6840 sources, maybe some (small?) changes are needed

fda77, 2021-08-07:
Try to use dts file for 6840, not 7490. Diff from 7.27 7490 source:

```
--- linux/arch/mips/boot/dts/Fritz_Box_6840.dts
+++ linux/arch/mips/boot/dts/Fritz_Box_HW185_310.dts
@@ -1,52 +1,46 @@
-/dts-v1/;
 #include <linux/avm_hw_config_def.h>
 #include <asm/mach-lantiq/common/ifx_gpio_def.h>
+#include "vr9.dtsi"

 / {
        avm-hw-revision{
                compatible = "avm,avm_hw_revision";
-               revision = "177";
+               revision = "185";
        };
        avm-gpio{
                compatible = "avm,avm_gpio_generic";
                gpio_avm_led_power{
-                       value = <32>;
-                       param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
-                       config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OUTPUT_SET>;
-                       module_id = <DEF_IFX_GPIO_MODULE_LED>;
-               };
-               gpio_avm_led_power_red{
-                       value = <33>;
+                       value = <45>;
                        param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
                        config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OUTPUT_SET>;
                        module_id = <DEF_IFX_GPIO_MODULE_LED>;
                };
                gpio_avm_led_internet{
-                       value = <38>;
+                       value = <47>;
                        param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
                        config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OUTPUT_SET>;
                        module_id = <DEF_IFX_GPIO_MODULE_LED>;
                };
-               gpio_avm_led_dect{
-                       value = <35>;
+               gpio_avm_led_festnetz{
+                       value = <36>;
                        param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
                        config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OUTPUT_SET>;
                        module_id = <DEF_IFX_GPIO_MODULE_LED>;
                };
                gpio_avm_led_wlan{
-                       value = <36>;
+                       value = <35>;
                        param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
                        config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OUTPUT_SET>;
                        module_id = <DEF_IFX_GPIO_MODULE_LED>;
                };
                gpio_avm_led_info{
-                       value = <47>;
+                       value = <33>;
                        param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
                        config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OUTPUT_SET>;
                        module_id = <DEF_IFX_GPIO_MODULE_LED>;
                };
                gpio_avm_led_info_red{
-                       value = <34>;
+                       value = <46>;
                        param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
                        config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OUTPUT_SET>;
                        module_id = <DEF_IFX_GPIO_MODULE_LED>;
@@ -81,6 +75,48 @@
                        config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_SET IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OD_SET>;
                        module_id = <DEF_IFX_GPIO_MODULE_PIGLET>;
[... 118 lines cut ...]
+               gpio_avm_offload_wasp_wakeup{
+                       value = <5>;
+                       param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_HIGH>;
+                       config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OD_SET IFX_GPIO_IOCTL_PIN_CONFIG_PUDSEL_SET>;
+                       module_id = <DEF_IFX_GPIO_MODULE_WLAN_OFFLOAD_WASP_RESET>;
+               };
+               gpio_avm_offload_wasp_reset{
+                       value = <34>;
+                       param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_LOW>;
+                       config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OD_SET IFX_GPIO_IOCTL_PIN_CONFIG_PUDSEL_SET>;
+                       module_id = <DEF_IFX_GPIO_MODULE_WLAN_OFFLOAD_WASP_RESET>;
+               };
+               gpio_avm_offload_scrpn_uart{
+                       value = <31>;
+                       param = <AVM_DEF_HW_PARAM_GPIO_OUT_ACTIVE_HIGH>;
+                       config = <IFX_GPIO_IOCTL_PIN_CONFIG_DIR_OUT IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL0_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_ALTSEL1_CLEAR IFX_GPIO_IOCTL_PIN_CONFIG_OD_SET IFX_GPIO_IOCTL_PIN_CONFIG_PUDSEL_SET>;
+                       module_id = <DEF_IFX_GPIO_MODULE_WLAN_OFFLOAD_SCRPN_UART>;
+               };
+               gpio_avm_boot_sel3{
+                       value = <4>;
+                       param = <AVM_DEF_HW_PARAM_GPIO_IN_ACTIVE_HIGH>;
+               };
        };
 };
```

PeterPawn, 2021-08-12:
It's the same as with other models ... changing a DTS source file doesn't work, if you want to get a working kernel.

Even if AVM is providing these files in source code format, they're never used. The flattened device trees for ALL HWSubrevision values (if there's more than one due to hardware layout changes) are taken from the `avm_kernel_config` area at runtime and it doesn't matter anything, what exists at compile-time.

The re-get the "Replace kernel" option (and actions), the extraction and processing of the mentioned avm_kernel_config area has to be enabled first - other patches (esp. patches for DTS files) are useless.

fda77, 2021-08-13:
Sorry, i used to murhc nromal Linux devices with dts support...
But im still wondering why avm
 - updates the dts(i)
 - publishes them in source package

It is know they do release at best what is mandatory

PeterPawn, 2021-08-13:
The configuration area contains **really** the compiled DTS info - but it's not enough to change the source file (or use another one), as long as those changes aren't later used (after something has been compiled to a FDT) while linking the new kernel. And the source code for exactly those parts is still missing from AVM's archives.

For the workaround it was easier, to re-use the FDTs already contained in the config area as BLOBs (means: as a blackbox containing some binary data) and to output an exact copy of them (or "of this", when only a single `HWSubrevision` is contained) again. But it's possible, too, to create these data from a `.dtc` file instead - but it's more complicated to link this together (with the other info from the config area used by AVM) to get a running kernel.

fda77, 2021-08-19:
"Fixed" as RK was disabled for this device.
There is still no tool by avm, even kernel-config-area is part of the kernel

closed 2021-08-19 by fda77