Freetz-NG/freetz-ng #337: Wireguard.ko IPv6 problems

linuxrrze, 2021-06-29:
Hi there!

I've been trying to add an IPv6 address to my wireguard wg0 interface.

In the web GUI I specified both an IPv4 and an IPv6 address like:

IPv4-Address: 10.101.2.9/24
IPv6-Address: fd01:101:20::9/48

However upon wireguard start I get several errors (and no IPv6 address on wg0).

While diagnosing this step by step I also tried to simply apply an IPv6 address to the wg0 interface by issuing

> ip -6 address add fd01:101:20::9/48 dev wg0

However all I got was this error message:

> RTNETLINK answers: Permission denied

Adding the IPv4 address however works fine.

For me it looks like all IPv6 related settings are getting refused, so I took a quick look at the source of the wireguard kernel module (related to IPv6 related stuff).

And there are several places where

`#if IS_ENABLED(CONFIG_IPV6)`

is present.
As I'm building for 7590 (and kernel building is disabled for this platform because of bootloader issues) I guess the kernel is build out-of-tree, right?
Which leads me to the next question: 

Is the kernel setting CONFIG_IPV6 honored when building the module this way or could this be the causing missing IPv6 support?
(Kernel itself is definitely IPv6 enabled, other devices do have IPv6 addresses).

Any help/hint is highly appreciated!

BTW: Thanx for freetz-ng - great work!

linuxrrze, 2021-06-29:
> ip -6 address add fd01:101:20::9/48 dev wg0

closed 2021-06-29 by linuxrrze

fda77, 2021-07-01:
Seems nobody ever routed ipv6 through wireguard... AVM uses for routing some weird closed-source binaries