Freetz-NG/freetz-ng #344: 7530 menuconfig does not offer wireguard-tools anymore

edeso, 2021-07-06:
selected
```
freetz-ng 18442-3c5b0c4e3 ng21030 2021-07-05
...
Hardware series (all)  --->
Hardware type (7530)  --->
Target region (avm - germany)  --->
Firmware version (FRITZ!OS 07.25+)  --->
>> This firmware fits on AVM and AVME devices
```
under Packages->W instead of
```
[*] wireguard-tools 1.0.20210424  
[X]   wireguard-linux-compat 1.0.20210424
```
menuconfig says
```
--- wireguard-tools (not available, needs kernel modules)
```
old checkout at
```
commit 754a0fbb38d832e8990c617b288057040cdf79a2
Author: fda77 <fda77@users.noreply.github.com>
Date:   Mon Jun 7 20:52:02 2021 +0200
    gcc5: fix missing sys/sdt.h
     - closes https://github.com/Freetz-NG/freetz-ng/issues/279
```
works fine. current checkout https://github.com/Freetz-NG/freetz-ng/commit/3c5b0c4e3e45fc779c8a303d0d29a33ddb128983 does not.

fda77, 2021-07-06:
Blame AVM for their user-unfreindliy source release management

closed 2021-07-06 by fda77

edeso, 2021-07-06:
thanks for the fix! sure AVM is not a stellar provider of modified GPLd sources, but in this case it seems they are not actually needed ;) or am i missing something?

fda77, 2021-07-06:
Exactly, nobody misses them except they are missing. So only a few people nag them by email requests and in some cases it took years until they released the sources

edeso, 2021-07-07:
well, the https://download.avm.de/fritzbox/fritzbox-7530/deutschland/fritz.os/info_de.txt says
"
Der Source Code der als Open Source verbreiteten Dateien kann
schriftlich angefordert werden über fritzbox_info@avm.de. 
"

i just wrote them a one liner. let's see :)

fda77, 2021-07-08:
Freetz uses 3 different sources for all avms IPQ40xx devices sources from 1200,4040,7530. All have FOS 7.25+, for 1200 source was just released and 7530 is yet! So for devices which use 4040 src modules should be disabled. But its a useless work to keep track of this release management of avm. And sometime older sources work (more or less) with older sources.

Simple solution: releasing sources together with a new fos version. Imho avm will never do as i think they do pack them manually, there are often things missing or not like in the release firmware

What you can do: Switch to openwrt https://openwrt.org/toh/avm/avm_fritz_box_7530
They give a shit on avm closed-source blobs and sources, so some features like dect or dsl may not work. But you have a kernel which is not 10 years old and could do multi-ssid, vlans, ful iptables with conntrack, sqm (bufferbloat)
If you like openwrt sell your avm stuff, there are devices which are fast and cost less than 50%, eg E8450 has like 7590 500mb ram and 128mb flash, but is sold for 90euro vs 250euro.
If you have no smartphone you may need an voip-phone additionally. But as AVM does not support voip with telekom hybrid with FOS 7.25+ anymore (broken diffserv support) its mandatory to use another voip server than fritzstuff

edeso, 2021-07-13:
quick update. no sources for the 7530 7.27 yet but
"
Sie haben vor einigen Tagen eine Anfrage an den AVM Support gestellt und wir hoffen, dass wir Ihnen mit unseren Informationen weiterhelfen konnten.

Da wir unsere Supportqualität stetig verbessern wollen, wären wir Ihnen für eine kurze Bewertung unseres Services dankbar. Klicken Sie dazu einfach auf einen der beiden Links:
"
from AVM Support <open-source@avm.zendesk.com>

the audacity! they really seem to package only if requested

wrt. OpenWrt. yes big fan, but used Fritzboxes for half or less the new price are pretty attractive. user interface and out of the box has neat DSL debugging and telephony system functions i'd sorely miss in OpenWrt. Not so on the 4040 of course :)

edeso, 2021-07-15:
seems to be online since yesterday.
https://osp.avm.de/fritzbox/fritzbox-7530/

in case even more sources are missing pls advise me. i'am happy to request them from AVM one by one :)

fda77, 2021-07-15:
AVM's user unfriendly release management sucks, like their buggy fritzos, pseudo-mesh and so on. As im moving to openwrt i dont need any sources anymore.
If you want to update the sourcen from 7.25 to 7.27, search for "delta" patches in make/linux/patches and do it this way

edeso, 2021-07-15:
does that mean Freetz-NG is going to die?

fda77, 2021-07-15:
No, its up to you ;-)

edeso, 2021-07-15:
then it probably is ;(.. happily contribute when stumbling over small issues. but that's all i'm capable of especially because of time restraints. anyway, thanks for your efforts, really appreciated!