Freetz-NG/freetz-ng #346: src für 3000 und 6000 und 3490 und 5491 und 7430

freetz-ng-mod, 2021-07-08:
3000
http://osp.avm.de/fritzwlan/fritzrepeater-3000/source-files-FRITZ.Repeater_3000-07.27.tar.gz
6000
http://osp.avm.de/fritzwlan/fritzrepeater-3000/source-files-FRITZ.Repeater_6000-qcaarmv8-07.28.tar.gz
http://osp.avm.de/fritzwlan/fritzrepeater-6000/source-files-FRITZ.Repeater_6000-07.28.tar.gz
3490
http://osp.avm.de/fritzbox/fritzbox-3490/source-files-FRITZ.Box_3490-scrpn-07.28.tar.gz
http://osp.avm.de/fritzbox/fritzbox-3490/source-files-FRITZ.Box_3490-vr9-07.28.tar.gz
5491
http://osp.avm.de/fritzbox/fritzbox-5491/source-files-FRITZ.Box_5491-vr9-07.27.tar.gz%20.gz
http://osp.avm.de/fritzbox/fritzbox-5491/source-files-FRITZ.Box_5491-scrpn-07.27.tar.gz
7430
http://osp.avm.de/fritzbox/fritzbox-7430/source-files-FRITZ.Box_7430-07.27.tar.gz

fda77, 2021-07-08:
Wird für 3000 nicht der src von 1200 verwendet?
Wird für 3490+7430 nicht der src von 7490 verwendet?
Das müsst im Umfeld von https://github.com/Freetz-NG/freetz-ng/blob/master/config/avm/source.in zu erkennen sein, bzw unter config/ allgemein

Von 6000 gibt es die 7.27 schon, da müsst nur jemand ein "diff" von 7.27 nach 7.28 machen, ähnlich wie https://github.com/Freetz-NG/freetz-ng/blob/master/make/linux/patches/3.10.73/7490_06.83/000-7490.06.83-7360.06.86-delta.patch Vielleicht gibt es auch gar keine Änderungen am kernel

freetz-ng-mod, 2021-07-29:
Du hast recht, Aber nicht bei allen.

fda77, 2021-07-31:
Iss möglich

fda77, 2021-08-19:
Es gibt dazu eine (die einzigste) angepinnte Diskussion

closed 2021-08-19 by fda77