Freetz-NG/freetz-ng #349: Start / Stop des OpenVPN service beim booten und über Freetz GUI funktioniert nicht richtig

stefanpuch, 2021-07-13:
Firmware: 113.07.27 rev885
Freetz: ng-18427-d1f4c28d1

Ich nutze in o.g. build OpenVPN 2.5.3 und stelle heute fest, dass eine Verbindung nicht möglich war. Einen Fehler in der config konnte ich schnell ausschließen, da lange Zeit etabliert. Ein Blick auf die console zeigt (so weit so gut) folgenden laufenden Prozess
```
root@vdslbox:/var/run# ps | grep openvpn 
17613 root      2628 S    openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon
17924 root      1832 S    grep openvpn
root@vdslbox:/var/run#
root@vdslbox:/var/run# cat /var/run/openvpn.pid
17613
root@vdslbox:/var/run#
```
Ein Verbindung per Client von "außen" (DSL Interface) ist nicht möglich. Ein stoppen des Dienstes über die Freetz Web GUI löscht zwar das PID file, der openvpn service läuft aber weiter. Ein restart startet folglich eine zweite Instanz.

Nach einem manuellen Start von der console (**ohne den Parameter --daemon**)
```
openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid
```
ist eine Verbindung per Client von "außen" (DSL Interface) ohne Probleme möglich. Ein strg + z bg (Prozess in den Hintergrund) funktioniert ebenfalls. Auffällig ist, dass der Prozess dann nicht als root sondern als user openvpn läuft
```
root@vdslbox:/var/run# ps | grep openvpn
18316 openvpn   2880 S    openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid
18330 root      1832 S    grep openvpn
root@vdslbox:/var/run#
```
In diesem Zustand funktioniert auch das Stoppen des Dienstes über die Freetz Web GUI. dabei wird sowohl das PID file als auch der Prozess (wie erwartet) beendet. 
```
root@vdslbox:/var/run# 2021-07-13 22:34:05 event_wait : Interrupted system call (code=4)
2021-07-13 22:34:05 Closing TUN/TAP interface
2021-07-13 22:34:05 net_addr_v4_del: 10.0.139.1 dev tun0
2021-07-13 22:34:05 sitnl_send: rtnl: generic error (-1): Operation not permitted
2021-07-13 22:34:05 Linux can't del IP from iface tun0
2021-07-13 22:34:05 SIGTERM[hard,] received, process exiting

[1]+  Done                       openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid
root@vdslbox:/var/run# cat /var/run/openvpn.pid
cat: can't open '/var/run/openvpn.pid': No such file or directory
root@vdslbox:/var/run#
```

Bei einem manuellen Start von der console **mit dem Parameter --daemon** ist wiederum keine Verbindung per Client von extern möglich, das Verhalten ist wie beim Bedienen über das Freetz Web GUI. Ob das beobachtete Verhalten an den rechten des ausführenden users liegt kann ich nicht beurteilen. Hat jemand eine Idee?

fda77, 2021-07-15:
Das log von einem fehlerhaften Start wäre besser. Je nachdem welche cgi du verwendest aktiviert man das anders.
Vielleicht musst du die "etablierte" config anpassen, weil sich bei openvpn was geändert hat?
Eine Verschlüsselung die 2000 top war ist heute evtl nicht mehr erlaubt weil sie unsicher ist. Das könnte auch mit openssl oder was auch immer du für eine ssl lib verwendest zusammenhängen.
Es ist jedenfalls kein grundsätzliches Problem, sonst gäbe es mehr Meldungen

stefanpuch, 2021-07-19:
Ich gehe davon aus, dass es die "Art und Weise" ist, wie openvpn beim Booten bzw. über das cgi (v1) gestartet wird. Wäre es ein Problem der "etablierten" config bzw. openssl dann würde der Dienst nicht beim Start per Kommandozeile ohne Probleme laufen und zwischen openvpn 2.5.2 und 2.5.3 gab es keine signifikanten Änderungen.

> Das log von einem fehlerhaften Start wäre besser.

Wird der Systemstart irgendwo direkt mitgeloggt? Wie gesagt, mit dem Parameter --daemon von der Kommandozeile kann ich es reproduzieren, aber das wird vermutlich nur der Hauptgrund sein. Wie kann ich einen Prozess im Kontext eines anderen users starten?

fda77, 2021-07-31:
Man kann ein Script mit "sh -x /etc/init.d/rc......" öffnen um die Befehle zu sehen. Es könnte am Environment liegen. Da bei anderen dieser Fehler aber nicht auftritt tippe ich eher auf die Config von Openvpn

stefanpuch, 2021-08-23:
So, urlaubsbedingt konnte ich nicht antworten. Die Config von OpenVPN kann ich ausschließen, denn dank DualBoot Manager habe ich am Wochenende die 07.12 (7490_07.12-freetz-master-20200626-26a753c89.de_20200627-000605) nochmals gebootet und auf der tritt das Problem nicht auf. Anschließend habe ich mit der neusten AVM FW und dem aktuellen Entwicklungszweig einen neuen Build erstellt (7490_07.28.all_freetz-ng-18511-614a813ac_20210820-184902) welcher wie zuvor berichtet das Problem aufweist. Ich kann bei Bedarf gerne noch die configs zur Reproduktion anhängen. So wie ich Zeit finde, kann ich versuchen das Problem weiter einzugrenzen, allerdings fehlt mir noch eine konkrete Idee.
Die OpenVPN Config würde ich ausschließen, sonst würde sie auch nicht 1:1 beim manuellen Aufruf auf der Console funktionieren.
[config.txt](https://github.com/Freetz-NG/freetz-ng/files/7031076/config.txt)

fda77, 2021-08-24:
Openvpn ist eines der Packages das vermutlich öfter genutzt wird. Da daran letzte Zeit nicht viel herumgebastelt wurde müsste es den Fehler schon länger geben - oder es liegt an deiner Konfiguration. Auch möglich wäre eine Änderung durch eine neue Openvpn-Version.
Jedenfalls gut zu wissen dass du den v1-Gui zum konfigurieren verwendest.
Allein durch wieder öffnen wird sich der Fehler allerdings nicht beheben

stefanpuch, 2021-12-29:
Ich habe endlich ein wenig Zeit gefunden, mir den Fehler nochmal mit aktuellem Build `7490_07.29.all_freetz-ng-18776-c715ed811_20211223-185105.image`anzusehen, komme aber nicht recht weiter. Der Aufruf des Startscripts von der Shell sieht soweit unauffällig aus.
```
root@fritzbox:/var/mod/root# sh -x /etc/init.d/rc.openvpn start
+ scriptname=/etc/init.d/rc.openvpn
+ DAEMON=openvpn
+ . /etc/init.d/modlibrc
+ . /var/env.mod.daemon
+ export 'PATH=/mod/sbin:/mod/bin:/mod/usr/sbin:/mod/usr/bin:/mod/etc/init.d:/sbin:/bin:/usr/sbin:/usr/bin'
+ export 'LD_LIBRARY_PATH=/mod/lib:/mod/usr/lib'
+ export 'LANG=en_US.UTF-8'
+ . /etc/init.d/loglibrc
+ : openvpn
+ : openvpn
+ : openvpn_ENABLED
+ : /mod/etc/openvpn.conf
+ : openvpn
+ : /var/run/openvpn.pid
+ : /tmp/flash/openvpn/openvpn.cfg
+ '[' -n openvpn ]
+ modlib_loadconfig
+ local 'CONF_FILE=/mod/etc/conf/openvpn.cfg'
+ '[' -r /mod/etc/conf/openvpn.cfg ]
+ . /mod/etc/conf/openvpn.cfg
+ export 'OPENVPN_ADDITIONAL=#push "register-dns" ; proto udp4 ; auth SHA512 ;'
+ export 'OPENVPN_AUTH_TYPE=static#certs'
+ export 'OPENVPN_AUTOSTART=#yes'
+ export 'OPENVPN_BOX_IP=#10.0.239.1'
+ export 'OPENVPN_BOX_MASK=255.255.255.0#255.255.255.0'
+ export 'OPENVPN_CIPHER=BF-CBC#AES-256-GCM'
+ export 'OPENVPN_CLIENT2CLIENT=#yes'
+ export 'OPENVPN_CLIENTS_DEFINED=#0'
+ export 'OPENVPN_CLIENT_INFO=#yes'
+ export 'OPENVPN_CLIENT_IPS=#'
+ export 'OPENVPN_CLIENT_MASKS=#'
+ export 'OPENVPN_CLIENT_NAMES=#'
+ export 'OPENVPN_CLIENT_NETS=#'
+ export 'OPENVPN_COMPLZO=yes#'
+ export 'OPENVPN_CONFIG_CHANGED=new#yes'
+ export 'OPENVPN_CONFIG_COUNT=1'
+ export 'OPENVPN_CONFIG_NAMES=DEFAULT#'
+ export 'OPENVPN_DEBUG=#'
+ export 'OPENVPN_DEBUG_TIME=10#10'
+ export 'OPENVPN_DHCP_CLIENT=#'
+ export 'OPENVPN_DHCP_RANGE=#10.0.239.2 10.0.239.11'
+ export 'OPENVPN_ENABLED=yes'
+ export 'OPENVPN_EXPERT=yes'
+ export 'OPENVPN_FILES2CP=#'
+ export 'OPENVPN_FLOAT=#'
+ export 'OPENVPN_IPV6=#'
+ export 'OPENVPN_KEEPALIVE=yes#yes'
+ export 'OPENVPN_KEEPALIVE_PING=10#10'
+ export 'OPENVPN_KEEPALIVE_TIMEOUT=120#120'
+ export 'OPENVPN_LOADTUN=yes'
+ export 'OPENVPN_LOCAL=#'
+ export 'OPENVPN_LOCAL_NET=#192.168.123.0 255.255.255.0'
+ export 'OPENVPN_LOGFILE=#yes'
+ export 'OPENVPN_MAXCLIENTS=1#10'
+ export 'OPENVPN_MGMNT=#'
+ export 'OPENVPN_MODE=server#server'
+ export 'OPENVPN_MTU=1500#'
+ export 'OPENVPN_NO_CERTTYPE=#'
+ export 'OPENVPN_OWN_KEYS=#'
+ export 'OPENVPN_PARAM_1=#'
+ export 'OPENVPN_PARAM_2=#'
+ export 'OPENVPN_PARAM_3=#'
+ export 'OPENVPN_PORT=#1193'
+ export 'OPENVPN_PROTO=udp#udp'
+ export 'OPENVPN_PULL=#'
+ export 'OPENVPN_PUSH_DNS=#192.168.123.2'
+ export 'OPENVPN_PUSH_DOMAIN=#home.local'
[... 209 lines cut ...]
2021-12-29 11:49:23 Socket Buffers: R=[229376->229376] S=[229376->229376]
2021-12-29 11:49:23 UDPv4 link local (bound): [AF_INET][undef]:1193
2021-12-29 11:49:23 UDPv4 link remote: [AF_UNSPEC]
2021-12-29 11:49:23 chroot to '/var/tmp/openvpn' and cd to '/' succeeded
2021-12-29 11:49:23 GID set to openvpn
2021-12-29 11:49:23 UID set to openvpn
2021-12-29 11:49:23 MULTI: multi_init called, r=256 v=256
2021-12-29 11:49:23 IFCONFIG POOL IPv4: base=10.0.239.2 size=10
2021-12-29 11:49:23 Initialization Sequence Completed
``` 
Ein `strg+z bg` mit anschließendem `ifconfig | grep tun` 
```
root@fritzbox:/var/mod/root# ifconfig | grep tun
tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
root@fritzbox:/var/mod/root#
```
zeigt, dass ein tun Interface geöffnet wurde. Der Dienst selbst läuft nach manuellen Start allerdings nicht als `root user` sondern als `openvpn user`
```
root@fritzbox:/var/mod/root# ps | grep openvpn
11483 openvpn   2736 S    openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid
11492 root      1832 S    grep openvpn
root@fritzbox:/var/mod/root#
```

@fda77 Wie kann ich das Problem weiter eingrenzen? Da bei Manuallem Start die Nutzung von OpenVPN ohne Probleme funktioniert, kann es aus meiner Sicht nicht an der OpenVPN Config liegen. Allerdings schreibst Du dass es bei Dir ohne Probleme funktioniert. Das Script zum Starten und Stoppen ` /etc/init.d/rc.openvpn` sollte ja unabhängig von der Gui (v1 bzw. v2) sein, daher frage ich mich, an welcher Stelle ich noch ansetzen kann? Ich habe jetzt verschiedene Entwicklungsstände getestet und kann nur nachvollziehen, dass es zuletzt mit einer 7.12 Firmware funktioniert hat. Beim Wechsel von 7.12 auf 7.2x gabe es je zahlreiche Änderungen wenn ich mich recht erinnere, können da in irgend einer Form noch "Altlasten" im Spiel sein? Ich wäre für jeden Ratschlag dankbar.

fda77, 2021-12-29:
> /etc/init.d/rc.openvpn sollte ja unabhängig von der Gui (v1 bzw. v2) sein,

Ne: `find make/openvpn*/ ! -type d`
Am besten würdest du das auch in den Überschrift dazumachem

Ich hab VPN auf dem Raspberry laufen, der ist schneller. Openvpn sollte aber eins der meistgenutzten Pakete sein, es gäbe bei "allgemeinen" Fehlern mehr Meldungen

> dass es zuletzt mit einer 7.12 Firmware funktioniert hat

Mit aktuellem git stand gebaut oder einem uralten? Es hat sich zb auch die openvpn Version geändert, vielleicht ein Bug, oder Änderung an den Optionen. Openssl ist mittlerweile auch neuer. Es kann auch sein dass es durch interne Änderungen von AVM nicht mehr geht. Bei einem DSL-Reconnect hat der dsld oder multid auch gern mal manuell gesetzte IPv6 gelöscht.

> Allerdings scheitert das Killen des Daemon mittels kill Befehl aus dem script. Der Prozess lebt weiter (obwohl die PID richtig ist), das pidfile wird gelöscht.

Schau was der Prozess so macht und wo er hängen bleibt, am besten im 2. Terminal vor dem "kill": `strace -fff -p DIE-PID`

> Beim Staten direkt von der Shell funktioniert es einwandfrei, wenn ich den Parameter --daemon weglasse:

Und wenn du ihn nicht weglässt? Wenn er auch hängt gleich so aufrufen: `strace -fff BEFEHL`

Im Script hat der daemon ein anderes Environment, durch das `env -` vornedran. Vielleicht fehlt da was

Ich hab bislang soweit möglich die Finger von den openvpn scirpten+cgis gelassen, die sind wesentlich komplizierter als die anderen

stefanpuch, 2021-12-29:
OK, hier hören meine Kenntnisse dann leider auf. Wenn Dir da nicht direkt was ins Auge springt, muss ich wohl mal ne aktuelle Version mit Gui v2 bauen und schauen, wie sich das verhält. Bisher ist alles immer auf die Gui v1 bezogen.

Ich habe im ersten Terminal per `sh -x /etc/init.d/rc.openvpn start` den daemon gestartet und im 2. Terminal ein strace attached. Dann habe ich zunächst mittels `sh -x /etc/init.d/rc.openvpn stop` versucht den daemon zu töten, dann ein zweites mal mit `kill 13258` und beim 3. mal mit `kill -9 13258` . Der Output dazu ist hier: 

```
root@fritzbox:/var/mod/root# strace -fff -p 13258
strace: Process 13258 attached
futex(0x6850d4, FUTEX_WAIT_PRIVATE, 0, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGTERM {si_signo=SIGTERM, si_code=SI_USER, si_pid=13270, si_uid=0} ---
rt_sigaction(SIGTERM, {sa_handler=0x452120, sa_mask=[TERM], sa_flags=SA_RESTART}, {sa_handler=0x452120, sa_mask=[TERM], sa_flags=SA_RESTART}, 16) = 0
sigreturn({mask=[]})                    = 4238
futex(0x6850d4, FUTEX_WAIT_PRIVATE, 0, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGTERM {si_signo=SIGTERM, si_code=SI_USER, si_pid=10334, si_uid=0} ---
rt_sigaction(SIGTERM, {sa_handler=0x452120, sa_mask=[TERM], sa_flags=SA_RESTART}, {sa_handler=0x452120, sa_mask=[TERM], sa_flags=SA_RESTART}, 16) = 0
sigreturn({mask=[]})                    = 4238
futex(0x6850d4, FUTEX_WAIT_PRIVATE, 0, NULL) = ?
+++ killed by SIGKILL +++
root@fritzbox:/var/mod/root#
```
Der Start mittels `strace -fff openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon` 
```
root@fritzbox:/var/mod/root# strace -fff openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon
execve("/mod/sbin/openvpn", ["openvpn", "--config", "/mod/etc/openvpn.conf", "--writepid", "/var/run/openvpn.pid", "--daemon"], 0x7ff411e8 /* 297 vars */) = 0
brk(NULL)                               = 0x683000
brk(0x683c7c)                           = 0x683c7c
set_thread_area(0x68a460)               = 0
brk(0x684908)                           = 0x684908
set_thread_area(0x68b0e0)               = 0
open("/dev/urandom", O_RDONLY)          = 3
read(3, "-k\310\342", 4)                = 4
close(3)                                = 0
ioctl(0, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
gettimeofday({tv_sec=1640805871, tv_usec=50042}, NULL) = 0
time(NULL)                              = 1640805871 (2021-12-29T20:24:31+0100)
brk(0x685908)                           = 0x685908
brk(0x686000)                           = 0x686000
futex(0x67c72c, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c720, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c718, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c634, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c6d8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c464, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c61c, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c710, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c6d0, FUTEX_WAKE_PRIVATE, 2147483647) = 0
getuid()                                = 0
geteuid()                               = 0
getgid()                                = 0
getegid()                               = 0
open("/mod/etc/ssl/openssl.cnf", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
futex(0x67c6f8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
rt_sigaction(SIGINT, {sa_handler=0x452120, sa_mask=[INT], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 16) = 0
rt_sigaction(SIGTERM, {sa_handler=0x452120, sa_mask=[TERM], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 16) = 0
rt_sigaction(SIGHUP, {sa_handler=SIG_IGN, sa_mask=[HUP], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 16) = 0
rt_sigaction(SIGUSR1, {sa_handler=SIG_IGN, sa_mask=[USR1], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 16) = 0
rt_sigaction(SIGUSR2, {sa_handler=SIG_IGN, sa_mask=[USR2], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 16) = 0
rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 16) = 0
open("/mod/etc/openvpn.conf", O_RDONLY|O_LARGEFILE) = 3
ioctl(3, TCGETS, 0x7fd39dcc)            = -1 ENOTTY (Inappropriate ioctl for device)
read(3, "# OpenVPN 2.5.5 mips-unknown-lin"..., 4096) = 1002
brk(0x687000)                           = 0x687000
open("/etc/gai.conf", O_RDONLY)         = -1 ENOENT (No such file or directory)
chdir("/var/tmp/openvpn")               = 0
brk(0x688000)                           = 0x688000
read(3, "", 4096)                       = 0
close(3)                                = 0
futex(0x67c798, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0x67c708, FUTEX_WAKE_PRIVATE, 2147483647) = 0
access("/tmp/flash/openvpn/dh.pem", R_OK) = 0
[... 398 lines cut ...]
brk(0x693000)                           = 0x693000
brk(0x694000)                           = 0x694000
brk(0x695000)                           = 0x695000
brk(0x696000)                           = 0x696000
brk(0x697000)                           = 0x697000
brk(0x698000)                           = 0x698000
gettimeofday({tv_sec=1640806529, tv_usec=57882}, NULL) = 0
write(1, "2021-12-29 20:35:29 Initializati"..., 542021-12-29 20:35:29 Initialization Sequence Completed
) = 54
_newselect(7, [5 6], [], NULL, {tv_sec=10, tv_usec=0}) = 0 (Timeout)
[... cut ...]
gettimeofday({tv_sec=1640806539, tv_usec=68363}, NULL) = 0
write(3, "Updated,2021-12-29 20:35:39\n", 28) = 28
write(3, "Common Name,Real Address,Bytes R"..., 67) = 67
write(3, "ROUTING TABLE\n", 14)         = 14
write(3, "Virtual Address,Common Name,Real"..., 50) = 50
write(3, "GLOBAL STATS\n", 13)          = 13
write(3, "Max bcast/mcast queue length,0\n", 31) = 31
write(3, "END\n", 4)                    = 4
_llseek(3, 0, [227], SEEK_CUR)          = 0
ftruncate64(3, 227)                     = 0
_newselect(7, [5 6], [], NULL, {tv_sec=10, tv_usec=0}
```

fda77, 2021-12-29:
Ich hab davon auch keine Ahnung, ich lese es einfach und versuche mir etwas zusammenzureimen... Mit Glück schaut @PeterPawn vorbei

Warum die "SIGTERM" nicht funktionieren versteh ich nicht. So sollte es wohl aussehen: https://gist.github.com/FruitieX/594f64e8736edfde5a2e Ob Openvpn das kill verhindert? Vielleicht gibt es dazu eine Option

Das "tun0" hat hier wohl geklappt:
```
open("/dev/net/tun", O_RDWR|O_LARGEFILE) = 5
ioctl(5, TUNSETIFF, 0x7ff91f48)         = 0
gettimeofday({tv_sec=1640806528, tv_usec=796738}, NULL) = 0
write(1, "2021-12-29 20:35:28 TUN/TAP devi"..., 472021-12-29 20:35:28 TUN/TAP device tun0 opened
ioctl(6, SIOCGIFINDEX, {ifr_name="tun0", ifr_ifindex=34}) = 0
write(1, "2021-12-29 20:35:28 net_iface_up"..., 462021-12-29 20:35:28 net_iface_up: set tun0 up
```

Es läuft nicht als root, da du wohl chroot aktiviert hast und dies zu "openvpn" wechselt
```
chroot("/var/tmp/openvpn")              = 0
chdir("/")                              = 0
gettimeofday({tv_sec=1640806529, tv_usec=19100}, NULL) = 0
write(1, "2021-12-29 20:35:29 chroot to '/"..., 732021-12-29 20:35:29 chroot to '/var/tmp/openvpn' and cd to '/' succeeded
) = 73
setgid(1000)                            = 0
gettimeofday({tv_sec=1640806529, tv_usec=24427}, NULL) = 0
write(1, "2021-12-29 20:35:29 GID set to o"..., 392021-12-29 20:35:29 GID set to openvpn
) = 39
setgroups(1, [1000])                    = 0
setuid(100)                             = 0
gettimeofday({tv_sec=1640806529, tv_usec=28062}, NULL) = 0
write(1, "2021-12-29 20:35:29 UID set to o"..., 392021-12-29 20:35:29 UID set to openvpn
```

UID: Dazu fällt mir https://freetz-ng.github.io/freetz-ng/NEWS.html#2021-04-10 ein
Wieso hat dein Benutzer die UID "100"? Auf meiner 7490 hat diese ID der Benutzer "smb", ich glaube der wurde von AVM so fest erstellt. Von Freetz erstellte beginnen ab 900 (oder früher 1000)
```
open("/etc/passwd", O_RDONLY)           = 6
ioctl(6, TCGETS, 0x7ff91ea4)            = -1 ENOTTY (Inappropriate ioctl for device)
read(6, "openvpn:x:100:1000:openvpn:/home"..., 4096) = 1411
```

Zwischen den letzten beiden straces gibt es den Unterschied
```
open("/var/log/openvpn.log", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0600) = 4   == scheinbar gut
open("/var/log/openvpn.log", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0600) = 3   == scheinbar schlecht
```
Das könnte bedeuten dass es Probleme mit den Zugriffsrechten gibt, evtl im Zusammenhang mit der UID oben.
Ist in der Date etwas drin? Von beiden Versuchen?

stefanpuch, 2021-12-29:
Das mit der UID ist ein guter Punkt, keine Ahnung warum die auf 100 gesetzt ist. Habe das mal auf 900.900 gesetzt (ist ja wohl jetzt der default Startwert für freetz und weitere user habe ich nicht.
Habe die Box zur Sicherheit mal neu gestartet, hat aber vom Verhalten her keinen Unterschied gemacht.

In `/var/log/openvpn.log`  legt openvpn die verbundenen Clients ab, wenn es welche gibt.

fda77, 2021-12-29:
Die ".900" ist die Gruppe. Gibt es ein mit GID 900? Falls user&gruppe fehlen werden die beim booten erstellt
Also ein openvpn.log ist leer?? War es nicht so dass verbundene in einer "status.log" sind? Und die logs in der openvpn.log? Die Berechtigungen von datein+verzeichnisen werden als UID GID gespeichert. Schau dass die alle zu neuen ID passen. Also logs, chroot-Verzeichis usw. Oder mach chroot mal aus (falls das geht)

stefanpuch, 2021-12-30:
Zu dem strace Unterschied:
> Zwischen den letzten beiden straces gibt es den Unterschied
```
open("/var/log/openvpn.log", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0600) = 4   == scheinbar gut
open("/var/log/openvpn.log", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0600) = 3   == scheinbar schlecht
```
Nach der Manpage gibt die Funktion open einen nicht negativen Wert zurück
```
RETURN VALUE 

       On success, open(), openat(), and creat() return the new file
       descriptor (a nonnegative integer).  On error, -1 is returned and
       errno is set to indicate the error.
```

[Quelle](https://man7.org/linux/man-pages/man2/open.2.html#RETURN_VALUE) 

--------------------------------------

Habe mir daher das Ganze nochmal mit debugging aus Sicht des openvpn Daemon angesehen und dazu in der config `verb 5` ergänzt, so das der Daemon dann nach `/var/tmp/debug_openvpn.out` das Log schreibt. 
Beide Varianten direkt von der shell gestartet, zunächst (wie im init script mit --daemon) danach ohne. 

Version1:  `openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon` 

``` 
root@fritzbox:/var/mod/root# cat /var/tmp/debug_openvpn.out
2021-12-30 18:14:41 us=156513 Current Parameter Settings:
2021-12-30 18:14:41 us=156814   config = '/mod/etc/openvpn.conf'
2021-12-30 18:14:41 us=156992   mode = 1
2021-12-30 18:14:41 us=157137   persist_config = DISABLED
2021-12-30 18:14:41 us=157269   persist_mode = 1
2021-12-30 18:14:41 us=157390   show_ciphers = DISABLED
2021-12-30 18:14:41 us=157514   show_digests = DISABLED
2021-12-30 18:14:41 us=157635   show_engines = DISABLED
2021-12-30 18:14:41 us=157759   genkey = DISABLED
2021-12-30 18:14:41 us=157881   genkey_filename = '[UNDEF]'
2021-12-30 18:14:41 us=158003   key_pass_file = '[UNDEF]'
2021-12-30 18:14:41 us=158123   show_tls_ciphers = DISABLED
2021-12-30 18:14:41 us=158245   connect_retry_max = 0
2021-12-30 18:14:41 us=158366 Connection profiles [0]:
2021-12-30 18:14:41 us=158489   proto = udp4
2021-12-30 18:14:41 us=158612   local = '[UNDEF]'
2021-12-30 18:14:41 us=158735   local_port = '1193'
2021-12-30 18:14:41 us=158859   remote = '[UNDEF]'
2021-12-30 18:14:41 us=158980   remote_port = '1193'
2021-12-30 18:14:41 us=159102   remote_float = DISABLED
2021-12-30 18:14:41 us=159226   bind_defined = DISABLED
2021-12-30 18:14:41 us=159348   bind_local = ENABLED
2021-12-30 18:14:41 us=159469   bind_ipv6_only = DISABLED
2021-12-30 18:14:41 us=159594   connect_retry_seconds = 5
2021-12-30 18:14:41 us=159716   connect_timeout = 120
2021-12-30 18:14:41 us=159839   socks_proxy_server = '[UNDEF]'
2021-12-30 18:14:41 us=159960   socks_proxy_port = '[UNDEF]'
2021-12-30 18:14:41 us=160083   tun_mtu = 1500
2021-12-30 18:14:41 us=160205   tun_mtu_defined = ENABLED
2021-12-30 18:14:41 us=160330   link_mtu = 1500
2021-12-30 18:14:41 us=160450   link_mtu_defined = DISABLED
2021-12-30 18:14:41 us=160622   tun_mtu_extra = 0
2021-12-30 18:14:41 us=160751   tun_mtu_extra_defined = DISABLED
2021-12-30 18:14:41 us=160875   mtu_discover_type = -1
2021-12-30 18:14:41 us=160996   fragment = 0
2021-12-30 18:14:41 us=161115   mssfix = 1450
2021-12-30 18:14:41 us=161238   explicit_exit_notification = 0
2021-12-30 18:14:41 us=161360   tls_auth_file = '[UNDEF]'
2021-12-30 18:14:41 us=161484   key_direction = not set
2021-12-30 18:14:41 us=161608   tls_crypt_file = '[UNDEF]'
2021-12-30 18:14:41 us=161735   tls_crypt_v2_file = '[UNDEF]'
2021-12-30 18:14:41 us=161873 Connection profiles END
2021-12-30 18:14:41 us=162010   remote_random = DISABLED
2021-12-30 18:14:41 us=162133   ipchange = '[UNDEF]'
[... 447 lines cut ...]
2021-12-30 18:09:21 us=205972 net_addr_v4_add: 10.0.239.1/24 dev tun0
2021-12-30 18:09:21 us=214571 Data Channel MTU parms [ L:1621 D:1450 EF:121 EB:406 ET:0 EL:3 ]
2021-12-30 18:09:21 us=215501 Socket Buffers: R=[229376->229376] S=[229376->229376]
2021-12-30 18:09:21 us=215854 UDPv4 link local (bound): [AF_INET][undef]:1193
2021-12-30 18:09:21 us=216141 UDPv4 link remote: [AF_UNSPEC]
2021-12-30 18:09:21 us=216501 chroot to '/var/tmp/openvpn' and cd to '/' succeeded
2021-12-30 18:09:21 us=216776 GID set to openvpn
2021-12-30 18:09:21 us=217103 UID set to openvpn
2021-12-30 18:09:21 us=217403 MULTI: multi_init called, r=256 v=256
2021-12-30 18:09:21 us=217975 IFCONFIG POOL IPv4: base=10.0.239.2 size=10
2021-12-30 18:09:21 us=218769 Initialization Sequence Completed
2021-12-30 18:09:55 us=497430 event_wait : Interrupted system call (code=4)
2021-12-30 18:09:55 us=498643 TCP/UDP: Closing socket
2021-12-30 18:09:55 us=499307 Closing TUN/TAP interface
2021-12-30 18:09:55 us=499707 net_addr_v4_del: 10.0.239.1 dev tun0
2021-12-30 18:09:55 us=500424 sitnl_send: rtnl: generic error (-1): Operation not permitted
2021-12-30 18:09:55 us=500867 Linux can't del IP from iface tun0
2021-12-30 18:09:55 us=544203 SIGTERM[hard,] received, process exiting
``` 

Beim direkten Vergleich fällt auf, dass die Variante mit Parameter `--daemon` ungefähr da stehenbleibt, wo die Variante ohne den Parameter die Dateien `dh.pem` und `crl.pem` verarbeiten. Das change root `chroot` kommt erst später, somit sollte das eigentlich nicht das Problem sein. Der Aufruf von der shell hat in beiden Varianten das gleiche environment, es sei denn, dass mittels Parameter `--daemon` implizit ein anderes genutzt wird.

Das einzige was mit auffällt ist, dass im Logfile beim Zugriff auf die crl.pem ein zusätzlicher `/` im Pfad enthalten ist. Da beide Aufrufe mit `strace` (siehe oben) aber `access("/var/tmp/openvpn//etc/crl.pem", R_OK) = 0` zeigen würde ich das als unwahrscheinlich erachten.

Mehr Ideen habe ich derzeit nicht. @PeterPawn müsste man vermutlich mal antriggern, denn in ein geschlossenes Ticket wird er vermutich nicht von selbst schauen oder unterschätze ich ihn da?

fda77, 2021-12-30:
Ich glaube er schaut gar nicht in die Issues hier sondern bekommt nur eine Meldung wenn man ihn ruft. So ähnlich wie beim Candyman  :) 
In der Überschrift steht immer noch nicht um welches cgi es sich handelt!

Hast du den daemon-modus schon ohne chroot probiert, falls es geht?
In freetz/ gibt es kein openssl 1.1, vielleicht wird eine openssl lib nicht richtig mitkopiert da das openvpn-script nicht angepasst wurde und aus dem chroot ist kein Zugriff darauf. Oder die Zertifkate sind nicht lesbar durch zuwenig Rechte

stefanpuch, 2021-12-30:
So, habe noch ein wenig rumprobiert. Habe das `chroot` mal rausgenommen und auch die Parameter welche die `user` und `group` auf openvpn setzen. Hat alles keinen Unterschied gemacht. Anschließend habe ich es mit der "Kopfstandmethode" versucht und mir das ganze einmal rückwärts angesehen, da ja auch das `kill` nicht funktioniert.

Mit `strace -fff -p DIE-PID`  am openvpnd und mit einem 2. Terminal mal `kill` mit unterschiedlichen Parametern gesendet. Nach der [OpenVPN Doku](https://openvpn.net/community-resources/controlling-a-running-openvpn-process/) soll sich der Daemon beenden, wenn er ein `SIGTERM` oder `SIGINT` bekommt.

```
root@fritzbox:/etc/init.d# strace -fff -p 12837
strace: Process 12837 attached
futex(0x6850c4, FUTEX_WAIT_PRIVATE, 0, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGTERM {si_signo=SIGTERM, si_code=SI_USER, si_pid=10274, si_uid=0} ---
rt_sigaction(SIGTERM, {sa_handler=0x452120, sa_mask=[TERM], sa_flags=SA_RESTART}, {sa_handler=0x452120, sa_mask=[TERM], sa_flags=SA_RESTART}, 16) = 0
sigreturn({mask=[RTMIN RT_1]})          = 4238
futex(0x6850c4, FUTEX_WAIT_PRIVATE, 0, NULL) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGINT {si_signo=SIGINT, si_code=SI_USER, si_pid=10274, si_uid=0} ---
rt_sigaction(SIGINT, {sa_handler=0x452120, sa_mask=[INT], sa_flags=SA_RESTART}, {sa_handler=0x452120, sa_mask=[INT], sa_flags=SA_RESTART}, 16) = 0
sigreturn({mask=[RTMIN RT_1]})          = 4238
futex(0x6850c4, FUTEX_WAIT_PRIVATE, 0, NULL
```
Dabei fällt das `To be restarted if SA_RESTART is set` auf, dass welches offenbar der Fall ist und den Prozess dazu bringt, "weiter zu laufen".  
`SA_RESTART (SVR4, BSD) – Wird ein Systemaufruf (Systemcall) durch dieses Signal unterbrochen, wird dieser automatisch neu gestartet.`  [Quelle](https://openbook.rheinwerk-verlag.de/linux_unix_programmierung/Kap08-004.htm)

Anschließend hängt strace wieder bei `futex` (Fast Userspace Mutex) fest.

Das Gleiche war [oben](https://github.com/Freetz-NG/freetz-ng/issues/349#issuecomment-1002749816) zu beobachten, beim Aufruf
`strace -fff openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon`  wo kein tun0 Interface geöffnet wird. 

Nun ist vermutlich tiefer gehende Kenntnis der [Threadprogrammierung ](https://weekly-geekly-es.imtqy.com/articles/de418705/index.html) erforderlich, ich habe aber eine Ahnung in welche Richtung das laufen könnte. Starte ich den openvpn Dienst ohne Parameter `-- daemon` wird kein Thread beim Aufruf 'geforked' und alles läuft wie es soll. Mit `strg +z bg`  wird zwar auch eine "Abspaltung"  vorgenommen, aber offensichtlich anders, als beim Start mit Parameter `-- daemon` (wie auch im rc Script). 

Den Grund dahinter kann ich (noch) nicht erklären, ich würde allerdings erwarten, dass es mit dem Linux "Grundsystem" zu tun hat und auch mit der GUI v2 (bei sonst identischer freetz config) genau so auftreten müsste. Ich hänge meine freetz config nochmal an, vielleicht bringt das etwas Licht ins Dunkel.

[config-7.29.txt](https://github.com/Freetz-NG/freetz-ng/files/7794526/config-7.29.txt)

fda77, 2021-12-30:
Mit (p)threads gab es vor ein paar Jahren öfter Problem bei Freetz. Ich hab mir das nie angeschaut und denen überlassen die sich damit auskennen. Leider ist eine Internetsuch nach (foren)-thread absolut nutzlos
Eins Suche auf https://trac.boxmatrix.info/freetz-ng/search?q=pthread hat auch zu nichts geführt.

Da mir nichts anderes einfällt: Teste doch mal mit der alten openvpn Version 2.4. In https://github.com/Freetz-NG/freetz-ng/blob/master/make/openvpn/Config.in in Zeile 20-34 die 6 '#' entfernen, dann kann man im menuconfig die alte auswählen

fda77, 2021-12-30:
In der uclib gibt es auch noch thread-optionen: "make uclibc-menuconfig", zb
```
toolchain/make/target/uclibc/configs/freetz/config-arm-1.0.14:UCLIBC_HAS_THREADS_NATIVE=y
toolchain/make/target/uclibc/configs/freetz/config-arm-1.0.37:UCLIBC_HAS_THREADS_NATIVE=y

toolchain/make/target/uclibc/configs/freetz/config-i686-0.9.33.2:UCLIBC_HAS_THREADS_NATIVE=y
toolchain/make/target/uclibc/configs/freetz/config-i686-1.0.14:UCLIBC_HAS_THREADS_NATIVE=y
toolchain/make/target/uclibc/configs/freetz/config-i686-1.0.37:UCLIBC_HAS_THREADS_NATIVE=y

toolchain/make/target/uclibc/configs/freetz/config-mips-0.9.29:LINUXTHREADS_OLD=y
toolchain/make/target/uclibc/configs/freetz/config-mips-0.9.32.1:LINUXTHREADS_OLD=y
toolchain/make/target/uclibc/configs/freetz/config-mips-0.9.33.2:LINUXTHREADS_OLD=y
toolchain/make/target/uclibc/configs/freetz/config-mips-1.0.14:UCLIBC_HAS_THREADS_NATIVE=y
toolchain/make/target/uclibc/configs/freetz/config-mips-1.0.37:UCLIBC_HAS_THREADS_NATIVE=y
```

PeterPawn, 2021-12-30:
Nur als Lebenszeichen ... ich habe das gesehen, aber im Moment (mind. bis 06.01.) keine Zeit, mir das wirklich gründlich zu Gemüte zu führen und die Zusammenhänge zu verstehen.

Die erste Frage, die sich mir beim Überfliegen stellt, wäre die, warum es mit der V2 (die ist flexibler zu konfigurieren) nicht ebenfalls "gut" wäre. Die meisten OpenVPN-Nutzer sind (afaik) auf die V2 umgestiegen ... und da hört man ja tatsächlich nur wenige Klagen, die auch immer relativ schnell zu klären waren.

Und ja, ich lese tatsächlich nicht aktiv irgendwo mit in diesem Projekt und nur in Ausnahmefällen mal die Liste der neuen Commits - auch zum hier: https://www.ip-phone-forum.de/threads/make-stoppt-mit-fehler-2.311975/post-2455982 behandelten Problem habe ich noch nicht in die Commits geschaut, ob Du (@fda77) da etwas unternommen hattest - mittlerweile ist wohl klar, daß es mit der 20.04.01 noch so funktioniert, wie es soll.

Wenn man mich also auf eine Diskussion oder ein Problem aufmerksam machen will, MUSS man mich tatsächlich direkt ansprechen - wobei das eben auch nicht immer eine Garantie dafür ist, daß ich dann zeitnah darauf reagiere (oder überhaupt). 

---

Jedenfalls könnte man hier ja auch mal die generierte OpenVPN-Konfiguration wegkopieren und mit dieser dann unter dem V2-Interface versuchen, die Verbindung(en) aufzubauen. Klappt das dort, liegt es wohl an irgendwelchen Vorbereitungsarbeiten im Start-Skript (das bei V1 iirc dem Benutzer deutlich "mehr Arbeit" abnehmen will/wollte, als bei V2, was die Möglichkeiten aber auch einschränkt) - schon die Frage, warum der Daemon beim Start über das Skript kein Log schreiben soll (so habe ich das jedenfalls bisher verstanden beim Überfliegen), würde mich da stutzig machen und man kann an der Stelle, wo das OpenVPN ansonsten als Daemon gestartet würde, ja auch noch einmal eine ausführliche "Dokumentation" der bis dahin erzeugten Dateien (mit Relevanz für diesen Service natürlich) erstellen lassen - mit allen Dateiinhalten und den zugehörigen Rechten/Besitzern. 

Denn auch die jeweiligen Sessions (interaktive Shell vs. Shell ohne steuerndes Terminal) unterscheiden sich ja beim manuellen und beim automatischen Start (letzterer dann aus dem Skript heraus) und das kann weitere Unterschiede in der Umgebung zur Folge haben - auch das Environment sollte man dann in die "Dokumentation" vor dem Start des Daemons einbeziehen.

Wenn das Vorgeschlagene schon abgearbeitet wurde, vergeßt den Vorschlag einfach - ich habe das bisher eben tatsächlich nur überfliegen können und habe mehr quer gelesen als richtig. Aber auf das Threading würde ich erst mal nicht abstellen als Ursache - das ist bei Verwendung von V2 nun mal identisch und daher KANN es eigentlich nicht an der Konfiguration/Erstellung des Binarys liegen, das ist - m.W. - bei beiden GUI-Angeboten eben dasselbe.

stefanpuch, 2021-12-31:
Ich werde die kommenden Tage mal mit der V2 testen. Habe heute auf meiner Testbox noch eine FW Version 
```
FritzOS 07.21 rev81779 (04.09.2020 14:53:45)
Freetz ng-17524M-caa08d4ed (07.12.2020 19:13:46)
```
gefunden, auf der es noch ohne Probleme mit der V1 und `OpenVPN 2.5.0 mips-unknown-linux-gnu [SSL (OpenSSL)] [LZO] [AEAD] built on Dec  5 2020`  läuft. Dank einfachem Partitionswechsel konnte ich so einfach zur
```
FritzOS 07.29 rev92201 (05.11.2021 11:15:11)
Freetz ng-18776-c715ed811 (23.12.2021 18:50:54)
```
und wieder zurück wechseln (die 7.21 geht noch, die 7.29 nicht). Da ich mit git aber nicht so fit bin, mir ein diesbezügliches diff anzusehen (wäre vermutlich auch nicht sonderlich zielführend) werde ich die Tage wie von Peter empfohlen mal den Umstieg auf die V2 in Angriff nehmen.

Kann man da eigentlich einfach eine neue FW mit V2 Build über eine bestehende V1 installieren oder sollte man dann gewisse Altlasten (z.B. Dateien der alten Version) löschen?

fda77, 2021-12-31:
Das hilft nicht besonders, da bei diesem 1 Jahr altem Image auf openvpn, openssl etc älter sind. Bau doch einfach mit der aktuellen revision ein 7.21 Image

@PeterPawn: Eigentlich gehts darum überhaupt zu finden worin der Fehler liegt. Ich weiss noch nicht ob es ein Bug ist oder an der Konfig liegt. Es könnte was mit "threads" sein. Zuerst funktioniert ein "kill" nicht, ausserdem hängt openvpn nur als "daemon" (fork), nicht aber im standalone mode. Siehe Logs in https://github.com/Freetz-NG/freetz-ng/issues/349#issuecomment-1003133363

---

Der ubuntu-21.10 bashing thread finde ich gehaltlos. Man erfährt nur dass es irgend ein Problem gibt. Es hat aber niemand geschafft die komplette Fehlermeldung zu posten. Wobei diese eh nicht öffentlich lesbar wäre. In dieser würde sogar stehen weshalb "error 2". Da ich grundsätzlich nur vorcompilierte Toolchains nutze fällt das selbst mit Fedora 35 nicht auf

PeterPawn, 2021-12-31:
Wieso soll das ein "bashing"-Thread sein? Da haben halt Leute mit Ubuntu 21 Probleme mit dem Build der Toolchain und da sie nicht wirklich wissen, wo das Problem liegt, fragen sie nach Hilfe.

Daß sie sich die Probleme (wahrscheinlich) erst durch Ubuntu 21 eingetreten haben, ist das eine ... und daß man nicht jedes Update für ein (nur zu diesem Zweck verwendetes) Build-System mitmachen muß, ist das andere. 

Du selbst hast ja offenbar auch die "Angewohnheit" von @gismotro übernommen, nur die jeweils neueste Image-Version für die "Freetz-VM" zu verlinken bzw. bereitzuhalten. 

Da können dann natürlich echte Newbies, die noch gar nichts von Freetz-NG wollten, als die letzten LTS-Versionen "aktuell" waren, auch nur schlecht auf ältere Images mit LTS-Versionen zugreifen, wenn die nicht irgendwo bereitgehalten werden.

Da ich gerade einen neuen (Dedicated-)Server mit dann 1 TB Storage einrichte (deshalb habe ich auch keine Zeit für anderes im Moment), könnte ich da zumindest ausgewählte Dateien noch bereithalten (dann unter der `yourfritz.de`-Domain), die in GitHub nicht so ohne weiteres gehostet werden können. 

Dazu könnten dann auch mehrere Versionen der Freetz-VM gehören (die dürften der jeweils dickste Brocken sein, aber eine 18.04-LTS und eine 20.04-LTS sollten eben auch reichen) - aber darüber kann man immer noch reden, wenn ich den Server fertig habe (was noch dauern wird).

Wenn Du aber meinen Hinweis, daß man für das Build-System nicht jedes Update mitmachen muß, als "bashing" verstanden hast, dann ist das zum Teil sogar richtig (wobei ich das eher als "rant" denn als "bashing" sehen würde) - warum ich das als sinnlos und als "mutwillige Fehlerquelle" erachte, habe ich ja dort aufgeschrieben - und vermutlich lag ich mit meiner Vermutung, daß es irgendein Update gewesen sein muß, was die Probleme beim Build des alten GCC verursacht, ja auch richtig, wenn das in der älteren VM noch problemlos klappt.

fda77, 2022-01-04:
Ich meine dein nicht-LTS bashing... Das weiss ich ja und wollt es mir nicht nochmal durchlesen. 
Dadurch hast du ausgelöst: Gismotro hat die neue VM version entfernt und bietet nur noch die ältere an. Dies bedeutet jeder generiert sich einen neuen Key und wieder ist die "Signierung" kaputt -.-

Die Ursache oder alleine die Fehlermeldung hat niemand gesucht/gepostet. -> bleibt also wie es ist, interessiert niemanden
Es gibt einfach einen Link auf das Verzeichnis von Gismotro. Welche Versionen er da reinpackt ist seine Sache. Ich hab mir auch ein paar seiner VMs gesichert, die älteste ist "Freetz-Linux-1.4.1-Ubuntu 15.04.04 x86.ova". Da ich aber nicht weiss ob er möchte dass die sonst jemand verbreitet fragst du ihn am besten selbst

PeterPawn, 2022-01-04:
Tja, ich bin aber auch nicht verantwortlich, wenn das als "entweder oder" verstanden wird - wobei ich nicht sehe, wer sich da (zumindest im IPPF) genauer mit dem Problem befassen sollte (oder das auch nur wollte), warum Freetz-NG mit Ubuntu 21 irgendwann nicht mehr funktioniert (das kann ja auch nicht gleich von der 21.04 an so gewesen sein, die ging ja wohl noch eine Weile).

Daß man schon die komplette Protokoll-Datei bräuchte, weil das Gezeigte alles nur Schrott ist, stand gleich am Beginn meiner Ausführungen (in diesem IPPF-Thread). Die Leute such(t)en aber sicherlich auch eher nach einer Lösung, denn nach einem Problem ... und wenn die ältere VM mit der LTS-Version das Problem nicht hat, IST das zumindest eine Lösung.

Wenn es Dich darüber hinaus (schon heute) interessiert, ob oder wie gut Freetz-NG mit der nächsten LTS-Version von Ubuntu zusammen funktioniert (das wäre dann ja die 22.04), dann sollte es nicht schwierig sein, sich eine passende VM zu erschaffen.

Ich wüßte jedenfalls nicht, daß Du im IPPF irgendjemandem in letzter Zeit geantwortet hättest - und ich habe auch nicht nachgesehen, ob Du das Problem aus dem IPPF irgendwie anders angegangen wärst. In den "NEWS" (das ist das Maximum an Zeit, die ich zu opfern bereit bin - daß Deine Commit-Kommentare niemandem weiterhelfen, behaupte ich ja nicht zum ersten Mal und ich habe einfach keine Lust, mich erst in einen Commit zu vertiefen, um die damit ausgelöste Änderung "zu verstehen") stand jedenfalls (zu diesem Zeitpunkt) nichts davon.

fda77, 2022-01-04:
Über Linux Editionen oder Version zu diskutieren mach keinen Sinn. Wenn jemand bessere Kommentare schreiben kann soll dieser es tun, ich drängele mich nicht vor

PeterPawn, 2022-01-04:
Vielleicht bin ich ja auch nur "zu dumm" um zu verstehen, welchen Standpunkt Du dabei (außer ggf. dem "Rundumschlag" gegen alle) nun einzunehmen gedenkst?

Einerseits stellst Du meine Kritik an der Updateritis bei der Freetz-VM in Frage (bzw. siehst sie als "bashing", was deutlich etwas anderes als "Kritik" ist) und andererseits klagst Du selbst darüber, daß jetzt niemand mehr den Fehler beim aktuellen Freetz-NG-Stand auf einer Ubuntu 21-Installation suchen würde.

So ganz verstehe ich jedenfalls nicht, worauf Du nun eigentlich hinaus willst - aber das ist hier auch irgendwas mit OpenVPN und da sollte dieses Thema schon deshalb nicht überhandnehmen ... erst recht nicht in einer Issue, wo das nach dem Schließen so oder so kein Mensch mehr liest.

fda77, 2022-01-07:
Es soll halt jeder das Linux nutzen was ihm gefällt. Wenn es Probleme damit gibt muss man/derjenige die halt lösen. Dass neuere Versionen immer Probleme bereiten können ist klar

PeterPawn, 2022-01-09:
> Wenn es Probleme damit gibt muss man/derjenige die halt lösen.

Und dann wäre da noch die Gruppe derjenigen, die solche Probleme halt nicht (alleine) lösen können.

Dann müßte die also jemand anderes für sie lösen ... oder ihnen passende Workarounds aufzeigen, mit denen sie dann ihr Ziel (und das ist ja nicht die Verwendung einer bestimmten Distribution für das Erstellen ihres Freetz-NG-Images, sondern das Image an sich) ebenso erreichen können. Und ganz offensichtlich gibt die Realität hier ja meiner Kritik an der Verwendung von Ubuntu 21 (was in Deinen Augen "bashing" sein mag, aber mir dann nur zeigt, daß wir da durchaus unterschiedliche Vorstellungen von der Bedeutung dieses Begriffes haben) recht - wenn das mit der 20.04.01 weiterhin funktioniert.

fda77, 2022-01-10:
Es gab keinen ubuntu-Fehler, sondern ein Variabelnnamen war einfach falsch

stefanpuch, 2022-02-16:
So, ich habe jetzt noch einmal ein komplett neues Testsystem auf einer 75(2)30 aufgebaut. Basis ist die aktuelle FW 7.29 freetz-ng-18872-c9c99eb10. dazu habe ich komplett neue SSL Zertifikate erstellt und die v2 GUI genutzt. Dazu das chroot weggelassen. Das einzige was sich geändert hat ist der Umstand, dass das Starten- und Stoppen des vpnd über die Freetz-GUI funktioniert.

Das Hauptproblem bleibt aber bestehen. Sowohl beim Aufruf über die FreetzGUI als auch über die Kommanozeile mit Parameter `--daemon` funktioniert nicht. Es wird kein tun Interface geöffnet und ein Aufruf mit `strace -fff openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon` bleibt wie bei der 7490 oben bei einem `futex` (Fast Userspace Mutex) stehen.
Ein Aufruf von der Kommandozeile ohne --daemon und dem strg+z bg funktioniert zuverlässig. Eine Verbindung mit einem Client kommt mit den hinterlegten Zertifikaten ohne Probleme zustande.

Ich werde das Gefühl nicht los, dass irgend etwas an meiner freetz-config "speziell" ist, da sonst niemand diesen Fehler hat.

[config-7530-0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8079049/config-7530-0729.txt)

PeterPawn, 2022-02-17:
Wird die `uClibc-ng 1.0.37` mit `pthread`-Support verwendet, werden bestimmte Operationen zwischen Threads automatisch serialisiert, dazu ersetzt/überlädt die `libpthread` einige Funktionen, u.a. auch `fork()` und `exit()`. 

Dadurch werden (u.a.) I/O-Operationen zwischen den Threads an den kritischen Stellen serialisiert - beim `fopen()` wird z.B. ein gemeinsam genutzter Speicherbereich verwendet, um eine (interne) Liste der geöffneten Dateien zu verwalten und der Zugriff auf diese Liste(n) (bzw. deren "Verwaltung" beim Entfernen/Hinzufügen von Einträgen) erfolgt dann "exklusiv": https://cgit.uclibc-ng.org/cgi/cgit/uclibc-ng.git/tree/libc/stdio/_fopen.c#n210 ... aber das dürfte hier auch nicht wirklich das Problem sein.

Nachdem der Daemon offenbar direkt nach dem Öffnen der Protokoll-Datei hängt (und angesichts der Tatsache, daß eben nur im Daemon-Modus dann auch ein - abgetrennter - Thread "geforkt" wird, was auch den Unterschied zum Aufruf als Vordergrund-Programm darstellt), ist das wohl irgendein Deadlock.

Was da warum nun exakt hängt, weiß ich auch nicht ... aber wenn Du Dir Deine beiden `strace`-Ausgaben oben ansiehst, ist da schon ein (sogar deutlicher) Unterschied zu sehen. 

Im Daemon-Modus geht der Thread aus irgendeinem Grund nach dem `open()` für die Protokoll-Datei und dem anschließenden `fcntl64()` (was eigentlich auch nur ein "close on exit" setzt) beim `futex()`-Aufruf in ein WAIT über - irgendetwas veranlaßt ihn offensichtlich, auf die "Freigabe" irgendeiner Ressource zu warten (die er als "in Benutzung" ansieht, warum auch immer), während beim Aufruf als Vordergrund-Prozess an dieser Stelle (mehrere) "Freigaben" für verschiedene Ressourcen zu sehen sind (WAKE - der erste Parameter ist ja jeweils die Adresse des "futex word", das sind also unterschiedliche "Futexe", für die ggf. wartende Thread geweckt würden). 

Das fehlende Ergebnis für den `futex(...WAIT...)`-Call dürfte daran liegen, daß das Warten nicht zum Ende kommt (im Fall des Starts als Daemon), das ist dann die Stelle, wo der Deadlock auftritt. 

Die unterschiedlichen FD-Handle sind jedenfalls auch nicht das Problem. Läuft das Programm als Daemon in einem gesonderten Thread, eröffnet der ursprüngliche Start-Thread noch einen Socket für `syslog` (nach `/dev/log`) und vererbt den an den geforkten Thread - daher ist in diesem Fall der FD 3 schon belegt und für das Status-File wird FD 4 verwendet.

Nach meiner Ansicht hängt der Daemon irgendwo nach dieser Stelle: https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/init.c#L4038 (da wird die Status-Datei geöffnet - in der aufgerufenen Funktion - und auch der `fcntl64()`-Aufruf ist dort zu finden - als `set_cloexec()`) und dann am ehesten beim Initialisieren der Crypto-Umgebung in Zeile 4095 (https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/init.c#L4095).

Denn beim Starten als Vordergrundprozess kommt als nächste nachvollziehbare Aktion (bei den `futex()`-Calls fehlt der Bezug, um welche Ressource(n) es sich konkret handelt) ja ebenfalls das Initialisieren der Crypto-Funktionen. Der Zugriff auf `/dev/urandom` (wenige Zeilen nach der Stelle mit dem `open()` für das Status-File) dürfte schon im OpenSSL-Code zu finden sein.

Da das `set_cloexec()` (https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/status.c#L107) auch noch im `strace` zu sehen ist (das ist eben das `fcntl64()`) und zwar als erfolgreich und beendet und in Anbetracht der Tatsache, daß das Changelog für die OpenSSL-Libraries auch eine Änderung für die "RAND facility" im Zusammenhang mit der Benutzung von POSIX-Threads beschreibt (und das Auslesen des PRNG in `/dev/urandom` dürfte dazu gehören): https://github.com/openssl/openssl/blob/master/CHANGES.md - nach `atfork` suchen - halte ich das tatsächlich für ein Problem, was erst seit OpenSSL 1.1.1 auftritt (https://github.com/openssl/openssl/blob/master/doc/man3/OPENSSL_fork_prepare.pod). Insofern würde ich das "bei anderen geht es genau so" noch gar nicht unterschreiben.

Schaut man etwas genauer in den OpenVPN-Quelltext, findet man u.a. das: https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/openvpn.c#L178 - ziemlich zeitnah zum Start des "Hauptprogramms" wird die Funktion `init_static()` aufgerufen, in der dann wiederum ebenfalls zeitnah die SSL-Library erst einmal initialisiert wird: https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/init.c#L792 -> https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/ssl.c#L342 -> https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/ssl_openssl.c#L86 + https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/crypto_openssl.c#L189

Da werden also `libcrypto` und `libssl` schon VOR DEM `fork()` für den Daemon initialisiert, denn der KANN ja auch erst erfolgen, wenn die Optionen/Parameter danach verarbeitet werden: https://github.com/OpenVPN/openvpn/blob/e8a2794824ded011a1ed58aeb95321332b1ae684/src/openvpn/openvpn.c#L214 ... und an dieser Stelle könnte/müßte dann die Änderung mit dem `atfork()` zuschlagen (da wohl aus Sicherheitsgründen jeder Thread seine eigene Entropie haben soll, was zuvor NICHT sichergestellt war) und ihrerseits dafür sorgen, daß bei diesem `fork()` , was sich hinter dem von OpenVPN verwe [... cut ...] nd anderes als Problem haben sollte (sowohl von den Freetz-Usern, aber viel wahrscheinlicher auch andere OpenVPN-Benutzer - wenn auch vielleicht nicht zwingend auf MIPS-Plattformen), muß man in die Tiefen des OpenSSL-Codes hinabsteigen, wenn man es genau wissen will ... andererseits kann man das OpenVPN notfalls auch mit einer anderen TLS-Library und statisch gelinkt bauen (wenn das tatsächlich ein Deadlock bei der OpenSSL-Initialisierung sein sollte, das ist bisher ja nur ein "educated guess").

fda77, 2022-02-17:
Seit ein paar Tagen wird uClibc-ng 1.0.40 verwendet: https://cgit.uclibc-ng.org/cgi/cgit/uclibc-ng.git/log/
Min 2 der ca 30 Commits seit .37 betreffen threads

> Wer nennt seine Status-Datei eigentlich openvpn.log

Die Freetz Webinterface+Script von openvpn sind sehr seltsam! Am besten nie was ändern...

stefanpuch, 2022-02-18:
Wenn ich die ausführlichen Hinweise von Peter richtig verstehe scheint es ein Problem im Zusammenhang mit OpenSSL zu sein. Meine weiteren Tests hinsichtlich unterschiedlicher FW Versionen haben zumindest keinen weiteren Ansatz ergeben.

Ich habe auf Basis der 7530_07.14.ger_freetz-ng-18872-c9c99eb10 eine Version mit OpenVPN 2.5.5 gebaut, da funktioniert aber auch der Modus ohne `--daemon` nicht.

[7.14_daemon-trace.txt](https://github.com/Freetz-NG/freetz-ng/files/8097762/7.14_daemon-trace.txt)
[7.14_ohne_daemon-trace.txt](https://github.com/Freetz-NG/freetz-ng/files/8097765/7.14_ohne_daemon-trace.txt)
[config-7530-0714.txt](https://github.com/Freetz-NG/freetz-ng/files/8097778/config-7530-0714.txt)

Da die Version 7.14 nicht mit OpenVPN 2.4.11 kompiliert, habe ich noch einer Version 7530_07.21.all_freetz-ng-18872M-c9c99eb10 gebaut, welche FW 7.21 und OpenVPN 2.4.11 nutzt. 

[config-7530-0721.txt](https://github.com/Freetz-NG/freetz-ng/files/8097786/config-7530-0721.txt)
Mit dieser Version funktioniert wie eingangs berichtet die Version ohne `--daemon` wie man es erwartet, die Version mit `--daemon` landet am `futex` (vermutlich im deadlock). Allen Version gemein ist OpenSSL 1.1.1m 

Als nächstes würde ich mir mal den Source Teil genauer anschauen.

stefanpuch, 2022-02-18:
Ein Schritt in die richtige Richtung! Ich habe die in https://github.com/Freetz-NG/freetz-ng/issues/349#issuecomment-1041354584 referenziert Minimalversion genommen und OpenSSL 1.1.1m gegen OpenSSL 1.0.2u ersetzt. Damit startet OpenVPN wie es sein sollte direkt beim Booten und auch das Starten / Stoppen über die Freetz-Gui funktioniert einwandfrei.

fda77, 2022-02-18:
3: vielleicht ist openvpn damit noch nicht kompatibel?
1.1: probier es mit no-threads beim configure

stefanpuch, 2022-02-18:
Da bräuchte ich etwas mehr Hilfe, an welcher Stelle das `no-threads`einzubauen wäre. Anstelle von `-pthread` in der config/.cache.in oder wie?

fda77, 2022-02-18:
Ich hab kein  "-pthread"  in der .mk von openssl gefunden. Soweit ich nachgelesen hab, werden threads in der neuen openssl Version aktiviert, solang man sie nicht bei configure deaktiviert.
Das SOLLTE in der .mk von openssl mit so einer Zeile gesetzt werden können
```$(PKG)_CONFIGURE_OPTIONS += no-threads```
Beim (neu)bauen von openssl (ZB `make openssl-recompile`)  sollte dazu irgend eine Melden auftauchen

PeterPawn, 2022-02-19:
Wenn weiterhin `strace` eingesetzt wird und das nicht eine ganz komische Version ist, deren Flags/Benutzung ich nicht kenne, sollte man den Aufruf noch einmal "überdenken".

Das normale `strace` sähe so aus: https://github.com/strace/strace - dessen "man page" findet man auch an verschiedenen Stellen im Netz; https://www.google.com/search?q=strace+manpage

Da findet man auch eine Option `-f` und auch ein `-ff` - aber kein `-fff` und es macht auch keinen Sinn, das als die (POSIX-konforme, afaik) Kurzform für `-ff -f` anzusehen, weil eben `-ff` die Auswirkungen von `-f` bereits beinhaltet.

Was aber Sinn machen kann, ist das Erfassen der `strace`-Ausgaben nach PIDs getrennt ... dazu muß man aber die Optionen `-ff` und `-o <filename>` miteinander kombinieren, womit dann pro PID eine Ausgabedatei mit dem Namen `<filename>.<pid>` erzeugt wird. 

Am besten kombiniert man das dann noch mit einer Option für einen Zeitstempel - `-r` wäre eine gute Wahl, solange es nicht noch um die Interaktion mit anderen Sessions oder gar Systemen geht, ansonsten kann man auch `-ttt` verwenden, wobei dann eben immer die Zeit als "epoch" vor JEDER Zeile steht, was bei `-r` stattdessen bei 0 beginnt zu zählen.

Findet man das dann zu unübersichtlich (weil man zwischen zwei/mehreren Dateien wechseln muß, sofern Threads parallel arbeiten), kann man sich das auch (mit einem entsprechenden Tool, das zum `strace` gehört: https://github.com/strace/strace/blob/master/src/strace-log-merge) wieder zu einer Datei zusammensetzen lassen. Wobei man da dann auch nur die tatsächlich interessanten Threads (hier gibt es ja bisher nur zwei davon) zusammensetzen lassen kann - das macht es gerade bei Programmen mit deutlich mehr Threads wieder übersichtlicher.

Ein (für mich erklärlicher) Aufruf von `strace` sollte also eher so aussehen: `strace -ff -o /tmp/strace.log <program>...` und dann muß man die Ausgaben auch nicht aus irgendeiner Konsole herauskopieren (wo manchmal auch längere Zeilen nicht komplett/richtig erfaßt werden), sondern findet die erzeugten Dateien (am besten löscht man bei mehrfachem Aufruf dann ältere Dateien, bevor man das `strace` erneut startet) am angegebenen Ort, jeweils als Datei `/tmp/strace.log.<pid>`.

Das aber nur nebenbei ...

---

Was mich noch interessieren würde ... wir haben hier ja bisher jeweils nur die Protokolle vom Initialisieren des OpenVPN-"Servers" gesehen - die Optionen sagen ja: `mode server` mit `proto udp`, wenn ich das oben richtig erinnere: https://linux.die.net/man/8/openvpn.

Was da aber noch fehlt, ist das, was in der Folge passieren müßte, wenn ein Client Verbindung mit dem Server aufnehmen will. 

Ich war bisher eigentlich der Ansicht, daß da pro Verbindung auch ein neuer Thread erstellt würde ... die Kontrolle eines eigenen OpenVPN-Servers (auch mit diesen "Eckdaten" - `mode server` + `proto udp`) zeigte mir, daß da tatsächlich auch mehrere Verbindungen IN DIESEM EINEN Thread bedient würden. 

Damit kommt wohl tatsächlich im OpenVPN (mit dieser Konfiguration) kein weiteres `fork()` für zusätzliche Prozesse (mal abgesehen von Hooks, die aber auch nicht dieselben OpenSSL-Bibliotheken benutzen würden, weil das i.d.R. verkappte system()-Calls - i.d.R. `execve()` - für irgendwelche Plugins/Shell-Skripte sind) zum Einsatz - das würde zumindest auch erklären, warum schon dieses erste `fork()`, wenn der Daemon als Hintergrundprozess abzweigt, einen Unterschied zwischen den beiden Modi bewirkt.

Die vorgeschlagenen zwei Änderungen im Source-Code von OpenVPN (das soll ja nur ein Test sein!) sind sicherlich nicht soo kompliziert - das Einzige, was das ggf. etwas komplizierter macht, ist der Umgang mit Source-Code und Patches in Freetz-NG ... wenn das immer noch so sein sollte, wie im "alten" Freetz. 

Denn dann würden üblicherweise die Quellen desöfteren neu entpackt und danach die Patches aus dem Unterverzeichnis im Paket unterhalb von `<freetz_dir>/make/<package>/patches` angewandt ... man muß/müßte dann also notwendige Änderungen als einen solchen Patch bereitstellen, was nicht immer ganz einfach ist und einiges an Kenntnissen, wie Freetz beim `make <something>` genau arbeitet, erfordert.

MIT diesen Kenntnissen ist das aber im Handumdrehen erledigt ... ich würde mich jedenfalls zunächst mal auf Änderungen im OpenVPN konzentrieren und das OpenSSL-Paket nicht anfassen (höchstens dann, wenn man da zusätzliche Ausgaben zum Tracen einbauen wollte). 

Auch mit OpenSSL 1.1.1 sollte es machbar sein, ein passendes OpenVPN zu bauen ... wobei es schon einige Parts in den OpenVPN-Quellen gibt, die nur bis 1.1.0 ausgeführt werden -> an diesen Stellen wird dann ggf. OPENSSL_VERSION (o.ä.) per `#define` mit einer Vorgabe verglichen.

Ich wäre/bin zwar genauso verblüfft, falls das zuvor nicht aufgefallen sein sollte ... gerade auch im OpenWRT-Projekt hätte ich da vermutet, daß das bekannt sein sollte, wenn es ein generelles Problem gibt. Aber ich habe da auch immer noch nicht gesucht - vielleicht sollte man (oder "Du" oder "Ihr" - je nachdem, wer sich "verantwortlich" fühlt) doch noch einmal eine entsprechende Suche im Netz veranstalten, bevor man selbst auf die Suche nach der Ursache und einer denkbaren Lösung geht.

Sollte beim Übersetzen von OpenSSL jedenfalls die Auswahl die Benutzung von POSIX-Threads einschließen, wäre ich einigermaßen verblüfft, wenn man die OpenSSL-Libraries dann auch ohne `pthread_atexit` linken kann - wobei auch das natürlich nicht vollkommen ausgeschlossen wäre, wenn das nur eine "we [... cut ...] das richtig verstanden habe, führt die aktive Option dazu, daß die OpenSSL-Libraries ihrerseits per `pthread_atfork()` die enthaltenen Fork-Handler auch registrieren. Wenn die Option automatisch mit gesetzt sein SOLLTE, schadet es aber auch nicht, sie noch einmal explizit zu "erwähnen" und damit zu setzen. Dabei aber beachten, wie mehrere Optionen zusammen gesetzt werden (mit "or" - das sind Bit-Flags): https://github.com/openssl/openssl/blob/OpenSSL_1_1_1-stable/doc/man3/OPENSSL_init_crypto.pod

fda77, 2022-02-19:
Das mit dem `strace -fffff` ist eine Angewohnheit wie `asterisk -rvvvvvvvv` - einfach ne Zeit lang auf der Taste bleiben ;-)

Patches werden bei Freetz wie bei Openwrt genutzt. Um sicher zu gehen dass der aktuelle Zustand genutzt wird ein "make PKG-recompile" (oder eines der *clean)
Beim Vergleichen mit Openwrt sollte man immer beachten dass Freetz ucllibc und Openwrt musl verwendet

Openssl: Freetz/ hat nur 0.9 und 1.0. Diese sind veraltet und ICH hab in NG 1.1 und 3 gemacht - es kann also sein dass daran was nicht stimmt! Früher gab es bei Freetz öfter Probleme mit diesen "threads" ich hab mich aber nie damit befasst
Da scheinbar sonst niemand mit Openvpn ein Problem hat könnte das auf openssl hindeuten

stefanpuch, 2022-02-19:
So, nächstes Update:
@PeterPawn Danke für Deine ausführlichen Hinweise. Teilweise gehen die jedoch so tief, dass da meine Expertise dann am Ende ist. Ich kann mir das zwar alles durchlesen, aber insbesondere der Teil mit dem forken von threads da bin ich dann raus.

Was ich sagen kann: Ein build wo ich in der Config.in.libs in make/openssl/ das pthreads gelöscht habe, hat im Ergebnis keinen Unterschied gebracht. Ich habe mich daher an den Vorschlag von Peter gemacht.
In der Tat ist es nicht so einfach die Änderungen am Code hinzubekommen, da "die Quellen desöfteren neu entpackt" werden und ich keine Expertise habe, die Änderungen über einen entsprechenden Patch in diesen ganzen Prozess einzuarbeiten.

Ich habe daher nach dem `make ovenvpn-clean`  abgewartet, bis er buildprozess das src Verzeichnis neu entpackt hat, dann abgebrochen, den Code wie oben vorgeschlagen geändert (und das `ssl_init_lib()` an die entsprechende Stelle verschoben). In der aktuellen Codebasis ist das dann Zeile 1250 direkt vor dem `ret = true;` 

Dann habe ich den vorgeschlagenen `strace -ff -o /tmp/strace.log <program>...`  genommen und noch einen Zeitstempel - `-r`
 eingefügt. Nach dem Start liefert ein `ps` dann 2 Prozesse, die ich von einer 2ten Console mit kill -9 getötet habe, um das strace zu beenden:
```
3467 root      2056 S    strace -ff -r -o /tmp/strace.log openvpn --config /mod/etc/openvpn.conf --writepid /var/run/
3471 root      2300 S    openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon
```
Die Dateien von `strace -ff -r -o /tmp/strace.log openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon` sind dann hier

[strace.log.3470.txt](https://github.com/Freetz-NG/freetz-ng/files/8102473/strace.log.3470.txt)
[strace.log.3471.txt](https://github.com/Freetz-NG/freetz-ng/files/8102474/strace.log.3471.txt)

Bei der Analyse bin ich dann aber endgültig raus. erkennen kann ich lediglich, dass der Prozess letztlich an einem Mutex wartet (worauf auch immer).

Eine Frage, die mir immer wieder in den Kopf kommt: OpenVPN ist doch ein relativ oft genutztes Paket. da wundert es mich, dass ich der einzige bin, der das Problem feststellt. Macht es einen evtl. einen Unterschied ob man `Link statically` nutzt (scheint per default gesetzt zu sein) oder das nicht nutzt?

PeterPawn, 2022-02-19:
> Da scheinbar sonst niemand mit Openvpn ein Problem hat könnte das auf openssl hindeuten

Ich habe mittlerweile noch das hier gefunden: https://github.com/OpenVPN/openvpn/blob/eded6a32edcb7d4378140fcbf984fae4d8735029/src/openvpn/init.c#L1228 ... wenn also OpenVPN 2.5 vom `systemd` gestartet wird (den der überwiegende Teil der Linux-Distributionen mittlerweile benutzt), dann wird auch bei angegebener Option `--daemon` eben NICHT geforkt. 

Das dürfte die Systeme, wo tatsächlich der `daemon()`-Call (https://github.com/OpenVPN/openvpn/blob/eded6a32edcb7d4378140fcbf984fae4d8735029/src/openvpn/init.c#L1240) zum Einsatz kommt, schon erheblich einschränken - für mich ein weiterer Grund dafür, daß es bisher nicht groß aufgefallen sein könnte. 

Denn ältere Systeme, die üblicherweise noch mit SysVinit arbeiten (oder andere, die "aus Prinzip" keinen `systemd` wollen), arbeiten auch nicht immer unbedingt mit den neuesten Paketen (auch wenn OpenSSL 1.1.1 nicht mehr brandneu ist und OpenVPN 2.5 ggf. auch nicht - die sind im Vergleich zu vielen anderen, älteren Distros schon recht "modern") - daher kann es gut sein, daß diese Kombination aus "nicht `systemd`, aber OpenVPN 2.5 mit OpenSSL 1.1.1" gar nicht so häufig vorkommt ... außerdem gerät OpenVPN seit dem Erfolg von WireGuard (das ja mittlerweile in den Linux-Kernel aufgenommen wurde) ein wenig ins Abseits und wird gar nicht mehr sooo oft verwendet.

Wie das beim Freetz-NG-Paket für OpenVPN aussieht, weiß man auch nicht genau (es fehlen belastbare Zahlen) - und wenn das nur in der Kombination für die 7490 (bzw. andere VR9-Boxen) auftritt, dürften das auch ständig weniger Boxen sein, denn die "Cracks" haben mittlerweile eben auch neuere Modelle ... so ganz "taufrisch" ist die VR9-Plattform ja nicht mehr und ich möchte auch nicht wissen, wieviele dieser Boxen noch mit einer älteren FRITZ!OS-Version (und älteren Freetz-NG-Paketen, wo das Problem in der Kombination auch noch nicht auftrat) arbeiten, weil ihre Besitzer eben nicht jedes FRITZ!OS-Update mitmachen und wenn man keine AVM-Service nutzen will, kommt man auch mit älteren FRITZ!OS-Versionen ja noch gut hin (da braucht's dann auch keine 07.29).

PeterPawn, 2022-02-19:
Und vielleicht war mein erster Vorschlag auch nicht ganz so schlau bzw. vollständig ... schaut man noch einmal in die Funktion `init_static()` hinein (und zwar auch im richtigen Branch/Tag): https://github.com/OpenVPN/openvpn/blob/eded6a32edcb7d4378140fcbf984fae4d8735029/src/openvpn/init.c#L822, dann findet man nach dem `init_ssl_lib()` ja auch noch das Initialisieren eines PRNG (also eines eigenen "Zufallszahlengenerators") und das sogar explizit mit dem Hinweis, daß man das nach einem `fork()` auch wiederholen soll(t)e - quasi eine eigene Vorkehrung im OpenVPN, damit zwei Threads nicht denselben Zufall (der dann auf einmal gar nicht mehr so zufällig ist, wenn er reproduzierbar doppelt auftritt) benutzen.

Daher würde ich die Zeilen 822 bis 827 aus der `init_static()`-Funktion nach `possibly_become_daemon()` (https://github.com/OpenVPN/openvpn/blob/eded6a32edcb7d4378140fcbf984fae4d8735029/src/openvpn/init.c#L1252) und zwar direkt vor das letzte `return`. Der bedingte Zweig `#ifdef ENABLE_SYSTEMD` sollte in Freetz-NG keine Rolle spielen und so kommt man bei der Ausführung des Codes zwangsläufig an dieser Stelle vorbei, egal ob da ein Hintergrundprozess geforkt wird (bzw. werden soll) oder nicht.

Man kann zwar sehen, daß Deine Änderungen "gewirkt" haben, denn im geforkten Thread tauchen jetzt auch `futex()`-Aufrufe auf, die zuvor nicht vorhanden waren (nach dem `dup2()`, wo die Standard-Handle alle auf `/dev/null` gesetzt werden, danach werden die Signal-Handler eingerichtet und dann kommen 6 Futex-Calls, die zuvor (im geforkten Prozess) nicht existierten. Damit kann man die m.E. ziemlich eindeutig der Initialisierung der Crypto-Backends zuordnen ... nur sieht man diese Initialisierungen auch noch parallel im Parent-Prozess (nach dem `close()` für die Konfigurationsdatei in Zeile 46), wenn ich mich nicht irre.

---

Allerdings habe ich auch einen Fehler gemacht weiter oben ... mit der Option `-r` beim `strace` muß man die Zeiten selbst aufaddieren, weil das jeweils das Intervall zwischen den Aufrufen angibt (Zeile 1 für PID 3471 bedeutet also, daß das `setsid()` 76 µs nach dem `clone()` (aus dem Parent) aufgerufen wurde) und damit das Mischen von Protokollen auch nicht funktioniert, weil das auf den Zeitstempeln basiert.

---

Da der im OpenVPN verwendete `daemon()`-Call auch den Parent-Prozess direkt abwürgt (via `exit()`-Aufruf), würde ich aber darauf verzichten, in diesem Prozess bereits irgendwie die OpenSSL-Libraries zu initialisieren (der Child-Prozess erbt ja den gesamten Status beim `clone()` (was vom verwendeten `daemon()` aufgerufen wird), weil da ggf. beim (implizien) `exit()` auch direkt wieder Aufräumarbeiten stattfinden und damit die Ressourcen, die von den OpenSSL-Libraries zuvor schon in diesem Prozess verwendet/initialisiert wurden, auch gleich wieder abgeräumt werden sollten. 

Wenn es am Ende eine dieser Ressourcen sein sollte, auf die der Child-Prozess warten möchte (das "futex word" in der letzten Zeile des Child-Prozesses taucht aber auch nirgendwo anders auf in der Ausgabe), dann dürfte es besser sein, wenn man im Parent-Prozess gar nicht erst die OpenSSL-Strukturen initialisiert/benutzt. 

Ich glaube auch nicht, daß zwischen dem `init_static()` (wo bisher OpenSSL initialisiert/benutzt wird) und der Stelle in `possibly_become_daemon()`, wohin Du das verschieben könntest, irgendwelche Operationen stattfinden, die irgendetwas mit Kryptographie zu tun haben. Der erste Thread braucht zwar auch für irgendwelche Zwecke ein wenig Zufall (https://github.com/OpenVPN/openvpn/blob/eded6a32edcb7d4378140fcbf984fae4d8735029/src/openvpn/init.c#L791), benutzt dazu aber noch keine Crypto-Libraries.

stefanpuch, 2022-02-19:
Wäre es dann nicht pragmatisch die Option `--daemon` wegzulassen und beim Start über das init script den Prozess anders abzutrennen so wie ich das beim manuellen Start mit `strg+z und bg` auch hinbekomme?
Ich habe das man getestet: wenn ich die Option `--daemon` einfach weglasse dann hängt natürlich das script an der Stelle fest und auch der Output landet nicht im syslog. Kann man das `strg+z und bg` durch einen Aufruf mit `&`  ersetzen?

fda77, 2022-02-19:
Da ist was dran. Ich hab Openvpn nur noch laufen "weil es schon konfiguriert ist" und tcp statt udp verwendet.  Normalerweise ist aber Wireguard im Einsatz. An Zahlen hab ich nur dies von Github, mit einem festen Zeitraum von 2 Wochen. Keine Ahnung wie die genau zu deuten sind, nicht-Cloner sind Fetcher? Es gibt aber noch die Bitbucket, Gitlab und Boxmatrix mirror. Und dazu noch die mit vorcompilierten Images.

[image]

@stefanpuch: Man könnte --daemon weglassen und den start-stop-daemon nutzen. Elegant ist aber was anderes :)

PeterPawn, 2022-02-19:
Theoretisch geht das auch ... wenn man nur nach einer pragmatischen Lösung und nicht nach der Ursache suchen will. Dann könnte man das z.B. auch über `start-stop-daemon` (https://github.com/brgl/busybox/blob/master/debianutils/start_stop_daemon.c), das sollte in der Freetz-NG-BusyBox enthalten sein, weil m.W. auch andere Freetz-Pakete dieses Applet nutzen, um Programme zu starten, die von sich aus nicht als Hintergrundprozess gestartet werden können.

stefanpuch, 2022-02-19:
OK, dann versuche ich mal den aktuellen Stand zusammen zu fassen:
- Das starten von OpenVPN (aktuell Version 2.5.5) mit der Option `--daemon` scheitert mit der aktuellen Version von OpenSSL 1.1.1m in Freetz-NG
- Als Workaround gibt es 2 Optionen:
1. OpenSSL 1.0.2u benutzen (ist noch in Freet-NG enthalten)
2. das Freetz-NG BusyBox Applet `start-stop-daemon` aktivieren und die Datei `make/openvpn-v2-cgi/files/root/etc/init.d/rc.openvpn` wie folgt ändern:
```
diff --git a/make/openvpn-v2-cgi/files/root/etc/init.d/rc.openvpn b/make/openvpn-v2-cgi/files/root/etc/init.d/rc.openvpn
index bb3447346..6db08c204 100755
--- a/make/openvpn-v2-cgi/files/root/etc/init.d/rc.openvpn
+++ b/make/openvpn-v2-cgi/files/root/etc/init.d/rc.openvpn
@@ -19,7 +19,7 @@ config() {
 }
 
 start() {
-       modlib_startdaemon ${DAEMON} --config /mod/etc/${DAEMON}.conf --writepid "$PID_FILE" --daemon
+       modlib_startdaemon start-stop-daemon -S -b -q -x ${DAEMON} -- --config /mod/etc/${DAEMON}.conf --writepid "$PID_FILE"
 }
 
 configs() {
```
Option 1. geht ohne manuelle Änderungen am Code
Option 2. hat den Vorteil, OpenSSL 1.1.1 nutzen zu können, allerdings auch den Nachteil, das der Output vom OpenVPN daemon nicht im syslog landet, wenn man den benutzen möchte und "Elegant ist aber was anderes :)"

-----------------------------------------------------------------------
Der sauberste Weg wäre wie @PeterPawn sagt "nach der Ursache suchen". 
Ich kann jetzt noch einmal versuchen, die beiden Vorschläge aus 
https://github.com/Freetz-NG/freetz-ng/issues/349#issuecomment-1042669132
und 
https://github.com/Freetz-NG/freetz-ng/issues/349#issuecomment-1046055547

zusammen umzusetzen. Mehr als Trial und Error ist das von meiner Seite aber nicht, da ich - wie schon gesagt - nicht so tief in der Threadprogrammierung  drin bin.

fda77, 2022-02-19:
Ich häte noch eine Option 3.: Fehler bei Openvpn melden

Zur Änderung in rc.openvpn gehört auf jeden Fall noch das Auswählen des Applets in der Config.in, wie https://github.com/Freetz-NG/freetz-ng/blob/master/make/rrdstats/Config.in#L5

stefanpuch, 2022-02-19:
@fda77 stimmt, mit der Änderung wird das Applet dann automatisch aktiviert, ich hatte das manuell gemacht

@PeterPawn 
```
diff -p ../source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/openvpn-2.5.5/src/openvpn/init.c ../init.c 
*** ../source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/openvpn-2.5.5/src/openvpn/init.c	2021-12-15 07:33:55.000000000 +0100
--- ../init.c	2022-02-19 21:49:33.096530612 +0100
*************** init_static(void)
*** 819,830 ****
  
      update_time();
  
!     init_ssl_lib();
  
      /* init PRNG used for IV generation */
      /* When forking, copy this to more places in the code to avoid fork
       * random-state predictability */
!     prng_init(NULL, 0);
  
  #ifdef PID_TEST
      packet_id_interactive_test();       /* test the sequence number code */
--- 819,830 ----
  
      update_time();
  
!     /* init_ssl_lib();*/
  
      /* init PRNG used for IV generation */
      /* When forking, copy this to more places in the code to avoid fork
       * random-state predictability */
!     /* prng_init(NULL, 0); */
  
  #ifdef PID_TEST
      packet_id_interactive_test();       /* test the sequence number code */
*************** possibly_become_daemon(const struct opti
*** 1249,1254 ****
--- 1249,1262 ----
  
          ret = true;
      }
+ 
+     init_ssl_lib();
+ 
+     /* init PRNG used for IV generation */
+     /* When forking, copy this to more places in the code to avoid fork
+      * random-state predictability */
+     prng_init(NULL, 0);
+ 
      return ret;
  }
```

Die Dateien von strace -ff -r -o /tmp/strace.log openvpn --config /mod/etc/openvpn.conf --writepid /var/run/openvpn.pid --daemon sind dann hier
[strace.log.3438.txt](https://github.com/Freetz-NG/freetz-ng/files/8103000/strace.log.3438.txt)
[strace.log.3439.txt](https://github.com/Freetz-NG/freetz-ng/files/8103001/strace.log.3439.txt)

Im Vergleich zu den vorherigen traces sehen sie anders aus, ob das bei der weiteren Analyse hilft kann ich nicht beurteilen. Zumindest bleibt der Prozess noch im `futex` hängen.

stefanpuch, 2022-02-19:
Ich habe jetzt mit den von Peter genannten Änderungen einmal statt der statisch gelinkten SSL Version die dynamische gelinkte Variante gebaut und kann feststellen: damit funktioniert alles wie es sollte. Zum Vergleich auch dazu einmal den `strace` Output.
[strace.log.3046.txt](https://github.com/Freetz-NG/freetz-ng/files/8103111/strace.log.3046.txt)
[strace.log.3047.txt](https://github.com/Freetz-NG/freetz-ng/files/8103112/strace.log.3047.txt)

Morgen werde ich daher noch einmal ohne die Änderungen eine dynamisch gelinkte Version testen.

PeterPawn, 2022-02-20:
1. Ich habe den "Umstieg" von der 7490 (um die es am Anfang ging) auf eine andere Plattform (jetzt ja wohl eine 7520 oder 7530) gar nicht richtig mitbekommen - erst die Dateiangabe in der `diff`-Ausgabe, wo auf einmal eine ARM-Plattform zu erkennen war, hat mich darauf gestoßen. Deshalb habe ich bisher auch immer nur von 7490 geschrieben. Weird ... aber zumindest ist damit ja die Frage, ob das nun plattform-spezifisch ist, auch geklärt.

2. Auch daß das OpenVPN statisch gelinkt wird/wurde, ist mir so nicht aufgefallen (obwohl ich auch in irgendeine config-Datei weiter oben gesehen hatte, um eine Vorstellung zu kriegen, welche Komponenten da verwendet wurden) - dann hätte es ja auch die Bemerkungen zum statischen Linken hier: https://github.com/Freetz-NG/freetz-ng/issues/349#issuecomment-1042669132 nicht gebraucht. Im Nachhinein fällt dann natürlich auch auf, daß da in den Ausgaben alles das fehlt, was üblicherweise beim dynamischen Laden von Libraries noch zu sehen wäre.

3. Die Ausgabe des `strace` für die PIDs 3438 und 3439 sieht dennoch komisch aus ... im Vordergrund-Thread (PID 3438) sind (ab Zeile 34) wieder/weiterhin mehrere `futex()`-Calls zu sehen, im Hintergrund-Thread (PID 3439) kein einziger (vor dem letzten, wo der Thread hängt). Eingefügt würden die aber - dem `diff` nach - noch vor dem Schreiben des PID-Files (Zeile 14 bei PID 3439), wenn sie im Zusammenhang mit der Initialisierung der Crypto-Backends stehen sollten: https://github.com/OpenVPN/openvpn/blob/eded6a32edcb7d4378140fcbf984fae4d8735029/src/openvpn/openvpn.c#L254 - direkt nach der Rückkehr aus dem `possibly_become_daemon()` wird das PID-File (neu) geschrieben.

Entweder ich liege mit meiner Annahme/Aussage, daß die `futex()`-Calls mit der Initialisierung der OpenSSL-Strukturen im Zusammenhang stehen, total daneben (was ich auch nicht ausschließen will - ich hatte oben ja betont, daß man diesen Calls leider nicht ansehen kann, woher sie kommen und welche Ressourcen damit "geschützt" werden) oder Deine Patches waren bei diesem Build gar nicht wirksam, weil Du Dich beim Patchen "von Hand" irgendwie vertan hast.

Beim darauf folgenden Versuch (mit den PIDs 3046 und 3047) sind dann zwar auch noch bei der ersten PID (3046) die `futex()`-Aufrufe zu sehen (ab Zeile 97), aber immerhin sind die dann im geforkten Thread (3047) auch wieder vorhanden ab Zeile 14 und die Tatsache, daß dahinter (vor dem Schreiben des PID-Files ab Zeile 26) auch versucht wird, eine Konfigurationsdatei für OpenSSL zu lesen (Zeile 24), bestätigt in meinen Augen die Annahme, daß die `futex()`-Calls beim Initialisieren der OpenSSL-Strukturen ausgeführt werden.

Damit bliebe von den zwei oben in Betracht gezogenen Ursachen für das Fehlen der Aufrufe bei der PID 3439 eigentlich nur noch übrig, daß Du Dich beim Patchen vertan hast und falsche/nicht gepatchte Quellen übersetzt wurden. Daher würde ich Dir - bevor Du weitere Tests machst - empfehlen, doch lieber passende Patch-Files zu erstellen und nach `<freetz-ng-dir>/make/openvpn/patches` zu kopieren (oder sie so umzubenennen, daß die Erweiterung nicht länger `.patch` ist, denn dann werden sie auch nicht genutzt - so kann man Patches temporär deaktivieren), damit die beim einem `make openvpn-precompiled` (ggf. mit einem vorherigen `make openvpn-clean`) dann auch automatisch angewandt werden.

Dazu muß man nur mal einen Blick in eines der bereits vorhandenen Patch-Files werfen: https://github.com/Freetz-NG/freetz-ng/blob/master/make/openvpn/patches/current/100-tun-tcp-inetd-nowait.patch - die Dateiangaben (mit denen nach der zu ändernden Datei gesucht wird) sind hier offensichtlich relativ zum Basis-Verzeichnis des entpackten Pakets, was unter `<freetz-ng-dir>/source/target[...]/<package>` zu finden wäre.

Um ein automatisch verwendbares Patch-File zu erzeugen für die eigenen Änderungen, könnte man also so vorgehen:

- `make openvpn-clean; make openvpn-unpacked` -> jetzt liegen die (bereits gepatchten) Quellen im o.a. Verzeichnis vor
- Kopieren der zu ändernden Dateien, damit man nach dem Editieren ein `diff` gegen den alten Stand machen kann (falls der verwendete Editor nicht automatisch Backup-Files erzeugt)
- Editieren der Datei(en), am besten nach vorherigem `cd` in das Verzeichnis mit den Quellen
- `diff -u <file>.old <file> >999-daemon-problem.patch` -> das letzte `>` ist kein Zeichen für einen anzupassenden Teil des Kommandos, sondern die Umleitung der Ausgabe in eine Datei und die Pfadangaben für `<file>` sollten relativ zum "Basisverzeichnis" des entpackten Pakets sein (siehe die ersten Zeilen in `100-tun-tcp-inetd-nowait.patch` oben)
- Kopieren des Patches nach `<freetz-ng-dir>/make/openvpn/patches` (vielleicht auch noch ein `current` im Pfad, so genau habe ich mir nicht angesehen, was da in Freetz-NG mit den beiden Unterverzeichnissen in `patches` gemacht wird), wobei der Dateiname so gewählt werden sollte, daß der eigene Patch NACH allen bereits vorhandenen angewandt wird (die werden in aufsteigender Folge der Namen ausgeführt, daher der Beginn mit `900-...`)

Damit sollte der Patch dann beim nächsten (impliziten oder expliziten) `make openvpn-source` auch einbezogen werden - bei den ersten Versuchen (wenn man's noch nicht so oft gemacht hat) kann man danach ja auch noch einmal nachschauen, ob die Änderungen tatsächlich wie gewünscht ausgeführt wurden - die Anzeige für den Patch (bei "gesprächigem" Freetz-NG) ist zwar ein Zeichen dafür, daß die Patch-Datei benutzt wurde, aber noch keine Garantie dafür [... cut ...] t man sicher sein kann, daß auch an beiden Stellen gepatcht war beim Übersetzen von OpenVPN.

In Anbetracht der Unterschiede zwischen den Protokollen für 3438/3439 und 3046/3047 würde ich jedenfalls vermuten, daß die Frage, ob das nun statisch oder dynamisch gelinkt wurde, gar nicht die einzige Änderung zwischen diesen beiden Versionen war ... in 3439 wurde OpenSSL gar nicht initialisiert im geforkten Thread (nicht ein `futex()`-Call VOR dem Deadlock in der `strace`-Ausgabe) nach meiner Ansicht.

PeterPawn, 2022-02-20:
Und ehe ich das doch noch vergesse: Die zu testenden Änderungen sind auch erst einmal nur ein Provisorium - wenn auch für den hier geplanten Einsatz passend. Denn das Verschieben der OpenSSL-Initialisierung in den Hintergrund-Prozess bereitet ggf. dann Probleme, wenn danach noch irgendwelche Zugriff auf STDIN/STDOUT erfolgen sollen, um eine - mit einem Kennwort gesicherte - Datei für die Crypto-Backends zu öffnen (z.B. irgendeinen RSA-Key bzw. dessen "private" Datei) ... der Hintergrund-Prozess hat ja nur noch `/dev/null` auf den FDs 0 bis 2.

Wenn man also sicher weiß, daß das Problem am `fork()` NACH der OpenSSL-Initialisierung liegt, erst dann kommt die "richtige Arbeit" (wenn man das tatsächlich zum Upstream geben wollte - ich habe immer noch nicht gesucht, ob das Problem bekannt oder gar schon gelöst ist), bei der man dann die Änderungen so umsetzen muß, daß sie auch tatsächlich für alle Fälle passen. 

Aber es könnte auch sein, daß irgendwo in der OpenVPN-Dokumentation explizit steht, daß man bei Verwendung von `--daemon` keine mit Kennwort gesicherten Key-Files benutzen kann ... da wäre also (vor einem Patch, den man am Upstream einreichen will) noch einiges zu klären, wenn man nicht irgendwelchen Wind um etwas machen will, was vielleicht schon gelöst ist.

Also die bisherigen Versuche und Überlegungen bitte auch richtig einordnen ... das ist im Moment nichts weiter als der Versuch, die Ursache zu verstehen und noch nicht zwingend auch die Suche nach der (für alle Lebenslagen passenden) "sauberen" Lösung.

fda77, 2022-02-20:
Vereinfachungen in NG: https://freetz-ng.github.io/freetz-ng/wiki/20_Advanced/make_targets.en.html
 - `make $(pkg)-recompile`    - compiliert immer komplett neu
 - `make $(pkg)-autofix`         - packt immer neu aus (und passt patches an wenn nötig)

stefanpuch, 2022-02-20:
Herzlichen Dank @PeterPawn und @fda77 für die weiteren Hilfestellungen. Ich werde mal versuchen, damit einen Patch zu bauen und somit das Debugging zu vereinfachen.
@PeterPawn da ich die 7490 nur über den Jahreswechsel zum testen hatte, musste ich eine andere mir verfügbare Box suchen, in diesem Falle eine 7520 auf der ich eine 7530 Firmware verwende. Ich bin mir gerade nicht zu 100% sicher meine aber, dass statisch gelinkt der Default in Freetz-NG ist (früher war das in Freetz glaube ich mal anders). Evtl. hätte ich da noch deutlicher drauf hinweisen können...

Nun zu der guten Nachricht. Ich habe nach einem `make distclean` nochmal die minimale Version mit OpenSSL dynamisch gebaut
[config-7530-0729.txt](https://github.com/Freetz-NG/freetz-ng/files/8104183/config-7530-0729.txt)

Damit läuft alles wie man es erwarten würde mit OpenVPN 2.5.5 und OpenSSL 1.1.1m. Die gleiche config bei der OpenVPN statisch gelinkt ist, zeigt das oben ausführlich dokumentierte Verhalten.  Damit hat man zu den o.g. Optionen noch die einfache Möglichkeit, einen (per Daufault) statischen Build abzuwählen. Ggf. nimmt man statisches Linken aus den möglichen Optionen von OpenVPN (temporär) raus, dann sollte kein anderer Nutzer in mein Problem laufen.
Worin liegt eigentlich der genaue Vorteil beim statischen linken? Das Binary wird größer, weil der Code direkt mit inkludiert wird. Ein Vorteil der sich mir daraus erschließt wäre, dass man das Binary einfach komplett gegen ein anderes austauschen kann. Gibt es sonst noch irgend einen Vorteil?

Nochmal zurück zu Debugging: Hilft es, wenn ich mit diesem Zusatzwissen noch einmal straces für beide Versionen ohne die o.g. Änderungen als direkten Vergleich paste (geht ja mit einfachem Reboot dank Peters Bootmanager) oder wäre der genannte Weg dennoch die erste Anlaufstelle?

PeterPawn, 2022-02-20:
@fda77:
Das (neue) `recompile` nutzt aber noch nichts, wenn man erst dabei ist, eigene Patches zu erstellen. Dann braucht man schon noch die oben beschriebenen Zwischenschritte, denn `recompile` faßt ja nur die Schritte `dirclean` und `precompiled` zusammen.

Manchmal braucht man aber durchaus auch `source` bzw. manchmal bräuchte es sogar ein "echtes" `unpack(ed)` (also ohne sofort anschließendes `PKG_PATCH`, was es m.W. so als Target auch in Freetz schon nicht gab: https://github.com/Freetz/freetz/blob/75984884e4353db3b45b04cd37707c6e929ca0ec/make/Makefile.in#L716, denn da sind `source` und `unpacked` ja nur Synonyme für dasselbe Ziel), falls man einen Patch hat, der nicht erst als letzter an der Reihe sein soll, weil er sich vielleicht auf die originalen Quelltexte bezieht oder als Shim dienen soll, damit die bisher existierenden Patches danach wieder funktionieren trotz tiefergreifender Änderungen am Original, die sich mit `autofix` nicht ohne weiteres beheben lassen.

Und wenn Du nicht die Abhängigkeiten für das Entpacken/Patchen bei neuen Files unterhalb von `patches` für ein Paket gestrichen hast, sollte eigentlich auch jedes geänderte Patch-File automatisch zum Löschen und neuen Entpacken/Patchen der Quellen für ein Paket führen.

---

Ich finde das ja auch sehr löblich, daß/wenn Du neue `make`-Targets einführst, die anderen (und sicherlich auch Dir) die Arbeit erleichtern können. 

Hinsichtlich der "Dokumentation" ist das dann aber (trotz Deiner ganzen Bemühungen, die ich durchaus auch sehe) doch eher dünn - ohne selbst ins `Makefile` zu schauen, um zu erkennen, welche Schritte (die hat man als Freetz-Benutzer ja im Laufe der Jahre oft genug gelesen und dafür gibt es auch Erklärungen irgendwo in IPPF-Threads, wie man das machen sollte, inkl. einer README-Datei an der passenden Stelle(!) im originalen Freetz) da mit den neuen Targets zusammengefaßt wurden, ist man auch nicht viel schlauer, wenn man nur unter dem o.a. Link von Dir schaut.

Denn als Du die beiden neuen Targets eingeführt hast (https://github.com/Freetz-NG/freetz-ng/commit/b799caca4f7226da91565c2167e01c14f04ffe6d), hast Du leider vergessen (die `README.Makefiles` von Freetz (https://github.com/Freetz/freetz/blob/master/make/README.Makefiles) ist bei Dir ja offenbar weggefallen und sollte durch andere Dateien/Informationen ersetzt werden - vermutlich diese: https://github.com/Freetz-NG/freetz-ng/blob/master/docs/wiki/60_Development/package_development_makefiles.en.md), die dazu passenden Dokumentationsdateien auch zu aktualisieren.

Daher braucht es schon eigene (zumindest rudimentäre) Kenntnisse vom Umgang mit `make` und `Makefile`s (und angesichts der Tatsache, daß die Targets für die einzelnen Pakete noch per Makro generiert werden (https://github.com/Freetz-NG/freetz-ng/blob/a2d76a3a9f5b963bf5d5798945a142fa2e87042f/make/Makefile.in#L625), wird "rudimentär" üblicherweise auch eher nicht reichen, denn das schließt Makros i.d.R. nicht ein), damit man überhaupt erkennen kann, was sich hinter diesem `recompile` tatsächlich verbirgt. Zumal man das als Freetz-NG-"Benutzer" auch eher selten brauchen wird ... als "Developer" aber schon eher und dann verstehe ich entweder Deine Aufteilung der Dokumentation(en) falsch oder Du hast das doch an der falschen Stelle (auch wenn "Developer" ein "Advanced user" einschließen sollte) untergebracht mit den neuen `make`-Targets.

---

@stefanpuch:
Ich schaue bei Gelegenheit noch einmal in die Trace-Ausgaben ... es interessiert mich schon, was da beim dynamischen Laden von Libraries anders laufen soll, als beim statischen Linken. Vermutlich hängt der DynLoader noch irgendwelche Handler in die Funktionen zum Starten/Beenden von Prozessen ein, die das "Problem" dann maskieren. 

Wobei es hier ja ohnehin kein echtes Problem mit realen Auswirkungen ist, weil in Wahrheit ja nicht wirklich zwei Threads mit demselben Entropie-Pool weiterarbeiten, denn der Vordergrund-Thread wird ja nach dem Forken des Hintergrundprozesses direkt beendet und macht nichts mehr mit weiterem Zufall.

Warum das bei Dir statisch gelinkt wird, müßtest Du aber selbst wissen. Denn die Standardkonfiguration dürfte das nicht sein: https://github.com/Freetz-NG/freetz-ng/blame/a2d76a3a9f5b963bf5d5798945a142fa2e87042f/make/openvpn/Config.in#L106 - da kann man auch sehen, welcher Commit einer Zeile die derzeit gültige Form gab und zumindest in diesem Kconfig-File für das OpenVPN-Paket steht das statische Linken auf `n`. Wenn das nicht durch irgendwelche anderen Pakete/Abhängigkeiten in den Kconfig-Files doch noch automatisch auf `y` geändert wird, dürfte es bei einer "frischen" Konfiguration auch bei Dir nicht so sein. Warum eine Einstellung letztlich gesetzt oder gelöscht ist als Voreinstellung, kann man sich bei der "Hilfe" (im `mconf`) für diese Einstellung anzeigen lassen.

---

Aber zumindest würde auch das wieder eine weitere Erklärung dafür liefern, wieso das Problem erst von Dir bemerkt wurde - wenn andere Freetz-NG-Benutzer mit der dynamisch gelinkten Version arbeiten und die funktioniert, dann haben die anderen eben auch dieses Problem nicht.

Ansonsten hat das statische Linken innerhalb von Freetz-NG nicht wirklich Vorteile (außer man linkt gegen Bibliotheken, die NUR für dieses eine Paket mit spezieller Konfiguration gebaut wurden). 

[... cut ...] ne Libraries ersetzen können oder bereits vorhandene von den dynamisch gelinkten Binaries auch genutzt werden können.

Wenn Du also Freetz-NG auch auf der Box benutzt, bringt es für Dich keinerlei Nachteile, wenn Du OpenVPN dynamisch gelinkt verwendest - zumal in Freetz-NG-Images ohnehin viele Libraries doppelt bzw. noch einmal gesondert für Freetz-Pakete vorliegen, weil die AVM-Versionen und die von Freetz auf unterschiedlichen Ständen (und manchmal sogar auf unterschiedlichen Paketen) beruhen.

fda77, 2022-02-20:
Hm, die README.Makefiles ist mir noch nie über den Weg gelaufen... Und wenn hätte ich sie vermutlich nach developer unter docs/ verschoben :)
Ich glaube es geht darin aber mehr um "Each package must define the following targets:"
Wenn man genauer hinschaut ist die aber auch nicht vollständig und zb distclean fehlt.

Die verlinkte Seite mit den aufrufbaren targets kann man sich auch einfach mit "make help" anzeigen lassen.

PeterPawn, 2022-02-20:
> Die verlinkte Seite mit den aufrufbaren targets kann man sich auch einfach mit "make help" anzeigen lassen.

Und das hilft dann dabei zu verstehen, was genau man sich unter `recompile` (oder auch unter `autofix`) vorzustellen hat? 🤔

fda77, 2022-02-20:
Fragst du mich? Klar hilft es mir, ich habs geschrieben :D
Optimal wäre wenn Wiki, Hilfe etc von sonst wem geschrieben würden. Solang das niemand macht muss es reichen

PeterPawn, 2022-02-20:
> Klar hilft es **mir**

Das war natürlich eher die Frage für die "Zielgruppe". Wenn's nur für Dich wäre, bräuchte es ja vermutlich keine der beiden neuen Zeilen in `make help` und wenn's dann doch für andere sein sollte, ist es eher witzlos, weil man ohne eigene Nachforschungen eben nicht erkennen kann (anhand Deiner erklärenden drei Worte), was da nun genau passieren würde, wenn man diese `make`-Targets benutzt. 

Zwar muß das auch nicht alles direkt in die Anzeige beim `make help` (sonst würde die auch zu lang), aber ein Link hier und da auf die passende Beschreibung unterhalb von `docs` (allerdings auch als HTTP(S)-Link, damit man den gleich anklicken/kopieren kann) würde sicherlich auch nicht schaden und wenn Du Probleme damit hast, den statisch auf einen einzelnen Anbieter für Git-Hosting zeigen zu lassen, mußt Du Dir eben etwas einfallen lassen, wie Du den dynamisch erzeugst, so daß er beim Bezug über GitLab oder Bitbucket dann auch dorthin zeigt. Aber der Git-Clone hat ja auch eine `remote`-Property ...

fda77, 2022-02-20:
Es gibt diverse Link zu github.io. Das Problem ist nur dass niemand inklusive mir dazu Lust hat

PeterPawn, 2022-02-20:
OK, das ist auch ein Standpunkt. 

Nur mußt Du Dich dann halt auch nicht wundern, wenn niemand von diesen Neuerungen so richtig Notiz nimmt. 

Wie ich schon mehrfach schrieb: Wenn man sich erst jede Änderung selbst dadurch erarbeiten muß, daß man alle neuen Commits Schritt für Schritt nachvollzieht, dann verliert man (zumindest ich) halt auch die Lust und wird dann auch mal von Änderungen überrascht (wie die neue Abhängigkeit von `ldconfig`, die Du m.E. noch einmal überdenken solltest, denn die Standard-Pakete der Distros installieren das eben üblicherweise an einer Stelle, wo es nur mit Superuser-Rechten zu benutzen wäre) bzw. erklärt Dritten etwas so, wie es VOR solchen Änderungen noch war und liegt damit falsch ... woraufhin man es dann besser gleich bleiben läßt (oder lassen sollte), außer man investiert die Zeit und macht sich mit allen diesen Änderungen vertraut, bevor man anderen zu helfen versucht.

fda77, 2022-02-20:
Was sollte man denn an ldconfig überdenken? Es wird zum prerequisites-check genutzt und ist damit nötig: https://github.com/Freetz-NG/freetz-ng/blob/master/tools/prerequisites#L17
Wenn es die Binary nicht gibt, gehen die Tests nicht

Mir ist klar dass 90% der Meinung sind ich hab 100%  zu erledigen, diese Meinung teile ich aber nicht

PeterPawn, 2022-02-22:
Entweder Du hast tatsächlich nicht verstanden, daß man bei den meisten Systemen das `ldconfig` nur mit zusätzlichen, eigenen Aktionen so installiert bekommt, daß es ein nicht-privilegierter Benutzer auch verwenden kann oder Du hast halt einige merkwürdige Einstellungen, was die "Zuverlässigkeit" des Freetz-NG-Projekts angeht. 

Wenn es von heute auf morgen durch eine Änderung am Projekt für den durchschnittlichen Linux-Benutzer nicht mehr möglich ist, die "Vorbedingungen" für die Verwendung von Freetz herzustellen, dann kratzt das in meinen Augen auch gehörig am Vertrauen, daß die Freetz-NG-Benutzer dem Projekt entgegen bringen (sollen). Da helfen dann auch alle denkbaren "Workarounds" nicht wirklich - auch die muß der Benutzer erst mal finden.

Ich habe jedenfalls (bisher und im Zusammenhang mit dem Commit, der `ldconfig` als Voraussetzung hinzugefügt hat) noch nichts gelesen, WIE das ein (meinetwegen auch etwas ungeübter) Linux-Benutzer nun so installieren soll, daß es auch von Freetz-NG gefunden wird und genutzt werden kann. 

Genau deshalb bezog ich mich oben auch auf Ubuntu "focal" und den Inhalt des dortigen `libc-bin`-Pakets, wo das Programm `ldconfig` nach `/sbin/ldconfig` installiert wird und mehr kann ein "durchschnittlicher Benutzer" dann auch mit der Paketverwaltung alleine auch nicht erreichen. 

Nun gäbe es ja unzählige Wege, wie man das auch für normale Benutzer zugänglich machen könnte ... von `sudoers` (womit man das aber auch passend aufrufen müßte aus Freetz heraus) über das Anlegen einer Kopie in einem erreichbaren Verzeichnis bis zu anderen Optionen, die ich vergessen habe zu erwähnen.

Zwar kann man sich jetzt auch auf den Standpunkt stellen: "Dann müssen sie sich halt selbst einen Kopf machen - ich bin ja nicht deren Kindermädchen." ... aber wie das dann genau dazu paßt, daß es in Freetz-NG irgendwelche Anleitungen gibt, wie man die Vorbedingungen installieren soll (https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md#ubuntu), erschließt sich mir nicht wirklich - zumindest dürfte das jetzt nicht mehr so richtig zum "realen Leben" passen (außer ich irre mich fürchterlich und `/sbin` ist unter Ubuntu eben doch auch ohne Superuser-Rechte zugänglich).

---

Aber das ist auch nicht meine Baustelle ... nimm's also einfach als meine Meinung oder einen Hinweis auf ein potentielles Problem von meiner Seite hin und entscheide selbst, ob Du Dir das noch einmal überlegen willst und da noch etwas kommt - entweder eine Lösung ohne `ldconfig` oder zumindest eine Beschreibung für den durchschnittlichen Freetz-Benutzer (und das ist i.d.R. kein "Linuxer"), wie man diese neue Voraussetzung bereitstellen kann.

Ansonsten vergiß es einfach ... für mich ist das Thema hiermit erst einmal wieder durch.

[18 more replies]

closed 2022-04-23 by fda77