Freetz-NG/freetz-ng #350: RRDstats SmartHome ohne Temperaturanzeige

leo-022, 2021-07-14:
Firmware: 154.07.27 rev88306 	
Freetz: ng-18456M-005524b6a

Leider verhält sich RRDstats SmartHome nicht so wie gewünscht. Alle Temperaturanzeigen fehlen, sowohl bei den Comet DECT als auch bei der FRITZ!DECT 200 (Spannung, Blindleistung usw. sind da).

*.rrd unf *.cvs werden erzeugt. Es scheint, dass die Generierung der png-Bilder in der entsprechenden Routine von `/usr/lib/cgi-bin/rrdstats/main.cgi` fehlschlägt, jedenfalls sind für die Temperaturen keine png vorhanden.

Ich habe wegen der vielen Variablen und Verzeigungen in `main.cgi` noch nicht herausfinden können, wie den nun der endültige Aufruf von `rrdstats` bzw. am Ende von `rrdtool` korrekt lautet, so dass ich es per Kommandozeile probieren könnte und evtl. Fehlermeldungen sehe.

Auch bin ich zu blöd, das Ganze so zu konfigurieren, dass `main.cgi` eine Fehler-Log erzeigen könnte.

fda77, 2021-07-15:
Grundsätzlich funktioniert es, bei mir mit Comet und Dect200 seit 6 Jahren. Daher vermute ich ein Problem in der config. Schau mal in /tmp/flash/rrdstats/. 
[image]

leo-022, 2021-07-15:
Dieses Bild (Wirkleistung, Blindleistung, Scheinleistung) wird bei mir auch korrekt angezeigt, nur bei Temeratur ist kein Bild vorhanden.

in /tmp/flash/rrdstats/ stehen 3 Dateien:
- smarthome.alias
- smarthome.color
- smarthome.kinds

smarthome.alias
```
116300011857|Steckdose
117950522216|Comet 1
109710644096|Comet 2
119590010136|Comet 3
119590061432|Comet 4
```
smarthome.color
```
#EA644A|Steckdose
#54EC48|Comet 1
#48C4EC|Comet 2
#A349A4|Comet 3
#CC7016|Comet 4
```
smarthome.kinds
```
116300011857|AKT
117950522216|HKR
109710644096|HKR
119590010136|HKR
119590061432|HKR
```

fda77, 2021-07-15:
Das sieht bei mir auch so aus, aber bei mir haben die Namen keine Leerzeichen. Es könnte sein dass irgendwo im Script das Quoting nicht sauber ist. Ersetz oder lösche alle Leerzeichen doch mal und dann "rc.rrdstats restart"

leo-022, 2021-07-15:
Das war die Lösung :smile: Leerzeichen in einem Namen verhindert die Erzeugung aller png-Dateien. Nach Änderung zeigt es auch alle anderen Werte an.

Noch eine Frage: Wenn der Zeitraum „Heizung aus“ eingestellt ist, liefert der Heizkörperregler dann auch Temperaturdaten?

Und noch ein Fehler: Obwohl in `sh-alert.def` eine Zeile DESCRIPTION steht, wird im Freetz-Webinterface nichts dergleichen angezeigt.

fda77, 2021-07-15:
Dann müsste mal jemand den Fehler suchen... wie du gemerkt hast sind die rrdstats Script etwas umfangreicher (und uralt).
Ich würde nie ein Leerzeichen im Namen verwenden, da schlecht programmiertes damit gern mal Probleme hat ;-)
Wirf einfach die kompletten AVM-Einstellungen raus. Die sind viel zu unflexibel. Das Mausgeschubse nervt auch. Mit "cron" und "aha.sh" kannst du unendlich viele Schaltzeitpunkte einstellen.
Ich glaub dazu gibt es noch keinen Link: http://fritz.box:90/cgi-bin/index.cgi?pdev=z&refresh=90
Die DESCRIPTION ging wohl bei den Änderungen für die LANG-Einstellungen kaputt

leo-022, 2021-07-18:
Besten Dank für die schnelle Korrektur von `sh-alert.def`.

Wegen des anderen Fehlers habe ich mal gesucht, aber noch leine endgültige Lösung gefunden. Problem ist die Übergabe der Variable `$_ALIAS` mit enthaltenen Leerzeichen (im Beispiel `Steckdose neu`) in `/usr/lib/cgi-bin/rrdstats/main.cgi` (Zeile 1058) an das Programm `rrdtool`.

Der am Ende erzeugte Aufruf von rrdtool (Zeile 1078) lautet dann wie folgt:
`/bin/nice rrdtool graph /var/media/ftp/uStor01/stat/rrdstats/aha_factLeistungsfaktor.png --title  --start now-1d --width 565 --height 226 --vertical-label Wirkfaktor --color SHADEA#ffffff --color SHADEB#ffffff --color BACK#ffffff --color CANVAS#eeeeee80 --slope-mode HRULE:0#000000 -l 0 -u 1 -W Generated on: Sun, 18 Jul 2021 18:45:00 +0200 DEF:fact0=/var/media/ftp/uStor01/stat/persistent/aha_60-116300011857.rrd:fact:AVERAGE LINE3:fact0#DF013A:Steckdose neu\t(min/avg/max/cur)  GPRINT:fact0:MIN:\t%8.3lf GPRINT:fact0:AVERAGE:%8.3lf GPRINT:fact0:MAX:%8.3lf GPRINT:fact0:LAST:\t%8.3lf\n`

Dieser ergibt die Fehlermeldung `ERROR: Could not make sense out of 'neu\t(min/avg/max/cur) '`, welche aus dem Aufruf `LINE3:fact0#DF013A:Steckdose neu\t(min/avg/max/cur) ` rührt. Im Manual gibt es viele Beispiele mit Leerzeichen im Namen; der Aufruf hat dann aber mit `LINE3:fact0#DF013A:"Steckdose neu\t(min/avg/max/cur)"` zu erfolgen.

Wenn ich nun in `main.cgi` die entsprechende Zeile in `LINE3:$kind$_SENSOR_CUR$_COLOR:\"$_ALIAS\t(min/avg/max/cur)\"$NBSP \` ändere, sieht der Aufruf korrekt aus, aber ich erhalte beim Ausführen die Fehlermeldung `ERROR: Could not make sense out of 'neu\t(min/avg/max/cur)" '`.

Merkwürdig ist, dass oben `-W Generated on: Sun, 18 Jul 2021 18:45:00 +0200` ohne " übergeben wird; an der Kommandozeile funktioniert das nur mit `-W "Generated on: Sun, 18 Jul 2021 18:45:00 +0200"`.

fda77, 2021-07-18:
Teste mal ob das so funktioniert. Die Leerzeichen werden durch ein Zeichen ersetzt das genau so aussieht, aber auch ohne Quoting kein neuer String ist. Das Quoting war in rrdstats schon immer seltsam, eigentlich müsste man alles mal überarbeiten

leo-022, 2021-07-18:
Manuell auf meinem Testsystem geändert klappt das so. Alle Namen mit Leerzeichen werden jetzt korrekt ausgewertet. Da werde ich gleich mal ein neues Image bauen.

leo-022, 2021-07-18:
Neues Image funktioniert aus. Besten Dank.

closed 2021-07-18 by leo-022