Freetz-NG/freetz-ng #351: DavFS2 startet nicht automatisch

leo-022, 2021-07-14:
Firmware: 154.07.27 rev88306 	
Freetz: ng-18456M-005524b6a

Ich habe DavFS2 für 3 Konten konfiguriert, Starttyp auf „Automatisch“. Leider erhalte ich nach dem Neustart der Box immer die Fehlermeldung `Starting DavFS2 ... cache-directory not available, failed`.

DavFS2 braucht ja, um zu funktionieren, zwingend einen Cache z. B. auf dem USB-Stick. Nun startet DavFS2 offensichtlich, bevor der USB-Stick erreichbar ist. Ein nachträgliches Starten von DavFS2 funktioniert einwandfrei.

Ich gehe davon aus, dass dieses Problem bei allen Nutzern auftritt. In der Startroutine müsste deshalb der Start so lange verzögert werden, bis das Cache-Verzeichnis auch erreichbar ist.

leo-022, 2021-07-15:
Das nicht erreichbare Cache-Verzeichnis ist nicht das einzige Problem.

Ich habe external installiert und dort den Start von davfs2 nach `rc.external` verlagert: `DavFS2 is disabled`. Es muss also zusätzlich noch gewartet werden bis eine Verbindung zum dav-Host besteht.

fda77, 2021-07-15:
Das betrifft alle Packages die Datein auf USB benötigen. Man kann external aktivieren und bei "eigene services" dort welche eintragen die erst zusammen mit USB gestartet werden sollen.
Leider hat es in den letzten Jahren noch niemand geschaffte das im Wiki zusammenzufassen.
Für remote Dateisysteme die nicht immer verfügbar sind wie nfs/cifs/davfs, autofs verwenden

closed 2021-07-15 by fda77