Freetz-NG/freetz-ng #353: [dnsmasq] chown: unknown user/group nobody:nobody

harryboo, 2021-07-20:
I have made an actual version of freetz for 7490.

on 2 of 5 updated machines i see this message in MOD-LOGS.
DNSMASQ works as i can see.

[image]

fda77, 2021-07-31:
Dnsmasq created this user if it does not exist: https://github.com/Freetz-NG/freetz-ng/blob/master/make/dnsmasq/files/root/etc/init.d/rc.dnsmasq#L102
But dnsmasq is started automatically before multid and maybe avm deletes the user again... Avm does not know how useres are created on linux and just replaces the whole passwd file like a beginner
You could check /etc/passwd and `/var/log/freetz.log` to have a timeline. Users are loaded by Freetz in MODLOAD since a short time

fda77, 2021-10-01:
As it occured only 1 time for 1 person it should not matter

closed 2021-10-01 by fda77