Freetz-NG/freetz-ng #357: voipd and Wireguard : sipiface_tunnel not working

Greg57070, 2021-07-30:
Hello guys,

I'm trying to connect the AVM voipd daemon to my SIP server over WireGuard.

As long as the wireguard interface is not renamed to "tun0" in /etc/init.d/rc.wireguard, it's not working (and only tun0 ! Even if the interface is called tun1 or tun_wg0, it's not working).

Is there any other sipiface that can handle communication over wg0 ? (I've tried with sipiface_automatic, sipiface_homenet, sipiface_tunnel with no sucess)

Best regards,

Thank you all for the good work with freetz-ng, keep up with it ! :)

fda77, 2021-07-31:
The whole routing things inside the closes-source avm daemons is very creepy. Maybe it helps to allow "accesses by internet" for the voip user?
I've now running asterisk which fixed all problems i had before with avm voipd ^^

PeterPawn, 2021-08-12:
The `sipiface_<something>` values are translated by `bin/voipd` itself to the final interface name ... one of `dsl`, `mta0`, `lan` or `tun0`.

You could try to patch the daemon file to use `wg0` instead - but beware AVM's copyright for this file. Under Germany's law such changes have to be made AFTER loading the file to RAM, so that no persistent changes are applied to the original file.

You could use the `devmem` applet of BusyBox with some magics (locating the load address from `procfs` entry for `voipd` or searching the whole memory (allocated to `voipd`) for the string `tun0´ or something else) to change this interface name. 

But I'd bet it would be easier (and imho still conforming to copyright laws) to patch the file at a volatile filesystem location (e.g. below `/var/tmp`) and call it from there (with some changes to the right shell scripts) - so it gets patched before any configuration file will be processed.

Greg57070, 2021-08-13:
To avoid binary modification, I've propose a pull request which allow defining interface name.
I just test since 2 weeks a WG connection named tun0 without any issue with voipd

fda77, 2021-08-13:
You could add some lines to https://freetz-ng.github.io/freetz-ng/make/wireguard.html to still know the possible names in some weeks and when it should be changed. The file is in docs/ directory

Greg57070, 2021-08-15:
With wg interface named as tun0, voipd works well.

I can read (almost) german, but I can't write nice sentences in german.
Do you know if there is a french community of freetz users ?

closed 2021-08-15 by Greg57070

fda77, 2021-08-15:
Then just write in english