Freetz-NG/freetz-ng #359: 7530 ax support possible ?

ds2k5, 2021-08-05:
Hi,
is it possible to get 7530 ax support in the future ?
thanks

fda77, 2021-08-05:
It is supported and if you cant find it you shluld not use it as nobody else tried it with real hardware and some bugs have to be fixed

closed 2021-08-05 by fda77

Surfer-76, 2021-12-09:
Wenn ich mich hier anschliessen kann.

Gibt es was neues? Bekomme bald eine günstige 7530AX und würde diese mit freetz-ng ausstatten. Im Moment habe ich beim Bauen Probleme mit libpython. Wusste auch nicht, dass es bei diesr Box ein neues Filesystem gibt, welches noch nicht so erforscht ist. Hat evtl. jemand für mich ein gebautes Image, dass wenn die Box kommt, flashen könnte?

fda77, 2021-12-10:
https://freetz-ng.github.io/freetz-ng/SUPPORT.html#untested