Freetz-NG/freetz-ng #361: Korrektur an aha.sh

leo-022, 2021-08-11:
Es gibt bei aha.sh einen kleinen Schönheitsfehler, woran ich bei meiner ersten Benutzung auch gescheitert bin, siehe #360.

Bei dcmd gibt es nur 2 zusätzliche Parameter, man kann command oder Zahlenwerte eingeben. Außerdem gibt es bei den HKR noch die Werte 253 für off und 254 für on. Das sollte man in der Usage-Zeile angeben:

`Usage: $0 <alias|fancy|small|battery|modus|gradc|translate <ain|name>|docmd <device> <command|-1|0|1|16-56|253|254|8,0-28,0>`

Laut neuer AVM-Dokumentation gibt es bei `functionbitmask` jetzt ein zusätzliches Bit 18 für Rollladensteuerung, Bit 16 sollte dann allgemeiner als z. B. "Level veränderbar" bezeichnet werden. Außerdem kann Bit 12 eindeutig als Kennzeichnung für eine auf der Box angelegt Gruppe von SmartHome-Geräten identifiziert werden. Hier müsste das Fragezeichen entfernt und der Text in "Gruppe" geändert werden.

Außerdem wäre es gut, wenn bei HKR auch on/off bzw. 1/0 akzeptiert werden würde (mit Umsetzung in die Werte 254 bzw. 253).

Perspektivisch sollte man darüber nachdenken, auch die Funktionen für Level- und Farbänderungen (Lampen/Rolläden) mit aufzunehmen. Mangels der entsprechenden Geräte kann ich das zur Zeit leider nicht testen.

fda77, 2021-08-11:
> Bei dcmd gibt es nur 2 zusätzliche Parameter, man kann command oder Zahlenwerte eingeben

Das steht doch so in der 1zeiligen Usage: `docmd <device> <command|-1|0|1|16-56|8,0-28,0> `

253+254 kannst du selbst (ohne git, im Browser) hinzufügen:
 - https://github.com/Freetz-NG/freetz-ng/blob/master/make/rrdstats/files/root/usr/bin/aha.sh#L281
 - oben den edit-Stift
 - ändern
 - preview-tab zum prüfen
 - save/commit/pr oder wie das heisst
 
Das erstellt im Hintergrund einen fork und branch für einen pr

>  Bit 16 sollte dann allgemeiner als z. B. "Level veränderbar"

Welche "Level" gibt es denn noch ausser Dimmbar? Meinst du "Rolladen 25%"? Also etwas wie ein "Potentiometer"?

> zusätzliches Bit 18

Ich mit dem Script angefangen als es noch keine Comet gab. Da kommt öfter was neues dazu

> HKR auch on/off bzw. 1/0

Eigentlich gute Idee, aber nutzt man das wirklich? Unter 16°C fängt es an zu schimmeln... :) Die 253/254 sind übrigens von AVM vorgegeben

Neue Geräte kann ich ohne die Hardware nicht hinzufügen. Ich geb für AVM Sachen aber kein Geld mehr aus. Da gibt es besseres was dazu noch günstiger ist

fda77, 2021-08-12:
Der Commit war schon vor 4 Wochen!
on/off für HKR ist mir auch zu stressig.. Für Steckdosen gibt es das, weil es von AVM aus der Befehl ist. Das gaze ist von avm leider nicht so einheitlich bzw durchdacht.
Der Rest von hier sollte in den aktuellen Commits drin sein