Freetz-NG/freetz-ng #377: Busybox 1.34 + crc32 patch

fda77, 2021-09-05:
@PeterPawn
Es gibt in der neuen Versionen https://github.com/Freetz-NG/freetz-ng/commit/9d8f4c92b8abc66d0afce9fdc00f74a7a7dd9660 ein Applet crc32 das sich auch in der Datei von chksum befindet. Dadurch hat dein Patch https://github.com/PeterPawn/YourFreetz/commit/1cba3b774adf4bf488facc60e4b4a0b1429a98a2 nicht mehr gepasst und ich hab ihn weggelassen.  Es wird scheinbar in Freetz nicht genutzt und für die Checksumme kann auch von https://github.com/PeterPawn/decoder berechnet werden.
Falls du ihn doch noch benötigst, bitte anpassen

PeterPawn, 2021-09-05:
Paßt schon ... der war tatsächlich nur dazu da, die Prüfsumme für eine Export-Datei berechnen zu können, z.B. hier: https://github.com/PeterPawn/YourFritz/tree/main/export - die parallel dazu liegende Shell-Implementierung braucht extrem lange (bei entsprechenden Datenmengen) und die ebenso vorhandenen C-Version braucht dann einen funktionierenden Compiler. Beides auf einer FRITZ!Box echte Hürden, daher gab es den Patch-Vorschlag, mit dem man `chksum` um die alternative Implementierung erweitern konnte. Ich habe mir `crc32` nicht angesehen (bisher), aber vermutlich macht das auch nur dasselbe - so viele unterschiedliche Polynome und "Schieberichtungen" für die Daten gibt es ja auch nicht.

fda77, 2021-09-07:
Ich hatte zwar dein Repo oben verlink, der Patch war aber schon lange in Freetz.
Dein alter Patch ist ja nicht so gross. Teile davon scheinen nun direkt in der Busybox zu sein. Wahrscheinlich ist es nur eine kleine Ändernung - falle es jemand braucht
Ich hab deine Beispiele aus "cksum_example_usage" angesehen, die neue Busybox berechnet aber was anderes:
```
$ echo -n 'The quick brown fox jumps over the lazy dog'  | ./busybox crc32
414fa339
```
Mit deinem Patch: `1095738169 43`

PeterPawn, 2021-09-08:
Wie geschrieben: paßt schon. Das neue Applet `crc32` gibt den Wert halt in hexadezimaler Darstellung aus, das Ergebnis ist aber dasselbe. Der zweite Wert beim `cksum` ist die Anzahl der Zeichen und ich wollte mit dem Patch so dicht wie möglich am Original bleiben (https://man7.org/linux/man-pages/man1/cksum.1.html - das Applet orientiert sich ja auch daran), deshalb dieselbe Form der Ausgabe. Das dann noch umzuwandeln (denn die Export-Datei verwendet ja auch den hexadezimalen Wert), ist ein Einzeiler - wobei meine Shell- und C-Implementierung auch beide schon in hexadezimal ausgeben.

fda77, 2021-09-08:
Ach, die Ausgabe ist in einem anderen Zahlensystem... Gut dann stimmen die beiden Ausgaben überein :)
Das kann man dann einfach in der Shell mit `echo $((0x414fa339))` umwandeln.
Dass 43 die Anzahl der Zeichen ist muss man aber wissen

closed 2021-09-08 by fda77