Freetz-NG/freetz-ng #381: freetz web stucks - maybe busybox 1.34

flosch-dev, 2021-09-07:
Ich habe busybox auf 1.34 umgestellt und nun hängt leider das web-interface. Ob es daran liegt kann ich leider noch nicht sagen.
Ich sehe noch die Kopfzeile, aber danach lädt es einfach bis zum timeout weiter ohne einen Inhalt zu empfangen.
Leider passiert auch das gleich beim laden der firmware update site :(

[image]

Gibt es einen Weg über SSH eine neue Firmware zu laden? Ich habe direkt vom gleichen build eine Version mit busybox 1.33 erstellt und wollte diese nun testen.

berndy2001, 2021-09-07:
Lebt das AVM-Webif noch? Dann probiers mal dort.

fda77, 2021-09-07:
Ist bei mir auch so. Der Aufbau des Menüs hängt: `/bin/sh -x /usr/lib/mod/menu-update`
Wenn du das Image mit dem gleichen Computer gebaut hast (mit dem gleichen Key signiert hast) kannst du Freetz über das AVM-Webif flashen.
Oder per SSH schauen: `cat /proc/sys/urlader/environment` und ändern: `echo "linux_fs_start 0oder1" > /proc/sys/urlader/environment` + reboot

flosch-dev, 2021-09-08:
Danke für den Hinweis, habe über SSH die anderen partition geladen und dann nochmal ein FW upgrade durchgeführt.
Schon eine Idee was da genau hängt?

fda77, 2021-09-08:
Ich hab keine Ahnung weshalb genau diese 2 `source` Befehle "hängen". In den Dateien befindet sich nur ein `title='abc'`. Von einem eigenen Script aus funktioniert es. Es kann ein Bug in BusyBox sein (die .0 wird da immer experimentell genannt) oder daran liegen wie das ganz Freetz-Webif geschrieben ist. Buehmann hatte überall sehr komplizierte Konstrukte verwendet.

fda77, 2021-10-01:
https://github.com/Freetz-NG/freetz-ng/issues/381

fda77, 2021-10-01:
Sollte behoben sein, müsste aber noch jemand testen!
 - https://git.busybox.net/busybox/commit/?h=1_34_stable&id=76ef4391548ded8db511e2f7f8f35a3010be7ec5
 - https://github.com/Freetz-NG/freetz-ng/commit/33e3edc254f06a61f07fff9aea895ced53335f50
 - https://github.com/Freetz-NG/freetz-ng/commit/05bd379b395f399a91661a1dbf449a28ecc1aef2

closed 2021-10-01 by fda77