Freetz-NG/freetz-ng #383: fakeroot 1.26 compilieren mit altem Linux

fda77, 2021-09-21:
Durch den Versionbump von fakeroot kann es nicht mehr mit zb Ubuntu 14 x64 gebaut werden, und somit auch keine dl-tools mehr. Vielleicht hat jemand Lust das Problem zu beheben oder die nächste Version hilft.
Ansonsten könnte man es reverten: [fakeroot-revert.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/7202780/fakeroot-revert.patch.txt) oder vorübergehend `pseudo` nutzen
 
Fehlermeldungen:
[1st-make.txt](https://github.com/Freetz-NG/freetz-ng/files/7202783/1st-make.txt)
[2nd-make.txt](https://github.com/Freetz-NG/freetz-ng/files/7202784/2nd-make.txt)

Fehlermeldungen mit [010-fix-xattr-prototypes.ubuntu.patch](https://github.com/Freetz-NG/freetz-ng/commit/5739a0f827c96863d7d93c782e8092bab0c07138#diff-ccb4f40d9468b018de469f4d4c6efea9f50f18071632cc6b3e66c0586f963bdb):
[1st-with--fix-xattr-prototypes.ubuntu.txt](https://github.com/Freetz-NG/freetz-ng/files/7202796/1st-with--fix-xattr-prototypes.ubuntu.txt)
[2ndt-with--fix-xattr-prototypes.ubuntu.txt](https://github.com/Freetz-NG/freetz-ng/files/7202797/2ndt-with--fix-xattr-prototypes.ubuntu.txt)

er13, 2021-09-21:
Hmm, warum hast Du [010-fix-xattr-prototypes.ubuntu.patch](https://raw.githubusercontent.com/Freetz-NG/freetz-ng/88b972a6283bfd65ae1bbf559e53caf7bb661ae3/tools/make/fakeroot-host/patches/010-fix-xattr-prototypes.ubuntu.patch) in https://github.com/Freetz-NG/freetz-ng/commit/5739a0f827c96863d7d93c782e8092bab0c07138 entfernt?

er13, 2021-09-21:
Der [010-fix-xattr-prototypes.ubuntu.patch](https://raw.githubusercontent.com/Freetz-NG/freetz-ng/88b972a6283bfd65ae1bbf559e53caf7bb661ae3/tools/make/fakeroot-host/patches/010-fix-xattr-prototypes.ubuntu.patch) wird weiterhin benötigt. Sein Inhalt soll zusätzlich wie folgt erweitert werden:

```diff
--- wrapdef.h
+++ wrapdef.h
@@ -145,22 +145,22 @@
 ssize_t (*next_fgetxattr)(int fd, const char *name, void *value, size_t size)=tmp_fgetxattr;
 #endif /* HAVE_FGETXATTR */
 #ifdef HAVE_SETXATTR
-ssize_t (*next_setxattr)(const char *path, const char *name, void *value, size_t size, int flags)=tmp_setxattr;
+int (*next_setxattr)(const char *path, const char *name, const void *value, size_t size, int flags)=tmp_setxattr;
 #endif /* HAVE_SETXATTR */
 #ifdef HAVE_LSETXATTR
-ssize_t (*next_lsetxattr)(const char *path, const char *name, void *value, size_t size, int flags)=tmp_lsetxattr;
+int (*next_lsetxattr)(const char *path, const char *name, const void *value, size_t size, int flags)=tmp_lsetxattr;
 #endif /* HAVE_LSETXATTR */
 #ifdef HAVE_FSETXATTR
-ssize_t (*next_fsetxattr)(int fd, const char *name, void *value, size_t size, int flags)=tmp_fsetxattr;
+int (*next_fsetxattr)(int fd, const char *name, const void *value, size_t size, int flags)=tmp_fsetxattr;
 #endif /* HAVE_FSETXATTR */
 #ifdef HAVE_REMOVEXATTR
-ssize_t (*next_removexattr)(const char *path, const char *name)=tmp_removexattr;
+int (*next_removexattr)(const char *path, const char *name)=tmp_removexattr;
 #endif /* HAVE_REMOVEXATTR */
 #ifdef HAVE_LREMOVEXATTR
-ssize_t (*next_lremovexattr)(const char *path, const char *name)=tmp_lremovexattr;
+int (*next_lremovexattr)(const char *path, const char *name)=tmp_lremovexattr;
 #endif /* HAVE_LREMOVEXATTR */
 #ifdef HAVE_FREMOVEXATTR
-ssize_t (*next_fremovexattr)(int fd, const char *name)=tmp_fremovexattr;
+int (*next_fremovexattr)(int fd, const char *name)=tmp_fremovexattr;
 #endif /* HAVE_FREMOVEXATTR */
 
 #ifdef HAVE_FSTATAT
```

Ungetestet, vom Lesen der von Dir angehängten Logs und des Diffs zwischen 1.25 und 1.26.

fda77, 2021-09-21:
Sieht gut aus! Mit Ubuntu 14 x64, Kernel 4.4:
```
l `find tools/build/ -name *fake*`
-rwxr-xr-x 1 freetz freetz 33646 Sep 21 23:13 tools/build/bin/faked
-rwxr-xr-x 1 freetz freetz  3978 Sep 21 23:13 tools/build/bin/fakeroot
-rwxr-xr-x 1 freetz freetz 52727 Sep 21 23:13 tools/build/lib32/libfakeroot-0.so
-rw-r--r-- 1 freetz freetz 45834 Sep 21 23:13 tools/build/lib32/libfakeroot.a
-rwxr-xr-x 1 freetz freetz   996 Sep 21 23:13 tools/build/lib32/libfakeroot.la
lrwxrwxrwx 1 freetz freetz    16 Sep 21 23:13 tools/build/lib32/libfakeroot.so -> libfakeroot-0.so
-rwxr-xr-x 1 freetz freetz 65603 Sep 21 23:13 tools/build/lib/libfakeroot-0.so
-rw-r--r-- 1 freetz freetz 63840 Sep 21 23:13 tools/build/lib/libfakeroot.a
-rwxr-xr-x 1 freetz freetz   994 Sep 21 23:13 tools/build/lib/libfakeroot.la
lrwxrwxrwx 1 freetz freetz    16 Sep 21 23:13 tools/build/lib/libfakeroot.so -> libfakeroot-0.so
```
Mit deinem noch älteren Kernel 3 System (was ist das denn genau?) muss du schauen ob das ausreicht
Ich hatte den 010-fix-xattr-prototypes.ubuntu.patch gelöscht weil mir nicht ganz klar war dass er für ältere Systeme ist und mit aktuellem Fedora es auch ohne lief

fda77, 2021-09-21:
Danke!

closed 2021-09-21 by fda77

er13, 2021-09-22:
> Mit deinem noch älteren Kernel 3 System (was ist das denn genau?)

Linux Mint 17.x basierend auf Ubuntu 14.04.6 LTS. Habe den Kernel inzwischen auch auf 4.4 upgedatet.

fda77, 2021-09-22:
I did not know Mint uses so old versions. As i needed to reinstall Fedora due to x32 -> x64 switch, i considered&tested Mint