Freetz-NG/freetz-ng #386: Tor makes system unresponsive (despite using softlimit)

jpsollie, 2021-09-30:
I'm completely out of ideas how to trace the cause here:
Tor makes my fritz 7950 unresponsive, even though I applied the softlimit 256M feature.
These are the latest kernel logs logged to external syslog server, before panic:
```
Sep 28 07:02:32 router/router kernel: [72927.632550][1][1]system-load  93% loadavg 0.21 0.8 0.2 - task runtime:2% max:kworker/0:1H 0%  softirqs:2.13% (ce_task
let [qca_ol] 100%) pgstat: sum=130308 free=23254 slab=6520 alloc=10283/s fault=3538/s ai_sys:0.70/s 0x88bc08                                                  
Sep 28 07:02:40 router/router kernel: [72935.210300][2][2]system-load  41% loadavg 0.65 0.17 0.6 - task runtime:63% max:tor 62%  softirqs:2.02% (do_cbm_taskle
t 73%) pgstat: sum=127725 free=20523 slab=6519 alloc=7605/s fault=3897/s (sleep 2)                                                                            
Sep 28 07:02:42 router/router kernel: [72937.629467][1][1]system-load  64% loadavg 0.65 0.17 0.6 - task runtime:98% max:tor 97%  softirqs:1.20% (ce_tasklet [q
ca_ol] 62%) pgstat: sum=129793 free=22567 slab=6519 alloc=1032/s fault=1038/s (sleep 4)                                                                       
Sep 28 07:02:44 router/router kernel: [72939.001677][3][3]system-load  99% loadavg 1.0 0.26 0.8 - task runtime:99% max:tor 98%  pgstat: sum=127468 free=20211 
slab=6521 alloc=7291/s fault=4487/s (sleep 0)                                                                                                                 
Sep 28 07:02:46 router/router kernel: [72940.972629][0][0]system-load  96% loadavg 1.0 0.26 0.8 - task runtime:99% max:tor 99%  softirqs:1.91% (TIMER 100%) pg
stat: sum=127280 free=20026 slab=6523 alloc=11655/s fault=4825/s (sleep 3)                                                                                    
Sep 28 07:02:50 router/router kernel: [72945.210325][2][2]system-load  99% loadavg 1.32 0.33 0.11 - task runtime:99% max:tor 97%  pgstat: sum=128122 free=2094
2 slab=6524 alloc=1256/s fault=1219/s (sleep 2)                                                                                                               
Sep 28 07:02:52 router/router kernel: [72947.632515][1][1]system-load  99% loadavg 1.32 0.33 0.11 - task runtime:99% max:tor 98%  softirqs:1.11% (ce_tasklet [
qca_ol] 67%) pgstat: sum=126529 free=19323 slab=6525 alloc=938/s fault=927/s (sleep 4)                                                                        
Sep 28 07:02:54 router/router kernel: [72949.001671][3][3]system-load  99% loadavg 1.62 0.41 0.14 - task runtime:99% max:tor 98%  pgstat: sum=128468 free=2126
9 slab=6525 alloc=721/s fault=732/s (sleep 0)                                                                                                                 
Sep 28 07:02:56 router/router kernel: [72950.972645][0][0]system-load  99% loadavg 1.62 0.41 0.14 - task runtime:99% max:tor 98%  softirqs:1.12% (TIMER 89%) p
gstat: sum=131008 free=23890 slab=6523 alloc=859/s fault=824/s (sleep 3)                                                                                      
Sep 28 07:03:00 router/router kernel: [72955.211068][2][2]system-load  73% loadavg 1.49 0.40 0.14 - task runtime:33% max:tor 32%  pgstat: sum=130723 free=2362
7 slab=6523 alloc=386/s fault=313/s (sleep 2)                                                                                                                 
Sep 28 07:03:02 router/router kernel: [72957.633868][1][1]system-load  50% loadavg 1.49 0.40 0.14 - task runtime:7% max:ctlmgr 2%  softirqs:1.16% (ce_tasklet 
[qca_ol] 71%) pgstat: sum=130741 free=23633 slab=6525 alloc=323/s fault=164/s (sleep 4)                                                                       
Sep 28 07:03:04 router/router kernel: [72959.004775][3][3]system-load  27% loadavg 1.37 0.40 0.13 - task runtime:1% max:kworker/0:1H 0%  pgstat: sum=130754 fr
ee=23640 slab=6525 alloc=297/s fault=131/s (sleep 0)                                                                                                          
Sep 28 07:03:06 router/router kernel: [72960.974043][0][0]system-load  30% loadavg 1.37 0.40 0.13 - task runtime:2% max:kworker/0:1H 0%  softirqs:1.43% (TIMER
 72%) pgstat: sum=130751 free=23638 slab=6526 alloc=161/s fault=21/s (sleep 3)                                                                                
Sep 28 07:06:21 router/router kernel: [73156.028757][1]/proc/tffs: info request: success                                                                      
Sep 28 07:07:00 router/router kernel: [73195.210423][2][2]system-load  53% loadavg 0.34 0.27 0.12 - task runtime:3% max:kworker/0:1H 0%  softirqs:3.58% (do_cb
m_tasklet 26%) pgstat: sum=119855 free=12885 slab=6695 alloc=3861/s fault=4791/s (sleep 2)                                                                    
Sep 28 07:07:02 router/router kernel: [73197.631536][1][1]system-load  52% loadavg 0.34 0.27 0.12 - task runtime:25% max:update_kernel 24%  softirqs:2.35% (ce
_tasklet [qca_ol] 42%) pgstat: sum=119873 free=12885 slab=6695 alloc=3839/s fault=4784/s (sleep 5)                                                            
Sep 28 07:07:04 router/router kernel: [73199.001635][3][3]system-load  37% loadavg 0.34 0.27 0.12 - task runtime:11% max:update_kernel 10%  softirqs:1.42% (TI
MER 81%) pgstat: sum=121532 free=14490 slab=6695 alloc=3885/s fault=4803/s (sleep 0)                                                                          
Sep 28 07:07:14 router/router kernel: [73209.001646][3][3]system-load  99% loadavg 0.44 0.29 0.13 - task runtime:25% max:update_kernel 24%  pgstat: sum=121492
 free=14465 slab=6693 alloc=468/s fault=1/s (sleep 0)                                                                                                         
Sep 28 07:07:24 router/router kernel: [73219.001725][3][3]system-load  71% loadavg 0.53 0.31 0.14 - pgstat: sum=121707 free=14669 slab=6693 alloc=589/s fault=
197/s (sleep 0)                                                                                                                                               
Sep 28 07:07:34 router/router kernel: [73228.870778][2]AVM_WATCHDOG_release(hdl=14 dsld): success                                                             
Sep 28 07:07:34 router/router kernel: [73229.001757][3][3]system-load  25% loadavg 0.52 0.32 0.15 - task runtime:3% max:kworker/0:1H 0%  pgstat: sum=137767 fr
ee=30588 slab=6687 alloc=1547/s fault=2313/s (sleep 0)                                                                                                        
Sep 28 07:07:37 router/router kernel: [73232.516447][1]ptm stop                                                                                               
Sep 28 07:07:37 router/router kernel: [73232.521798][1]device ptm0 left promiscuous mode                                                                      
Sep 28 07:07:37 router/router kernel: [73232.521944][1]audit: type=1700 audit(1632805657.688:14): dev=ptm0 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses
=4294967295                                                                                                                                                   
Sep 28 07:07:37 router/router kernel: [73232.531651][1]device xfrm left promiscuous mode                                                                      
Sep 28 07:07:37 router/router kernel: [73232.531809][1]audit: type=1700 audit(1632805657.696:15): dev=xfrm prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses
=4294967295                                                                                                                                                   
Sep 28 07:07:37 router/router kernel: [73232.537581][1]internet: ipmasq_flush: start: 396 mappings, 578 entries.                                              
Sep 28 07:07:37 router/router kernel: [73232.551008][1]internet: ipmasq_flush: end: 10 mappings, 0 entries.                                                   
Sep 28 07:07:37 router/router kernel: [73232.577287][1]internet: ipmasq_flush: start: 10 mappings, 0 entries.                                                 
Sep 28 07:07:37 router/router kernel: [73232.577602][1]internet: ipmasq_flush: end: 10 mappings, 0 entries.                                                   
Sep 28 07:07:37 router/router kernel: [73232.583555][1]avmvpn: ipmasq_flush: start: 1 mappings, 1 entries.                                                    
[... cut ...]
Sep 28 07:07:42 router/router kernel: [73236.989174][3]Emergency Remount complete                                                                             
Sep 28 07:07:42 router/router kernel: [73237.084631][2]UBIFS (ubi0:3): un-mount UBI device 0                                                                  
Sep 28 07:07:42 router/router kernel: [73237.084708][2]UBIFS (ubi0:3): background thread "ubifs_bgt0_3" stops
```
Anyone? an idea how I could at least see what happens? no matter if I have to make it panic again, but I need to know why it crashes!

fda77, 2021-09-30:
It seems tor causes a high syslemload and the avm watchdog causes a reboot. Are you runningn an exit node?

This could be the opened connections, you could check with "netstat"
```
ipmasq_flush: start: 396 mappings, 578 entries.
```

jpsollie, 2021-09-30:
> It seems tor causes a high syslemload and the avm watchdog causes a reboot. Are you runningn an exit node?
No, but I planned to once it seems stable ^^.  BTW: the watchdog doesn't cause a reboot: I have to power cycle manually
> 
> This could be the opened connections, you could check with "netstat"
any idea how I could get a dump from it before avm watchdog triggers / while it triggers?
> 
> ```
> ipmasq_flush: start: 396 mappings, 578 entries.
> ```
that indeed looks pretty evil ... any idea how to limit it? or is a control group the only solution here?

fda77, 2021-10-01:
May you should first check if there are relly so much opened connections, and by which program.
Older Fritzboxens had a limit around ~300 connections when the router stop responding

jpsollie, 2021-10-03:
> May you should first check if there are relly so much opened connections, and by which program. Older Fritzboxens had a limit around ~300 connections when the router stop responding

Absolutely. But is there any way I could trigger that solution? Ideal solution would be I could log it to syslogd, who will forward it to logging server

fda77, 2021-10-03:
I dont know. The classic "stress test programs" where eDonkey and BitTorrent :)