Freetz-NG/freetz-ng #387: Conflict between multid and dnsmasq

snowball77, 2021-09-30:
I was upgrading from freetz with 7.12 on a 7560 to freez-ng with 7.28. I ran into the problem that with the new version multid and dnsmasq are in conflict. In the old image it was ok to start dnsmasq before multid and then multid simply did not serve DNS. However, in 7.28 multid crashed when the 53 port is already used by dnsmasq. That is detected by the watchdog and then the box reboots after approx. 5min. Disabling dnsmasq or moving it to a different port works as workaround. However, now my client machines use multid as dns proxy.

Any hints or ideas how to resolve it. I tried to disable DNS in multid but I was unable to find the option in menuconfig.

fda77, 2021-09-30:
Dnsmasq on 53 works for me with all devices, eg 7490 and 7590

snowball77, 2021-09-30:
Maybe it would be good to know the FW version here, as it worked for me with 7.12 but stopped with 7.28

fda77, 2021-09-30:
```
root@FRITZ:/var/mod/root# netstat -anp|grep dnsm
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      3006/dnsmasq
tcp        0      0 :::53                   :::*                    LISTEN      3006/dnsmasq
udp        0      0 0.0.0.0:53              0.0.0.0:*                           3006/dnsmasq
udp        0      0 :::53                   :::*                                3006/dnsmasq
unix  2      [ ]         DGRAM                      4740 3006/dnsmasq
root@FRITZ:/var/mod/root# w
 13:31:42 up 7 days, 12:43,  load average: 0.20, 0.21, 0.16

root@FRITZ:/var/mod/root# cat /etc/.version
07.28
```

snowball77, 2021-10-01:
Ok, this gives hope that it can be made to work. I suggest that I try to compile a firewire again which failed for me before and that I retrieve the failure stack trace from multid and/or AVM watchdog to post it. Maybe from there, we can get an idea that does not work here.

fda77, 2021-10-02:
I think its something with your configuration, eg disable avms dhcp by avm-webif if you want dnsmasq act as dhcp. Or dont enable DoT

thestr4ng3r, 2022-05-22:
I seem to be running into the same issue at the moment. When I stop the multid service, start dnsmasq listening on 53, and run multid by hand, it segfaults like this, even with a nice stacktrace:
```
# multid -d -f
multid: /var/flash/vpn.cfg: is empty
multid: load_config(vpn): file empty - factory default loaded
multid: startup ($Revision$$CompileDate: Oct 26 2021 14:15:22 $)
multid: HWRevision=185
multid: PROFIVOIP detected, we are CPU 1
multid: using 192.168.179.1 on cpu
multid: IP Client detected
multid: opt_without_tr069discover=1, tr069disc=0, AR7CFG->vinax_hsi_lan_ports=(null), multid.tcom_targetarch=0, is_eth_switch_installed()=0, multid.cpmac_no_ata_possible=0, pppoeiface=eth0
multid: bridge_configured: nqos_bridge_active=0, bridge_port_set=0x0, igmp_flt_portset=0x0
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: list[0]=lan
multid: br_get_interfaces(lan)=5
multid: list[1]=guest
multid: mrouter: using IGMPv3 for upstream interface eth0
multid: mrouter: using IGMPv3 for other interfaces
multid: DHCPC on lan
multid: DHCPD on lan:0 skipped, is virtual interface
multid: DHCPC: lan: state init
multid: DHCPC: lan reset ip address
multid: DHCPC: lan: state init, got event discover-sent
multid: DHCPC: lan: state selecting
multid: pcpc_create_ctx: no pcpserver found
multid: interface lo up.
multid: interface lo carrier up
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: interface eth0 up.
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: interface eth1 up.
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: interface eth2 up.
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: interface eth3 up.
multid: interface wasp up.
multid: interface wasp carrier up
multid: DHCPC on lan
multid: interface lan up.
multid: DHCPC on lan
multid: interface lan carrier up
multid: interface guest up.
multid: interface guest carrier up
multid: interface wlan up.
multid: list[0]=lan
multid: list[1]=guest
multid: br_get_interfaces(guest)=1
multid: interface wlan carrier up
[... 29 lines cut ...]
multid: [bt]  772b8254 dnsd_dot_set_enabled (/sbin/multid at 00040254)
multid:                         Code: 8fb0001c 03e00008 27bd0030 <03e00008> a0852f00 03e00008 90822f00 3c1c0006 279c9dbc
multid: [bt]  7728e0f4 [7728de28] <0+0x7728de28>+0x2cc (/sbin/multid at 00015e28/0x160f4)
multid:                         Code: 8e450048 8e040018 8f9981a8 <0320f809> 0005282b 8e420048 10400004 8fbc0018 24030001
multid: [bt]  77294240 [77293ddc] <0+0x77293ddc>+0x464 (/sbin/multid at 0001bddc/0x1c240)
multid:                         Code: 8fbc0028 8f998034 27395e28 <0411e6f9> 00000000 8fbc0028 24050001 8f90803c 8f998b6c
multid: [bt]  7729f0ac [7729efbc] <0+0x7729efbc>+0xf0 (/sbin/multid at 00026fbc/0x270ac)
multid:                         Code: 8e0400cc afa20014 8e02001c <0320f809> afa20010 1000007b 8fbc0020 00a02025 0411e5a7
multid: [bt]  772a08e4 [7729f574] <0+0x7729f574>+0x1370 (/sbin/multid at 00027574/0x288e4)
multid:                         Code: 02202025 8f99807c 27394310 <0320f809> 00000000 10000229 8fbc0020 8fa20a2c 10400227
multid: [bt]  771a23b0 packetsock_handle_one_packet+0x4d8 (/lib//libewnwlinux.so.2 at 00008ed8/0x93b0)
multid:                         Code: 00001025 8e390020 02002825 <0320f809> 03c02025 8fbc0018 1000ff41 24020001 8e060004
multid: [bt]  770e0108 [770e0090] <0+0x770e0090>+0x78 (/lib//libavmcsock.so.2 at 00047090/0x47108)
multid:                         Code: 8fbc0010 8e1902d4 8e0502c0 <0320f809> 02002025 8fbc0010 8fbf0024 8fb0001c 02202025
multid: [bt]  770e2334 csock_select_with_msecs+0x3ec (/lib//libavmcsock.so.2 at 00048f48/0x49334)
multid:                         Code: 10400015 00003025 8eb901d4 <0320f809> 02a02025 8fbc0010 8f99859c 0320f809 02802025
multid: [bt]  77286368 main+0x2848 (/sbin/multid at 0000bb20/0xe368)
multid:                         Code: 54600001 ac4030c4 8f998d84 <0320f809> 00000000 1000ffda 8fbc0028 8f84803c 8f998938
multid: [bt] finished.
```

No idea why it works for you.

`/etc/.version` is `07.29` and the box is a 7490. I see there is `libmultid` for the exact purpose of making multid bind to different ports: https://github.com/Freetz-NG/freetz-ng/blob/master/make/libs/libmultid/src/libmultid.c However I was not able to get it enabled from menuconfig as it seems to depend on `FREETZ_AVM_HAS_AVMDAEMON_PRELOAD`, which depends on `!FREETZ_SEPARATE_AVM_UCLIBC`, seems to be force-enabled on that config.
I am considering to patch the `multid` binary now, will see how it goes...

thestr4ng3r, 2022-05-22:
Patching the binary works for me, this is what I do:
```
diff --git a/patches/scripts/9999-change_multid_dns_port.sh b/patches/scripts/9999-change_multid_dns_port.sh
new file mode 100644
index 000000000..f64f26f90
--- /dev/null
+++ b/patches/scripts/9999-change_multid_dns_port.sh
@@ -0,0 +1,12 @@
+
+echo1 "patch multid to not listen on 53 for dns"
+
+shasum=($(sha256sum build/original/filesystem/sbin/multid))
+if [ "$shasum" != "d8cf49061b9179112c0780452abb402c20302d4f1368f8ff916bb28ec8bd2e9c" ]; then
+       error 1 "multid sha256sum mismatch, this is specific to FRITZ.Box_7490-07.29.image"
+fi
+
+# patch `addiu a1, zero, 0x35` to use some other port instead. Be careful, because this sign-extends!
+set -xe
+rizin -w -Qc "wx 0036 @ sym.dnsd_init+0x27e; wx 0036 @ sym.dnsd_init+0x422; wx 0036 @ sym.dnsd_init+0x46a" "${FILESYSTEM_MOD_DIR}/sbin/multid"
+set +xe
```

But obviously this is not a general solution, so would still be interesting to know why it works in @fda77's setup or how to get libmultid working. Regarding settings, my box is configured (through the avm settings) to act only as a dhcp client, so server is disabled there.

fda77, 2022-05-22:
Interesting solution, but only for this 1 firmware. For all, this could prevent many compilications. Maybe you want also block other ports eg 5353?

Since FOS 7.20 (and somtime befor 6 or so) dnsmasq is started befort multid and so the port is not used, multid just skips. 

libmultid was needed as multid crashed when 53 is blocked, luckyli this is gone. With 7.20 freetz does not use anymore the uclibc of avm because avm uses uclibc/musl/glibc in different versions. Preloading the lib did not work anymore with different loader

snowball77, 2022-05-23:
But that is not what is happing - at least not on some of the boxes. When dnsmasq takes the port as it starts first, multid crashes with the stack trace we posted. And then the FOS watchdog reboots the box. I never got it to work and I am happy now that my box is not the only one showing this behavior.

fda77, 2022-05-23:
I did not know there are more people with this problem
But my 7490 is running with the same FOS like @thestr4ng3r  : 7490 07.29 ! So im sure at least this device works with the correct start order of the services

```
ns|grep -E "multid|dnsma"
tcp        0      0 0.0.0.0:53                                          0.0.0.0:*                                           LISTEN      2967/dnsmasq
tcp        0      0 :::53                                               :::*                                                LISTEN      2967/dnsmasq
udp        0      0 0.0.0.0:53                                          0.0.0.0:*                                                       2967/dnsmasq
udp        0      0 :::53                                               :::*                                                            2967/dnsmasq
```

```
ps|grep -E "multid|dnsma"
S   900  2967     1  1856   568 0:0   Feb14 00:00:00 dnsmasq --pid-file=/var/run/dnsmasq/dnsmasq.pid -p 53
S     0  3013     1  8076  1948 0:0   Feb14 00:07:21 multid -d -i
R     0 32644 30637  1852   464 pts0  11:37 00:00:00 grep -E multid|dnsma
```

The crash log above shows `dnsd_dot_set_enabled `. I have first disabled every dns related stuff in avm-webif, like "DNS over TLS (DoT)" - "Verschlüsselte Namensauflösung im Internet (DNS over TLS)" to reduce maybe problems.