Freetz-NG/freetz-ng #388: 7170 v2 schlägt fehl -> ld-uClibc.so.0 fehlt oder falscher Pfad

MartinX3, 2021-10-02:
Beim kompilieren sagt er mir
```
Traceback (most recent call last):
  File "/home/developer/opensource/github/freetz-ng/tools/build/bin/mklibs", line 509, in <module>
    for symbol in provided_symbols(lib):
  File "/home/developer/opensource/github/freetz-ng/tools/build/bin/mklibs", line 196, in provided_symbols
    raise Exception("Cannot find lib" + obj)
Exception: Cannot find lib/home/developer/opensource/github/freetz-ng/build/modified/filesystem/usr/lib/freetz/ld-uClibc.so.0
```

Allerdings existiert die Lib unter:
```
./toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.29/mipsel-linux-uclibc/lib/ld-uClibc.so.0
./build/original/filesystem/lib/ld-uClibc.so.0
./build/modified/filesystem/lib/ld-uClibc.so.0
./packages/target-mipsel_gcc-4.6.4_uClibc-0.9.29/root/lib/ld-uClibc.so.0
```

Irgendwie habe ich das gefühl falsch abgebogen zu sein, aber unter `Shared libraries -> uClibc -> ld-uClibc-x.y.z.so` ist ein `x`.

(Ich suche parallel weiter nach der Ursache)

MartinX3, 2021-10-02:
Es lag an `Strip options -> Run mklibs on the built root filesystem`
Dadurch ist zwar im Dateisystem immernoch alles beim alten, aber ohne diese Option sucht er die `ld-uClibc.so.0` im richtigen Pfad.

closed 2021-10-02 by MartinX3

fda77, 2021-10-02:
Bei neuerem Fritzos liegt die uclibc in /usr/lib/freetz/, der Pfad ist falsch

MartinX3, 2021-10-02:
Ui, dankeschön, das teste Morgen oder Montag. :)

fda77, 2021-10-03:
Gut zu wissen dass ein so altes Gerät überhaupt noch mit freetz funktioniert, ich hab schon lange keine Geräte mehr zum testen

MartinX3, 2021-10-03:
> Gut zu wissen dass ein so altes Gerät überhaupt noch mit freetz funktioniert, ich hab schon lange keine Geräte mehr zum testen

Kompilieren ja, ich muss es nur noch lauffähig machen.
Vielleicht habe ich zu viel gestriped oder removed oder es lag an den Freetz-Userdaten von 2020, die ich nicht gewiped hatte.
Gestern ging nur die PowerLED an und die 7170 war nicht erreichbar. Sobald ich wieder daheim bin, experiementier ich weiter rum. :)

fda77, 2021-10-04:
Der Test auf der Hardware ist das spannende ;-)

MartinX3, 2021-10-05:
> Der Test auf der Hardware ist das spannende ;-)

Da hast du recht, aber leider, da ich kein Low-Level spreche, muss ich dich mit Bug Tickets nerven :P