Freetz-NG/freetz-ng #389: 7170 v2 - Standardeinstellungen sind nicht baufähig

MartinX3, 2021-10-05:
Wenn man die `.config` löscht und in der `make menuconfig` nur die 7170 auswählt und das ganze bauen will, erhält man:

```
make[4]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/binutils“ wird betreten
make[4]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gas“ wird betreten
make[5]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gas“ wird betreten
make[5]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/binutils“ wird betreten
/bin/sh ./libtool --tag=CC --mode=link gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -D_GNU_SOURCE -fno-stack-protector   -o as-new  app.o as.o atof-generic.o cond.o depend.o dwarf2dbg.o dw2gencfi.o ecoff.o ehopt.o expr.o flonum-copy.o flonum-konst.o flonum-mult.o frags.o hash.o input-file.o input-scrub.o listing.o literal.o macro.o messages.o output-file.o read.o remap.o sb.o stabs.o subsegs.o symbols.o write.o tc-mips.o obj-elf.o atof-ieee.o itbl-parse.o itbl-lex.o itbl-ops.o  e-mipself.o ../opcodes/libopcodes.la ../bfd/libbfd.la ../libiberty/libiberty.a   
make[5]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/binutils“ wird verlassen
make[4]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/binutils“ wird verlassen
make[3]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/binutils“ wird verlassen
make[3]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gprof“ wird betreten
make  all-recursive
make[4]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gprof“ wird betreten
make[5]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gprof“ wird betreten
/bin/sh ./libtool --tag=CC --mode=link gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -D_GNU_SOURCE -fno-stack-protector   -o gprof  basic_blocks.o call_graph.o cg_arcs.o cg_dfn.o cg_print.o corefile.o gmon_io.o gprof.o hertz.o hist.o source.o search_list.o symtab.o sym_ids.o utils.o i386.o alpha.o vax.o tahoe.o sparc.o mips.o flat_bl.o bsd_callg_bl.o fsf_callg_bl.o ../bfd/libbfd.la ../libiberty/libiberty.a  
libtool: link: gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -D_GNU_SOURCE -fno-stack-protector -o as-new app.o as.o atof-generic.o cond.o depend.o dwarf2dbg.o dw2gencfi.o ecoff.o ehopt.o expr.o flonum-copy.o flonum-konst.o flonum-mult.o frags.o hash.o input-file.o input-scrub.o listing.o literal.o macro.o messages.o output-file.o read.o remap.o sb.o stabs.o subsegs.o symbols.o write.o tc-mips.o obj-elf.o atof-ieee.o itbl-parse.o itbl-lex.o itbl-ops.o e-mipself.o  ../opcodes/.libs/libopcodes.a ../bfd/.libs/libbfd.a ../libiberty/libiberty.a
/usr/bin/ld: dwarf2dbg.o:(.bss+0x0): multiple definition of `dwarf2_loc_mark_labels'; as.o:(.bss+0x158): first defined here
/usr/bin/ld: write.o:(.bss+0x0): multiple definition of `dwarf2_loc_mark_labels'; as.o:(.bss+0x158): first defined here
/usr/bin/ld: tc-mips.o:(.bss+0x0): multiple definition of `dwarf2_loc_mark_labels'; as.o:(.bss+0x158): first defined here
/usr/bin/ld: obj-elf.o:(.bss+0x0): multiple definition of `dwarf2_loc_mark_labels'; as.o:(.bss+0x158): first defined here
collect2: error: ld returned 1 exit status
libtool: link: gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -D_GNU_SOURCE -fno-stack-protector -o gprof basic_blocks.o call_graph.o cg_arcs.o cg_dfn.o cg_print.o corefile.o gmon_io.o gprof.o hertz.o hist.o source.o search_list.o symtab.o sym_ids.o utils.o i386.o alpha.o vax.o tahoe.o sparc.o mips.o flat_bl.o bsd_callg_bl.o fsf_callg_bl.o  ../bfd/.libs/libbfd.a ../libiberty/libiberty.a
make[5]: *** [Makefile:2209: as-new] Fehler 1
make[5]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gas“ wird verlassen
make[4]: *** [Makefile:2254: all-recursive] Fehler 1
make[4]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gas“ wird verlassen
make[3]: *** [Makefile:2140: all] Fehler 2
make[3]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/toolchain-mipsel_gcc-3.4.6/binutils-2.18-build/gas“ wird verlassen
make[2]: *** [Makefile:5437: all-gas] Fehler 2
```

Anbei:
[Standard 7170 config.txt](https://github.com/Freetz-NG/freetz-ng/files/7283674/config.txt)
[Stark modifizierte config.7170v2.txt, die zwar baut, aber die 7170 ist nicht erreichbar und hat nur eine PowerLED](https://github.com/Freetz-NG/freetz-ng/files/7283684/config.7170v2.txt)

MartinX3, 2021-10-05:
Das passiert, warum auch immer, weil ich default eine selbst kompilierte Toolchain nehmen muss.

In meiner stark modifizierten .config wird mir der Download der Toolchain angeboten.

By default gibt's keine Download-Option, also habe ich wohl beim rumfummeln irgendwas "repariert", damit er die downloaded.

MartinX3, 2021-10-06:
Ahhhh, danke für die info.
Ich wusste nicht, dass APTGuru das freischaltet (und ich glaube, dass stand auch nicht im Hilfetext dazu).

Mein Computer ist mit Arch Linux definitiv zu neu. Hihi
Ich sehe das dann mal als Anwenderfehler bzw. mir fehlt der Toolchainhinweis im APTGuru und ich schließe das Ticket mal.

closed 2021-10-06 by MartinX3

fda77, 2021-10-06:
Der Hinweis fehlt nicht: https://github.com/Freetz-NG/freetz-ng/blob/master/config/ui/user-competence.in#L35

MartinX3, 2021-10-06:
Ups, dann habe ich das leider vergessen. Sorry.
Aber gelesen hatte ich jeden Hilfe-Hinweis. :)

MartinX3, 2021-10-06:
> Klar ist die baubar! Entweder
> 
>     * eine altes Linux nutzen. Für die Toolchain hab ich Ubuntu 14 verwendet, damit sind auf jeden Fall alle compilierbar
> 
>     * oder die Fehler wie oben beschrieben beheben
> 
> 
> Da ich keine so alte Hardware mehr hab kann ich dazu nichts sagen. Am besten kritisches wie replace kernel weglassen, und zuerst nur unbedingt nötige Pakete installieren, zb ssh+syslog. Aber keine dnsmasq etc

Ja klar, wobei man heutzutage dafür explizit container in linux verwendet (podman oder kubernetes).

Ich habe jetzt nur dsld entfernt (Platzmangel bei default settings) und die precompiled toolchain genutzt, aber leider das selbe Problem. Die PowerLED ist an, die Box hat aber kein Netzwerk, also kann ich nichts debuggen.
Einen funktionierenden Commitstand zu finden wird jetzt schwierig. Erstmal gucken, ob freetz im letzten Stand das selbe Problem hatte.

> Oder auch eine ein paar Jahre alte Revision (alte ng revision ist das gleiche wie freetz ohne ng) nehmen und schauen ob es damit noch funktioniert

Nur, dass diesem fork 470 commits von freetz fehlen, irritiert mich.

fda77, 2021-10-07:
> nicht einmal mit einer Kneifzange an.

Dem stimmt ich vollumfänglich zu. Für mich kommt zu deiner Aufzählung aber noch uralte Hardware die vor über 15 Jahren veröffentlicht wurde ;-)
Wenn es dir zu stressig ist die Ursache zu suchen nimm halt ein Freetz von vor 10 Jahren

> Oder du hattest deren Commit-Hash geändert, was man nicht machen sollte. (Forks sollte man entweder auf Upstream rebasen oder via merge commit Änderungen von Upstream reinholen)

Hashes werden automatisch generiert. Aber wie bereits geschrieben: 

> Dann pick doch mal 1 heraus und erklär beispielhaft daran was genau irritiert