Freetz-NG/freetz-ng #390: 7170v2 - Defekt -> Strip options - Run mklibs on the built root filesystem - EXPERIMENTAL

MartinX3, 2021-10-05:
```
Making all in mklibs-readelf
make[4]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/host-tools/mklibs-0.1.34/src/mklibs-readelf“ wird betreten
g++ -DHAVE_CONFIG_H -I. -I../..     -g -O2 -MT elf.o -MD -MP -MF .deps/elf.Tpo -c -o elf.o elf.cpp
g++ -DHAVE_CONFIG_H -I. -I../..     -g -O2 -MT main.o -MD -MP -MF .deps/main.Tpo -c -o main.o main.cpp
In Datei, eingebunden von main.cpp:12:
elf.hpp:52:56: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   52 |       const section &get_section(unsigned int i) const throw (std::out_of_range) { return *sections.at(i); };
      |                                                        ^~~~~
elf.hpp:62:47: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   62 |       static file *open(const char *filename) throw (std::bad_alloc, std::runtime_error);
      |                                               ^~~~~
elf.hpp:65:38: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   65 |       file(uint8_t *mem, size_t len) throw (std::bad_alloc) : mem(mem), len(len) { }
      |                                      ^~~~~
elf.hpp:68:52: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   68 |         static file *open_class(uint8_t *, size_t) throw (std::bad_alloc, std::runtime_error);
      |                                                    ^~~~~
elf.hpp:131:55: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  131 |         std::string get_string(uint32_t offset) const throw (std::bad_alloc)
      |                                                       ^~~~~
elf.hpp:266:39: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  266 |       std::string get_version() const throw (std::bad_alloc);
      |                                       ^~~~~
elf.hpp:267:44: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  267 |       std::string get_version_file() const throw (std::bad_alloc);
      |                                            ^~~~~
elf.hpp:269:44: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  269 |       std::string get_name_version() const throw (std::bad_alloc);
      |                                            ^~~~~
elf.hpp:308:29: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  308 |       version_requirement() throw (std::bad_alloc);
      |                             ^~~~~
make[4]: *** [Makefile:278: main.o] Fehler 1
make[4]: *** Es wird auf noch nicht beendete Prozesse gewartet....
In Datei, eingebunden von elf_data.hpp:24,
                 von elf.cpp:21:
elf.hpp:52:56: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   52 |       const section &get_section(unsigned int i) const throw (std::out_of_range) { return *sections.at(i); };
      |                                                        ^~~~~
elf.hpp:62:47: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   62 |       static file *open(const char *filename) throw (std::bad_alloc, std::runtime_error);
      |                                               ^~~~~
elf.hpp:65:38: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   65 |       file(uint8_t *mem, size_t len) throw (std::bad_alloc) : mem(mem), len(len) { }
      |                                      ^~~~~
elf.hpp:68:52: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   68 |         static file *open_class(uint8_t *, size_t) throw (std::bad_alloc, std::runtime_error);
      |                                                    ^~~~~
elf.hpp:131:55: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  131 |         std::string get_string(uint32_t offset) const throw (std::bad_alloc)
      |                                                       ^~~~~
elf.hpp:266:39: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  266 |       std::string get_version() const throw (std::bad_alloc);
      |                                       ^~~~~
elf.hpp:267:44: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  267 |       std::string get_version_file() const throw (std::bad_alloc);
      |                                            ^~~~~
elf.hpp:269:44: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  269 |       std::string get_name_version() const throw (std::bad_alloc);
      |                                            ^~~~~
elf.hpp:308:29: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  308 |       version_requirement() throw (std::bad_alloc);
      |                             ^~~~~
In Datei, eingebunden von elf.cpp:21:
elf_data.hpp:97:42: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
   97 |         file_data(uint8_t *, size_t len) throw (std::bad_alloc, std::runtime_error);
      |                                          ^~~~~
elf_data.hpp:112:43: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  112 |         virtual void update(const file &) throw (std::bad_alloc);
[... 99 lines cut ...]
      |                                               ^~~~~
elf.cpp:453:47: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  453 | std::string symbol::get_name_version () const throw (std::bad_alloc)
      |                                               ^~~~~
elf.cpp:478:98: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  478 | void symbol_data<_class, _data>::update_string(const section_type<section_type_STRTAB> &section) throw (std::bad_alloc)
      |                                                                                                  ^~~~~
elf.cpp:484:83: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  484 | void symbol_data<_class, _data>::update_version(const file &file, uint16_t index) throw (std::bad_alloc)
      |                                                                                   ^~~~~
elf.cpp:531:110: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  531 | void version_definition_data<_class, _data>::update_string(const section_type<section_type_STRTAB> &section) throw (std::bad_alloc)
      |                                                                                                              ^~~~~
elf.cpp:537:44: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  537 | version_requirement::version_requirement() throw (std::bad_alloc)
      |                                            ^~~~~
elf.cpp:561:65: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  561 | update_string(const section_type<section_type_STRTAB> &section) throw (std::bad_alloc)
      |                                                                 ^~~~~
elf.cpp:596:65: Fehler: ISO C++17 erlaubt keine dynamischen Ausnahmespezifikationen
  596 | update_string(const section_type<section_type_STRTAB> &section) throw (std::bad_alloc)
      |                                                                 ^~~~~
make[4]: *** [Makefile:278: elf.o] Fehler 1
make[4]: Verzeichnis „/home/developer/opensource/github/freetz-ng/source/host-tools/mklibs-0.1.34/src/mklibs-readelf“ wird verlassen
```

fda77, 2021-10-05:
Wie binutils in https://github.com/Freetz-NG/freetz-ng/issues/389: ISO C++17 nach c++03

MartinX3, 2021-10-06:
Ich schließe das Ticket auch mal, da https://github.com/Freetz-NG/freetz-ng/issues/389#issuecomment-935563292

closed 2021-10-06 by MartinX3