Freetz-NG/freetz-ng #404: 001-regular_dot_config_location.patch

freetz-ng-mod, 2021-11-14:
-export AVM_KCONFIG_CONFIG = avm/conf/linux-4.9.grx5

Wird für die 7590ax inhaus 7.39 gebraucht

fda77, 2021-11-14:
Ich bin jetzt nicht so sicher wohin das soll und ob es geändert oder hinzugefügt werden woll
Bootet diese neue 7.39 inhaus gundlegend?

freetz-ng-mod, 2021-11-14:
Ich traue mich noch nicht, die inhaus darauf zu flashen. Da viele mit der Version probs haben.

--- linux-4.9/.kernelvariables
+++ linux-4.9/.kernelvariables
@@ -35,7 +35,6 @@
 CP ?= /bin/cp
 export CP
 
-export AVM_KCONFIG_CONFIG = avm/conf/linux-4.9.grx5
 
 # No variables exported via $(kernelvariables_extra)

fda77, 2021-11-14:
Jemand müsste die Inhaus mal ausprobieren :)
Mach die Patches doch zumindest in Codetags, ich hab die in den Post obendrüber eingefügt.
Wenn du mit git keine Patches erstellen kannst, häng doch einfach die geänderten Dateien an, `git status` zeigt sie.
Dann alles packen mit `tar cvf meine.tar  dirA/fileB dirC/fileD dirE/fileF`
Die .tar hier dann anhängen, falls die Dateiendung nicht erlaubt ist noch ein ".txt" dranhängen

fda77, 2021-11-14:
Ah, das sind Patches für die Kernelsourcen. Da AVM noch keine veröffentlicht hat gibt es noch nichts zu patchen 
@all AVM via Mail nerven

closed 2021-11-14 by fda77

freetz-ng-mod, 2021-11-14:
Dann Bau mal ein Image, und du wirst sehen das du beide patche brauchst. 
Habe die ja nicht um sonst gepostet.

fda77, 2021-11-14:
Hast du mit der neuesten Revision ausprobiert?

freetz-ng-mod, 2021-11-14:
Ich habe schon lange meine eigenen Patche drin. Und die laufen durch.

fda77, 2021-11-14:
Mit den Sourcen für 5530? Die hat eine PRX300 CPU statt GRX5.
Module bauen kannst du bauen, ob du damit auf der Box was anfangen kannst ist was anderes.
Es gibt halt noch keine Sourcenj -> nachfragen