Freetz-NG/freetz-ng #407: 833-show_dsl_spectrum-minmax_07_4X.patch

freetz-ng-mod, 2021-11-14:
--- usr/www/all/internet/dsl_spectrum.lua
+++ usr/www/all/internet/dsl_spectrum.lua
@@ -258,7 +258,7 @@
 import Chartist from "/js/chartist.js";
 import { ajaxGet, stopXhr } from "/js/ajax.js";
 var g_spectrum = <?lua box.out(js.table(g_data)) ?>;
-var g_show_MinMax=<?lua if box.get.min_max then box.out(" true ") else box.out(" false ") end ?>;
+var g_show_MinMax=<?lua if not box.get.min_max then box.out(" true ") else box.out(" false ") end ?>;
 function yAxisFormatterSnr(val, axis) {
 if (val % 10 == 0) {
 return val;
@@ -701,7 +701,7 @@
 </div>
 
 <div class="formular">
-<a class="textlink nocancel" id="uiToggleMinMax">{?7888:534?}</a><br>
+<a class="textlink nocancel" id="uiToggleMinMax">{?7888:534?}</a><br>  
 <a class="hide" id="uiMinMaxReset">{?7888:4099?}</a>
 </div>
 </div>

freetz-ng-mod, 2021-11-14:
833-show_dsl_spectrum-minmax.sh

[ "$FREETZ_MODIFY_DSL_SPECTRUM" == "y" ] || return 0
echo1 "patching dsl spectrum"

if [ "$FREETZ_AVM_VERSION_07_4X_MIN" == "y" ] ; then
	PVER="07_4X"
elif [ "$FREETZ_AVM_VERSION_07_0X_MIN" == "y" ] ; then
	PVER="07_0X"
elif [ "$FREETZ_AVM_VERSION_06_8X_MIN" == "y" ] ; then
	PVER="06_8X"
elif [ "$FREETZ_AVM_VERSION_06_5X_MIN" == "y" ] ; then
	PVER="06_5X"
fi

modpatch "$FILESYSTEM_MOD_DIR" "${PATCHES_COND_DIR}/833-show_dsl_spectrum-minmax/833-show_dsl_spectrum-minmax_${PVER}.patch"

fda77, 2021-11-14:
Da wird nichts geändert ausser einem Kommentar hinzugefügt. In der Funktion (falls es sie noch gibt) steht die passende string-ID, ehemals 518
```
-<a class="textlink nocancel" id="uiToggleMinMax">{?7888:534?}</a><br>
+<a class="textlink nocancel" id="uiToggleMinMax">{?7888:534?}</a><br>  <!-- from function OnMinMax() -->
```

closed 2021-11-14 by fda77

freetz-ng-mod, 2021-11-27:
Der Patch lauft richtig auf der 7590ax so wie der 831-extend-dsl_line_settings-speed_07_4X.patch auch. Habe nun meine Box mit freetz und fw 7.39 geflasht. 

Damit ist nun getestet

fda77, 2021-11-27:
Du meinst das UNTESTED soll weg?
Wenn dann doch irgend was nicht funktioniert schreien wieder alle herum, dass doch alles funktionieren müsste weil keine untested/devel dran steht