Freetz-NG/freetz-ng #408: 5530 Kernel 4.9.231

freetz-ng-mod, 2021-11-14:
Bei der 5530 ist aus dem Kernel 4.9.218 Kernel 4.9.231 geworden

fda77, 2021-11-14:
Ja: https://github.com/Freetz-NG/freetz-ng/blob/master/config/.img/5530--07_25.in#L9
Falls du damit sagen willst dass es keinen Sourcecode gibt: https://freetz-ng.github.io/freetz-ng/SUPPORT.html#quellcode

fda77, 2021-11-14:
Und https://github.com/Freetz-NG/freetz-ng/discussions/134

closed 2021-11-14 by fda77

freetz-ng-mod, 2021-11-14:
Wenn du schon Sachen aus dem Kernel löschen lassen willst, brauchst du ein Kernel 4.9.231 Verzeichnis 
https://github.com/Freetz-NG/freetz-ng/blob/master/make/linux/patches/4.9.218/001-regular_dot_config_location.patch

fda77, 2021-11-14:
Der commit comment lautet "add 5530 src"...

freetz-ng-mod, 2021-11-14:
so
das soll aus der .kernelvariables gelösdcht werden
-export AVM_KCONFIG_CONFIG = avm/conf/linux-4.9.prxI
aber wir brauchen für Kernel 4.9.218 aber das
export AVM_KCONFIG_CONFIG = avm/conf/linux-4.9.grx5
Und wie ich schon sagte die 5530 ist nun 4.9.231

fda77, 2021-11-14:
Sagtest du schonmal ebenso ohne Zusammenhang und ich antwortete in https://github.com/Freetz-NG/freetz-ng/issues/408#issuecomment-968327314 darauf. Ich hab aber noch immer nicht durchblickt was du damit sagen willst. Was soll das denn heissen?