Freetz-NG/freetz-ng #410: nfs-utils build error

e6e7e8, 2021-11-16:
Bislang habe ich freetz-ng auf einer VM mit Ubuntu gebaut. Funktioniert einwandfrei. Auf Fedora f35 gibt es beim Bauen von nfs-utils aber folgende Fehlermeldung:

`/bin/sh ../../libtool  --tag=CC   --mode=link /home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -Wall -Wextra -Wstrict-prototypes  -pipe -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/include/tirpc   -o mountd mountd-mountd.o mountd-mount_dispatch.o mountd-auth.o mountd-rmtab.o mountd-cache.o mountd-svc_run.o mountd-fsloc.o mountd-v4root.o ../../support/export/libexport.a ../../support/nfs/libnfs.a ../../support/misc/libmisc.a  -lwrap   -ldl -L/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib -ltirpc  
libtool: link: /home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -Wall -Wextra -Wstrict-prototypes -pipe -march=34kc -mtune=34kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/include/tirpc -o mountd mountd-mountd.o mountd-mount_dispatch.o mountd-auth.o mountd-rmtab.o mountd-cache.o mountd-svc_run.o mountd-fsloc.o mountd-v4root.o  ../../support/export/libexport.a ../../support/nfs/libnfs.a ../../support/misc/libmisc.a -lwrap -ldl -L/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib /usr/lib/libtirpc.so -lpthread -Wl,-rpath -Wl,/usr/lib -Wl,-rpath -Wl,/usr/lib
/usr/lib/libtirpc.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:515: mountd] Error 1
make[3]: Leaving directory '/home/e6e7e8/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/nfs-utils-1.3.4/utils/mountd'
make[2]: *** [Makefile:445: all-recursive] Error 1
make[2]: Leaving directory '/home/knop/fritzbox/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/nfs-utils-1.3.4/utils'
make[1]: *** [Makefile:469: all-recursive] Error 1
make[1]: Leaving directory '/home/knop/fritzbox/freetz-ng/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/nfs-utils-1.3.4'

ERROR: Build failed.
make: *** [make/nfs-utils/nfs-utils.mk:43: source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/nfs-utils-1.3.4/utils/exportfs/exportfs] Error 1
`
Vergleicht man das mit dem Ubuntu-Logfile fällt auf, dass vor /usr/lib/libtirpc.so der Prefix-Pfad
`/home/knop/fritzbox/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc`
fehlt.

fda77, 2021-11-16:
Ich nutze schon lange Fedora, aktuell aber noch 34.
Weiter oberhalb deines Logauszugs könnte noch Hinweise sein, zB wie configure den falschen Pfad findet. Poste da mal noch.
Dann müsste man in der .mk Datei noch den Pfad vorgeben

e6e7e8, 2021-11-20:
Habe leider nichts gefunden. Die log-Datei sieht bei Ubuntu und Fedora (f34 und f35) ziemlich gleich aus. Irgendwas wandelt in Fedora den Parameter -ltirpc in /usr/lib/libtirpc.so statt /Pfad/zur/freetz/Version/libtirpc.so um. Funktionierende Workarounds sind
1. Ubuntu verwenden.
2. Einen symbolischen Link /usr/lib/libtirpc.so -> /Pfad/zur/freetz/Version/libtirpc.so anlegen.
3. In den Makefiles der Verzeichnisse mound, nfsd,showmount und statd die Zeile
`LIBTIRPC = -L/Pfad/zur/freetz/Version -ltirpc`
durch
`LIBTIRPC = /Pfad/zur/freetz/Version/libtirpc.so`
ersetzen.

fda77, 2021-11-21:
Ein Fedora spezifisches Problem sollte es nicht sein, da es bei mir mit F34 funktioniert. Es dürfte mehr davon abhängen was auf dem Hostsystem installiert ist und beim cross-build fälschlicherweise verwendet wird. Am besten vergleicht man mal deine und meine Ausgaben mit Fedora.

3. Ich weiss nicht ob das die korrekte Behebung des Problems ist. Ein `-L` wird auch in anderen Packages (lua) und libs (libavmhmac) in Freetz verwendet. Wo hast du das `LIBTIRPC = -L/Pfad/zur/freetz/Version -ltirpc` überhaupt her?

e6e7e8, 2021-11-24:
`/Pfad/zur/freetz/Version`  war nur meine Abkürzung für
`/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/mips-linux-uclibc/usr/lib`

Ich habe nun ein frisches fedora f35 (workstation) in einer VM aufgestetzt. Leider geht es jetzt gar nicht mehr. Ohne jede Auswahlen bricht der Compiler ab:
` In file included from /usr/include/c++/11/stdlib.h:36,
                 from ../../../Common/StringConvert.cpp:8:
/usr/include/c++/11/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.`

fda77, 2021-11-27:
Jetzt fehlt vermutlich irgend ein devel Paket einer Lib. Vielleicht hilft eine Suche nach der Fehlermeldung.
Durch diesen kurzen Auszügen aus den Logs kann ich sonst nichts dazu sagensagen

e6e7e8, 2021-11-30:
Ich bin etwas weiter gekommen. Es fehlte tatsächlich das Paket `libstdc++-devel.x86_64`. Sollte man vielleicht noch in die prerequisites schreiben.

Leider kompiliert nfd-utils immer noch nicht.

`
make[3]: Entering directory '/home/e6e7e8/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/nfs-utils-1.3.4/tools/rpcgen'
gcc -DHAVE_CONFIG_H -I. -I../../support/include  -I/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/usr/include/tirpc  -D_GNU_SOURCE  -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/include/tirpc -MT rpcgen-rpc_clntout.o -MD -MP -MF .deps/rpcgen-rpc_clntout.Tpo -c -o rpcgen-rpc_clntout.o `test -f 'rpc_clntout.c' || echo './'`rpc_clntout.c
gcc -DHAVE_CONFIG_H -I. -I../../support/include  -I/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/usr/include/tirpc  -D_GNU_SOURCE  -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -I/home/e6e7e8/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/include/tirpc -MT rpcgen-rpc_cout.o -MD -MP -MF .deps/rpcgen-rpc_cout.Tpo -c -o rpcgen-rpc_cout.o `test -f 'rpc_cout.c' || echo './'`rpc_cout.c
as: unrecognized option '--trap'
cc1: error: bad value (‘34kc’) for ‘-march=’ switch
`
Wenn das richtig verstehe, wird der host-Compiler und nicht der guest-Compiler verwendet.

fda77, 2021-12-02:
Du hattest installiert was in https://github.com/Freetz-NG/freetz-ng/blob/master/docs/PREREQUISITES.md steht? Dort gibt es "libstdc++-devel.i686". Ich hab auf meine, Fedora 34 nachgesehen, dort sind libstdc++-devel bzw libstdc++-devel.x86_64 und rpcgen installiert, vermutlich als Abhängigkeiten von irgendwas.
Damit make bereits früher abbricht könnte man diese Datei erweitern: https://github.com/Freetz-NG/freetz-ng/blob/master/.prerequisites#L48 . Dort kommt auch rein falls es nur für bestimmte Pakete Abhängigkeiten sind.
Auf Github gibt es übrigens bei beiden verlinkten Dateien oben rechts ein Stift-Symbol zum editieren

e6e7e8, 2021-12-02:
Alles, was ich weiß, ist, dass libstdc++-devel.x86_64 und rpcgen auf einem frisch installiereten fedora nicht vorhanden sind. Auch nicht nach der Installation der Prerequisites. Ich versuch's mal mit dem editieren.

fda77, 2021-12-02:
Wenn es bereits einen PR gibt kann man zu diesem Commits hinzufügen. Ob das auch über die Github Oberfläche funktioniert weiss ich aber nicht.
Sollte rpcgen nicht auch zu PREREQUISITES.md ?

e6e7e8, 2021-12-03:
Man kann rpcgen auch in in PREREQUISITES.md  stecken. Ich habe das nur nicht gemacht, weil es meines Wissens nach nur fuer nsf-utilds und autofs gebraucht wird. Zugegebenermassen habe ich nicht alle Pakete durchprobiert.

fda77, 2021-12-03:
```
grep RPCGEN  make/* -r -i
make/autofs/patches/abandon/009_autofs-5.0.5-fix-ext4-fsck-at-mount.patch: RPCGEN = @PATH_RPCGEN@
make/autofs/patches/abandon/140-conf.patch:+RPCGEN = rpcgen
make/autofs/patches/abandon/050_autofs-5.0.5-include-krb5-library.patch: PATH_RPCGEN
make/autofs/patches/abandon/050_autofs-5.0.5-include-krb5-library.patch: RPCGEN
make/autofs/patches/abandon/050_autofs-5.0.5-include-krb5-library.patch:@@ -145,8 +145,9 @@ AF_CHECK_PROG(RPCGEN, rpcgen, , $searchpath)
make/libs/libevent/patches/010-no_underscore_libs.freetz.patch: @INSTALL_LIBEVENT_TRUE@dist_bin_SCRIPTS = event_rpcgen.py
make/libs/libevent/patches/010-no_underscore_libs.freetz.patch:         test/rpcgen_wrapper.sh test/print-winsock-errors.c \
make/nfs-utils/patches/900-no_locktest.patch:-DIST_SUBDIRS = locktest rpcdebug nlmtest mountstats nfs-iostat rpcgen
make/nfs-utils/patches/900-no_locktest.patch:+DIST_SUBDIRS = rpcdebug nlmtest mountstats nfs-iostat rpcgen
```

Bleibt die Frage um welches autofs es bei dir geht. Es könnte sein dass die neuere Version durch den wechsel zu libtirpc kein rpcgen mehr braucht

closed 2022-04-16 by fda77