Freetz-NG/freetz-ng #426: Not possible to flash or recover from linux (7270v3

MasterPlexus, 2021-12-11:
Hi all,

it could be that my problem is more linked to the problem that the network is not correct set, but I'm not sur what to do additionally.

Problem is:
Nor the make push-firmware or the make recover is able to find the box. on a other Windowssystem I was able to flash with the official avm recover tool, so generally I think it should be possible...

I checked that I have the correct network device with ifconfig. And I deactivate wlan, and set with 
```
dirk@Moms-old:~/freetz-ng$ sudo ifconfig enp2s0 down
dirk@Moms-old:~/freetz-ng$ sudo ifconfig enp2s0 192.168.178.2 up
```
the network in the correct state (as far as I know, but now I'm not sure...)
This is my network information with ifconfig after the previous commands:
```
enp2s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500                                                                                                                                            
        inet 192.168.178.2  netmask 255.255.255.0  broadcast 192.168.178.255
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 766  bytes 248069 (248.0 KB)
        RX errors 0  dropped 100  overruns 0  frame 0
        TX packets 482  bytes 88258 (88.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 244  bytes 19426 (19.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 244  bytes 19426 (19.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

and this is the output of the booth tools:

```
dirk@Moms-old:~/freetz-ng$ umask 0022
dirk@Moms-old:~/freetz-ng$ make push-firmware

 * Analyzing 'images/7270v3_06.06.ger_freetz-ng-18722M-6e2a97c09_20211205-175236.image' ... 

 * Using command: ftp
 * Target host: 192.168.178.1
 * Outgoing IP: 192.168.178.2
 * Flash mode: single-boot

 !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!!
 !!!  THERE IS NO WARRANTY AT ALL !!! USE AT YOUR OWN RISK   !!!

 * Are you sure, that you want to flash this file to the device?
   images/7270v3_06.06.ger_freetz-ng-18722M-6e2a97c09_20211205-175236.image
   Proceed? (y/[n]) y

 * You should now reboot your box (192.168.178.1). Waiting for shut down.
   Switch off, if reboot is not detected because it happens too quickly.
   Some newer bootloader versions allow to flash on power-cycle only.
   
 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting ...............................................................................................................................................^C
aborted
make: *** [Makefile:369: push-firmware] Error 1

dirk@Moms-old:~/freetz-ng$ make recover images/7270v3_06.06.ger_freetz-ng-18722M-6e2a97c09_20211205-175236.image
This can help if your box is not booting any more
(Power LED on and flashing of all LEDs every 5 secs).

Make sure that there is only one box in your subnet.

Are you sure you want to recover filesystem and kernel
from dl/fw/7270_v3-de.detected.image? (y/n) y

If this fails try to specify a local IP adress. Your
local IP has to be in the 192.168.178.0/24 subnet.
e.g. make recover LOCALIP=192.168.178.20

Looking for Fritz!Box ooooooooooooooooooooooooooooooooooooooooo^Z
[1]+  Stopped                 make recover images/7270v3_06.06.ger_freetz-ng-18722M-6e2a97c09_20211205-175236.image
```
Of course tried several times with new starts, and also with reboots of the fritzbox.

Many thanks for any help!!
BR

fda77, 2021-12-11:
You have pulled out and plugged in the power cable during "Waiting"? Is your PC directly connected to the Fritzbox? In this case maybe your PC is to slow to reenable the network device, another switch (ot a fritzbox) between them could help.
You could run "ping 192.168.178.1" in a separate terminal windows.
Btw, it should work to add a 2nd IP to your device to keep current connections by: `sudo ifconfig enp2s0:0 192.168.178.2 up`

MasterPlexus, 2021-12-11:
Thanks for your suggestions. Yes I also plugged power off and on during waiting, and the boys is directly connected (as suggested) via lan1 to the laptop. The :0 approach I also tried, but I was not sure if it makes sense as maybe the network is not detected right...

As I did not have a switch, but an real old Fritzbox. I assume that I have to start this then with an other IP, so that it will not interfear with the box wich I want to Update? I'll try and come back. But I really wonder that Linux is such slow in the device for the network...

MasterPlexus, 2021-12-11:
Ok, the way how I was able to do it:

- I use a old fritz 7171
- I set this fritz to IP 192.168.178.111 and set dns range to 2-3 (just to take care and less possibilitys
- as the 7171 has only one network connection, I connected the to flash Fritzbox to lan, and my laptop I connected to elan to the old box (7171)
- then in shell I tried without the make directly the tools/push_firmware with dedicated -ip option for the to flash box. But there I get after a long wait only a short connect, but then forbidden (may because the box is in bool loop and to slow for the Uploadmode)
- then I use the recover tool in tools directory with -i for ip address and - l for the image load file, and this needs shirt time, but it worked.

Reboot the box and it worked! The box is now bootloop free, and I can use it at minimum as a repeater... This was my idea. The box had the bootloop after a thunder-flash in the naiborhood. I had read that this could destroy the telephone Chip, so I build the freetzinage without the telephone module's, and it looks like that this worked fine... Just to document it also here if it could help other's ;)

Thanks for all the help!

closed 2021-12-11 by MasterPlexus

fda77, 2021-12-12:
MY recommendation is not connect PC + Fritzbox directly as the PC could be to slow in detection of the network status.
The DNS and gateway settings does not matter. Keep them as they are, delte them or set something random. The device is connected by IP without routing.
Very important: No other device should use the IP 192.168.178.1 which is default for AVm bootloaders