Freetz-NG/freetz-ng #427: haproxy - source code was miscompiled by the compiler

berndy2001, 2021-12-14:
Bei mir mag haproxy nicht mehr starten.
7590 mit aktuellem Trunk, aber das Problem besteht schon länger.

```
root@fritz:/var/mod/root# haproxy
FATAL ERROR: invalid code detected -- cannot go further, please recompile!
The source code was miscompiled by the compiler, which usually indicates that
some of the CFLAGS needed to work around overzealous compiler optimizations
were overwritten at build time. Please do not force CFLAGS, and read Makefile
and INSTALL files to decide on the best way to pass your local build options.

Build options :
  TARGET  = custom
  CPU     = generic
  CC      = /home/user/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc
  CFLAGS  = -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -ffunction-sections -fdata-sections
  OPTIONS = USE_EPOLL=1 USE_PCRE=1 USE_OPENSSL=1
  DEBUG   =

```

Danke.

fda77, 2021-12-14:
Sachen gibts...
Ich hab mal nach der Fehlermeldung gesucht, 1. Treffer
https://bugs.archlinux.org/task/71861?project=0&order=id&sort=desc&status%5B0%5D=closed&string=apr
Da ich kein Gerät zum testen hab, probier du das mal aus:
https://github.com/Freetz-NG/freetz-ng/blob/master/make/haproxy/haproxy.mk#L31

berndy2001, 2021-12-14:
Ich verstehe davon wenig bis gar nichts, aber `-fwrapv` (und `make haproxy-clean`) hat geholfen.

```
root@fritz:/var/mod/root# haproxy -vv
HAProxy version 2.5.0-f2e0833 2021/11/23 - https://haproxy.org/
Status: stable branch - will stop receiving fixes around Q1 2023.
Known bugs: http://www.haproxy.org/bugs/bugs-2.5.0.html
Running on: Linux 4.9.198 #1 SMP 2021-07-20 mips
Build options :
  TARGET  = custom
  CPU     = generic
  CC      = /home/user/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc
  CFLAGS  = -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -ffunction-sections -fdata-sections -fwrapv
  OPTIONS = USE_EPOLL=1 USE_PCRE=1 USE_OPENSSL=1
  DEBUG   =

...
```

fda77, 2021-12-14:
Das war ja einfach :) Ich hab davon auch keine Ahnung!

closed 2021-12-14 by fda77