Freetz-NG/freetz-ng #428: Iksemel 1 error

NeunMuenze92001, 2021-12-24:
Hallo, ich kriege beim erstellen eines Images für meine FRITZ!Box 7560 folgenden Fehler: 
filter.c: In function 'iks_filter_packet':
filter.c:128:19: warning: comparison between 'enum ikstype' and 'enum ikspaktype' [-Wenum-compare]
    if (rule->type == pak->type) score += 1; else fail = 1;
                   ^~
libtool: compile:  /home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I../include -I../include -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -Wall -I/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/usr/include -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -Wall -MT libiksemel_la-filter.lo -MD -MP -MF .deps/libiksemel_la-filter.Tpo -c filter.c -o libiksemel_la-filter.o >/dev/null 2>&1
tls-openssl.c: In function 'my_bio_create':
tls-openssl.c:27:3: error: dereferencing pointer to incomplete type 'BIO' {aka 'struct bio_st'}
  b->init = 1;
   ^~
tls-openssl.c: At top level:
tls-openssl.c:92:1: error: variable 'my_bio_method' has initializer but incomplete type
 static BIO_METHOD my_bio_method = {
 ^~~~~~
tls-openssl.c:93:2: warning: excess elements in struct initializer
  ( 100 | 0x400 ),
  ^
tls-openssl.c:93:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:94:2: warning: excess elements in struct initializer
  "iksemel transport",
  ^~~~~~~~~~~~~~~~~~~
tls-openssl.c:94:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:95:2: warning: excess elements in struct initializer
  my_bio_write,
  ^~~~~~~~~~~~
tls-openssl.c:95:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:96:2: warning: excess elements in struct initializer
  my_bio_read,
  ^~~~~~~~~~~
tls-openssl.c:96:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:97:2: warning: excess elements in struct initializer
  my_bio_puts,
  ^~~~~~~~~~~
tls-openssl.c:97:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:98:2: warning: excess elements in struct initializer
  my_bio_gets,
  ^~~~~~~~~~~
tls-openssl.c:98:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:99:2: warning: excess elements in struct initializer
  my_bio_ctrl,
  ^~~~~~~~~~~
tls-openssl.c:99:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:100:2: warning: excess elements in struct initializer
  my_bio_create,
  ^~~~~~~~~~~~~
tls-openssl.c:100:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:101:2: warning: excess elements in struct initializer
  my_bio_destroy
  ^~~~~~~~~~~~~~
tls-openssl.c:101:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:92:19: error: storage size of 'my_bio_method' isn't known
 static BIO_METHOD my_bio_method = {
                   ^~~~~~~~~~~~~
make[2]: *** [Makefile:565: libiksemel_la-tls-openssl.lo] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
mv -f .deps/libiksemel_la-filter.Tpo .deps/libiksemel_la-filter.Plo
make[2]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/iksemel-978b733462/src“ wird verlassen
make[1]: *** [Makefile:454: all-recursive] Fehler 1
make[1]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/iksemel-978b733462“ wird verlassen

ERROR: Build failed.
make: *** [make/libs/iksemel/iksemel.mk:27: source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/iksemel-978b733462/src/.libs/libiksemel.so.3.1.1] Fehler 1

Ich bitte um eine Lösung

Danke.

fda77, 2021-12-24:
Was willst du mit der 10 Jahre alten lib? Vermutlich passt die nicht zu der von dir ausgewählten openssl version. Teste mal diesen seit 6 Jahren nicht gemergten pull-request https://github.com/meduketto/iksemel/pull/46

NeunMuenze92001, 2021-12-24:
Hab das Image von Gismotro.

NeunMuenze92001, 2021-12-24:
Wie kann ich das 6 alte Jahre stück integrieren?

NeunMuenze92001, 2021-12-24:
Es ist leider der selbe Fehler. 
 error: dereferencing pointer to incomplete type 'BIO' {aka 'struct bio_st'}
  b->init = 1;
   ^~
tls-openssl.c: At top level:
tls-openssl.c:92:1: error: variable 'my_bio_method' has initializer but incomplete type
 static BIO_METHOD my_bio_method = {
 ^~~~~~
tls-openssl.c:93:2: warning: excess elements in struct initializer
  ( 100 | 0x400 ),
  ^
tls-openssl.c:93:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:94:2: warning: excess elements in struct initializer
  "iksemel transport",
  ^~~~~~~~~~~~~~~~~~~
tls-openssl.c:94:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:95:2: warning: excess elements in struct initializer
  my_bio_write,
  ^~~~~~~~~~~~
tls-openssl.c:95:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:96:2: warning: excess elements in struct initializer
  my_bio_read,
  ^~~~~~~~~~~
tls-openssl.c:96:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:97:2: warning: excess elements in struct initializer
  my_bio_puts,
  ^~~~~~~~~~~
tls-openssl.c:97:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:98:2: warning: excess elements in struct initializer
  my_bio_gets,
  ^~~~~~~~~~~
tls-openssl.c:98:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:99:2: warning: excess elements in struct initializer
  my_bio_ctrl,
  ^~~~~~~~~~~
tls-openssl.c:99:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:100:2: warning: excess elements in struct initializer
  my_bio_create,
  ^~~~~~~~~~~~~
tls-openssl.c:100:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:101:2: warning: excess elements in struct initializer
  my_bio_destroy
  ^~~~~~~~~~~~~~
tls-openssl.c:101:2: note: (near initialization for 'my_bio_method')
tls-openssl.c:92:19: error: storage size of 'my_bio_method' isn't known
 static BIO_METHOD my_bio_method = {
                   ^~~~~~~~~~~~~
make[2]: *** [Makefile:565: libiksemel_la-tls-openssl.lo] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
mv -f .deps/libiksemel_la-sha.Tpo .deps/libiksemel_la-sha.Plo
make[2]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/iksemel-978b733462/src“ wird verlassen
make[1]: *** [Makefile:454: all-recursive] Fehler 1
make[1]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/iksemel-978b733462“ wird verlassen

ERROR: Build failed.
make: *** [make/libs/iksemel/iksemel.mk:27: source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/iksemel-978b733462/src/.libs/libiksemel.so.3.1.1] Fehler 1

fda77, 2021-12-25:
Dann nimm halt die openssl version die zur Zeit der Veröffentlichung von Iksemel aktuell war. Vielleicht gibt es eine readme in der die Kompatibilität steht. Wenn die Version dir zu unsicher ist und lieber ne neuere möchtest musst du dort nachfragen, der patch bringt ja scheinbar nichts. Alternativ ne andere Libs nutzen falls möglich, wofür du die brauchst hattest du nicht beantwortet. Vielleicht braucht man von diesem Programm auch eine neuere Version. Oder ein anderes Programm kann es besser.
So weiss man halt nichts

pfichtner, 2022-01-28:
Uuups, sorry, hatte übersehen, dass es hier schon nen Issue gibt, dann hätte ich #449 hier angehängt

fda77, 2022-01-29:
Ach prima, dann behebt dein PR dieses Issue :)

closed 2022-01-29 by fda77