Freetz-NG/freetz-ng #429: Mosquitto 1.6.15 fails with message error: macro "pthread_testcancel" passed 1 arguments, but takes just 0

pfichtner, 2021-12-26:
With be43d906a5a13e9933156f79d2c7c53ed9303e8b mosquitto was upgraded from 1.6.8 to 1.6.15.  
```
make[2]: Entering directory '/freetz/source/target-mipsel_gcc-4.6.4_uClibc-0.9.32.1/mosquitto-1.6.15/src'
/freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_BRIDGE -DWITH_PERSISTENCE -DWITH_SYS_TREE -DWITH_EC -DWITH_EPOLL -Ideps -march=4kc -mtune=4kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -DVERSION="\"1.6.15\"" -DWITH_BROKER -c mosquitto.c -o mosquitto.o
/freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_BRIDGE -DWITH_PERSISTENCE -DWITH_SYS_TREE -DWITH_EC -DWITH_EPOLL -Ideps -march=4kc -mtune=4kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 --std=gnu99 -DVERSION="\"1.6.15\"" -DWITH_BROKER -c ../lib/alias_mosq.c -o alias_mosq.o
In file included from /freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/lib/gcc/mipsel-linux-uclibc/4.6.4/../../../../mipsel-linux-uclibc/include/bits/uClibc_mutex.h:15:0,
                 from /freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/lib/gcc/mipsel-linux-uclibc/4.6.4/../../../../mipsel-linux-uclibc/include/bits/uClibc_stdio.h:107,
                 from /freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/lib/gcc/mipsel-linux-uclibc/4.6.4/../../../../mipsel-linux-uclibc/include/stdio.h:72,
                 from ../lib/memory_mosq.h:20,
                 from ../lib/alias_mosq.c:21:
/freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/lib/gcc/mipsel-linux-uclibc/4.6.4/../../../../mipsel-linux-uclibc/include/pthread.h:622:37: error: macro "pthread_testcancel" passed 1 arguments, but takes just 0
make[2]: *** [Makefile:82: alias_mosq.o] Error 1
```

Downgrading mosquitto to the previous previous version 1.6.8 (and removing 100-pthread_setname_np.patch) fixed the problem, so the errors seems to be in relation to the upgrade.

fda77, 2021-12-28:
Seems to work 
```

$ make mosquitto-precompiled
mkdir -p packages/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15/root
if test -d make/mosquitto/files; then   tools/tar-gnu -cf - -C make/mosquitto/files --exclude=.svn --exclude=.git --exclude=.gitignore --exclude=.build-prereq-checked --exclude=.unpacked --exclude=.configured --exclude=.compiled --exclude=.installed . | tools/tar-gnu -xf - -C packages/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15; fi
---> package/mosquitto ... preparing ... mkdir -p source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15; tools/gunzip -c dl/mosquitto-1.6.15.tar.gz | tools/tar-gnu -x -C source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15 --transform='s|^./\+||' --strip-components=1
set -e; shopt -s nullglob; for i in make/mosquitto/patches/*.patch*; do case $i in *.patch|*.patch.gz|*.patch.bzip2|*.patch.bz2|*.patch.bz|*.patch.xz|*.patch.lz|*.patch.lzma|*.patch.Z) ;; *) continue ;; esac; tools/freetz_patch source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15 $i; done;
    applying patch file make/mosquitto/patches/010-no_cxx_lib.patch
    patching file lib/Makefile
    ----------------------------------------------------------------------
    applying patch file make/mosquitto/patches/100-pthread_setname_np.patch
    patching file lib/thread_mosq.c
    ----------------------------------------------------------------------
cmd() { PATH="/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin:/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0/mips-unknown-linux-gnu/bin:/home/freetz/.local/bin:/home/freetz/bin:/usr/lib64/ccache:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/freetz/.composer/vendor/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make -j9  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };   if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -ne "\e[48;5;56mbuilding\e[49m ... "; touch source/.echo_item_build; fi; cmd -C source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15 CC="/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CFLAGS="-march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1" WITH_DOCS=no WITH_SRV=no WITH_UUID=no WITH_TLS=no WITH_MEMORY_TRACKING=no
building ... make[1]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15“ wird betreten
set -e; for d in lib client src; do make -C ${d}; done
make[2]: Verzeichnis „/home/freetz/freetz-ng/source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15/lib“ wird betreten
/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_THREADING -DWITH_SOCKS -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fPIC -c mosquitto.c -o mosquitto.o
/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_THREADING -DWITH_SOCKS -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fPIC -c actions.c -o actions.o
/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_THREADING -DWITH_SOCKS -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fPIC -c callbacks.c -o callbacks.o
/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_THREADING -DWITH_SOCKS -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fPIC -c connect.c -o connect.o
/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_THREADING -DWITH_SOCKS -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fPIC -c handle_auth.c -o handle_auth.o
/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_THREADING -DWITH_SOCKS -march=34kc -mtune=34kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wl,-I/usr/lib/freetz/ld-uClibc.so.1 -fPIC -c handle_connack.c -o handle_connack.o
/home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-gcc  -I. -I.. -I../lib -I../src/deps -DWITH_THREADING -DWITH_SOCKS -march=34kc -m [... cut ...] ; cp -a source/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15/lib/libmosquitto.so* packages/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15/root/usr/lib/freetz/; /home/freetz/freetz-ng/toolchain/build/mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mips-linux-uclibc/bin/mips-linux-uclibc-strip --remove-section={.comment,.note,.pdr} packages/target-mips_gcc-8.3.0_uClibc-1.0.37-nptl_kernel-4.9/mosquitto-1.6.15/root/usr/lib/freetz/libmosquitto.so.1;
done.
```

pfichtner, 2021-12-28:
I guess the problem is `uClibc-0.9.32.1` vs `uClibc-1.0.37`. Do you have any suggestions what I could try to get deeper into it?

fda77, 2021-12-29:
The uclibc version depends on your device and fritzos version

pfichtner, 2021-12-29:
Am I right that the problem is due to different uClibc versions? If so your answer means that mosquitto 1.6.8 will compile with the uClibc version used on my box while 1.6.15 won't anymore. If so: Anything I can do to get it working 1.6.15 working?

fda77, 2021-12-29:
No, it couold be everything. A bug in the new mosquitto version, removed backward compatibility, uclibc, gcc, error in crosscompiler/host thing are used etc

pfichtner, 2021-12-29:
Fixed it manually, this made it work

```
freetz@693eb6861300:/freetz$ diff -Naur /freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/include/pthread.h.old /freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/include/pthread.h    
--- /freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/include/pthread.h.old	2021-12-29 22:58:35.649726659 +0100
+++ /freetz/toolchain/build/mipsel_gcc-4.6.4_uClibc-0.9.32.1/mipsel-linux-uclibc/include/pthread.h	2021-12-29 22:59:33.881536051 +0100
@@ -619,7 +619,7 @@
 /* Test for pending cancellation for the current thread and terminate
    the thread as per pthread_exit(PTHREAD_CANCELED) if it has been
    cancelled.  */
-extern void pthread_testcancel (void);
+extern void pthread_testcancel ();
 
 
 /* Install a cleanup handler: ROUTINE will be called with arguments ARG
```

But I don't know how and when to apply the patch and I don't know if this will lead to other problems (my firmware builds fine from scratch)

fda77, 2021-12-29:
If you change uclibc, other programs may fail to compile which use this function.
Isnt it better to patch mqtt and add 1 argument?
```
grep pthread_testcancel  . -r
./lib/loop.c:                   pthread_testcancel();
./lib/loop.c:                   pthread_testcancel();
./lib/dummypthread.h:#define pthread_testcancel()
```
Or: If you check the files you see they are in a `#ifdef HAVE_PTHREAD_CANCEL`. So it should compile if this is unset.
In version 1.6.11 - 2020-08-11 there where a change of HAVE_PTHREAD_CANCEL

As im not a c developer i cant say whats the best solution. You could ask in mqtt forum/bugtracker.
If asked: uclib version could not be changed, as avm closed-source binary need this version

fda77, 2021-12-29:
https://github.com/eclipse/mosquitto/blame/master/lib/loop.c
https://github.com/eclipse/mosquitto/commit/b3c2ac20dc5fdaac75baa046134d716e0f5ae846
https://github.com/eclipse/mosquitto/commit/4ef48269f5dd98ef20aee209813222c13c5a5072
https://github.com/eclipse/mosquitto/issues/117
https://github.com/eclipse/mosquitto/pull/1748
https://github.com/eclipse/mosquitto/issues/1547

pfichtner, 2021-12-30:
Thanks for your help and your investigation @fda77! 

As I am not a c developer, too, I played around in mosquitto sources without knowing what I am doing. 

This works for me: 
```
freetz@e149213ae41d:/freetz/source/target-mipsel_gcc-4.6.4_uClibc-0.9.32.1/mosquitto-1.6.15$ diff -Naur lib/dummypthread.h.orig lib/dummypthread.h
--- lib/dummypthread.h.orig	2021-06-09 16:06:23.000000000 +0200
+++ lib/dummypthread.h	2021-12-30 11:56:21.898242877 +0100
@@ -4,7 +4,7 @@
 #define pthread_create(A, B, C, D)
 #define pthread_join(A, B)
 #define pthread_cancel(A)
-#define pthread_testcancel()
+#define pthread_testcancel(void)
 
 #define pthread_mutex_init(A, B)
 #define pthread_mutex_destroy(A)
```

Mosquitto switches between the original pthread include and the dummy, see 
https://github.com/eclipse/mosquitto/blob/3cbe805e71ac41a2a20cc9b2ea6b3b619f49554a/lib/mosquitto_internal.h#L36
While my pthread.h files have a signature of `extern void pthread_testcancel (void);` while mosquitto's dummythread.h has a define `#define pthread_testcancel()`
After adapting it (see patch) to have the same signature everything is fine. 

Do you have an idea now how this could be integrated cleanly into freetz? Otherwise I'll ask on the mosquitto mailing list or will ask some C developers.

fda77, 2021-12-30:
You could put your patch local into make/mosquitto/patches/
As it depends on uclib versions, it sould be conditional like https://github.com/Freetz-NG/freetz-ng/blob/master/make/tinc/tinc.mk#L8
These are used in freetz:
```
	default "0.9.28"    if FREETZ_TARGET_UCLIBC_0_9_28
	default "0.9.29"    if FREETZ_TARGET_UCLIBC_0_9_29
	default "0.9.32.1"  if FREETZ_TARGET_UCLIBC_0_9_32
	default "0.9.33.2"  if FREETZ_TARGET_UCLIBC_0_9_33
	default "1.0.14"    if FREETZ_TARGET_UCLIBC_1_0_14
	default "1.0.37"    if FREETZ_TARGET_UCLIBC_1_0_37
```

Much better: Open an issue here https://github.com/eclipse/mosquitto/issues
When its fixed, the new release version could be used in freetz - if support of old uclib will be fixed

pfichtner, 2021-12-30:
Issue filed: https://github.com/eclipse/mosquitto/issues/2419

fda77, 2021-12-31:
I think "pthread_testcancel" if used form libc/uclibc OR dummypthread. It seems "threads" could be disabled, but I dont know if then some functions are recuced. 
I hope they dont close the issue because they dont want to support anymore a 10 year old uclibc

pfichtner, 2022-01-01:
If I am right this is indeed a bug in the mosquitto sources that is covered on modern compilers, see my [latest comment](https://github.com/eclipse/mosquitto/issues/2419#issuecomment-1003623974). If so [my patch](https://github.com/Freetz-NG/freetz-ng/issues/429#issuecomment-1002987960) could be applied generally. But let's wait for the mosquitto devs reply.

fda77, 2022-01-04:
Yes, even latest uclibc-ng 1.0.39 has `pthread_testcancel (void)` https://elixir.bootlin.com/uclibc-ng/v1.0.39/source/libpthread/nptl/pthread_testcancel.c
So your patch should fit for all uclibc and gcc versions

pfichtner, 2022-01-04:
Applied my patch without any condition and build against older and newer uClibc version without any problem. This reinforces my opinion that it is a bug in mosquitto. But let's wait and see the reaction of the mosquitto team. 
On the other hand, we could of course also fix mosqitto version 1.6.5 in Freetz-NG directly without waiting what the mosquitto team will release in the future.

fda77, 2022-01-07:
Fixed by https://github.com/Freetz-NG/freetz-ng/commit/a575b08d2b3c49c293dcac27394ad68053360999 , if there is a fix upstream the patch could be deleted again

closed 2022-01-07 by fda77