Freetz-NG/freetz-ng #438: Error building wireguard-linux-compat for 4040

mbfoo, 2022-01-06:
wireguard-linux-compat build fails building for FritzBox 4040. I'm unsure if the KERNEL_VERSION-redefines matter, therefore I did not strip these messages from the log. git is at current HEAD

```
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:6:0,
                 from include/linux/avm_pa.h:40,
                 from include/linux/skbuff.h:43,
                 from include/linux/ip.h:20,
                 from include/linux/if_tunnel.h:4,
                 from include/net/ip_tunnels.h:4,
                 from include/net/udp_tunnel.h:4,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/udp_tunnel/udp_tunnel_partial_compat.h:143,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:73,
                 from <command-line>:0:
include/generated/uapi/linux/version.h:1:0: warning: "LINUX_VERSION_CODE" redefined
 #define LINUX_VERSION_CODE 263228
 ^
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:10:0,
                 from <command-line>:0:
/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:10:0: note: this is the location of the previous definition
 #define LINUX_VERSION_CODE KERNEL_VERSION(COMPAT_VERSION, COMPAT_PATCHLEVEL, COMPAT_SUBLEVEL)
 ^
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:6:0,
                 from include/linux/avm_pa.h:40,
                 from include/linux/skbuff.h:43,
                 from include/linux/ip.h:20,
                 from include/linux/if_tunnel.h:4,
                 from include/net/ip_tunnels.h:4,
                 from include/net/udp_tunnel.h:4,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/udp_tunnel/udp_tunnel_partial_compat.h:143,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:73,
                 from <command-line>:0:
include/generated/uapi/linux/version.h:2:0: warning: "KERNEL_VERSION" redefined
 #define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + (c))
 ^
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:10:0,
                 from <command-line>:0:
/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:8:0: note: this is the location of the previous definition
 #define KERNEL_VERSION(a, b, c) (((a) << 24) + ((b) << 16) + (c))
 ^
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:6:0,
                 from include/linux/avm_pa.h:40,
                 from include/linux/skbuff.h:43,
                 from include/linux/ip.h:20,
                 from include/linux/if_tunnel.h:4,
                 from include/net/ip_tunnels.h:4,
                 from include/net/udp_tunnel.h:4,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/udp_tunnel/udp_tunnel_partial_compat.h:143,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:73,
                 from <command-line>:0:
include/generated/uapi/linux/version.h:1:0: warning: "LINUX_VERSION_CODE" redefined
 #define LINUX_VERSION_CODE 263228
 ^
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:10:0,
                 from <command-line>:0:
/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:10:0: note: this is the location of the previous definition
 #define LINUX_VERSION_CODE KERNEL_VERSION(COMPAT_VERSION, COMPAT_PATCHLEVEL, COMPAT_SUBLEVEL)
 ^
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:6:0,
                 from include/linux/avm_pa.h:40,
                 from include/linux/skbuff.h:43,
                 from include/linux/ip.h:20,
                 from include/linux/if_tunnel.h:4,
                 from include/net/ip_tunnels.h:4,
                 from include/net/udp_tunnel.h:4,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/udp_tunnel/udp_tunnel_partial_compat.h:143,
                 from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:73,
                 from <command-line>:0:
include/generated/uapi/linux/version.h:2:0: warning: "KERNEL_VERSION" redefined
 #define KERNEL_VERSION(a,b,c) (((a) << 16) + ((b) << 8) + (c))
 ^
In file included from /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/compat.h:10:0,
                 from <command-line>:0:
/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/compat/version/linux/version.h:8:0: note: this is the location of the previous definition
 #define KERNEL_VERSION(a, b, c) (((a) << 24) + ((b) << 16) + (c))
 ^
/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/socket.c: In function 'wg_socket_send_buffer_to_peer':
/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/socket.c:198:2: error: implicit declaration of function 'skb_put_data' [-Werror=implicit-function-declaration]
  skb_put_data(skb, buffer, len);
  ^
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:258: /home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src/socket.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:1406: _module_/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src] Error 2
make[2]: *** [GNUmakefile:28: modules] Error 2
make[1]: *** [Makefile:26: module] Error 2
make[1]: Leaving directory '/home/mark/freetz-ng/source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20210606/src'

ERROR: Build failed.
```

fda77, 2022-01-07:
Die Fehlermeldung selbst:
```
source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20211208/src/compat/siphash/siphash.c: In function '__siphash_unaligned':
source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20211208/src/compat/siphash/siphash.c:112:36: error: 'fallthrough' undeclared (first use in this function)
  case 7: b |= ((u64)end[6]) << 48; fallthrough;
                                    ^
source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20211208/src/compat/siphash/siphash.c:112:36: note: each undeclared identifier is reported only once for each function it appears in
source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20211208/src/compat/siphash/siphash.c: In function '__hsiphash_unaligned':
source/target-arm_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-4.4/wireguard-linux-compat-1.0.20211208/src/compat/siphash/siphash.c:451:36: error: 'fallthrough' undeclared (first use in this function)
  case 3: b |= ((u32)end[2]) << 16; fallthrough;
                                    ^
```

Wo das "fallthrough" abgeblieben ist weiss ich nicht, kernel 4.4 den die 4040 verwendet hat bereits einen Patch:
https://github.com/Freetz-NG/freetz-ng/blob/master/make/wireguard-linux-compat/wireguard-linux-compat.mk#L10
https://github.com/Freetz-NG/freetz-ng/blob/master/make/wireguard-linux-compat/patches/07_2X_MIN/100-avm_has-skb_put_data.patch

fda77, 2022-01-30:
Im Kernel wurde `fallthrough` "gefixt" indem es einfach gelöscht wurde, scheint überflüssig: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=6a9dc5fd6170d0a41c8a14eb19e63d94bea5705a

closed 2022-01-30 by fda77