Freetz-NG/freetz-ng #443: Build for 7390 ends with md4.ko and cmac.ko error

Darkyputz, 2022-01-11:
Hello...
I was trying to build an image with and without kernel replacement but all i get is:
`installing and stripping modules
  cifs.ko ...........................   95.83 Kb 
  jbd2.ko ...........................   23.41 Kb 
  ext4.ko ...........................  101.43 Kb 
  ecb.ko ............................    1.48 Kb 
  arc4.ko ...........................    0.94 Kb 
  sha256_generic.ko .................    3.54 Kb 
  crc16.ko ..........................    1.22 Kb 
WARNING: File of kernel module not found: cmac.ko
WARNING: File of kernel module not found: md4.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
Makefile:306: die Regel für Ziel „firmw`
Found similar issue already in the closed section of issues, so i am not sure if it reappeared or never got resolved...
Any help would be great

fda77, 2022-01-14:
Hallo, lange nichts von dir gehört. Hast du deutsch schon verlernt ;-)
As long as nobody fixes it, it keeps unfixed... How to fix is in the "ERROR" line. Freetz hast many different kernel, and many devices with same version has even different config file. As nobody wants to reconfig all of these and test if they still compile the info about how to fix was added. So if you need it for your device, you have to fix it

closed 2022-01-14 by fda77

Darkyputz, 2022-01-14:
Hallo cuma, nix verlernt...nur nicht sicher was hier haus und hof sprache iss. Habe jetzt ne firmware gebaut bekommen. Problem haengt mit cifsmount zusammen, cmac und md4 sind in kernel fs im regulaeren menuconfig ausgewahelt aber werden nicht gebaut. Durch 80 libs und kos musste ich noch den kernel replacen...werds dann mal flashen.
Wie kommts das es zu diesem fork kam? Sehr gepsannt

fda77, 2022-01-15:
AVM sieht diese kernelmodule nicht vor, und da deren .config genutzt wird muss irgend jemand die gewünschten Module aktivieren. Fall dieser danach noch einen pull-request mach muss es sonst nie wieder jemand für dieses gerät+fos machen...
Das betrifft Module für alles mögliche wie nfs, cifs, dateisysteme, usb-treiber usw
Zum Fork gibt es im freetz/ repo ne alte Diskussion (bis sie von kriegaex gesperrt wurde). Ich hatte dort keine Schreibrechte und mir war es zu mühselig für jeden commit einen pr zu machen und immer zu warten bis jemand (da war schon fast niemand mehr aktiv) es commited

Darkyputz, 2022-01-16:
Soooo....
Nu bin ich grad mal nen bissel in den make files von cifsmount unterwegs gewesen, aber das scheint nicht die korreckte addresse zu sein um dem problem her zu werden...
Kannst du mich in die richtige richtung stubsen? werde dann auch bissel modelpflege betreiben.
Aber doch schon boese ahnungslos am herumstochern Grins

fda77, 2022-01-16:
Du bist ja nicht der erste der fragt.. Ich versteh nicht weshalb jedes mal sogar die Meldung mitgepostet wird in der steht wie man das selbst beheben kann:

```
WARNING: File of kernel module not found: cmac.ko
WARNING: File of kernel module not found: md4.ko
ERROR: Enable missing with 'make kernel-menuconfig' as (M)odule (and push a pull request). Use '/' to search in menuconfig.
```
Auf deutsch: aufrufen - suchen - auswählen. Gut, dass man anschliessend speichern soll steht nicht dabei

Darkyputz, 2022-01-16:
Cuma...du kennst mich, wenn es md4 und oder cmac bei dieser make kernel menuconfig gegeben haette und ich damit weiter gekommen waere, dann haett ich dich nicht belaestigt. Werd nochmal gucken aber das letzte mal war nix da

fda77, 2022-01-16:
Dann sag doch einfach dass es das Modul in dieser Kernelversion nicht gibt! Das ist was anderes...

Sieht nach ab kernel 3.10 aus: https://cateee.net/lkddb/web-lkddb/CRYPTO_CMAC.html
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/crypto/cmac.c?h=v4.0.1&id=93b5e86a6d13c5dec18c6611933fb38d7d80f0d2

Das ist dann ein Fehler in Freetz unter config/

fda77, 2022-01-16:
Probier mal aus ob das so funktioniert.

Darkyputz, 2022-01-19:
Soooo...endlich zeit gefunden. compiliert und hat deine md4 aenderung und cmac aenderung genommen...
Muss heut abend nur noch flashen und gucken ob das jetyt wieder mit cifs mount klappt...
Danke und sorry fuer die verwirrung