Freetz-NG/freetz-ng #448: Upload error (Fritzbox 6890 LTE)

tadoopai, 2022-01-23:
Hallo,

ich habe eine Fritzbox 6890 LTE (v2) und habe ein Problem beim upload.
Ich have folgenden guide verwendet: [LINK](https://web.archive.org/web/20190529205946/https://www.ip-forum.eu/howto-fritz-box-firmware-images-auch-unsignierte-ueber-den-bootloader-installieren-577#p3852)

1. Ich habe das Image erfolgreich (über die freetz VM) erstellt.
2. Ich habe daraus mein in-memory Image erstellt (Fritzbox 6890 LTE hat ja so wie ich das gelesen habe einen NAND Speicher)
3. Ich habe versucht das in-memory image hochzuladen, aber immer mit einem Error. (Ich habe auch schon einen anderen PC ausporbiert.

Die Fritzbox startet kurz darauf immer in ihr ursprüngliches FritzOS neu.

Powershell  log:

```
PS D:\temp\eva_tools> ./EVA-FTP-Client.ps1 -Verbose -Debug -ScriptBlock { BootDeviceFromImage D:\temp\eva_tools\images\firmware.image.in-memory }
DEBUG: Response:
220 ADAM2 FTP Server ready

================
DEBUG: Sent
USER adam2
================
DEBUG: Response:
331 Password required for adam2

================
DEBUG: Sent
PASS adam2
================
DEBUG: Response:
230 User adam2 successfully logged in

================
DEBUG: Sent
SYST
================
DEBUG: Response:
215 AVM EVA Version 1.3292 0x0 0x46409

================
DEBUG: Sent
GETENV memsize
================
DEBUG: Response:
memsize               0x20000000

200 GETENV command successful

================
DEBUG: Memory size found    : 0x20000000 (512 MB)
DEBUG: Memory size used     : 0x08000000 (128 MB)
DEBUG: Image size found     : 0x02c26b00
DEBUG: Set memory size to   : 0x053d9500
DEBUG: Set MTD RAM device to: 0x853d9500,0x88000000
DEBUG: Sent
SETENV memsize 0x053d9500
================
DEBUG: Response:
200 SETENV command successful

================
DEBUG: Sent
SETENV kernel_args_tmp mtdram1=0x853d9500,0x88000000
================
DEBUG: Response:
200 SETENV command successful

================
DEBUG: Sent
TYPE I
================
DEBUG: Response:
200 Type set to BINARY

================
DEBUG: Sent
MEDIA SDRAM
================
DEBUG: Response:
200 Media set to MEDIA_SDRAM

================
DEBUG: Uploading file 'D:\temp\eva_tools\images\firmware.image.in-memory' to '0x853d9500 0x88000000' ...
DEBUG: Sent
P@SW
================
DEBUG: Response:
227 Entering Passive Mode (192,168,178,1,12,29)

================
DEBUG: Sent
STOR 0x853d9500 0x88000000
================
DEBUG: Response:
150 Opening BINARY data connection

================
DEBUG: Sent
SETENV memsize 0x20000000
================
DEBUG: Response:
226 Transfer complete
200 SETENV command successful

================
DEBUG: Sent
UNSETENV kernel_args_tmp
================
DEBUG: Response:
501 environment variable not set

================
DEBUG: Sent
QUIT
================
DEBUG: Response:
221 Thank you for using the FTP service on ADAM2
221 Goodbye.

================
Ausnahme beim Aufrufen von "Invoke" mit 0 Argument(en):  "Error uploading image file."
In D:\temp\eva_tools\EVA-FTP-Client.ps1:638 Zeichen:21
+                     $ScriptBlock.Invoke()
+                     ~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : RuntimeException
```

tadoopai, 2022-01-23:
Okay ich habe nun auf einem Linux System push_image verwendet und es hat sofort funktioniert!

closed 2022-01-23 by tadoopai

fda77, 2022-01-24:
Es gibt 5 Arten wie eine Fritzbox zu flashen ist, 1 davon ist "inmemory". Welche für die 6890 die richtige ist weiss ich nicht. Da pushfirmware selbst die richtige auswählt braucht man das auch nicht zu wissen.
Mit den Scripten von PeterPawn hab ich keine Erfahrung und kann dazu nichts sagen, am besten direkt hier fragen: https://github.com/PeterPawn/YourFritz