Freetz-NG/freetz-ng #456: ibusb1-1.0.25 Error

freetz-ng-mod, 2022-02-02:
cc1: some warnings being treated as errors
make[3]: *** [Makefile:649: os/linux_usbfs.lo] Fehler 1
make[3]: Verzeichnis „/home/user//tools/freetz/test1/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/libusb1-1.0.25/libusb“ wird verlassen
make[2]: *** [Makefile:480: all-recursive] Fehler 1
make[2]: Verzeichnis „/home/user//tools/freetz/test1/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/libusb1-1.0.25“ wird verlassen
make[1]: *** [Makefile:385: all] Fehler 2
make[1]: Verzeichnis „/home/user//tools/freetz/test1/source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/libusb1-1.0.25“ wird verlassen

ERROR: Build failed.
make: *** [make/libs/libusb1/libusb1.mk:29: source/target-mips_gcc-5.5.0_uClibc-1.0.37-nptl_kernel-3.10/libusb1-1.0.25/libusb/.libs/libusb-1.0.so.0.3.0] Fehler 1

fda77, 2022-02-02:
1) als code formatieren
2) die warnings nennen

freetz-ng-mod, 2022-02-02:
Es kann kein Image gebaut werden.
os/linux_usbfs.c:1432:6: error: 'active_config' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  int active_config;
      ^
cc1: some warnings being treated as errors
mehr steht da nicht

fda77, 2022-02-02:
Dass eine Image nicht gebaut wird wenn ein Teil von make fehlschlägt ist klar.
Es könnte sein dass auch hier ein `--std=gnu99` in die .mk muss, wie hier: https://github.com/Freetz-NG/freetz-ng/commit/6e60f1299f12dae0fe0e10aecbd1f4433baf8ccb Das müsste man irgend wie anders in Freetz lösen.
Das Problem ist der uralten Compiler etc die AVM für deine (unbekannte) Box nutzt

harryboo, 2022-02-03:
Das ist nur eine Warnung.
Probier mal bitte folgenden Patch (einfach den Ordner 'patches' unter 'make\libs\libusb1' anlegen und die Datei reinkopieren, vorher die Extension '.txt' entfernen).

[activeconfig.patch.txt](https://github.com/Freetz-NG/freetz-ng/files/7993612/activeconfig.patch.txt)

fda77, 2022-02-03:
Das Initialisieren wird das "maybe-uninitialized" wohl verhindern.
Bei neuerem GCC ist das ganze aber scheinbar kein Problem da es mit der von mir gewählten Box compiliert
Man könnte die Überprüfung aber auch ganz im configure mit einem `CONFIGURE_PRE_CMDS` in der makefile ausschalten.
Ich weiss nicht so recht was das beste wäre

fda77, 2022-02-03:
Auch eine Möglichkeit wäre es den "error" hier zu melden: https://github.com/libusb/libusb ...

harryboo, 2022-02-03:
Als Ordentlicher Programmierer sorgt man eigentlich immer für eine Initialisierung, wenn die Variable sonst nur in irgendwelchen Bedingungen gesetzt wird. Neuere Compiler machen dies automatisch.
Ich würde den Check nicht ausschalten.

freetz-ng-mod, 2022-02-03:
Mit dem Patch von harryboo lauft es wieder durch

fda77, 2022-02-04:
Wer erstellt ein Issue bei libusb?

WXbet, 2022-02-06:
https://github.com/libusb/libusb/pull/1062

fda77, 2022-02-07:
Thx

Dort wird gefragt:
 >Just wondering what is the Linux distro and C compiler version which trigger this warning. Thanks.

fda77, 2022-02-09:
Okay

closed 2022-02-09 by fda77