Freetz-NG/freetz-ng #460: Sundtek DVB-C/T/T2 driver funktioniert nicht

leo-022, 2022-02-10:
Ich wollte den Sundtek-Treiber installieren (7590). Make läuft auch reibunglos durch.
Leider bekomme ich dann beim Start die Fehlermeldung`/usr/bin/sundtek-mediasrv: can't load library 'librt.so.0'`

Eine librt ist nirgendwo auf dem System vorhanden, auch kann ich sie bei `make menuconfig` nicht auswählen.

fda77, 2022-02-11:
librt war eine Datei der uclibc bei älteren Versionen, bei neueren ist alles in 1 Datei. Ich rate mal dass du ein aktuelles FOS ausgewählt hast, dafür müsste man eine andere sundtek binary nehmen.
Falls die libc von AVM verwendet wird, muss es je nach Gerät+FOS für musl/glibc/uclibc sein...

Lade dir doch http://www.sundtek.de/media/sundtek_installer_210803.071224.sh direkt auf die Box. Mit
`dd if=sundtek_installer_210803.071224.sh skip=1 bs=66145 of=sundtek_installer_210803.071224.tar.gz` die Datei auspackbar machen. Darin gibt es dann fürhttps://github.com/Freetz-NG/freetz-ng/blob/master/make/sundtek/Config.in#L30 : 

```
32bit
32bit23
64bit
64bitmusl
arm4hfmusl32
arm4uclibc
arm64
arm64musl
armoabi
armsysv
armsysvhf
c3
ddwrtmips
ddwrtmips2
dreambox
mips
mipsel
mipsel2
mipselbcm
openwrtarm4
openwrtmipsr2
openwrtmipsr3
openwrtppc32
openwrtppc32msb2015
ppc32
ppc64
sh4
```

Mit Glück funktioniert eine davon, evtl muss man den RPATH und Loader-Path noch mit patchelf anpassen

fda77, 2022-02-11:
ea66ca8797eb309d1d4128c6bb27e5accc0adde0 ist nur ein erster Scrhitt

leo-022, 2022-02-11:
Ja, FritzOS 07.29. Die Dateien habe ich noch nicht getestet. Aber sundtek_installer_210803.071224.sh lässt sich ja auch auf der Box ausführen. Es spuckt u. a. das Folgende aus:
```
checking system... OpenWRT/Fritzbox support enabled
DD-WRT Compatible System detected (2015)
```
"DD-WRT Compatible System detected (2015)" bedeutet laut sh-Datei `SYSTEM="ddwrtmips2"`

leo-022, 2022-02-12:
Verwende ich ddwrtmips2 aus dem Installer-Paket, bekomme ich beim Start von `sundtek-mediasrv` ein "Segmentation fault".

In der Freetz-Beschreibung zu Sundtek steht: "Um den Treiber benutzen zu können muss vorher dieser Befehl ausgeführt werden: `export LD_PRELOAD=/usr/lib/libmediaclient.so`". 

Die `libmediaclient.so` wird gar nicht in das Image übernommen. Mache ich das manuell, bekomme ich auch beim `export LD_PRELOAD` ein "Segmentation fault".

fda77, 2022-02-13:
Wie gesagt, es gibt verschieden libc. Falls du das als package mit freetz installiert hast, wurde der loader-pfad vermutlich auf die von freetz umgeschrieben, `readelf` sollte das zeigen. Die Binary muss halt zu dieser passen. Oder die .sh direkt auspacken, dann sollte die von AVM genutzt weren. Die Binary muss dann auch wieder zu dieser passen. Ich hab keine Ahnung ob sundtek überhaupt mit neueren uclibc funktioniert. Mit `ldd libmediaclient.so` kann man schauen welche libs von einer lib/binary verwendet werden.
Das `LD_PRELOAD` muss man vor Programmen die den Treiber verwenden wollen ausführen. Segfaults bedeuten dass was nicht passt :)

Da AVM mittlerweile nicht mehr bei Kernel 2.6 ist sondern teilweise bei 4 könnte man auch mal schauen ob es dafür Treiber direkt im kernel gibt. zb das Gerät an einen Linux-PC anschliessen und schauen welche Module geladen werden (syslog)

leo-022, 2022-02-13:
Ich sehe nicht ganz durch, was da beim make gemacht wird. Ich hatte einfach in die Config.in `default "ddwrtmips2"` reingeschrieben, aber er nimmt irgendeine andere Datei (man sieht das ja nur an der Größe, da alle denselben Namen haben - ene Datei mit dieser Größe habe ich nirgemdwo im Paket 210803.071224 gefunden), deshalb auch Segfault.

Wenn ich manuell die beiden Dateien mediasrv und mediaclient auf die Box kopiere und per `mount -o bind` auf `/usr/bin/sundtek-mediasrv` und `/usr/bin/sundtek-mediaclient` verweise, funktioniert es, auch ohne irgeneine lib, also ohne `LD_PRELOAD`.

Es ist jetzt nur die Frage, wie ich Config.in beibringe, bei der 7590 ddwrtmips2 zu verwenden.

fda77, 2022-02-13:
Achso, sag das doch...
Es wird wohl die Datei genommen die entpackt wurde - das package muss komplett neu "gebaut" werden. 
 * `make help`
 * `make $(pkg)-distclean`

> Das LD_PRELOAD muss man vor Programmen die den Treiber verwenden wollen ausführen. 

Kommt auf das Programm an das du verwendest

leo-022, 2022-02-14:
Nochmals getestet: Die mit freetz installierten `sundtek-mediasrv` und `sundtek-mediaclient` (`/usr/lib/freetz/ld-uClibc.so.1`) funktionieren leider nicht, aber die originalen ddwrtmips2 aus 210803.071224 extrahierten (`/lib/ld-musl-mips-sf.so.1`) laufen einwandfrei.

Was spricht dagegen, diese in Freetz direkt zu verwenden?

fda77, 2022-02-15:
Probier die openwrtmipsr3 jetzt so mal aus

leo-022, 2022-02-15:
Geht leider nicht. Die einzigen beiden, die mit uClibc 1.0.40 keine anderen Fehler produzieren, sind openwrtmipsr2 und openwrtmipsr3, allerdings melden beide "can't load library 'librt.so.0'". Es ist also in dem Sundtek-Paket nichts enthalten, was mit der aktuellen uClibc funktioniert.

fda77, 2022-02-17:
Binary-only Dinge machen über die Zeit immer Probleme, das sieht man auch an den ganzen AVM Sachen, zb Kernelmodulen. Ohne diese Blobs könnten Fritzboxen einen weniger als 10 Jahre alten Kernel nutzen UND es würde Dect, Telefon etc funktionieren.

Versuch doch die fehlende(n) "librt.so.0" in /usr/lib/freetz/ als Link auf die uclibc (ld-uClibc-1.0.40.so oder libuClibc-1.0.40.so?) zu erstellen. Vielleicht funktioniert das.

In Freetz ist nicht bekannt welche Geräte/FOS musl etc nutzen, dh man könnte sundtek einfach nur für alles ab ca FOS 7.2x ("FREETZ_SEPARATE_AVM_UCLIBC") deaktivieren

sundtek, 2022-04-09:
Habt ihr das hinbekommen oder braucht ihr da noch Hilfe?

fda77, 2022-04-15:
Nein, ich weiss auch nicht wie.

AVM hat vor version 7.20 auf arm/mips(le)/x86 überall eine uralte uclibc-ng 1.0.14 oder noch älter verwendet. Dies war noch  multi-lib  mit den ganzen libm, libdl etc
Ab 7.20 ist AVM umgestiegen auf musl ODER glibc ODER neuere uclibc-ng (single-lib). Jeweils auch verschieden Versionen, siehe auch https://boxmatrix.info/wiki/System-Selector
Vor 7.20 hat freetz die libc von AVM mitgenutzt. Danach war es durch die ganzen verschiedenen zu aufwändig und freetz nutzt immer und überall eine eigen uclibc-ng 1.0.40 (die Geräte haben mittlerweile auch genug Flash dafür). Diese liegt wie auch alle anderen Libs von Freetz in /usr/lib/freetz/, die AVM binaries erkennen diese nicht. Die Binaries von Freetz haben den loader-path und rpath auf dieses Verzeichnis gesetzt.

Wenn man den sundtek Treiber jetzt für alle Geräte haben wollte müsste man für die 3 neuen libcs JE cpu architektur die passende sundtek Binary haben.
Oder sundtek-Binaries für arm/mips(le)/x86 für neue single-lib uclibc-ng. Hier aber mit angepassten Pfaden. Wenn ich es richtig in Erinnerung hab gibt es einer Veränderungserkennung im Treiber und man kann mit patchelf nicht nachträglich Pfade ändern

closed 2022-05-02 by fda77

sundtek, 2022-05-02:
Also im Grunde braucht ihr eigentlich nur eine Version für uclibc-ng 1.0.40?

fda77, 2022-05-02:
Genau, Freetz nutzt auf allen Architekturen uclibc-ng 1.0.40 bei aktuellem FritzOS.
Alle Libs von Freetz befinden sich in /usr/lib/freetz/ , dh der Loader muss dort zuerst gesucht werden (nachträglich mit zb patchelf) und evtl benötigte libs auch (RPATH, zur Not LD_LIBRARY_PATH)
Beispiel:
```
$ ldd $(which mtr)
        libncurses.so.6 => /usr/lib/freetz/libncurses.so.6 (0x77ab7000)
        libc.so.0 => /usr/lib/freetz/libc.so.0 (0x779e1000)
        ld-uClibc.so.1 => /usr/lib/freetz/ld-uClibc.so.1 (0x77b12000)
```

Das ist unabhängig von der libc (welche auch immer) die AVM im "normalen" Pfad hat. 
Die von AVM sind teilweise ungewöhnlich gepatcht, veraltet oder anders konfiguriert und werden daher ignoriert