Freetz-NG/freetz-ng #461: cant start nfsd

tobit6795, 2022-02-13:
Hi,

after building and installing freetz with nfs i get the following error:

```
rpcbind: another rpcbind is already running. Aborting
nfsd: Unable to access /proc/fs/nfsd errno 2 (No such file or directory).
Please try, as root, 'mount -t nfsd nfsd /proc/fs/nfsd' and then restart nfsd to correct the problem
Starting nfsd ... failed.
```

```

fda77, 2022-02-13:
Maybe you need replace kernel for your device

tobit6795, 2022-02-13:
i wasnt aware of nfs4 support might be a problem. so i rebuild the image with nfs3 only.
process can be started, but now i have another problem:

```

fda77, 2022-02-13:
Maybe logfiles/syslog shows some infos
See also https://github.com/Freetz-NG/freetz-ng/pull/441

fda77, 2022-02-13:
https://github.com/Freetz-NG/freetz-ng/blob/master/make/nfsd-cgi/files/root/etc/init.d/rc.nfsd#L3

tobit6795, 2022-02-14:
i modified the unit and rebuild the fw:

```
root@fritz:/var/mod/root# grep -i "^RPCPROG=" $(find / 
-name "rc.nfs*")
/etc/init.d/rc.nfsd:RPCPROG='rpcbind'
/var/mod/etc/init.d/rc.nfsd:RPCPROG='rpcbind'
```
```
root@fritz:/var/mod/root# netstat -tulpen | grep "rpcbi
nd\|port"
tcp        0      0 169.254.1.1:111         0.0.0.0:*               LISTEN      1218/portmap
tcp        0      0 127.0.0.1:111           0.0.0.0:*               LISTEN      1218/portmap
tcp        0      0 :::111                  :::*                    LISTEN      5277/rpcbind
udp        0      0 169.254.1.1:111         0.0.0.0:*                           1218/portmap
udp        0      0 127.0.0.1:111           0.0.0.0:*                           1218/portmap
udp        0      0 :::57404                :::*                                5277/rpcbind
udp        0      0 :::111                  :::*                                5277/rpcbind
```
still having portmap running in parallel

tobit6795, 2022-02-19:
i have two 6591 boxes to compare:

one with fritzos 7.29 + Firmware: 161.07.29 rev92035 / Freetz: ng-18835M-5ad1652e3

another with fritzos07.13(also tested 7.22) Firmware: 161.07.13 rev75441 / Freetz: ng-18882M-6ea279b3e

the second box is the one with the portmap issue, the first one does not have this problem atm

fda77, 2022-02-19:
AVM added a systemd-like thing between 7.13 and 7.29 and much more changes

closed 2022-04-23 by fda77