Freetz-NG/freetz-ng #465: Repeater 6000, 5530 und 7530AX Image Bau-Fehler

freetz-ng-mod, 2022-02-19:
STEP 1: UNPACK
source firmware file: dl/fw/FRITZ.Repeater_6000-07.29.image
checking signature: valid
unpacking firmware image
abort()ing pseudo client by server request. See https://wiki.yoctoproject.org/wiki/Pseudo_Abort for more details on this.
Check logfile: /home/xxxx/60xx/.fakeroot-cache/pseudo.log
./fwmod: Zeile 694: 562490 Abgebrochen             "$TAR_GNU" -xif "$FIRMWARE" -C "$FIRMWARE_DIR"
FINISHED

log
debug_logfile: fd 2
pid 562451 [parent 562450], doing new pid setup and server start
Setup complete, sending SIGUSR1 to pid 562450.
path mismatch [2 links]: ino 47454209 db '/home/xxxx/60xx/build/original/firmware/var/tmp' req '/proc/self/fd/5'.
db cleanup for server shutdown, 13:36:51.745
memory-to-file backup complete, 13:36:51.745.
db cleanup finished, 13:36:51.745

Bei allen Boxen immer der gleiche Fehler.
Kann es sein das alle FIT-Image betroffen sind.

fda77, 2022-02-19:
Diese 3 Geräte sind alle als "DEVELOPER" gekennzeichnet.  Hast du davon schon mal ein geflasht und kannst einen Screenshot von "AVM-Info" anhängen?

freetz-ng-mod, 2022-02-19:
Ich habe versucht denn 6000 zu Freetzen mit push_firmware nur das geht nicht. Glaube da sind die Adressen falsch für 
`
 * MAPSTART=0x00000000
 * FULLSIZE=0x18000000	 (384 MB)
 * MAPLIMIT=0x18000000
 * FILESIZE=0x01550000	 (~21 MB)
 * FREESIZE=0x16ab0000	 (~362 MB)
 * MTDSTART=0x16ab0000
`
`
` * Analyzing '/home/xxxx/freetz-images/images/6000/6000_07_29_all_freetz-ng-18764MO-bfb5ad36f-repeater_openssl_1_1_x.image' ... 

 * Using command: ncftpput
 * Target host: 192.168.178.1
 * Outgoing IP: 192.168.178.32
 * Flash mode: fit-boot
 * Allowed memory size: 384 MB
 * Designated linux_fs_start: <other>

 !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!!
 !!!  THERE IS NO WARRANTY AT ALL !!! USE AT YOUR OWN RISK   !!!

 * Are you sure, that you want to flash this file to the device?
   /home/xxxx/freetz-images/images/6000/6000_07_29_all_freetz-ng-18764MO-bfb5ad36f-repeater_openssl_1_1_x.image
   Proceed? (y/[n]) y

 * You should now reboot your box (192.168.178.1). Waiting for shut down.
   Switch off, if reboot is not detected because it happens too quickly.
   Some newer bootloader versions allow to flash on power-cycle only.
   
 * MAPSTART=0x00000000
 * FULLSIZE=0x18000000	 (384 MB)
 * MAPLIMIT=0x18000000
 * FILESIZE=0x01550000	 (~21 MB)
 * FREESIZE=0x16ab0000	 (~362 MB)
 * MTDSTART=0x16ab0000

 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting ........ found!

 * Detected linux_fs_start: '0'
 * Designated linux_fs_start: 1

 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting ........ found!

 * Box is back up again, initiating transfer.

LibNcFTP 3.2.5 (January 17, 2011) compiled for linux-x86_64-glibc2.27
Uname: Linux|xmg-ultra|5.13.0-10027-tuxedo|#29~20.04.1tux1 SMP Thu Jan 20 10:52:53 UTC 2022|x86_64
Contents of /etc/debian_version:
  bullseye/sid
Contents of /etc/issue:
  Ubuntu 20.04.3 LTS \n \l
Glibc: 2.31 (stable)
220: ADAM2 FTP Server ready
Connected to 192.168.178.1.
Cmd: USER adam2
331: Password required for adam2
Cmd: PASS xxxxxxxx
230: User adam2 successfully logged in
Logged in to 192.168.178.1 as adam2.
Cmd: MEDIA SDRAM
200: Media set to MEDIA_SDRAM
Cmd: SETENV memsize 0x16ab0000
200: SETENV command successful
Cmd: SETENV kernel_args_tmp "avm_fwupdate mtdram1=0x16ab0000,0x18000000 mtdparts_ext=update-image.0:0x01550000@0x0(fit-image)"
200: SETENV command successful
Cmd: TYPE I
200: Type set to BINARY
Cmd: MLST 0x16ab0000 0x18000000
502: Command not implemented
Cmd: PASV
227: Entering Passive Mode (192,168,178,1,43,206)
Cmd: STOR 0x16ab0000 0x18000000
150: Opening BINARY data connectiont
`
Denn wenn man hier schaut
https://boxmatrix.info/wiki/FIT-Image#6000
brauche ich dafür aber 0x7ece0000,0x80000000

fda77, 2022-02-19:
Dafür gibt es eine Option:
```
$ push_firmware  --help
-map <hex>        Only ram-boot and fit-boot mode: Start of mapped memory (PHYS_OFFSET).
With Broadcom devices (7581+7582) you have to use like avm: '-ram 512 -map 0x00000000'
```

Um es 100% zu wissen wie es richtig geht, mach mit dem avm-tool ein recovery und schneide die ftp Verbindung mit (command-Kanal reicht)

freetz-ng-mod, 2022-02-19:
Die habe ich gesehen,  aber wo trage ich die ein ;-)

fda77, 2022-02-19:
Vielleicht ist etwas zu wenig ram zugewiesen. Laut Boxmatrix nutzt avm etwa 1gb 
```
SETENV memsize 0x3ece0000     = 1052770304 dezimal   /1024/1024  == 1004megabyte
```
PF nutzt 384mb. Also beim Aufruf noch "-ram " dazu, 1024 vielleicht
```
Einfach mit ram&map etwas herumspielen...
Man muss auch nicht immer (versuchen zu) flashen, die Werte werden am Anfang angezeigt

fda77, 2022-02-19:
Nach etwas experimentieren:
```
....  -ram 1024 -map 0x40000000
 * MAPSTART=0x40000000
 * FULLSIZE=0x40000000   (1024 MB)
 * MAPLIMIT=0x80000000
 * FILESIZE=0x014a0000   (~20 MB)
 * FREESIZE=0x3eb60000   (~1003 MB)
 * MTDSTART=0x7eb60000

```

freetz-ng-mod, 2022-02-19:
``` 
* Analyzing '/home/xxxx/freetz-images/images/6000/6000_07_29_all_freetz-ng-18764MO-bfb5ad36f-repeater_openssl_1_1_x.image' ... 

 * Using command: ncftpput
 * Target host: 192.168.178.1
 * Outgoing IP: 192.168.178.50
 * Flash mode: fit-boot
 * Allowed memory size: 1004 MB
 * Designated linux_fs_start: <other>

 !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!!
 !!!  THERE IS NO WARRANTY AT ALL !!! USE AT YOUR OWN RISK   !!!

 * Are you sure, that you want to flash this file to the device?
   /home/xxxx/freetz-images/images/6000/6000_07_29_all_freetz-ng-18764MO-bfb5ad36f-repeater_openssl_1_1_x.image
   Proceed? (y/[n]) y

 * You should now reboot your box (192.168.178.1). Waiting for shut down.
   Switch off, if reboot is not detected because it happens too quickly.
   Some newer bootloader versions allow to flash on power-cycle only.
   
 * MAPSTART=0x70000000
 * FULLSIZE=0x3ec00000	 (1004 MB)
 * MAPLIMIT=0xaec00000
 * FILESIZE=0x01550000	 (~21 MB)
 * FREESIZE=0x3d6b0000	 (~982 MB)
 * MTDSTART=0xad6b0000

 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting ............ found!

 * Detected linux_fs_start: '0'
 * Designated linux_fs_start: 1

 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting ....... found!

 * Box is back up again, initiating transfer.

LibNcFTP 3.2.5 (January 17, 2011) compiled for linux-x86_64-glibc2.27
Uname: Linux|xmg-ultra|5.13.0-10027-tuxedo|#29~20.04.1tux1 SMP Thu Jan 20 10:52:53 UTC 2022|x86_64
Contents of /etc/debian_version:
  bullseye/sid
Contents of /etc/issue:
  Ubuntu 20.04.3 LTS \n \l
Glibc: 2.31 (stable)
220: ADAM2 FTP Server ready
Connected to 192.168.178.1.
Cmd: USER adam2
331: Password required for adam2
Cmd: PASS xxxxxxxx
230: User adam2 successfully logged in
Logged in to 192.168.178.1 as adam2.
Cmd: MEDIA SDRAM
200: Media set to MEDIA_SDRAM
Cmd: SETENV memsize 0x3d6b0000
200: SETENV command successful
Cmd: SETENV kernel_args_tmp "avm_fwupdate mtdram1=0xad6b0000,0xaec00000 mtdparts_ext=update-image.0:0x01550000@0x0(fit-image)"
200: SETENV command successful
Cmd: TYPE I
200: Type set to BINARY
Cmd: MLST 0xad6b0000 0xaec00000
502: Command not implemented
Cmd: PASV
227: Entering Passive Mode (192,168,178,1,42,2)
Cmd: STOR 0xad6b0000 0xaec00000
150: Opening BINARY data connection
/tmp/freetz_ivg/fitboot.flash:     ETA:  19:48    0.12/ 21.31 MB   18.26 kB/s -Remote write failed after 131072 bytes had been sent: Connection reset by peer.
Could not read reply from control connection -- timed out.
/tmp/freetz_ivg/fitboot.flash:                          21.31 MB  922.19 B/s  
ncftpput 0xad6b0000 0xaec00000: socket write error.
```

fda77, 2022-02-19:
Schau mal den Post 6 sekunden vor deinem

freetz-ng-mod, 2022-02-19:
``` 
 * Analyzing '/home/xxxx/freetz-images/images/6000/6000_07_29_all_freetz-ng-18764MO-bfb5ad36f-repeater_openssl_1_1_x.image' ... 

 * Using command: ncftpput
 * Target host: 192.168.178.1
 * Outgoing IP: 192.168.178.50
 * Flash mode: fit-boot
 * Allowed memory size: 1024 MB
 * Designated linux_fs_start: <other>

 !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!! WARNING !!!
 !!!  THERE IS NO WARRANTY AT ALL !!! USE AT YOUR OWN RISK   !!!

 * Are you sure, that you want to flash this file to the device?
   /home/xxxx/freetz-images/images/6000/6000_07_29_all_freetz-ng-18764MO-bfb5ad36f-repeater_openssl_1_1_x.image
   Proceed? (y/[n]) y

 * You should now reboot your box (192.168.178.1). Waiting for shut down.
   Switch off, if reboot is not detected because it happens too quickly.
   Some newer bootloader versions allow to flash on power-cycle only.
   
 * MAPSTART=0x40000000
 * FULLSIZE=0x40000000	 (1024 MB)
 * MAPLIMIT=0x80000000
 * FILESIZE=0x01550000	 (~21 MB)
 * FREESIZE=0x3eab0000	 (~1002 MB)
 * MTDSTART=0x7eab0000

 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting .ping: connect: Das Netzwerk ist nicht erreichbar
.......... found!

 * Detected linux_fs_start: '1'
 * Designated linux_fs_start: 0

 * No reply from box, assuming switch-off or restart. Trying to re-detect box.
   Waiting .ping: connect: Das Netzwerk ist nicht erreichbar
.. found!

 * Box is back up again, initiating transfer.

LibNcFTP 3.2.5 (January 17, 2011) compiled for linux-x86_64-glibc2.27
Uname: Linux|xmg-ultra|5.13.0-10027-tuxedo|#29~20.04.1tux1 SMP Thu Jan 20 10:52:53 UTC 2022|x86_64
Contents of /etc/debian_version:
  bullseye/sid
Contents of /etc/issue:
  Ubuntu 20.04.3 LTS \n \l
Glibc: 2.31 (stable)
220: ADAM2 FTP Server ready
Connected to 192.168.178.1.
Cmd: USER adam2
331: Password required for adam2
Cmd: PASS xxxxxxxx
230: User adam2 successfully logged in
Logged in to 192.168.178.1 as adam2.
Cmd: MEDIA SDRAM
200: Media set to MEDIA_SDRAM
Cmd: SETENV memsize 0x3eab0000
200: SETENV command successful
Cmd: SETENV kernel_args_tmp "avm_fwupdate mtdram1=0x7eab0000,0x80000000 mtdparts_ext=update-image.0:0x01550000@0x0(fit-image)"
200: SETENV command successful
Cmd: TYPE I
200: Type set to BINARY
Cmd: MLST 0x7eab0000 0x80000000
502: Command not implemented
Cmd: PASV
227: Entering Passive Mode (192,168,178,1,39,225)
Cmd: STOR 0x7eab0000 0x80000000
150: Opening BINARY data connection
/tmp/freetz_ZGv/fitboot.flash:     ETA:   0:00    0.03/ 21.31 MB  303.40 MB/s  226: Transfer complete
/tmp/freetz_ZGv/fitboot.flash:                          21.31 MB    2.22 MB/s  
Cmd: SETENV linux_fs_start 0
200: SETENV command successful
Cmd: QUIT
221: Thank you for using the FTP service on ADAM2
Cmd: QUIT
221: Goodbye.

done
``` 
Aber es ist kein freetz auf dem 6000

fda77, 2022-02-19:
Das sieht doch schon mal gut aus! Passt das "linux_fs_start"? Nicht dass bei den "fit" sich da was geändert hat. Mit dem Parameter "-lfs 0" oder 1 kann man eins fest auswählen. Es könnte aber auch sein dass Freetz nicht richtig compiliert/gepackt wurde.
Hast du noch eine andere, ältere AVM Firmware des 6000? Flash die doch mal und schau ob sich die Version ändert, also ob wirklich geflasht wird.

Um es richtig zu erkennen: Eine serielle Console anschliessen. Als ich das mir meinem 1200 gemacht hab, hatte ich die 3 oder 4 Pins einfach mit Klebestreifen festgepappt, aufschrauben war auch kein Problem

freetz-ng-mod, 2022-02-19:
Das image ist von 14.12.21. Denn da habe ich das letzte mal eins gebaut.

fda77, 2022-02-19:
"Flashen" sind hier 2 Schritte: in den ram hochladen + dieses system wird gestartet und flasht sich selbst
Es ist nicht ganz klar ob der 2. Schritt auch funktioniert hat. Falls es in Freetz einen Fehler gab könnte das Schreiben abgebrochen worden sein. Wenn du ein altes AVM flasht kann man erkennen ob das flashen grundsätzlich funkioniert, da avm wissen sollte wie das Image korrekt zusammengepackt wird

freetz-ng-mod, 2022-02-19:
Also sie hat nicht neu gestartet nach dem Hochladen. Habe 15 min gewartet

fda77, 2022-02-19:
Es könnte an push firmware noch etwas falsch sein, oder an der modifikation des Images -> mit einem (alten) von avm ausprobieren

freetz-ng-mod, 2022-02-19:
Werde ich später testen. Denn habe noch die FRITZ.Repeater_6000-07.28.image

fda77, 2022-02-19:
Falls an Freetz irgend etwas nicht passt, wird es übrigens ohne Console sehr schwer herauszufinden woran es liegt und damit es zu behenen. Solche Adapter gibts schon für ein paar Euro

freetz-ng-mod, 2022-02-19:
Hast mal ein Link für so ein Adapter

fda77, 2022-02-19:
Ich hatte mir diesen vor kurzem gekauft um auf einem Speedport "root zugriff" zu bekommen (auf 3,3v jumpern!): https://www.amazon.de/gp/product/B07G87GW4K
Ich bin aber kein Elektriker. Frag doch bei Hippie im IRC nach, der kennt sich da super aus.
Fritzboxen sind wesentlich freundlicher als so ein Speedport, denn AVM packt immer 4 Löcher auf die Platine in die man eine Stifleist löten kann - oder temporär kurze leitende schrauben mit drangeknoteten Drähtchen mit Tesa fixiert reinstecken.
Ich vermute auf diesem Bild rechts in der mitte:  https://boxmatrix.info/wiki/File:FRITZ!Repeater_6000-Image47.jpg
Bei AVM ist das eckige nicht wie sonst GND, sondern GND ist das vom eckigen entfernteste. Die reihenfolge der Pins ist bei allen AVM Geräten die ich kenne gleich

fda77, 2022-02-19:
Leg am besten das "pseudo" Problem als neues Issue an. Das kam wohl durch den bump vor kurzem

freetz-ng-mod, 2022-02-20:
So ich habe nun 20 mal versucht denn 6000 zu flashen einmal habe ich es gehabt das die FW 7.28 darauf war ansonsten ist es immer ohne freetz.

Was mich aber auch immer wundert, dass das immer da steht 
Cmd: MLST 0x7eac0000 0x80000000
502: Command not implemented

fda77, 2022-02-20:
Die 7.28 war dann aber eine von AVM, oder? Bist du sicher dass push firmware nicht nur "linux_fs_start" geändert hat und deshlab die 7.28 gestartet hat?

freetz-ng-mod, 2022-02-20:
Ja ich habe 3 Mal oder so das original Image  per push versucht aufzuspielen.

aus-sua, 2022-02-20:
OT?:
Ich durfte neulich 'mal "schnell" einen 3000er flashen.
Erst bin ich an meiner eigenen Vergeßlichkeit (sic: -ram 256) gescheitert, dann/dabei ging es auch selbst nach einem AVM-Recover (das ich explizit vermeiden wollte...) nicht.
Erst als ich nach dem AVM-Recover die aller erste originale AVM-Anmeldeprozedur per WEB-IF durchlaufen hatte (da sind Land und Sprache explizit festzulegen, der 3000er boote auch danach), gab es beim flashen mit push_firmware keine Probleme.
Ich habe das nicht weiter verfolgen können, s.o., es mußte schnell gehen, sorry...

fda77, 2022-02-20:
@freetz-ng-mod; Gut, dann kann man davon aus gehen dass push firmware soweit funktioniert und das gebaute Image nicht okay ist. Es ist im "fit" Format, wie die  1200ax 4060 5530 5590 6000 7530ax 7510.
Ich weiss jetzt nicht ob jemals irgendwo ein Freetz image auf so einem Gerät schon lief. Das mit Rückmeldungen ist so ein Sache
Es ist damit möglich das das "fitimg" tool von hippie noch nicht ganz passt, oder dass in Freetz ein Problem ist

@aus-sua: Das sollte egal sein, das der Bootloader auf den zugegriffen wird gar nicht weiss ob/was konfiguriert ist. Es kann auch gar kein System oder ein kaputtes im Flash sein. Es ist aber bekannt dass manche neuere Bootloader ein flashen nur akzeptieren, nachdem der Stromstecker eingesteckt wurde, aber nicht nach einem "reboot" Befehl

freetz-ng-mod, 2022-02-21:
So ich habe nun Freetz darauf
[image]

Ich habe 2 mal die FRITZ.Repeater_6000-07.28-recover.exe ausgeführt und dann 2-mal psuh sodass er beide linux_fs beschreibt.

Ein Update über freetz geht nicht
[image]

freetz-ng-mod, 2022-02-21:
Der Boot-Manager geht über AVM und freetz Seite auch nicht
[image]
[image]

fda77, 2022-02-21:
Wow, das sieht doch gut aus. Die ganzen Fehler kommen durch FIT welches in den Scripten (noch?) nicht unterstützt wird.
Die "bootmanager" deaktiviere ich vorerst bis die repariert sind. Ohne ein passendes Gerät zum testen ist das aber blöd.
Alternativ zum Flashen über Freetz nimm das AVM Interfce. Das zu flashende Image muss mit deinem Key signiert sein der zu deinem öffentlichen Key im installierten Image passt. Wenn du auf dem gleichen Computer/VM nochmacl ein Image erstellst sollte das so sein.
Hast du eine Ahnung wo die fehlenden 150MB ram abgeblieben sind und was AVM damit macht? Evtl ist es auch nur ein Anzeigefehler, schau dich mal im System um ("free", /proc/ etc).
CPU ARMv7 passt nicht ganz, es ist ein ARMv8. Beim Raspberry ist es aber auch so dass ein v8 mit 32bit Betriebssystem v7 anzeigt

freetz-ng-mod, 2022-02-21:
``` 
root@6000-repeater:/var/mod/root# cat /sys/devices/virtual/thermal/thermal_zone0
/temp
40
cat /proc/avm/powermanagmentressourceinfo/powe
rdevice_temperature
0 °C 
cat /proc/avm/temp_sensors
kommt nichts
``` 
``` 
free
              total        used        free      shared  buff/cache   available
Mem:         877156       79988      516380        3252      280788      759704
Swap:             0           0           0
``` 
so aber wo die 150 MB sind kein plan ;-)

Du meinst unter Funknetz ja da werden mir 3 Mac Adressen angezeigt

fda77, 2022-02-21:
Das mit dem RAM ist komisch, eigentlich hat dieses Gerät keine weitere CPU der der zugeteilt werden sein könnte. 
Schau mal bei Gelegenheit ob der Punkt hinter der Temperatur nun weg ist.
Wenn "WLAN3" passt es, ich hatte noch nie ein Gerät mit 3 Wlans
Genau, wenn man  ~/.freetz-signature sichert kann man immer über AVM-gui flashen. Für UIMG und FIT müsste sich das jemand anschauen und anpassen...

freetz-ng-mod, 2022-02-22:
```
Prozessor
Temperatur    43 °C
BogoMIPS    51.34 / 51.34 / 51.34 / 51.34
```
Der Punkt ist weg, ist mir gar nicht aufgefallen bis zu dem das du es gesagt hast. 

Brauchst du sonst noch welche Informationen.

fda77, 2022-02-22:
Dass das Flashen über Freetz nicht funktioniert ist blöd, ich hab mich mit den Flash-Scripten nciht beschäftgt und das irgendwie remote zu machen ist noch komplizierter.
Für die Bootmanager-Anzeigen könntest du PeterPawn hier ein paar Infos posten: https://github.com/PeterPawn/YourFritz/issues/45
wie "cat" von /proc/cpuinfo, /proc/mtd, /proc/avm_partitions und ob es /proc/sys/urlader gibt und was darin ist.
Am besten gleich ein komplettes "support data", da können aber private Infos drin sein

fda77, 2022-02-23:
Flashen und sogar booten funktioniert. Wenn du magst kannst du "flashen" auf die Wishlist setzen.

closed 2022-02-23 by fda77

fda77, 2022-03-07:
Ja, es ist weg weil es nicht funktioniert. Wenn jemand das Script angepasst hat kann man es wieder aktivieren.