Freetz-NG/freetz-ng #468: pseudo: abort()ing pseudo client by server request

fda77, 2022-02-20:
### Discussed in https://github.com/Freetz-NG/freetz-ng/discussions/467

<div type='discussions-op-text'>

<sup>Originally posted by **NDiGH** February 19, 2022</sup>
Hallo,

wenn man für eine 7520/7530 eine Firmware bauen möchte scheitert es an "Step 1: Unpack":

```
STEP 1: UNPACK
source firmware file: dl/fw/FRITZ.Box_7530-07.29.image
checking signature: valid
unpacking firmware image
abort()ing pseudo client by server request. See https://wiki.yoctoproject.org/wiki/Pseudo_Abort for more details on this.
Check logfile: /home/Downloads/freetz-ng_2022-02-18_IPQ/.fakeroot-cache/pseudo.log
./fwmod: Zeile 694:  4923 Abgebrochen             (Speicherabzug geschrieben) "$TAR_GNU" -xif "$FIRMWARE" -C "$FIRMWARE_DIR"
unpacking filesystem image
    Filesystem on build/original/firmware/var/tmp/filesystem.image is xz compressed (4:0)
    Parallel unsquashfs: Using 6 processors
    9645 inodes (10620 blocks) to write
    created 8963 files
    created 539 directories
    created 681 symlinks
    created 1 devices
    created 0 fifos
unpacking var.tar
abort()ing pseudo client by server request. See https://wiki.yoctoproject.org/wiki/Pseudo_Abort for more details on this.
Check logfile: /home/Downloads/freetz-ng_2022-02-18_IPQ/.fakeroot-cache/pseudo.log
./fwmod: Zeile 694:  5060 Abgebrochen             (Speicherabzug geschrieben) "$TAR_GNU" -xipf "$VARTAR" -C "$VARTAR_DIR"
FINISHED
```

Hier die pseudo.log:
```
debug_logfile: fd 2
pid 4884 [parent 4883], doing new pid setup and server start
Setup complete, sending SIGUSR1 to pid 4883.
path mismatch [2 links]: ino 16884141 db '/home/Downloads/freetz-ng_2022-02-18_IPQ/build/original/firmware/var/tmp' req '/proc/self/fd/5'.
path mismatch [2 links]: ino 16893771 db '/home/Downloads/freetz-ng_2022-02-18_IPQ/build/original/kernel/var.tar/var/dev' req '/proc/self/fd/5'.
db cleanup for server shutdown, 22:10:43.010
memory-to-file backup complete, 22:10:43.016.
db cleanup finished, 22:10:43.016
```

Das Build-System ist ein Arch-Linux:
```
[user@device ~]$ uname -a
Linux device 5.15.24-1-lts #1 SMP Wed, 16 Feb 2022 16:04:21 +0000 x86_64 GNU/Linux
```</div>

fda77, 2022-02-20:
Wie https://github.com/Freetz-NG/freetz-ng/issues/465, was aber in eine andere Richtung verlief.

Das Problem wurde durch den letzte bump von pseudo verursacht: https://github.com/Freetz-NG/freetz-ng/commit/19a3a39036697f1ab8b4bb8b2ca1d838b95b7c4e. Um genau zu sein einer dieser beiden Commits:
 - https://git.yoctoproject.org/pseudo/commit/?h=oe-core&id=300d7570720bc14d01f49d285498d71cce3ce44e
 - https://git.yoctoproject.org/pseudo/commit/?h=oe-core&id=d34f2f6cedccf8488730001bcbde6bb7499f8814

Das Problem betrifft jedes Gerät und nur wenn man die vorcompilierten Host-Tools verwendet. Diese wurden unter Ubuntu 14 x64 mit https://github.com/Freetz-NG/freetz-ng/blob/master/tools/dl-hosttools erstellt.

Workarounds
 - tools 2013-02-10 verwenden (https://github.com/Freetz-NG/freetz-ng/commit/8e70e25864ec4ee92081b0fdaded76e8510b7df5)
 - tools selbst bauen (menuconfig)
 - fakeroot statt pseudo verwenden (menuconfig)

@PeterPawn Hast du eine Idee?

PeterPawn, 2022-02-20:
Bisher nicht - aber m.W. hat `pseudo` auch noch ein paar Einstellungen, mit denen man ein ausführlicheres (Debug-)Protokoll erzeugen kann (über eine Environment-Variable `PSEUDO_DEBUG`).

Wenn ich mir die Log-Datei ansehe, geht es ja offenbar darum, daß da die Ausgabe von Daten zweimal nach `/proc/self/fd/5` erfolgt (das dürfte beim Entpacken mit `tar` für JEDES File gelten), aber `pseudo` dann nicht entscheiden kann, wohin da der Symlink (denn die Einträge im `procfs` für die File-Handle sind alles auch nur Symlinks) zeigt bzw. der aktuell vorhandene Zustand nicht mit dem bis hierhin von `pseudo` protokollierten übereinstimmt - woraus `pseudo` dann den Schluß zieht, daß zwischendrin Änderungen außerhalb des `pseudo`-Kontexts erfolgt sein müssen.

Ich würde also genau den Commit als "Schuldigen" ausmachen ... ohne den Patch dürfte der Aufruf für diese beiden Verzeichnisse hier landen: https://git.yoctoproject.org/pseudo/tree/pseudo_util.c?h=oe-core&id=300d7570720bc14d01f49d285498d71cce3ce44e#n709 und die Änderung am Zustand der `pseudo`-Umgebung mehr oder weniger ignoriert werden.

Wobei mir der Patch im Commit ohnehin merkwürdig vorkommt ... das Hinzufügen der Definition für `O_PATH` verstehe ich nicht (den ersten Hunk), denn ich finde in der Datei auch nach dem Patch gar keine Stelle, wo diese Definition benutzt würde: https://git.yoctoproject.org/pseudo/tree/pseudo_util.c?h=oe-core&id=300d7570720bc14d01f49d285498d71cce3ce44e#n32

Da das ja zuvor wohl keine Probleme im Freetz-NG-Kontext bereitete, würde ich den Patch halt einfach auslassen - ich gehe mal davon aus, daß Du das spätere Packen der geänderten Dateien auch dahingehend kontrolliert hast, daß die Dateisystemstruktur danach auch die gewünschte ist. 

Wobei beim späteren Packen ja m.W. ohnehin alle User- und Group-IDs auf `0` gesetzt werden, weil mit `-all-root` gepackt wird ... damit sind die ggf. nicht durch `pseudo` korrigierten UID/GID-Attribute am Ende für `fwmod` auch nicht ausschlaggebend. Letztlich braucht's das `pseudo` (oder auch `fakeroot`) ja eigentlich nur dafür, die Dateien im AVM-Image auch ohne `root`-Rechte zu entpacken (was ansonsten für "special files" nicht funktioniert) - beim Einpacken ist das ohnehin alles egal.

---

Ich hatte irgendwann mal einige Änderungen am `unsquashfs` begonnen, mit denen das tatsächliche Entpacken solcher kritischen Dateien durch das Eintragen in eine Dateiliste, die dann später beim `mksquashfs` wieder (mit der Option `-pf`) als "Pseudo-File"-Definitionsdatei verwendet werden kann, ersetzt wird: https://github.com/PeterPawn/YourFritz/tree/main/squashfs - mehr bräuchte es auch im Freetz-NG gar nicht, um die Modifikationen auch ohne Superuser-Rechte ausführen zu können. 

Allerdings werden diese Patches vermutlich heute auch nicht mehr auf die aktuellen Quellen passen - zumindest bis zur 4.3 konnte ich damit aber auch ohne `root`-Rechte beim Entpacken der Firmware eine Struktur (plus die zusätzliche Datei mit den Definitionen der Pseudo-Files) erstellen, die beim Einpacken wieder der AVM-Struktur entsprach ... und das ganz ohne die zusätzliche Abhängigkeit von einem (weiteren) externen Projekt/Paket (was z.B. auf einer FRITZ!Box auch nicht vorhanden wäre - selbst wenn es da gar nicht gebraucht wird, weil im FRITZ!OS ohnehin (fast) alles `root` ist).

---

Aber zurück zum Problem ... dann laß doch einfach diesen Patch aus (oder füge einen eigenen hinzu, der den aus dem Upstream rückgängig macht, wenn Du die neue Version dennoch auschecken willst), denn im Freetz-NG-Kontext wird er ja eigentlich nicht benötigt. 

Wenn man ganz neugierig ist, kann man die Struktur auch mal (innerhalb eines `pseudo`-Kontexts oder auch nach dem Packen (im `pseudo`-Kontext) ohne `-all-root`-Option) anschauen ... da sollten dann die beiden erwähnten Verzeichnisse auch nur mit der UID/GID des aktuellen Benutzers (und nicht mit `root`) auftauchen, wenn sie von `pseudo` tatsächlich ignoriert wurden beim Entpacken der AVM-Firmware.

---

Wobei das hier wohl auch ein Problem wäre, was man bei `pseudo` am Upstream einfach nur melden könnte (bis zur Lösung müßte man sich aber ohnehin eigene Gedanken für einen Workaround machen) ... das dürfte ein reproduzierbares Problem der Abfolge beim Entpacken der AVM-Firmware sein und da kann man den Maintainern ja dann die AVM-Firmware als Grundlage für das Reproduzieren des Problems "anbieten".

fda77, 2022-02-20:
>  bei pseudo am Upstream einfach nur melden könnte

Es betrifft (jedenfalls bei mir, von den 2 anderen weiss ich es nicht) nur die mit Ubuntu 14 vorcompilierte pseudo binary. Die mit dem System (Fedrora 35 bei mir) selbst compilierte hat den Fehler nicht. Ob es für so ein altes System noch Support gibt glaube ich eher nicht...

Hier hatte ich extra das DEBUG=0 eingebaut um es bei Bedarf schneller zu finden. Diese Log ist aber im "make" extrem ausführlich! https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L370

Deshlab Verkürzungen:

Ich teste mit dieser Datei (ist aber egal welche)
```
tar tvf dl/fw/FRITZ.Box_5590-07.27-94541-Inhaus.image
drwxr-xr-x 0/0               0 2022-02-10 18:59 ./var/
-rwxr-xr-x 0/0            1298 2022-02-10 18:59 ./var/install
-rwxrwxrwx 0/0          284888 2022-02-10 18:57 ./var/tzupdate
drwxr-xr-x 0/0               0 2022-02-10 18:59 ./var/tmp/
-rw-r--r-- 0/0        44925631 2022-02-10 18:59 ./var/tmp/fit-image
-rw-r--r-- 0/0            2807 2022-02-10 18:59 ./var/info.txt
-rw-r--r-- 0/0              14 2022-02-10 18:57 ./var/install-features
-rw-r--r-- 0/0             347 2022-02-10 18:59 ./var/content
-rw-r--r-- 0/0              40 2022-02-10 18:58 ./var/version
-rw-r--r-- 0/0             128 2022-02-10 18:59 ./var/signature
```

Initialisieren vor jedem Testlauf:
```
rm -rf var/
TOOLS_DIR="tools"
FAKEROOT_CACHE_DIR=".fakeroot-cache"
PSEUDO_PREFIX="$TOOLS_DIR/build" PSEUDO_LOCALSTATEDIR="$FAKEROOT_CACHE_DIR"                  "$TOOLS_DIR/build/bin/pseudo"  -S    #halt
mkdir -p "$FAKEROOT_CACHE_DIR"
rm -f $FAKEROOT_CACHE_DIR/*

PSEUDO_PREFIX="$TOOLS_DIR/build" PSEUDO_LOCALSTATEDIR="$FAKEROOT_CACHE_DIR"                  "$TOOLS_DIR/build/bin/pseudo"  -B  #scan
# considering table logs
# considering table migrations
# considering table files
# considering table xattrs
# considering table migrations
# considering table files
# considering table xattrs
# considering table migrations
# db cleanup for server shutdown, 13:11:24.704
# memory-to-file backup complete, 13:11:24.704.
# db cleanup finished, 13:11:24.704
```

vorcompiliertes tar-gnu wie in make, mit Fehler.
Ohne make/fwmod drumherum hat sich die Fehlermeldung von /proc/self/fd/5 auf /proc/self/fd/4 geändert
```
PSEUDO_PREFIX="$TOOLS_DIR/build" PSEUDO_LOCALSTATEDIR="$FAKEROOT_CACHE_DIR" PSEUDO_DEBUG="1" "$TOOLS_DIR/build/bin/pseudo" '--' tools/tar-gnu xvf dl/fw/FRITZ.Box_5590-07.27-94541-Inhaus.image
./var/
./var/install
./var/tzupdate
./var/tmp/
./var/tmp/fit-image
./var/info.txt
abort()ing pseudo client by server request. See https://wiki.yoctoproject.org/wiki/Pseudo_Abort for more details on this.
Check logfile: .fakeroot-cache/pseudo.log
Abgebrochen (Speicherabzug geschrieben)

cat .fakeroot-cache/pseudo.log
debug_logfile: fd 2
pid 788618 [parent 788617], doing new pid setup and server start
Couldn't open .fakeroot-cache/pseudo.pid: No such file or directory
warning: couldn't write pid file.
Setup complete, sending SIGUSR1 to pid 788617.
Failed: unable to open database file
Error getting 'logs.db' database.
Failed: unable to open database file
Error getting 'files.db' database.
[... 56 lines cut ...]
openat_creat: /home/fritz/freetz-ng/var/version -> 0600
openat(path /home/fritz/freetz-ng/var/version), flags 2004701, stat rc 0, stat mode 100600
./var/signature
openat_creat: /home/fritz/freetz-ng/var/signature -> 0600
openat(path /home/fritz/freetz-ng/var/signature), flags 2004701, stat rc 0, stat mode 100600

cat .fakeroot-cache/pseudo.log
debug_logfile: fd 2
pid 788850 [parent 788849], doing new pid setup and server start
Couldn't open .fakeroot-cache/pseudo.pid: No such file or directory
warning: couldn't write pid file.
Setup complete, sending SIGUSR1 to pid 788849.
Failed: unable to open database file
Error getting 'logs.db' database.
Failed: unable to open database file
Error getting 'files.db' database.
pdb_log_msg: database error.

```

--> Das neueste pseudo kommt mit der tar ohne selinux Support nicht klar!?
Zur Erinnerung: pseudo/fakeroot können die xattrs nicht und werfen mit Warnungen um sich
Das Busybox-tar kann das verwendete "-i" (Blöcke mit Nullen im Archiv ignorieren) nicht

Es könnte also auch ein Problem von tools/make/tar-host/ sein das nur durch pseudo sichtbar wird

fda77, 2022-02-20:
Man kann das Problem auch umgehen indem man tar-gnu selbst compiliert und die vorcompilierte überschreibt: `make tar-host-precompiled`. Das funktioniert auch mit der von Ubuntu 14 vorcompilierten pseudo-binary

NDiGH, 2022-02-20:
Was ist eigentlich der Grund, dass das noch mit Ubuntu 14.(04 LTS) kompiliert wird?

fda77, 2022-02-20:
Damit die Binaries auf möglichst vielen Systemen funktionieren

NDiGH, 2022-02-20:
Ich verwende weder Ubuntu noch Fedora (und dennoch haben zumindest bisher die vorkompilierten host-tools und toolchain funktioniert). Und wenn man wegen Kompatibilität auf Ubuntu setzt, sollte es wohl nicht mehr Ver. 14.04 sein, denke ich.

Gibt es (bekannte) Probleme wenn man mit Fedora baut und die vorkompilierten Tools dann mit Ubuntu oder anderen Distributionen verwenden möchte?

freetz-ng-mod, 2022-02-20:
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

Linux xmg-ultra 5.13.0-10027-tuxedo #29~20.04.1tux1 SMP Thu Jan 20 10:52:53 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

PeterPawn, 2022-02-20:
> Damit die Binaries auf möglichst vielen Systemen funktionieren.

Sollten die dazu nicht besser statisch gelinkt sein? Oder sind sie das gar? 

Es ist gar nicht so einfach zu ermitteln (unter meinem `openSUSE Tumbleweed`), wo und wie genau das Download-Paket für die Host-Tools gebaut wird.

Das geht schon damit los, daß hier die "prerequisites" - ziemlich unvermittelt - ein unerwartetes Ergebnis liefern:

```
peh@vidar:~> mkdir -p /tmp/freetz-ng-check
peh@vidar:~> cd /tmp/freetz-ng-check/
peh@vidar:/tmp/freetz-ng-check> git clone https://github.com/Freetz-NG/freetz-ng.git
Cloning into 'freetz-ng'...
remote: Enumerating objects: 185774, done.
remote: Counting objects: 100% (1266/1266), done.
remote: Compressing objects: 100% (776/776), done.
remote: Total 185774 (delta 634), reused 899 (delta 475), pack-reused 184508
Receiving objects: 100% (185774/185774), 67.79 MiB | 2.21 MiB/s, done.
Resolving deltas: 100% (127339/127339), done.
peh@vidar:/tmp/freetz-ng-check> cd freetz-ng/
peh@vidar:/tmp/freetz-ng-check/freetz-ng> make help
ERROR: The program 'ldconfig' was not found in path
Makefile:136: *** Some build prerequisites are missing! See '.prerequisites' for why. Please install the missing packages before trying again. See https://freetz-ng.github.io/freetz-ng/PREREQUISITES for installation hints.  Stop.
peh@vidar:/tmp/freetz-ng-check/freetz-ng> cnf ldconfig

Program 'ldconfig' is present in package 'glibc', which is installed on your system.

Absolute path to 'ldconfig' is '/usr/sbin/ldconfig', so running it may require superuser privileges (eg. root).

peh@vidar:/tmp/freetz-ng-check/freetz-ng> sudo which ldconfig
[sudo] password for root:
/usr/sbin/ldconfig
peh@vidar:/tmp/freetz-ng-check/freetz-ng>
```

Das offizielle Paket, in dem hier das `ldconfig` enthalten wäre, installiert das also unter einem Pfad, wo es nur mit Superuser-Rechten zu finden ist. Was mache ich denn jetzt (nach diesem Commit: https://github.com/Freetz-NG/freetz-ng/commit/f0e0abfc9bbd88093a04bd35020ce92ae73337ef) als Benutzer, der keine richtige Ahnung von Linux hat? Das Naheliegendste wäre es m.E., wenn ich die benötigte Datei selbst irgendwo innerhalb der Freetz-Struktur bereitstelle:

```
peh@vidar:/tmp/freetz-ng-check/freetz-ng> sudo cp /usr/sbin/ldconfig tools/
peh@vidar:/tmp/freetz-ng-check/freetz-ng> make help
ERROR: The program 'ldconfig' was not found in path
Makefile:136: *** Some build prerequisites are missing! See '.prerequisites' for why. Please install the missing packages before trying again. See https://freetz-ng.github.io/freetz-ng/PREREQUISITES for installation hints.  Stop.
peh@vidar:/tmp/freetz-ng-check/freetz-ng>
```

Das hilft schon mal nicht, weil das `tools`-Verzeichnis (obwohl es das Staging-Verzeichnis für die Tools wäre, wo die später auch alle zu finden sind und von wo sie aufgerufen werden) bei der Suche/Prüfung gar nicht einbezogen wird.

```
peh@vidar:/tmp/freetz-ng-check/freetz-ng> sudo cp /usr/sbin/ldconfig /home/peh/bin/
peh@vidar:/tmp/freetz-ng-check/freetz-ng> make help
..................................................................................................

Targets of Freetz's "make"

help                                    Shows docs/wiki/20_Advanced/make_targets.en.md

menuconfig                              Configuration of Freetz
menuconfig-nocache                      Configuration without caching of .in files
menuconfig-single                       Alternative configuration
oldconfig                               Updates existing .config file interactive
olddefconfig                            Updates existing .config file automatically
reuseconfig                             Removes device and toolchain related settings from .config file
allnoconfig                             Sets everything to (n)o
allyesconfig                            Sets everything to (y)es

config-clean-deps                       Deselects everything not mandatory
config-clean-deps-keep-busybox          Deselects all except BusyBox applets
config-clean-deps-modules               Deselects all kernel modules
config-clean-deps-libs                  Deselects all libraries
config-clean-deps-busybox               Deselects all BusyBox applets
[... 36 lines cut ...]
Das ist auch nur ein Beispiel für die "Nebenwirkungen", die von solchen Patches verursacht werden - vor diesem Patch hat das auch unter meinem System noch funktioniert, OHNE daß man sich zusätzliche Gedanken machen mußte. Dieser Patch, der mich gar nicht betreffen würde, wenn ich anstelle von `pseudo` einfach weiterhin `fakeroot` benutzen wollte (sofern man überhaupt diesen Aufstand machen muß, wenn man am Ende ohnehin mit `-all-root` alles wieder einpackt), macht jedenfalls plötzlich, daß es "nicht mehr funktioniert" und das auch, ohne daß ich als Freetz-Benutzer jetzt den Zusammenhang erkennen könnte.

Die ganze Geschichte mit dem Download der Host-Tools ist nicht wirklich "rund" ... ich verstehe ja den Ansatz dahinter (und es KANN tatsächlich Zeit und Ressourcen sparen, wenn man es nicht ständig selbst übersetzt), aber so funktioniert das eben auch nicht wirklich.

---

Nun soll dieser Ansatz ja wohl verhindern (was ich sehr richtig finde), daß man wegen jedem "Sch..ß", für den man die Toolchain benutzen will, immer gleich den kompletten Build für die `host-tools` anwerfen muß.

[... cut ...]  installiert haben werden), kommen sie da nicht dran, ohne sich entweder Zugriff auf `/sbin` zu verschaffen (für den Benutzer, den sie zum Freetz-Build verwenden wollen) oder eine andere Kopie (wie ich weiter oben) davon zu erstellen.

Und das alles nur deshalb (wenn ich den Commit und #450 richtig verstanden habe), damit ein vorkompiliertes `pseudo`-Paket im Download-Archiv für die Host-Tools geprüft werden kann? Sind das nicht die berühmten Kanonen und hat denn niemand Mitleid mit den Spatzen?

fda77, 2022-02-20:
Alle prerequisites loswerden: ./.prerequisites leeren, siehe 
> Makefile:136: *** Some build prerequisites are missing! See '.prerequisites' for why.

Der Pfad ./tools/path/ wird mit in den $PATH genommen, ich bin aber nicht sicher ob das da schon greift

Das ldconfig kam halt in die Prerequisites, da damit  geprüft wird ob andere Dinge vorhanden sind

Script zum Host-Tools bauen (erstellt die Datei in dl/): tools/dl-hosttools

> Sollten die dazu nicht besser statisch gelinkt sein? Oder sind sie das gar?

Wenn ich mich recht erinnere hatte ich danach mal genschaut, und die Abhängigkeiten waren "gering" und der Rest bereitete irgend welche Problem...

Ja, da mit der Reihenfolge, wie .image zuerst downloaden hatten wir schonmal. Das ist so wegen der Usernbeschwerden: 1h bauen, und dann bricht es doch ab nur weil avm ein Verzeichnis oder Datei umbenannt hat

pfichtner, 2022-02-21:
Ich baue meine Freetz(-NG) images schon seit geraumer Zeit auf meinem CI Server in einem Docker Container. Mein Docker Image habe ich kürzlich auch [öffentlich gemacht](https://github.com/pfichtner/pfichtner-freetz). Wenn in diesem ein Kommando aufgerufen wird welches nicht installiert ist wird es automatisch installiert. Dazu habe ich cnf im Image gepatcht. 
Ich hatte auch mal testhalber which "gepatcht" so dass auch beim Aufruf von `which` das Paket installiert in dem das binary zu finden ist. Das führt aber beim Ausführen von tools `tools/prerequisites` natürlich dazu, dass man anschließend dreiviertel der Distro installiert hat. 
Dies nur zur Info falls es was hilft. Sollte ich mit dem Docker Image etwas unterstützendes beitragen können kommt gerne auf mich zu.

fda77, 2022-02-21:
Das mit den tools passt jetzt noch weniger. Durch den bump von patchelf baut dies wegen zu altem gcc auch nicht mehr mit Ubuntu14 -.- Am einfachsten nimmt man wohl ein neuere Linux
Mit docker hab ich keine Erfahrung und absolut keine Idee was man mit dem von dir verlinkten machen kann :) Wenn es sinvoll ist könnte man das hier verlinkten

pfichtner, 2022-02-21:
Du rufst einfach im ausgecheckten freetz-ng Verzeichnis einfach `docker run --rm -it -v $PWD:/workspace "$IMAGE" /bin/bash` auf und hast damit eine Shell in einer Buildumgebung auf Basis eines[ ubuntu 20.04](https://github.com/pfichtner/pfichtner-freetz/blob/main/Dockerfile#L1) in dem [all die Dinge bereits installiert sind](https://github.com/pfichtner/pfichtner-freetz/blob/13facc059c4470aaefe556e4be55d4ace3c6bf59/Dockerfile#L15), die man zum Bau von freetz-ng images benötigt. Ist wie wenn Du in der bash die bash startestm nur dass diese bash im Container läuft (und wie jede andere shell nach `exit` verlassen wird). 
Durch das `$PWD:/workspace` steht das aktuelle Verzeichnis des Hosts im Container als `/workspace` zur Verfügung (in dem man auch direkt automatisch landet). Das funktioniert auf jedem System auf dem Docker läuft, egal welches OS, egal welche Distribution, Du hast damit immer eine einheitliche Umgebung.

pfichtner, 2022-02-21:
Ist quasi wie freetz linux, nur in einfach, leichtgewichtig, überschaubar, quelloffen/transparent und mit Technolgie aus den letzten 10 Jahren :smirk:

fda77, 2022-02-21:
Ich muss mir diesen neumodischen Kram mal anschauen... Werden gewisse Dinge die man gerne sichern möchte "extern" gespeichert, zb .config oder signaturkeys

pfichtner, 2022-02-21:
```
> head -2 /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=21.10
DISTRIB_CODENAME=impish
DISTRIB_DESCRIPTION="Ubuntu 21.10"
> docker run --rm -it -v $PWD:/workspace pfichtner/freetz /bin/bash
builduser@myhostname:~$ head -2 /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
builduser@installme:~$ exit
> 
```

pfichtner, 2022-02-21:
> Ich muss mir diesen neumodischen Kram mal anschauen... Werden gewisse Dinge die man gerne sichern möchte "extern" gespeichert, zb .config oder signaturkeys

Ich mounte $PWD als /workspace in den container (rw), d.h. alles was du veränderst wird direkt im Dateisystem des Hosts verändert und bleibt damit auch genauso erhalten, wie wenn Du es direkt dort geändert hättest.

pfichtner, 2022-02-21:
D.h. alles was innerhalb Deines freetz-ng Verzeichnis geschrieben/verändert/gelöscht wird, wird auch direkt dort verändert, wie wenn Du es **nicht** innerhalb eines Docker containers machen würdest.

pfichtner, 2022-02-21:
```
> ls
addon  build  config  dl  docs  graphics  images  include  kernel  make  packages  patches  source  toolchain  tools   COPYING  fmake  fwmod  fwmod_custom  Makefile  README.md
> docker run --rm -it -v $PWD:/workspace pfichtner/freetz /bin/bash
builduser@myhostname:~$ ls
COPYING  Makefile  README.md  addon  build  config  dl	docs  fmake  fwmod  fwmod_custom  graphics  images  include  kernel  make  packages  patches  source  toolchain  tools
builduser@installme:~$ echo "from container" >hello.txt
builduser@installme:~$ exit
> ls
addon  build  config  dl  docs  graphics  images  include  kernel  make  packages  patches  source  toolchain  tools  unpacked  COPYING  fmake  fwmod  fwmod_custom  hello.txt  Makefile  README.md
> cat hello.txt
from container
> rm hello.txt
> 
```

fda77, 2022-02-21:
Ich glaub das wäre in einer eigene Diskussion besser aufgehoben.
Das kling ganz um zb alle Abhängigkeiten zu haben wenn man ein Image erstellen will. Für mich selbst bin ich nicht so sicher. Ich hab zb einen paar Gb grossen ccache, der auch beschreibbar sein müsste. Und die Firmware-Downloads hab ich auch extern, einige die benötigt werden gibt es bei AVM nicht mehr zb für config/generate_img.sh

pfichtner, 2022-02-21:
Das habe ich ebenso, den Cache reiche ich ebenfalls rw in den Container: 
docker run --rm -it -v $PWD:/workspace **-v /the/dir/where/your/dl-cache/is:/workspace/dl** pfichtner/freetz /bin/bash

Diskussion gerne an geeigneter Stelle, z.B. disucssion, gerne auch via PM o.ä. wenn Du masgt

PeterPawn, 2022-02-22:
Macht doch wirklich mal eine passende Diskussion dafür auf - da kann man dann auch mal ein paar Containerfiles (mittlerweile gewinnt `podman` ja die Oberhand über `docker` :wink: ) diskutieren, die auf der Basis eines Ubuntu-(Docker-)Images (auch die LTS-Versionen gibt's ja in den diversen Registries) im Handumdrehen ein aktuelles(!) `podman`-Image bauen lassen. 

Das geht in vielen Fällen auch noch schneller, als der Download einer fertigen VM ... und ist vor allen Dingen auch so weit automatisiert, daß da IMMER derselbe Stand beim `podman build` heraus kommt und die ganzen Aktionen, die man ansonsten noch "von Hand" machen soll (https://www.ip-phone-forum.de/threads/freetz-ng-fehler-beim-bauen-f%C3%BCr-unterschiedliche-konfigurationen.308030/#post-2387896), lassen sich "skripten" (ja, das geht bei solchen Builds fast nicht anders).

Von der Portabilität entsprechender Images mal ganz zu schweigen ... auch gibt es dann wieder diverse Optionen, um solche Container auf andere Maschinen zu schieben oder auch auf andere Architekturen (z.B. Docker Desktop unter Windows oder auch die WSL2). Selbst dann, wenn man auch ansonsten mit Linux arbeiten sollte und sogar mit einem passenden System, auf dem sich ein Freetz-NG-Build direkt starten ließe, kann sich die zusätzliche Abstraktionsschicht eines Containers noch positiv auswirken.

Bis hin zu der Option, solche Container auch auf Service-Basis "in the cloud" ausführen zu lassen, wo sie dann vor sich hinarbeiten können (bis zum kompletten Freetz-Build - oben habe ich ja schon mal auf ein anderes Projekt verlinkt, was "vollautomatisiert" die Host-Tools über einen Freetz-Build in einem Container erstellt), ohne den eigenen PC zu blockieren oder sogar durch "nur Windows"-Benutzer ein Linux-Container bei einem Cloud-Anbieter genutzt werden kann. 

Da haben "die Großen" (MS Azure, Amazon WS, Google Cloud) i.d.R. sogar kostenlose Angebote - meist für eine gewisse Zeit "auf Probe" und/oder mit beschränkten Ressourcen ... aber selbst wenn man dafür etwas zahlen muß, lohnt sich das häufig immer noch, denn man braucht ja i.d.R. keine Echtzeit-Instanzen oder größeren Speicherplatz oder Traffic - das sind dann die Faktoren, die eine Cloud-Instanz teuer machen können. Ohne dieses, rechnet sich das meist schon durch eingesparte Arbeitszeit und elektrische Energie - von zusätzlicher Hardware, die dadurch überflüssig wird, ganz zu schweigen.

All die Unterschiede, die man sich heute noch mit der Frage, welches Basis-Linux man eigentlich für seinen Freetz-Build benutzt, eintritt, sind damit auch mehr oder weniger Geschichte und angesichts der Größe von Container-Images und deren Flexibilität (das können auch mal unterschiedliche Features "übereinander gestapelt" werden auf der Basis derselben Images - das sind keine Monolythen wie die Images für eine "echte VM"), reibt man sich bei der Größe von fast 3,5 GB für die Freetz-VM auch schon mal die Augen und fragt sich, wie das alles auch mit deutlich weniger Traffic zurechtkommen kann.

fda77, 2022-02-22:
Das klingt alles ganz gut. Mit so einem Container könnte man auch die tools+toolchains Pakete erstellen lassen, und das ursprüngliche Problem hier lösen. Für diese dann aber ein etwas älteres Linux, aber kein Ubuntu 14 mehr...

pfichtner, 2022-02-22:
Selbst  ubuntu 14 sollte gehen. Inweit das Sinn macht bleibt offen

fda77, 2022-02-22:
Schau dir mal an um was es in diesem Issue eigentlich geht ;-)

Docker -> https://github.com/Freetz-NG/freetz-ng/discussions/475

pfichtner, 2022-02-22:
Du beziehst Dich auf 
> Das Problem betrifft jedes Gerät und nur wenn man die vorcompilierten Host-Tools verwendet. Diese wurden unter Ubuntu 14 x64 mit https://github.com/Freetz-NG/freetz-ng/blob/master/tools/dl-hosttools erstellt.
oder? 

Ich bleibe bei meiner Aussage: Das kann man auch weiterhin (auch im Jahr 2022) auf einem aktuellen System in einem auf ubuntu14 basierten Container jederzeit neu und reproduzierbar bauen. Ob das Sinn macht (auf ubuntu14 zu bleiben) oder den Aufwand wert ist auf ein anderes/neueres/Linux zu wechseln bleibt offen.

fda77, 2022-02-22:
Das unter Ubuntu14 erstelle Binary tar-gnu läuft nicht korrekt auf aktuellen Systemen, Fehlermeldung im OP.
Ausserdem baut patchelf überhaupt nicht mehr seit dem bump unter U14

pfichtner, 2022-02-22:
> Das unter Ubuntu14 erstelle Binary tar-gnu läuft nicht korrekt auf aktuellen Systemen, Fehlermeldung im OP. Ausserdem baut patchelf überhaupt nicht mehr unter U14 seit dem bump

Danke, jetzt hab auch ich das Problem verstanden :smile:

fda77, 2022-02-22:
Um dieses host-tools Paket zu erstellen überleg ich ein Ubuntu >16< x64 docker zu verwenden statt der 14er VM. Ich hab keine passende 16er VM dafür und will nicht noch eine vorhalten. Vielleicht könnte man das später auch irgendwie mit github automatisieren...
Es muss nur tools/dl-hosttools ausgeführt werden.
Und aktuell in tools/make/Makefile.in am Ende das "until ..." rausgenommen werden

pfichtner, 2022-02-22:
Dauert selbst mit einem blanken ubuntu image in welches man erstmal noch den gcc installiert nur ein paar Sekunden, 20.04 kann natürlich durch jede andere Version ersetzt werden (eventuell müssen dann aber andere Pakete installiert werden) (oder natürlich auch andere distros, dann muss natürlich der jeweilge Paketmanager statt apt/dpkg aufgerufen werden). 

```
> ls -ltrh source/host-tools/tar-1.34/src/tar
ls: cannot access 'source/host-tools/tar-1.34/src/tar': No such file or directory

> docker run --rm -it -v $PWD:/tmp/foo ubuntu:20.04 /bin/bash -c 'apt update; apt install -y make gcc libacl1-dev; cd /tmp/foo/source/host-tools/tar-1.34; make'
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
[...]
Get:18 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [25.2 kB]
Fetched 21.2 MB in 2s (9331 kB/s)                         
[...]
The following additional packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu cpp cpp-9 gcc-9 gcc-9-base libasan5 libatomic1 libattr1-dev libbinutils libc-dev-bin libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0 libctf0
  libgcc-9-dev libgomp1 libisl22 libitm1 liblsan0 libmpc3 libmpfr6 libquadmath0 libtsan0 libubsan1 linux-libc-dev manpages manpages-dev
[...]
Get:32 http://archive.ubuntu.com/ubuntu focal/main amd64 libattr1-dev amd64 1:2.4.48-5 [22.8 kB]
Get:33 http://archive.ubuntu.com/ubuntu focal/main amd64 libacl1-dev amd64 2.2.53-6 [77.4 kB]
Fetched 33.4 MB in 2s (18.6 MB/s)       
debconf: delaying package configuration, since apt-utils is not installed
[...]
Setting up gcc-9 (9.3.0-17ubuntu1~20.04) ...
Setting up gcc (4:9.3.0-1ubuntu2) ...
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
make  all-recursive
make[1]: Entering directory '/tmp/foo/source/host-tools/tar-1.34'
Making all in gnu
make[2]: Entering directory '/tmp/foo/source/host-tools/tar-1.34/gnu'
make  all-recursive
make[3]: Entering directory '/tmp/foo/source/host-tools/tar-1.34/gnu'
make[4]: Entering directory '/tmp/foo/source/host-tools/tar-1.34/gnu'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34/gnu'
make[3]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34/gnu'
make[2]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34/gnu'
Making all in lib
make[2]: Entering directory '/tmp/foo/source/host-tools/tar-1.34/lib'
make  all-am
make[3]: Entering directory '/tmp/foo/source/host-tools/tar-1.34/lib'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34/lib'
make[2]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34/lib'
Making all in src
make[2]: Entering directory '/tmp/foo/source/host-tools/tar-1.34/src'
  CCLD     tar
make[2]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34/src'
make[2]: Entering directory '/tmp/foo/source/host-tools/tar-1.34'
make[2]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34'
make[1]: Leaving directory '/tmp/foo/source/host-tools/tar-1.34'

> ls -ltrh source/host-tools/tar-1.34/src/tar
-rwxr-xr-x 1 root root 575K Feb 22 22:59 source/host-tools/tar-1.34/src/tar
```

Mit 16.04 scheiterte o.g. Aufruf mit 
```
In file included from tempname.c:52:0:
./sys/random.h:40:30: fatal error: sys/random.h: No such file or directory
compilation terminated.
Makefile:2222: recipe for target 'tempname.o' failed
```
auf die Schnelle habe ich nicht gefunden, welches Paket man dafür noch installieren muss.

fda77, 2022-02-23:
Es fehlen irgend welche Header, das Script prüft nichts.
Ich hab versucht mit Ubuntu 16 patchelf zu bauen, der gcc ist darin aber auch zu alt. Es wird C++17 benötigt. Mit Ubuntu 18 ist es dann okay.
Allerdings bringt ein tar-gnu+pseudo unter Ubuntu 18 gebaut auf Fedora noch immer die Fehler aus dem OP.
Mir ist das ganze zu blöd, ich glaub ich reverte einfach den pseudo+patchelf bump

fda77, 2022-02-23:
So,"funktioniert" wieder. Vielleicht beheben irgendwann neuere Versionen von pseudo oder tar das Problem. Das soll natürlich niemanden davon abhalten der die Ursache suchen möchte

closed 2022-02-23 by fda77