Freetz-NG/freetz-ng #470: Mosquitto build schlägt fehl

pfichtner, 2022-02-21:
Ich erfass mal einen Issue zu meinem eigenen PR #469, da ich alles andere als ein Makefile-Experte bin und vielleicht/hoffentlich jemand auf die Schnelle helfen kann. 

mosquitto_passwd wird von mosquitto gebaut, wenn man mosquitto mit TLS (`WITH_TLS`) baut. https://github.com/Freetz-NG/freetz-ng/blob/master/make/mosquitto/mosquitto.mk versucht allerdings, egal ob mosquitto_passwd exisitiert zu kopieren. Ich dachte das mit dem [exclude](https://github.com/Freetz-NG/freetz-ng/blob/0fd87c4be7941499cb5e24dc0ce177bc9e7e0598/make/mosquitto/mosquitto.mk#L20) ausgeschlossen zu haben, bei welchem übrigens offensichtlich der Pfad fehlt, aber auch das brachte keine Besserung. 

Hier der Auszug, bei welchem nur die zwei client binaries, also ohne Broker gebaut werden sollen
```
FREETZ_PACKAGE_MOSQUITTO=y
# FREETZ_PACKAGE_MOSQUITTO_BROKER is not set
FREETZ_PACKAGE_MOSQUITTO_CLIENTS=y
FREETZ_PACKAGE_MOSQUITTO_CLIENT_PUB=y
FREETZ_PACKAGE_MOSQUITTO_CLIENT_SUB=y
FREETZ_PACKAGE_MOSQUITTO_WITH_UUID=y
# FREETZ_PACKAGE_MOSQUITTO_WITH_SSL is not set
```

Dies schlägt dann die folgt fehl: 
```
make[2]: Entering directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/src'
make[1]: Leaving directory '/workspace/source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15'
mkdir -p packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/root/usr/bin/; cp source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/src/mosquitto_passwd packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/root/usr/bin/mosquitto_passwd; [ "" == "y" ] && tools/patchelf --set-interpreter /usr/lib/freetz/ld-uClibc.so.1 packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/root/usr/bin/mosquitto_passwd || true; /workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-strip --remove-section={.comment,.note,.pdr} packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/root/usr/bin/mosquitto_passwd;
cp: cannot stat 'source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/src/mosquitto_passwd': No such file or directory
/workspace/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-strip: 'packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/root/usr/bin/mosquitto_passwd': No such file
make: *** [make/mosquitto/mosquitto.mk:52: packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mosquitto-1.6.15/root/usr/bin/mosquitto_passwd] Error 1
```

Ich nehme an, die excludes sind nur dafür da, später beim Übernehmen ins Image Dateien auszuscließen. 
(und mosquitto_extra.def habe ich vergessen in https://github.com/Freetz-NG/freetz-ng/blob/0fd87c4be7941499cb5e24dc0ce177bc9e7e0598/make/mosquitto/mosquitto.mk#L19 aufzunehmen)

Wenn mir jemand einen Tipp geben kann, wie ich das o.g. Problem beim build fixen kann fixe ich dies gerne (und die zwei weiteren genannten Dinge).

fda77, 2022-02-21:
Es sollte helfen in der precompiled-Zeile https://github.com/Freetz-NG/freetz-ng/blob/master/make/mosquitto/mosquitto.mk#L65 um den ganzen _passwd-Ausdruck ein `$(if ...` wie bei EXCLUDE zu machen.
Wie wäre es in menuconfig passwd immer anzuzeigen, und bei Auswahl TLS auszuwählen?

pfichtner, 2022-02-21:
Danke @fda77, genau das hat geholfen! Habe die drei genannten Punkte gefixt und erfolgreich unterschiedliche Konstellationen durchgetestet. 
Deine Frage habe ich nicht ganz verstanden. `mosquitto_passwd` hängt aktuell davon ab, ob TLS ausgewählt wurde (denn nur dann baut das mosquitto Makefile das binary). Ob man das binary ins image übernehmen will kann man in menuconfig auswählen, sobald man TLS ausgewählt hat. Den default hierfür habe ich mal "nicht aktiviert" (n) gewählt.

fda77, 2022-02-21:
> Ob man das binary ins image übernehmen will kann man in menuconfig auswählen, sobald man TLS ausgewählt hat

Das meinte ich. Solange TLS deaktiviert ist >weiss man nicht< dass es auch passwd gibt. Deshlab finde ich es geschickter passwd immer anzuzeigen (ohne depend) und wenn es ausgewählt ist "select" von TLS.
Ist aber deine Sache!

pfichtner, 2022-02-21:
Ah ok. Klingt, finde ich, erstmal sinnvoll! 
Dann aber: Wenn man `mosquitto_passwd` auf die gleiche Stufe hebt wie den broker (server) und die zwei client binaries, würde es bedeuten, dass wenn man `mosquitto_passwd` selektiert würde dadurch der eventuell nicht selektierte broker selektiert und für den dann TLS, das widerum fände ich unlogisch. Deshalb würde ich dazu tendieren es so zu lassen wie es derzeit ist.

fda77, 2022-02-21:
Zu 
```
select FREETZ_PACKAGE_MOSQUITTO_BROKER if (!FREETZ_PACKAGE_MOSQUITTO_CLIENT_PUB && !FREETZ_PACKAGE_MOSQUITTO_CLIENT_SUB)
```
muss dann auch noch das passwd in die Klammer dazu :)

Das passwd müsste wenn es nicht von ausgewähltem SSL abhängen soll, zusätzlich von openssl>=1 abhängen 

Aber mach das wie es dir passt, ich will da nciht reinreden. Hattest du gesehen dass ich dir eine Eiladung geschickt hatte die abgelaufen ist?

pfichtner, 2022-02-21:
> ```
> select FREETZ_PACKAGE_MOSQUITTO_BROKER if (!FREETZ_PACKAGE_MOSQUITTO_CLIENT_PUB && !FREETZ_PACKAGE_MOSQUITTO_CLIENT_SUB)
> ```
> muss dann auch noch das passwd in die Klammer dazu :)

Ich glaube nicht, weil passwd hängt unter TLS, welches unter broker hängt, Du kannst damit passwd nur selektieren, wenn Du den broker (und TLS) selektiert hast. 

> Das passwd müsste wenn es nicht von ausgewähltem SSL abhängen soll, zusätzlich von openssl>=1 abhängen

Da bin ich überfordert :flushed:

> Aber mach das wie es dir passt, ich will da nicht reinreden. Hattest du gesehen dass ich dir eine Einladung geschickt hatte die abgelaufen ist? 

Nein, leider nicht, sorry! Habe weder was in der github UI gesehen noch irgendeine Benachrichtigung erhalten :worried:

pfichtner, 2022-02-21:
Schliesse den issue, da mit merge von PR #473 die Fehler weg sind. Diskussion bezüglich menuconfig natürlich gerne weiterhin.

closed 2022-02-21 by pfichtner

fda77, 2022-02-22:
Ich weiss nicht wo/wie das angezeigt wird
[image]

So meinte ich: [patch.txt](https://github.com/Freetz-NG/freetz-ng/files/8112642/patch.txt)

pfichtner, 2022-02-22:
> Ich weiss nicht wo/wie das angezeigt wird [image]

Kannst Du bitte nochmal versuchen/erneuern? Ich suche mich dann nochmal durch die github UI. Eventuell ist auch einfach nur eine notification irgendwo verloren gegangen. 

> So meinte ich: [patch.txt](https://github.com/Freetz-NG/freetz-ng/files/8112642/patch.txt)

Das mosquitto Paket besteht eigentlich aus zwei Teilen: Dem Broker (Server), den man betreiben kann und Client Programmen mit denen man mit beliebigen MQTT-Brokern (z.B. mosquitto) auf beliebigen Servern kommunzieren kann (`mosquitto_pub` und `mosquitto_sub`). 
Ich habe FritzBoxen auf denen ich nur den Broker betreibe
und FritzBoxen auf denen ich nur das `mosquitto_pub` binary habe um Nachrichten an einen (auf einem anderen Rechner laufenden) MQTT-Broker zu schicken. 

`mosquitto_passwd` hingegen ist ein Utility Programm um passwd Dateien erzeugen zu können die der Broker dann nutzt (liest). Meines Erachtens macht es wenig Sinn, `mosquitto_passwd` zu kompilieren ohne dass ich den Broker kompiliere. Deswegen habe ich dieses binary nicht auf der Stufe der anderen zwei binaries gesehen. 

Beim rumspielen mit menuconfig finde ich aber trotzdem Deinen Ansatz logischer, nämlich, dass TLS selektiert wird wenn man `mosquitto_passwd` auswählt (statt `mosquitto_passwd` nur auswählbar zu haben, wenn TLS selektiert wurde). 

Übernimmst Du dann Deinen patch auf den master?

pfichtner, 2022-02-22:
> Ich hab das heut morgen bereits wieder geschickt. Schau mal nach dem Adblocker

Der Spamfilter hat die Mail tatsächlich gefressen und die Invites sind offensichtlich (im Gegensatz zu allem anderen) nicht im github dashboard einsehbar. Es tut mir leid, aber ich müsste Dich nochmals bitten den Invite zu verschicken, diesmal kommt er aber auch an :smile: Sorry für die Mühe!

pfichtner, 2022-02-22:
> Ne, mach du das. Wenn man versucht für andere mitzugenken (passwd nur für broker) kommt bestimmt irgend 

Danke, erledigt. Danke für den patch und vor allem Danke für das Vertrauen!

fda77, 2022-02-23:
Jeder der was beitragen möchte das auch tun können sollte. Ich hab auch nicht Lust/Zeit immer mich in jedes Package hineinzudenken. Mosquitto hatte ich mal mit einem ShellyPlug ausprobiert, fand das aber nicht so spannend und benutze sie aktuell einfach mit wget.
Ich hatte auch schon vorher mal dem einen oder ein Einladung geschickt, vielleicht sind die gar nicht alle bemerkt worden