Freetz-NG/freetz-ng #472: Unable to update Fritz OS

Pixelramp, 2022-02-21:
Currently, my Fritzbox 6591 Cable is running on 161.07.28 (07.28).

I'm trying to update to 07.29. But after uploading the new Image, I'll receive this message.
[image]

Pixelramp, 2022-02-21:
> 

It's actually not a downgrade.  I have no idea why Freetz calls it a downgrade. 

It should be an upgrade from 7.28 to 7.29

Updating via the avm gui results in this Error : 

firmwarecfg: ERROR
Error: opening page ../html/tools/update_result_noreboot.html

fda77, 2022-02-21:
Tried without ticking the "downgrade" checkbox?
By AVM gui it works only if the signature matches

Pixelramp, 2022-02-21:
Yes
[image]
Same result without the downgrade

Is a mismatching signature the cause for the firmwarecfg: ERROR  error?

fda77, 2022-02-21:
Okay, then freetz-flash does not work. But it always work to flash an avm image by avm gui

Pixelramp, 2022-02-21:
It sould work but it results into

firmwarecfg: ERROR
Error: opening page ../html/tools/update_result_noreboot.html

But i think thats more a AVM problem than a Freetz problem.. 

It's sad that there is no recover tool for the Cable versions.. 

It's been a while since i flashed the freetz image, so i need to look into it to flash the normal Image

fda77, 2022-02-21:
Maybe you made some branding "magic"?
Try to flash with tools/push_firmware or https://bitbucket.org/fesc2000/ffritz/src/6591/README-6591.md

Pixelramp, 2022-02-21:
Ok, I've tried pushing the avm image via the push_firmware tool. But the tool seems to be stuck here 
[image]

The info led lights up red and the power led is blinking green.

Should I wait longer or is there something wrong?

Pixelramp, 2022-02-21:
Ok, i successfully flashed the avm image via the flashtool. 

Thank you for the help! Now I can do the flashing for freetz again and hopefully updating will work the next time!

closed 2022-02-21 by Pixelramp