Freetz-NG/freetz-ng #478: avm-firewall not available in newer versions

oli-z, 2022-02-23:
Hi,

I wanted to ask if there is any reason, why the avm-firewall package is not available with newer FRITZ!OS versions. If I comment out the dependency it is available again.

Is there a reason why this old version dependency is there and are there any problems with commenting it out and using the package anyway? Just curious.

Thanks

fda77, 2022-02-23:
-> https://freetz-ng.github.io/freetz-ng/make/avm-portfw.html

closed 2022-02-23 by fda77

oli-z, 2022-02-25:
> -> https://freetz-ng.github.io/freetz-ng/make/avm-portfw.html

This works for port forwardings, but is there also a way to block outgoing connections to certain IPs?

fda77, 2022-03-07:
No. For simple DROP/REJECT you could use iptables, which works as long as you dont state/conntrack