Freetz-NG/freetz-ng #484: Wireguard Status reported inaccurately on "Services"-Overview

hoewer, 2022-02-28:
For several months now the status of Wireguard on the services overview is reported as stopped, even so it is up and running flawlessly.
[image]

On the detailed Wireguard overview the status is reported correctly
[image]

I'm on FB 7490, freetz-18925MA.

It is only a minor problem, but I decided to report it anyway.

fda77, 2022-03-07:
https://github.com/Freetz-NG/freetz-ng/blob/master/make/wireguard/files/root/etc/init.d/rc.wireguard#L5

> DAEMON_CHECK=wg-crypt-$WIREGUARD_INTERFACE

Can you check which processes are running? Maybe `WIREGUARD_INTERFACE` is not yet set there

hoewer, 2022-03-08:
Wireguard interface seems to be up and running, connections are built up easily:

4792 root 0 SW< [wg-crypt-wg0]

<details>
  <summary>Click to expand full process list!</summary>
  
  ## Result of 'ps'
  PID USER       VSZ STAT COMMAND
    1 root      1840 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [migration/0]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [rcu_sched]
   10 root         0 SW   [watchdog/0]
   11 root         0 SW   [watchdog/1]
   12 root         0 SW   [migration/1]
   13 root         0 SW   [ksoftirqd/1]
   15 root         0 SW<  [kworker/1:0H]
   20 root         0 SW<  [khelper]
   21 root         0 SW   [kdevtmpfs]
  134 root         0 SW<  [writeback]
  137 root         0 SW<  [bioset]
  138 root         0 SW<  [crypto]
  140 root         0 SW<  [kblockd]
  241 root         0 SW   [kswapd0]
  242 root         0 SWN  [ksmd]
  243 root         0 SW   [fsnotify_mark]
  250 root         0 SW   [temp_reader_CPU]
  772 root         0 SW   [pm_info]
  777 root         0 SW<  [eventsink_wq]
  780 root         0 SWN  [avm_debugd]
  896 root         0 SW<  [ipv6_addrconf]
  903 root         0 SW   [tffsd]
  904 root         0 SW<  [deferwq]
  907 root         0 SW<  [avmnet_workqueu]
  908 root         0 SW<  [PhyWaspHeartbea]
  924 root         0 SW<  [avmnet_timer]
  930 root         0 SW<  [loop0]
  959 root         0 SW   [cleanup_timer_f]
  960 root         0 SW   [avm_connect_eve]
  961 root         0 SW<  [eventsink_wq]
  962 root         0 SW<  [eventsink_wq]
  977 root      1528 S    supervisor /lib/systemd/system
  994 root         0 SW   [yaffs-bg-1]
 1130 root      1840 S    init
 1183 root     30400 S    /usr/bin/untrustedd
 1678 root      3244 S    /sbin/udevd
 2495 root         0 SW<  [eventsink_wq]
 2516 root         0 SW<  [capi_pipew]
 2517 root         0 SW<  [capi_schedw]
 2518 root         0 SW<  [eventsink_wq]
 2519 root         0 SW<  [kworker/1:1H]
 2520 root         0 SW   [pcmlink_ctrl/1]
 2522 root         0 SW   [capitransp]
 2535 root         0 SWN  [kneightrackd]
 2567 root         0 SW<  [avm_dect_thread]
 2644 root         0 SW   [yaffs-bg-1]
 2682 root      3836 S    /bin/avmipcd
 2688 root      1488 S    /bin/run_clock -c /dev/tffs -d
 2689 root      2512 S    /sbin/psetd /etc/psetd.cfg
 2696 root      2764 S    /usr/bin/boxnotifyd
 2717 root      2796 S    /bin/configd
 2794 root      7160 S    avmnexusd
 2819 root      4608 S    l2tpv3d
 2850 root     10268 S    dsl_control -i10_00_10_40_00_04_01_07 -f/lib/modules
 2856 root      8056 S    dsl_monitor -d
 2863 root     32468 S    ctlmgr
[... 45 lines cut ...]
 5519 nobody    3228 S    tinyproxy -c /mod/etc/tinyproxy.conf
 5520 nobody    3208 S    tinyproxy -c /mod/etc/tinyproxy.conf
 5521 nobody    3220 S    tinyproxy -c /mod/etc/tinyproxy.conf
 5522 nobody    3220 S    tinyproxy -c /mod/etc/tinyproxy.conf
 5523 nobody    3228 S    tinyproxy -c /mod/etc/tinyproxy.conf
 6234 root      3880 S    eth_oam -ieoam -D
 6236 root      8720 S    dpm_manager -ilan -C -D -T80 -U4 -V12 -W0 -X0
 6338 root      6664 S    /bin/avmike
 6445 root      1556 S    /sbin/chronyd -n -f /var/tmp/chrony.conf
 6607 root         0 SW   [kworker/0:2]
13459 root         0 SW   [kworker/1:1]
14953 root         0 SW   [kworker/0:0]
15024 root         0 SW   [kworker/1:0]
15192 root         0 SW   [kworker/u4:1]
17663 tor      29796 S    tor --runasdaemon 1
21525 root         0 SW   [kworker/u4:2]
22432 root      1380 S    dropbear -p 22 -0
22729 root      1408 S    dropbear -p 22 -0
22730 root      3428 S    /usr/lib/sftp-server
22736 root      1400 R    dropbear -p 22 -0
22737 root      1804 S    -sh
22864 root      1784 R    ps
</details>

Anything else I can check out for you?

fda77, 2022-03-08:
This is broken since https://github.com/Freetz-NG/freetz-ng/commit/face6c15567047d67e623521a308679664d5b7df

hoewer, 2022-03-10:
Works fine!
Thanx not only for caring for cosmetic problems like this one, but for your overall work on freetz, which expands even an average users opportunities by far.

Greg57070, 2022-03-25:
Removing personalized interface name broke the possibility to connect FritzBox VoIP through WireGuard.
in voip.cfg, sipiface_tunnel only work when interface name is tun0

Is there another possibility to use sipiface_tunnel and wg0 as interface name ?

fda77, 2022-03-30:
Ungetestet, falls es nicht klappt wird sich schon jemand melden

closed 2022-03-30 by fda77