Freetz-NG/freetz-ng #502: [6660 Cable] - LAN ports do only 100 MBit/s

nextgen-networks, 2022-03-25:
It seems latest changes introduced after build ng-18977 (2022-03-11) and before/until 2022-03-24 created a unwanted behavior:
Box will start regularly with Ports set to max. speed (usually 1 GBit/s).
After a couple of minutes all Ports are set to 100 MBit/s mode with no possibility to change in my setup (LAN1 as WAN port).

After revert back to build ng-18977 (created 2022-03-11) everything works as expected and all Ports work in 1 GBit/s mode.

Please share your experiences. Do you face similar issues?
@dev(s): need more input?

Thanks to everyone that support this project!

fda77, 2022-03-26:
Is this the same Fritzos version?

nextgen-networks, 2022-03-26:
@fda77 jepp, FritzOS hast been kept with 7.29.
Only FreetzNG build has been changed.

closed 2022-04-16 by fda77