Freetz-NG/freetz-ng #505: short heads up: just requested the Sources for 7.29 on FB6490, 6660, 7530

edeso, 2022-03-29:
Source Code FRITZ!OS 7.29 FRITZ!Box 6490 Cable was just uploaded
  https://osp.avm.de/fritzbox/fritzbox-6490-cable/

edeso, 2022-03-30:
Source Code FRITZ!OS 7.29 FRITZ!Box 7530
http://osp.avm.de/fritzbox/fritzbox-7530/

fda77, 2022-03-30:
The 6490 uses 6590 7.20 sources. Are there relevant changes to 6490 7.20?

closed 2022-03-30 by fda77

edeso, 2022-03-30:
> The 6490 uses 6590 7.20 sources. Are there relevant changes to 6490 7.20?

dunno, just noticed they were not released, so decided to request em for these 3 models i'm actually owning.

there are labor/betas 7.39 available. would you maybe interested in those too?

fda77, 2022-03-30:
Asking avm for sources of every versions help to teach them they shold automatic release them...
The 6660 labor uses kernel 4.9.279 and for this version are no sources available, so yes they are much more interesting.
But AVM ignores most time the GPL and just does not publish sources for labor versions. But you can try

edeso, 2022-04-01:
asking for release versions worked flawlessly so far. although they took up to two weeks to provide them.

for the beta/labor-version the first answer was NO, albeit they are accompanied by trhe same Source-Code  Offer as the release versions. i pointed that and the conditions of GPL2 out to them in the hope óf the first answer being just a fluke.

fda77, 2022-04-01:
As the labor of 6660 has another kernel version, the stable sources could not fit, just simple to verify... seems avm people are silly.
For labor src requests it seems to me they are wasting time by slow answer + writing some shit to reach the point when the stable is released and then take again some weeks for this sources...
Someone should raise a lawsuite, i have no time and money for that
Additional many times thing in the kernel-.config does not match the provided sources (additions or removals) which is also not okay with GPL
Btw, my consequence is to switch to a better and less expensive router without avm but with openwer. So i get rid of the crap avm closed-source daemons which i dont trust and with every new version has new bugs

edeso, 2022-04-05:
Source Code FRITZ!OS 7.29 FRITZ!Box 6660 Cable is now avail on
https://osp.avm.de/fritzbox/fritzbox-6660-cable/

fda77, 2022-04-05:
Freetz uses currently 7.26 sources of 6660: https://github.com/Freetz-NG/freetz-ng/blob/master/config/avm/source.in#L138=
So just download 7.26 and 7.29 and make a "diff" of the kernel sources and put it in make/linux/patches/ , i think here https://github.com/Freetz-NG/freetz-ng/tree/master/make/linux/patches/4.9.175.x86/

edeso, 2022-04-06:
running `diff -durN --no-dereference 6660.07.26/linux-4.9/ 6660.07.29/sources/kernel/linux-4.9/ > 6660.07.26-6660.07.29-delta.patch` throws some warnings in the beginning (see below)
resulting delta.xz is attached (remove gz extension, github didn't allow to attach a *.xz)

```
 Datei 6660.07.26/linux-4.9/arch/arm/boot/dts/include/dt-bindings ist ein symbolische Verknüpfung, während Datei 6660.07.29/sources/kernel/linux-4.9/arch/arm/boot/dts/include/dt-bindings ein Verzeichnis ist.
Datei 6660.07.26/linux-4.9/arch/arm64/boot/dts/include/dt-bindings ist ein symbolische Verknüpfung, während Datei 6660.07.29/sources/kernel/linux-4.9/arch/arm64/boot/dts/include/dt-bindings ein Verzeichnis ist.
Datei 6660.07.26/linux-4.9/arch/cris/boot/dts/include/dt-bindings ist ein symbolische Verknüpfung, während Datei 6660.07.29/sources/kernel/linux-4.9/arch/cris/boot/dts/include/dt-bindings ein Verzeichnis ist.
Datei 6660.07.26/linux-4.9/arch/metag/boot/dts/include/dt-bindings ist ein symbolische Verknüpfung, während Datei 6660.07.29/sources/kernel/linux-4.9/arch/metag/boot/dts/include/dt-bindings ein Verzeichnis ist.
Datei 6660.07.26/linux-4.9/arch/mips/boot/dts/include/dt-bindings ist ein symbolische Verknüpfung, während Datei 6660.07.29/sources/kernel/linux-4.9/arch/mips/boot/dts/include/dt-bindings ein Verzeichnis ist.
Datei 6660.07.26/linux-4.9/arch/powerpc/boot/dts/include/dt-bindings ist ein symbolische Verknüpfung, während Datei 6660.07.29/sources/kernel/linux-4.9/arch/powerpc/boot/dts/include/dt-bindings ein Verzeichnis ist.
diff -durN --no-dereference 6660.07.26/linux-4.9/arch/x86/configs/atomp7/intelsdk.config 6660.07.29/sources/kernel/linux-4.9/arch/x86/configs/atomp7/intelsdk.config
--- 6660.07.26/linux-4.9/arch/x86/configs/atomp7/intelsdk.config        1970-01-01 01:00:00.000000000 +0100
+++ 6660.07.29/sources/kernel/linux-4.9/arch/x86/configs/atomp7/intelsdk.config 2021-11-03 15:44:13.000000000 +0100
@@ -0,0 +1,3627 @@
+#
..
[6660.07.26-6660.07.29-delta.patch.xz.gz](https://github.com/Freetz-NG/freetz-ng/files/8427377/6660.07.26-6660.07.29-delta.patch.xz.gz)
.

fda77, 2022-04-06:
Thx! AVM does all the time such small silly changes to keep people busy. With latest 7530ax sources there where also empty directories replaced by link to other empty directories
I've changed the patch to patchlevel 0.

edeso, 2022-04-06:
the commit https://github.com/Freetz-NG/freetz-ng/commit/a51d29a7ed99b519da548ada31593737263d44ee just adds the big patch. do i understand correctly that AUTOFIX didn't need to update any patches then?

fda77, 2022-04-07:
autofix rewrites a patch to make it fit 100% (fuzz, lines moved) - if it does not yet before

edeso, 2022-05-01:
Source Code FRITZ!OS **7.39 Labor** FRITZ!Box 6660 Cable is now avail on
https://osp.avm.de/fritzbox/fritzbox-6660-cable/

unfortunately they did not specify the actual build version. i will request why they ommitted this crucial information.

fda77, 2022-05-01:
See https://github.com/Freetz-NG/freetz-ng/commit/d329a54e3ba0153aa53561fa40f8c06861da3d4d
You could try to fix the warnings because of https://github.com/Freetz-NG/freetz-ng/commit/a71123ad79cd7e4d4b9fb1931b8802f3375b2661 with `make kernel-menuconfig` and find the correct selections. Eg search where the missing symbols in the source are and enable/disable them...

edeso, 2022-05-01:
let me complain to AVM about the missing/wrong .config file. maybe it's just an oversight they are willing to fix? the whole process of putting the sources together seems improvised and changing. obviously they don't care, but so far they complied after insisting thoroughly enough.

fda77, 2022-05-01:
I "oversight" (or gpl violation) is well known by avm. Some examples: 9493063213ca68c42126b78347a8fab23e5cd4a2 and 2688d20d65be847eef80fb98b593902b69ef8334

edeso, 2022-05-01:
i know :) and i feel your pain. let's not go there. obviously you and me are still using AVM products and work with and on their firmware. 

afaics this is the first time they released a beta labor version at all.  a step in the right direction! let me point out the GPL requirement of "plus any associated interface definition files, plus the scripts used to control compilation and installation of the executable" which includes proper `.config` and see if they improve on the next delivery.

fda77, 2022-05-01:
No, i'm now using a good router OS (without closed source avm binary blobs) i can trust!
So i could use iptables/conntrac/state, multi-SSID, custom routing, VLANs etc - as the stuff not working with bad avm software

[image]

I use FOS no longer IN my lan, only external as modem without pppoe

fda77, 2022-05-01:
Btw, this is not the 1st time with labor sources, eg 7490_06.98.55294
But avm all the time does silent updates of sources, linke 7490_06.51 7369_06.33 or 7590_07.19
This silent-update crap is also done with release-firmwares like FRITZ.Box_Fon_WLAN_7360.124.06.00 or FRITZ.Box_5490-07.12

edeso, 2022-05-01:
> So it would be good when there are more developers of freetz!

surely. can't help you as much as this project deserves unfortunately.

> Btw, this is not the 1st time with labor sources, eg 7490_06.98.55294

good to hear. they even managed to add proper build numbers there :)

> But avm all the time does silent updates of sources, linke 7490_06.51 7369_06.33 or 7590_07.19
> This silent-update crap is also done with release-firmwares like FRITZ.Box_Fon_WLAN_7360.124.06.00 or FRITZ.Box_5490-07.12

you mean retroactively changing released sources? sure, not supposed to happen. would be interesting why they did.

and again. AVM fails in a lot of places, but i'm hopeful that it might get better, if only enough users execute their right to request sources and point out where AVM fails to fulfill their responsibilities, ideally publicly. you're welcome to disagree of course :)

fda77, 2022-05-01:
This is the name freetz uses for the kernel sources, i'm not sure anymore how avm named it.
AVM does not care much about GPL, except in 1 point: The sources have to be provided for AT LEAST 3 years - so avm removed then EXACTLY after 3 years ;-) It's just nasty.

They not only silent-update sources but also the release firmwares. Sometimes they include some .xlsx charts -.-

> but i'm hopeful that it might get better

I did this also for many years, last summer i switched to openwrt.
In the 6660-labor sources i found that (some?) current avm devices uses efuses for secure-boot, so it seems getting worser. Lock-in instead bugfixes and improvements

edeso, 2022-05-01:
> This is the name freetz uses for the kernel sources, i'm not sure anymore how avm named it. AVM does not care much about GPL, except in 1 point: The sources have to be provided for AT LEAST 3 years - so avm removed then EXACTLY after 3 years ;-) It's just nasty.

no doubt!

> They not only silent-update sources but also the release firmwares. Sometimes they include some .xlsx charts -.-

huh, sounds like Apple *lol*

> > but i'm hopeful that it might get better
> 
> I did this also for many years, last summer i switched to openwrt. In the 6660-labor sources i found that (some?) current avm devices uses efuses for secure-boot, so it seems getting worser. Lock-in instead bugfixes and improvements

surely they can make running other stuff difficult, as other manufacturers do already. if they make it impossible interest in the sources might also dry up might be a speculation of theirs. anyway, 
still they at least publish something, whilst others (especially chinese manufacturers) completely ignore GPL requests.

for now i'm comfortable exercising my right as user even if it needs 3 mails to reach anything.

fda77, 2022-05-02:
For freetz it would be enough to have a diff to the vanilla kernel and the .config of the kernel and busybox. The other things do not matter. For sure it should be the correct .config files.
A good company would automate it (and this prevents excel files in the packed file)
But its very boring all the time people have to write mails an beg for source

edeso, 2022-05-06:
didn't point it out to them so far. so probably yeah.
checking the archive `source-files-FRITZ.Box_6660_Cable-atom-07.39-95784.tar.gz` i can't find a `.config` but running `make menuconfig` results in the below, so i'd guess the `../linux/avm/conf/linux-4.9.atomp7` is just a template and freetz-ng should run `make ...` on it to create the actual `.config` ?

```
:~/source-files-FRITZ.Box_6660_Cable-atom-07.39-95784/sources/kernel/linux$ make menuconfig
UPD ../linux/avm/conf/linux-4.9.atomp7 -> .config
make[1]: Entering directory '/raider/foo/gluon/source-files-FRITZ.Box_6660_Cable-atom-07.39-95784/sources/kernel/linux'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/mconf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTCC  scripts/kconfig/lxdialog/checklist.o
  HOSTCC  scripts/kconfig/lxdialog/util.o
  HOSTCC  scripts/kconfig/lxdialog/inputbox.o
  HOSTCC  scripts/kconfig/lxdialog/textbox.o
  HOSTCC  scripts/kconfig/lxdialog/yesno.o
  HOSTCC  scripts/kconfig/lxdialog/menubox.o
  HOSTLD  scripts/kconfig/mconf
scripts/kconfig/mconf  Kconfig
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

make[1]: Leaving directory '/raider/foo/gluon/source-files-FRITZ.Box_6660_Cable-atom-07.39-95784/sources/kernel/linux'
Linux config has changed compared to ../linux/avm/conf/linux-4.9.atomp7:
+AVM_COMPANION_PANIC n
+AVM_PA_SCH_CAKE n
+BATTERY_RT5033 n
+TFFS_V31_CONVERT n
+UNIX_SCM y

:~/source-files-FRITZ.Box_6660_Cable-atom-07.39-95784/sources/kernel/linux$ head -n 10 .config
#
# Automatically generated file; DO NOT EDIT.
# Linux/x86 4.9.279 Kernel Configuration
#
# CONFIG_64BIT is not set
CONFIG_X86_32=y
CONFIG_X86=y
CONFIG_INSTRUCTION_DECODER=y
CONFIG_OUTPUT_FORMAT="elf32-i386"
CONFIG_ARCH_DEFCONFIG="arch/x86/configs/i386_defconfig"

edeso, 2022-05-06:
yeah right it's in `.kernelvariables` . 

so seeing, that make "fixes" or better creates a proper `.config` i'd say freetz-ng needs to be fixed. my kernel-compiling-foo is weak though, so no clue which command should be run to create it proper ;(

fda77, 2022-05-06:
start with "make help"

fda77, 2022-05-06:
Btw, using olddefconfig for a kernel by avm just means avm did not was they had to prvide due to gpl.
You cant know if the automatically selected things are set in the way avm did it / the fritzbox needs it to not show errors and is able to load the module. So if it does not work just change the options and try until there are no more errors :D

edeso, 2022-05-06:
anyway. is the [fix crappy 6660 src](https://github.com/Freetz-NG/freetz-ng/commit/a71123ad79cd7e4d4b9fb1931b8802f3375b2661) solved with this information or should we patch something to make source updates easier?

edeso, 2022-05-06:
> Btw, using olddefconfig for a kernel by avm just means avm did not was they had to prvide due to gpl. You cant know if the automatically selected things are set in the way avm did it / the fritzbox [...]

wasn't there a way to ask a running kernel for the config used to compile it or so? for now we do not know if AVM does not actually build the kernel using this way (if so, they fulfilled their obligations).

fda77, 2022-05-08:
AVM can not build with this .config, because it will ask y/n/m questions during build.
A kernel config could be embedded or not, you could check the .config if avm published it ;-)
Its called something like "ikconfig"

fda77, 2022-05-08:
> anyway. is the [fix crappy 6660 src issue](https://github.com/Freetz-NG/freetz-ng/commit/a71123ad79cd7e4d4b9fb1931b8802f3375b2661) solved with this information

I just hit enter for every question, i dont know if default answers are correct! Seems not as there are many warnings wiht depmod

edeso, 2022-05-08:
> AVM can not build with this .config, because it will ask y/n/m questions during build. 

unless they run `make oldconfig` on it, which neither of us knows.

> A kernel config could be embedded or not, you could check the .config if avm published it ;-) Its called something like "ikconfig"

sorry, what do you mean?

edeso, 2022-05-08:
> > anyway. is the [fix crappy 6660 src issue](https://github.com/Freetz-NG/freetz-ng/commit/a71123ad79cd7e4d4b9fb1931b8802f3375b2661) solved with this information
> 
> I just hit enter for every question, i dont know if default answers are correct! Seems not as there are many warnings wiht depmod

after you used the `.config` created via `make olddefconfig` ?

fda77, 2022-05-08:
After "olddefconfig" the .config fits to the sources, it uses defaults

edeso, 2022-05-08:
> After "olddefconfig" the .config fits to the sources, it uses defaults

right, for unknown symbols, which in this case are "only"
```
Linux config has changed compared to ../linux/avm/conf/linux-4.9.atomp7:
+AVM_COMPANION_PANIC n
+AVM_PA_SCH_CAKE n
+BATTERY_RT5033 n
+TFFS_V31_CONVERT n
+UNIX_SCM y
```
can we prove that these are not the configuration AVM built the kernel with and that's why they didn't update `linux-4.9.atomp7` ?

fda77, 2022-05-08:
I dont know, only avm does. Check if its embedded in the kernel. Or try to find a config which does not show warnings
Btw, 1 symbol can "unlock" many other dependeing...
PA is important, its the packet-accelerator, tffs for saving.
The package by avm is just crap, like many other. Digging into every source-package (for each device and again for each fritzos) is just to time consuming. Just get a lawyer and remember avm about gpl

edeso, 2022-05-08:
> I dont know, only avm does. Check if its embedded in the kernel. Or try to find a config which does not show warnings 

do you know how to check if it is embedded in the kernel?

>Btw, 1 symbol can "unlock" many other dependeing... PA is important, its the packet-accelerator, tffs for saving. The package by avm is just crap, like many other. Digging into every source-package (for each device and again for each fritzos) is just to time consuming. 

indeed. nobody wants to do that. that's why i ask - is there an easy way to find out in their finished product?

>Just get a lawyer and remember avm about gpl

how about asking nicely first? i find the money spent on lawyers is often wasted as an open dialog first could have resolved issues without involving lots of people with a law degree.

fda77, 2022-05-08:
> do you know how to check if it is embedded in the kernel?

As i wrote above, google for "ikconfig"

> is there an easy way to find out in their finished product?

No, every kernel for every device end every fritzos is different.  It takes much time and needs someone who realy want it :)
This problem with avm exists since several years

> how about asking nicely first?

I'm using non-avm as my router meanwhile as im bored asking avm. This was even cheaper and has more features which avm will never manage to make working, eg answer ping to 2 DIFFERENT devices PARALLEL

I stopped asking avm , but feel free and tell if/what the answer is
best to request only these things
 - kernel PATCH to vanilla kernel + config
 - busybox patches to vanilla + config
 - maybe libc 
that are about 5mb, and not 1gb unneeded stuff. Fore sure automated

edeso, 2022-05-09:
> > do you know how to check if it is embedded in the kernel?
> 
> As i wrote above, google for "ikconfig"

thanks, will have a look if time permits

> > is there an easy way to find out in their finished product?
> 
> No, every kernel for every device end every fritzos is different. It takes much time and needs someone who realy want it :) This problem with avm exists since several years

i'll just assume then that i'll build a freetz-ng, flash it, ssh to it and then try to whatever `ikconfig` is locally to find out what symbols the kernel waas build with.

> I'm using non-avm as my router meanwhile as im bored asking avm. This was even cheaper and has more features which avm will never manage to make working, eg answer ping to 2 DIFFERENT devices [...]

that's the plan.

> * kernel PATCH to vanilla kernel + config
> * busybox patches to vanilla + config
> * maybe libc
>   that are about 5mb, and not 1gb unneeded stuff. Fore sure automated

yeah, no. not gonna do that as they are unfortunately free to deliver sources in any form they deem fits. also i can relate that they want to dump everything as to not manually do source-"releases" fitting for each and every request they might get.

but again, i may ask for a "proper" Kernel `.config` or at least the information how these symbols are to be set next time.

fda77, 2022-05-10:
I think most source requests are because of freetz. If someone needs more they could then still do it. As AVM does source release manually this saves time

edeso, 2022-05-10:
> I think most source requests are because of freetz. If someone needs more they could then still do it. As AVM does source release manually this saves time

doesn't matter. not our place to request a specific form or fashion :)

just requested 6660/7530 sources for latest Labor version and pointed out that missing `.config` issue. let's see what that results in.

edeso, 2022-05-17:
wrt. the ".config issue" i got an answer. it seems and i checked that in recent deliveries (e.g. source-files-FRITZ.Box_6660_Cable-atom-07.39-95784) there is a file `sources/kernel/linux/README.avm-fos-kernel` which in detail explains (in a very polite tone btw.) how to compile the kernel. reading that i'd say the instructions are clear and that issue resolved or?

i'll just copy/paste it here verbatim, it's short

> AVM FRITZ!OS kernel - README
> ============================
> 
> AVM FRITZ!OS kernel is a Linux derivate.  If you want to rebuild your FRITZ!OS
> kernel, exactly, please go ahead this way:
> 
>  * Enable the settings for SOURCE_DATE_EPOCH, KBUILD_BUILD_USER,
>    KBUILD_BUILD_HOST and KBUILD_BUILD_TIMESTAMP in .kernelvariables by setting
>    'reproducibility' to a nonempty value.
> 
>  * Adjust the toolchain variables, to match your cross toolchain as required:
>    CROSS_COMPILE (e.g.  mips-linux-gnu-), CC (e.g. mips-linux-gnu-gcc-5),
>    HOSTCC (e.g. gcc).
> 
>  * Run 'make olddefconfig' to let kconfig regenerate the original .config file;
>    remember to add 'O=../path/to/build/dir' to all make calls, if you want to
>    build out-of-source.  Official FRITZ!OS kernel builds do 'olddefconfig', too,
>    and with the exception of puma6 based boards, we always build out-of-source.
> 
> Now you can build the kernel almost as ususal ('make reproducibility=1 all
> install modules_install headers_install dtbs_install ...').
> 
> Thanks for caring for your open-source kernel.
> 
> Yours,
> the FRITZ!OS kernel team.

fda77, 2022-05-20:
Yes, im not more polite as this crap with avm-sources lasts more than 10 years. I see only 2 cases
 - The developers @avm are to silly to pack the source package
 - Or its planned to release incorrect stuff
 
The "olddefconfig" is what i did, but as you see there are much missing symbols, so the default are not okay.
Beside that GPL want to include the USED .config, which is not the case as its still a different version than the kernel itself

edeso, 2022-05-20:
Where do you see "much missing symbols"? When?
Also the GPL does not state "USED .config" but "scripts used to control compilation". As i read it, **if** the scripts generate a fitting `.config` that would be satisfied, but only if.

In other news. AVM stopped delivering Labor-Version-Sources and demands now a prestamped envelope and storage media via snail mail on which they would supposedly burn/copy the data in an obvious attempt to avoid or at least make it more difficult to request the sources.
this seems a pretty obvious GPL violation so i guess it's time to find some legal help. will keep you up to date :)

fda77, 2022-05-20:
"depmod" in fwmod shows "unresolved/unknown symbols", if there are any: https://github.com/Freetz-NG/freetz-ng/blob/master/fwmod#L1436
This runs only if replace kernel is enabled or at least 1 kernel module selected