Freetz-NG/freetz-ng #512: bind BIND 9.16.27 crashes on startup

cawidtu, 2022-04-05:
I just flashed freetz-ng dc3962803f5e2e01c83ee3d365f68b5b5937fbbd (April 5) for my 5491 (Firmware 07.29) and found out that bind 9.16 does not start up any longer despite unchanged configuration. The previous version 9.11 worked fine. A syslog complains about a problem with sockets, see below. Any ideas?

`Apr  5 15:11:10 fritz daemon.info named[15054]: found 2 CPUs, using 2 worker threads
Apr  5 15:11:10 fritz daemon.info named[15054]: using 2 UDP listeners per interface
Apr  5 15:11:10 fritz daemon.info named[15054]: using up to 1024 sockets
Apr  5 15:11:10 fritz daemon.info named[15054]: loading configuration from '/etc/named.conf'
Apr  5 15:11:10 fritz daemon.info named[15054]: unable to open '/etc/bind.keys'; using built-in keys instead
Apr  5 15:11:10 fritz daemon.info named[15054]: using default UDP/IPv4 port range: [32768, 61000]
Apr  5 15:11:10 fritz daemon.info named[15054]: listening on IPv4 interface lo, 127.0.0.1#10054
Apr  5 15:11:10 fritz daemon.info named[15054]: listening on IPv4 interface lan, 192.168.178.2#10054
Apr  5 15:11:10 fritz daemon.info named[15054]: generating session key for dynamic DNS
Apr  5 15:11:10 fritz daemon.info named[15054]: sizing zone task pool based on 3 zones
Apr  5 15:11:10 fritz daemon.info named[15054]: using built-in root key for view _default
Apr  5 15:11:10 fritz daemon.info named[15054]: set up managed keys zone for view _default, file 'managed-keys.bind'
Apr  5 15:11:10 fritz daemon.info named[15054]: configuring command channel from '/etc/rndc.key'
Apr  5 15:11:10 fritz daemon.notice named[15054]: couldn't add command channel 127.0.0.1#953: file not found
Apr  5 15:11:10 fritz daemon.info named[15054]: managed-keys-zone: loaded serial 1
Apr  5 15:11:10 fritz daemon.info named[15054]: zone ...: loaded serial 2021111601
Apr  5 15:11:10 fritz daemon.notice named[15054]: all zones loaded
Apr  5 15:11:10 fritz daemon.notice named[15054]: running
Apr  5 15:11:10 fritz daemon.crit named[15054]: socket.c:3147: REQUIRE(readable || writeable) failed
Apr  5 15:11:10 fritz daemon.crit named[15054]: exiting (due to assertion failure)`

fda77, 2022-04-05:
Falsch konfiguriert. Einfach mal nach der Fehlermeldung suchen!
PS: Ich hab keine Lust immer hinter dir her zu formatieren

fda77, 2022-04-05:
Ist vielleicht ein Folgefehler von

https://forums.freebsd.org/threads/couldnt-add-command-channel-127-0-0-1-953-file-not-found.51540/
> Also, make sure you have a proper rndc-key generated. I'm not sure but I think older versions will just skip it if it's not there, later versions started actually requiring it.

Oder der Port ist einfach schon belegt (durch avm?)?

Ich benutze dnsmasq+unbound, auf der Fritzbox nur "dig"

cawidtu, 2022-04-05:
Ich habe die Remotekonfiguration jetzt abgeschaltet und der Syslog sieht wie folgt aus:
````Apr  5 17:59:07 fritz daemon.info named[7619]: found 2 CPUs, using 2 worker threads
Apr  5 17:59:07 fritz daemon.info named[7619]: using 2 UDP listeners per interface
Apr  5 17:59:08 fritz daemon.info named[7619]: using up to 1024 sockets
Apr  5 17:59:09 fritz daemon.info named[7619]: loading configuration from '/etc/named.conf'
Apr  5 17:59:09 fritz daemon.info named[7619]: unable to open '/etc/bind.keys'; using built-in keys instead
Apr  5 17:59:09 fritz daemon.info named[7619]: using default UDP/IPv4 port range: [32768, 61000]
Apr  5 17:59:09 fritz daemon.info named[7619]: listening on IPv4 interface lo, 127.0.0.1#10054
Apr  5 17:59:09 fritz daemon.info named[7619]: listening on IPv4 interface lan, 192.168.178.2#10054
Apr  5 17:59:09 fritz daemon.info named[7619]: generating session key for dynamic DNS
Apr  5 17:59:09 fritz daemon.info named[7619]: sizing zone task pool based on 3 zones
Apr  5 17:59:09 fritz daemon.info named[7619]: using built-in root key for view _default
Apr  5 17:59:09 fritz daemon.info named[7619]: set up managed keys zone for view _default, file 'managed-keys.bind'
Apr  5 17:59:09 fritz daemon.info named[7619]: managed-keys-zone: loaded serial 1
Apr  5 17:59:09 fritz daemon.info named[7619]: zone ....: loaded serial 2021111601
Apr  5 17:59:09 fritz daemon.notice named[7619]: all zones loaded
Apr  5 17:59:09 fritz daemon.notice named[7619]: running
Apr  5 17:59:09 fritz daemon.crit named[7619]: socket.c:3147: REQUIRE(readable || writeable) failed
Apr  5 17:59:09 fritz daemon.crit named[7619]: exiting (due to assertion failure)```

Der Server kommt also in den Zustand "running", bevor er wegen der fehlgeschlagenen Assertion aussteigt. Port sind nach meinem besten Wissen nicht belegt; insbesondere startet bind 9.11 mit derselben Konfiguration.

fda77, 2022-04-05:
Ich dachte du wärst jemand anderes der hier öfter schriebt und von dem ich weiss dass er Code formatieren kann...

https://github.com/isc-projects/bind9/blob/v9_16/lib/isc/unix/socket.c#L3146=
Irgend ein "fd" passt wohl nicht. Ich weiss nicht ob es an deiner Konfiguration (irgend was mit "socket" Kommunikation?), dem Compilieren/bump oder der Fritzbox liegt.

Starte named mal manuell wie hier: https://github.com/Freetz-NG/freetz-ng/blob/master/make/bind/files/root/etc/init.d/rc.bind#L19=
Vermutlich ist das nur "named"
Davor aber noch ein strace: "strace -fff named". Damit sollte man sehen um welches "fd" es geht und du kannst schauen wie das auf der Fritzbox angelegt ist und ob es das überhaupt gibt
Vielleicht hat named auch Parameter für mehr verbose/debug

fda77, 2022-04-06:
Schwer da was zu finden, ich kenne mich damit nicht gut aus!

Der Inhalt von /var/log/named.log ist was du am Anfang gepostet hast?
Es wird scheinbar noch eine "named.run" angelegt, was laut https://forums.cpanel.net/threads/how-do-you-stop-named-from-creating-named-run-files.629395/ wohl Debuginfos enthalten soll. Es steht aber kein Pfad dabei, sie könnte in einem "data" Verzeichnis sein. 

Ist dieser Port 41913 bereits vorher geöffnet? Das sollte unwahrscheinlich sein.
Mit den Sockets gibt es _nach_ der aktuellen Version (https://github.com/isc-projects/bind9/commits/v9_16) noch ein paar Änderungen für "SO_REUSEPORT_LB", siehe Kommentar von https://github.com/isc-projects/bind9/commit/8993ebc01a5f7035ac6374299215dd2867f68e10

Den Pfad "/var/media/ftp/uStor01/bind" von Freetz in der conf hast du angepasst?

Es gibt noch diverse "No such file or directory", zb für abc1/2.de Zonen, ich weiss nicht ob bind ohne diese starten will. bind.keys ist jedenfalls optional.

Ganz am Ende
```
pid 10030] --- SIGIOT {si_signo=SIGIOT, si_code=SI_TKILL, si_pid=10025, si_uid=0} ---
```
Also hat Prozess 10025 wohl einen Fehler

Dieser 10025 führt 2x ein "eventfd2" aus und ausserdem startet er wohl pid 10029+10030, irgend was mit isc-socket-0+isc-socket-1. Von diesem 10030 kommt am ende das IOT

... ich blick nicht durch

cawidtu, 2022-04-08:
Der Anhang war nur die Ausgabe von syslog. Ich habe jetzt auch in die named.log geschaut und dort Folgendes gefunden:

```
08-Apr-2022 08:24:21.485 general: debug 10: sendmsg: Invalid argument
08-Apr-2022 08:24:21.485 resolver: debug 11: sending packet to 8.8.8.8#53
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id:  60716
;; flags: rd cd; QUESTION: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1
;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags: do; udp: 512
; COOKIE: cbef8059699fd13f
;; QUESTION SECTION:
;.                              IN      DNSKEY
```

Im log von bind 9.11 taucht das nicht auf. Anschließend bin ich auf folgenden Thread gestoßen: https://serverfault.com/questions/1039137/bind-not-forwarding-query-after-upgrade-from-9-11-to-9-16

Insbesondere wird dort die Option 
```
##  bind916 options
    dnssec-validation no;
```

Ich habe `dnssec-validation no;` in meiner Konfigurationsdatei von bind 9.16 ergänzt und nun startet der Server. Ganz klar ist mir nicht, warum es ohne diese Option nicht mehr geht.

fda77, 2022-04-08:
Die 9.11 kann mit oder ohne openssl gebaut werden, bei der 9.16 wird es immer benötigt (ist auch hier so drin: https://github.com/Freetz-NG/freetz-ng/commit/014ac3dc1c504104de9284328a1add7e3d7ac15a)
Dh evtl war diese Validation vorher bei dir nicht aktiv da im menuconfig nicht ausfgewählt, konnte also keine Problem verursachen.
Auch wenn das Abschalten den daemon wieder starten lässt würde ich das nicht als endgültige Lösung ansehen. Ich nutze unbound weil ich ua die Validation haben will. Es wäre besser die Ursache herauszufinden weshalb das nicht funktioniert, es fehlt vielleicht nur eine Datei auf der Fritzbox, bzw ein Link wie diese 3 https://github.com/Freetz-NG/freetz-ng/tree/master/make/bind/files/root/etc  oder etwas in der Konfiguration (default: https://github.com/Freetz-NG/freetz-ng/blob/master/make/bind/files/root/etc/default.bind/named_conf)
Ich hab oben in den Logs etwas gelesen, dass bind.keys (https://www.isc.org/bind-keys/) nicht gefunden wird und eine interne Liste verwendet wird, das sollte okay sein

cawidtu, 2022-04-21:
Ich habe das weiter untersucht. OpenSSL ist aktiv und im Image enthalten. Auch sind die Konfigurationsdateien, die Du erwähnst, vorhanden (bis auf die von rndc, was ich nicht benutze). Interessanterweise läuft der Server sowohl, wenn ich `dnssec-validation no;` als auch wenn ich `dnssec-validation yes;` in der Konfigurationsdatei habe. Nur wenn ich die Option weglasse, stürzt der Server beim Start ab.

fda77, 2022-04-21:
Hm, komische Sache.
https://www.zytrax.com/books/dns/ch7/security.html#dnssec-validation sagt
> Since 9.5 the default value is dnssec-validation yes;

Vielleicht ist das ein Bug in bind? Ich habs mal aktualisiert mit https://github.com/Freetz-NG/freetz-ng/commit/4f16f5b0c , da ist irgend was mit dnssec: https://bind9.readthedocs.io/en/v9_16/notes.html#notes-for-bind-9-16-28

Ansonsten könnte man ein Anmerkung auf https://github.com/Freetz-NG/freetz-ng/blob/master/docs/make/bind.md packen oder die default-config (wenn man es noch nie auf der Fritzbox installiert hatte) erweitern:  https://github.com/Freetz-NG/freetz-ng/blob/master/make/bind/files/root/etc/default.bind/named_conf

closed 2022-05-20 by fda77