Freetz-NG/freetz-ng #514: Fehler beim Bauen des libtirpc-1.3.2

freetz-ng-mod, 2022-04-08:
```
In file included from clnt_dg.c:60:0:
/home/freetz-ng/75xx/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/arm-linux-uclibcgnueabi/include/linux/errqueue.h:8:16: error: redefinition of 'struct timespec'
         struct timespec {
                ^
In file included from /home/freetz-ng/75xx/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/arm-linux-uclibcgnueabi/include/sched.h:31:0,
                 from /home/freetz-ng/75xx/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/arm-linux-uclibcgnueabi/include/pthread.h:24,
                 from clnt_dg.c:35:
/home/freetz-ng/75xx/toolchain/build/arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/arm-linux-uclibcgnueabi/include/time.h:119:8: note: originally defined here
 struct timespec
        ^
make[3]: *** [Makefile:700: libtirpc_la-clnt_dg.lo] Fehler 1
make[3]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[3]: Verzeichnis „/home/freetz-ng/75xx/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/libtirpc-1.3.2/src“ wird verlassen
make[2]: *** [Makefile:517: all-recursive] Fehler 1
make[2]: Verzeichnis „/home/freetz-ng/75xx/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/libtirpc-1.3.2“ wird verlassen
make[1]: *** [Makefile:402: all] Fehler 2
make[1]: Verzeichnis „/home/freetz-ng/75xx/source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/libtirpc-1.3.2“ wird verlassen

ERROR: Build failed.
make: *** [make/libs/libtirpc/libtirpc.mk:19: source/target-arm_gcc-5.5.0_uClibc-1.0.40-nptl_kernel-4.1/libtirpc-1.3.2/src/.libs/libtirpc.so.3.0.0] Fehler 1
```

fda77, 2022-04-08:
Schreib doch die Box+FOS immer dazu, dann muss ich mir das nicht zusammensuchen, wozu ich zu faul bin

freetz-ng-mod, 2022-04-09:
7530AX

fda77, 2022-04-15:
Commit https://github.com/Freetz-NG/freetz-ng/commit/179fba31db6b148af32cb06ac4eb92ffeab3d1a9

closed 2022-04-15 by fda77