Freetz-NG/freetz-ng #518: Fehler beim Bauen von davfs2

freetz-ng-mod, 2022-04-17:
Bei den Boxen 6430 6490 6590 bekommt man denn Fehler 
ERROR: Build failed.
make: *** [make/davfs2/davfs2.mk:37: source/target-i686_gcc-5.5.0_uClibc-1.0.40-nptl/davfs2-1.5.2/.configured] Fehler 1

fda77, 2022-04-17:
"geht nicht" ist zu wenig

closed 2022-04-17 by fda77

freetz-ng-mod, 2022-04-17:
Mehr wird nicht angezeigt. Dann baue ein besserer Fehler anzeige ein dann kann man dir auch bessere Fehlermeldungen Posten ;-)

fda77, 2022-04-17:
So ein Bullshit

freetz-ng-mod, 2022-04-17:
Mehr kommt aber nicht, teste es selber doch. Der Fehler kommt noch bei so einigen anderen Boxen.

fda77, 2022-04-17:
Es kommt auf jeden Fall der Befehl der ausgeführt wurde. Dann zu 99,9% eine Fehlermeldung die diese zurückgegeben "1" verursacht hat

freetz-ng-mod, 2022-04-17:
```
---> package/davfs2 ... configuring ... (conf_cmd() { ./configure  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; cd source/target-mipsel_gcc-5.5.0_uClibc-0.9.32.1/davfs2-1.5.2;  rm -f config.{cache,status};    PATH="/home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin:/home/72xx/toolchain/build/mipsel_gcc-4.6.4/mipsel-unknown-linux-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-17-oracle/bin:/usr/lib/jvm/java-17-oracle/db/bin" FREETZ_LIBRARY_DIR="/usr/lib/freetz" FREETZ_KERNEL_VERSION_MAJOR="2.6.32" FREETZ_TARGET_ARCH="mips" FREETZ_TARGET_ARCH_ENDIANNESS_DEPENDENT="mipsel" CC="/home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc" CXX="/home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-g++-wrapper" CFLAGS="-march=4kc -mtune=4kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CXXFLAGS="-march=4kc -mtune=4kc -msoft-float -Ofast -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" LDFLAGS="" PKG_CONFIG_PATH="/home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/../lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/../lib/pkgconfig" GLOBAL_LIBDIR=/home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/usr/lib  CONFIG_SITE=/home/72xx/include/config.site/mipsel-linux-uclibc conf_cmd   --cache-file=/dev/null --target=mipsel-linux --host=mipsel-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --datadir=/usr/share --includedir=/usr/include --infodir=/usr/share/info --libdir=/usr/lib --libexecdir=/usr/lib --localstatedir=/var --mandir=/usr/share/man --sbindir=/usr/sbin --sysconfdir=/etc --with-gnu-ld --disable-nls    )
configure: loading site script /home/72xx/include/config.site/mipsel-linux-uclibc
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for mipsel-linux-strip... mipsel-linux-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for mipsel-linux-gcc... /home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc accepts -g... yes
checking for /home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of /home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc... gcc3
checking whether ln -s works... yes
checking whether NLS is requested... no
checking for msgfmt... /usr/bin/msgfmt
checking for gmsgfmt... /usr/bin/msgfmt
checking for xgettext... /usr/bin/xgettext
checking for msgmerge... /usr/bin/msgmerge
checking build system type... x86_64-pc-linux-gnu
checking host system type... mipsel-unknown-linux-gnu
checking for ld used by GCC... /home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/mipsel-linux-uclibc/bin/ld
checking if the linker (/home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/mipsel-linux-uclibc/bin/ld) is GNU ld... yes
checking for shared library run path origin... (cached) done
checking how to run the C preprocessor... /home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/mipsel-linux-uclibc-gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for CFPreferencesCopyAppValue... no
checking for CFLocaleCopyCurrent... no
checking whether to use NLS... no
checking for neon-config... /home/72xx/toolchain/build/mipsel_gcc-5.5.0_uClibc-0.9.32.1/mipsel-linux-uclibc/bin/neon-config
checking linking against neon... yes

ERROR: Build failed.
make: *** [make/davfs2/davfs2.mk:37: source/target-mipsel_gcc-5.5.0_uClibc-0.9.32.1/davfs2-1.5.2/.configured] Fehler 1
```

ICH sage doch da steht nichts.

freetz-ng-mod, 2022-04-17:
Denn Fehler produziert 
https://github.com/Freetz-NG/freetz-ng/commit/5de3cfae9f04d8c02dd9d076660fc525f1582bf9
so weit bin ich schon mal.

fda77, 2022-04-17:
Geht doch. Wie hast du das "Mehr wird nicht angezeigt [als diese 2 Zeilen]" nochmal genau gemeint?
Wie du selbst merkst wurde dir nach dem posten der Ausgabe selbst klar dass es was mit "neon"  sein muss.
Als nächstes könntest du diese lib neu bauen, vielleicht gibt es irgend welche Hinweise in der Ausgabe

freetz-ng-mod, 2022-04-17:
NEIN ich habe mir davfs2 genauer angeschaut und gesehen das es nur mit den alten libs da geht. 
Warum sollen immer andere Leute deine Fehler ausbessern. Du hast es doch verbockt. Nicht ich. 

Auch wenn ich nicht jede Box habe kann man zwischendurch doch mal ein paar Test Images bauen um zusehen, ob das überhaupt noch lauft.

fda77, 2022-04-17:
Du willst mich doch verarschen!
Ich hab meine Zeit für dich verschwendet und es das mal selbst gebaut. Es ist klar zu sehen dass du mit Absicht die 2 letzten und aufschlussreichen Zeilen weggelassen hast:
```
checking for CFLocaleCopyCurrent... no
checking whether to use NLS... no
checking for neon-config... /home/cam/freetz-ng/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.40-nptl/i686-linux-uclibc/bin/neon-config
checking linking against neon... yes
configure: incompatible neon library version 0.32.1: wanted 0.27 28 29 30
configure: error: could not find neon

ERROR: Build failed.
make: *** [make/davfs2/davfs2.mk:37: source/target-i686_gcc-5.5.0_uClibc-1.0.40-nptl/davfs2-1.5.2/.configured] Fehler 1
```

> Gewisse Boxen kommen nur mit 0.27 28 29 30 klar

Deine "Diagnose ist natürlich Käse. 

> configure: incompatible neon library version 0.32.1: wanted 0.27 28 29 30
configure: error: could not find neon

Wie da steht braucht die alte davfs2 auch eine alte neon Version. Damit ist deine Aufgabe klar: davfs2 aktualisieren

Lass das mit der "Analyse" am besten und poste einfach den kompletten und unveränderten Fehler in Zukunft

freetz-ng-mod, 2022-04-17:
Unterstelle mir nicht das ich was weggelassen habe. Denn das ist nicht der fall, weil warum sollte man das machen. 
Bei mir kam der Fehler nicht so.

Dann fix deine Fehler doch einfach und gut ist es.

fda77, 2022-04-17:
Es ist doch ganz klar dass du lügst!
Wie solltest du denn auf dieses Geschwurbel 
> Gewisse Boxen kommen nur mit 0.27 28 29 30 klar

kommen ohne diese 2 "fehlenden" Zeilen gelesen zu haben??

> Dann fix deine Fehler doch einfach und gut ist es.

Du mich auch